technyon/nuki_hub #481: Ring detection not working

Mkyuu10, 2024-10-07:
### PROBLEM DESCRIPTION
Since updating to version 9.00 the ring detection for the Opener does not work anymore. The protocol of the Opener lists the ringing, but nukihub does not publish anything on the /lock/ring topic as it should do (the topic doesn't even exists in MQTT Explorer) and the /lock/binaryRing topic does not change (but exists at least). It doesn't work in either mode (RTO, CM or locked) and was working in previous versions.

### REQUESTED INFORMATION
- [x] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [x] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [x] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [x] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [x] Device used (e.g., ESP32-S3): ESP32-S3
- [x] Nuki Hub binary firmware version number used: 9.00
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Provide the output of http://nukihubIP/info:
```
System Information
Nuki Hub version: 9.00
Nuki Hub build: 10235422676.1.1
Nuki Hub build type: Release
run: true
confVersion: 900
deviceId: 1445228237
deviceIdOp: 1445228237
nukiId: ***
nukidOp: ***
mqttbroker: homeassistant.local
mqttport: 1883
mqttuser: ***
mqttpass: ***
mqttlog: false
checkupdates: true
websrvena: false
lockena: true
lockpin: 1
mqttpath: nuki
openerena: true
openerpin: 1
openercont: false
mqttoppath: nukiopener
maxkpad: 
opmaxkpad: 
maxtc: 
opmaxtc: 
enabtlprst: false
mqttca: 
mqttcrt: 
mqttkey: 
hassdiscovery: homeassistant
hassConfigUrl: 
buffsize: 4096
dhcpena: true
ipaddr: 
ipsub: 
ipgtw: 
dnssrv: 
nwhw: 1
nwwififb: true
rssipb: 60
hostname: nukihub
nwbestrssi: false
nettmout: 60
restdisc: true
rstbcn: 60
lockStInterval: 1800
tcPerEntry: false
kpPerEntry: false
configInterval: 3600
batInterval: 1800
kpInterval: 1800
kpCntrlEnabled: true
kpInfoEnabled: false
kpPubCode: false
tcCntrlEnabled: false
tcInfoEnabled: false
cnfInfoEnabled: true
regAsApp: false
regOpnAsApp: false
nrRetry: 3
rtryDelay: 100
crdusr: ***
crdpass: ***
disnonjson: false
pubAuth: false
pubdbg: false
prdtimeout: 
offHybrid: false
hybridTimer: 600
hybridAct: false
hybridRtry: false
hasmac: false
macb0: 
macb1: 
macb2: 
latest: 9.00
tsksznetw: 12288
tsksznuki: 8192
authmaxentry: 5
kpmaxentry: 10
tcmaxentry: 10
updMqtt: true
showSecr: false
bleTxPwr: 9
recNtwMqttDis: true
MQTT connected: Yes
Lock firmware version: 3.10.7
Lock hardware version: 6.11
Lock paired: Yes
Lock valid PIN set: Yes
Lock has door sensor: No
Lock has keypad: No
Lock ACL (Lock): Allowed
Lock ACL (Unlock): Allowed
Lock ACL (Unlatch): Allowed
Lock ACL (Lock N Go): Allowed
Lock ACL (Lock N Go Unlatch): Allowed
Lock ACL (Full Lock): Allowed
Lock ACL (Fob Action 1): Allowed
Lock ACL (Fob Action 2): Allowed
Lock ACL (Fob Action 3): Allowed
Lock config ACL (Name): Disallowed
Lock config ACL (Latitude): Disallowed
Lock config ACL (Longitude): Disallowed
Lock config ACL (Auto Unlatch): Disallowed
Lock config ACL (Pairing enabled): Disallowed
Lock config ACL (Button enabled): Allowed
Lock config ACL (LED flash enabled): Disallowed
Lock config ACL (LED brightness): Disallowed
Lock config ACL (Timezone offset): Disallowed
Lock config ACL (DST mode): Disallowed
Lock config ACL (Fob Action 1): Disallowed
Lock config ACL (Fob Action 2): Disallowed
Lock config ACL (Fob Action 3): Disallowed
Lock config ACL (Single Lock): Disallowed
Lock config ACL (Advertising Mode): Disallowed
Lock config ACL (Timezone ID): Disallowed
Lock config ACL (Unlocked Position Offset Degrees): Disallowed
Lock config ACL (Locked Position Offset Degrees): Disallowed
Lock config ACL (Single Locked Position Offset Degrees): Disallowed
Lock config ACL (Unlocked To Locked Transition Offset Degrees): Disallowed
Lock config ACL (Lock n Go timeout): Disallowed
Lock config ACL (Single button press action): Disallowed
Lock config ACL (Double button press action): Disallowed
Lock config ACL (Detached cylinder): Disallowed
Lock config ACL (Battery type): Disallowed
Lock config ACL (Automatic battery type detection): Disallowed
Lock config ACL (Unlatch duration): Disallowed
Lock config ACL (Auto lock timeout): Disallowed
Lock config ACL (Auto unlock disabled): Disallowed
Lock config ACL (Nightmode enabled): Disallowed
Lock config ACL (Nightmode start time): Disallowed
Lock config ACL (Nightmode end time): Disallowed
Lock config ACL (Nightmode auto lock enabled): Disallowed
Lock config ACL (Nightmode auto unlock disabled): Disallowed
Lock config ACL (Nightmode immediate lock on start): Disallowed
Lock config ACL (Auto lock enabled): Disallowed
Lock config ACL (Immediate auto lock enabled): Disallowed
Lock config ACL (Auto update enabled): Disallowed
Opener firmware version: 1.11.0
Opener hardware version: 4.17
Opener paired: Yes
Opener valid PIN set: Yes
Opener has keypad: Yes
Opener ACL (Activate Ring-to-Open): Allowed
Opener ACL (Deactivate Ring-to-Open): Allowed
Opener ACL (Electric Strike Actuation): Allowed
Opener ACL (Activate Continuous Mode): Allowed
Opener ACL (Deactivate Continuous Mode): Allowed
Opener ACL (Fob Action 1): Allowed
Opener ACL (Fob Action 2): Allowed
Opener ACL (Fob Action 3): Allowed
Opener config ACL (Name): Disallowed
Opener config ACL (Latitude): Disallowed
Opener config ACL (Longitude): Disallowed
Opener config ACL (Pairing enabled): Disallowed
Opener config ACL (Button enabled): Allowed
Opener config ACL (LED flash enabled): Disallowed
Opener config ACL (Timezone offset): Disallowed
Opener config ACL (DST mode): Disallowed
Opener config ACL (Fob Action 1): Disallowed
Opener config ACL (Fob Action 2): Disallowed
Opener config ACL (Fob Action 3): Disallowed
Opener config ACL (Operating Mode): Disallowed
Opener config ACL (Advertising Mode): Disallowed
Opener config ACL (Timezone ID): Disallowed
Opener config ACL (Intercom ID): Disallowed
Opener config ACL (BUS mode Switch): Disallowed
Opener config ACL (Short Circuit Duration): Disallowed
Opener config ACL (Eletric Strike Delay): Disallowed
Opener config ACL (Random Electric Strike Delay): Disallowed
Opener config ACL (Electric Strike Duration): Disallowed
Opener config ACL (Disable RTO after ring): Disallowed
Opener config ACL (RTO timeout): Disallowed
Opener config ACL (Doorbell suppression): Disallowed
Opener config ACL (Doorbell suppression duration): Disallowed
Opener config ACL (Sound Ring): Disallowed
Opener config ACL (Sound Open): Disallowed
Opener config ACL (Sound RTO): Disallowed
Opener config ACL (Sound CM): Disallowed
Opener config ACL (Sound confirmation): Disallowed
Opener config ACL (Sound level): Disallowed
Opener config ACL (Single button press action): Disallowed
Opener config ACL (Double button press action): Disallowed
Opener config ACL (Battery type): Disallowed
Opener config ACL (Automatic battery type detection): Disallowed
Network device: Built-in Wi-Fi
BSSID of AP: ***
Uptime: 132 minutes
Heap: 69300
Stack watermarks: nw: 8200, nuki: 4812
Restart reason FW: RestartOnDisconnectWatchdog
Restart reason ESP: ESP_RST_SW: Software reset via esp_restart.
```

### TO REPRODUCE
Ring the doorbell and inspect the /lock/ring topic for the Opener.

### EXPECTED BEHAVIOUR
The string "ring" should published to this topic when a doorbell ring is detected while RTO or CM is active or "ringlocked" when both are inactive.

### SCREENSHOTS
[image]

### ADDITIONAL CONTEXT
-

**(Please, remember to close the issue when the problem has been addressed)**

mundschenk-at, 2024-10-08:
I can confirm the issue with 

```
Version: 9.01-master16
Build: 10644385152.62.1
```

technyon, 2024-10-14:
@iranl I had a look at the code, the condition to detect a ring hasn't changed as far as I can see, but there's a second condition now for "Nuki opener: Ring detected (Locked)". I don't think this should make a difference. Can you verify this? Unfortunately I don't have an opener at this time.

anthonws, 2024-10-14:
+1. I noticed recently this.

This is what I have exposed in HA at the moment:

[image]

It doesn't seem to be picking up state changes.

iranl, 2024-10-14:
I'll take a look at a fix for inclusion in 9.02. Will need to figure out how to fake rings on my opener (as I don't own an intercom)

technyon, 2024-10-15:
Switch to analog intercom and shortcut the ring wires (green and yellow I think)

hahnet, 2024-10-16:
i can confirm it with my installation aswell.

do i see it right? 9.02 is expected in month? So if i really need the ring, i need to go back to 8.xx?

adrianipopescu, 2024-11-04:
updated last night to the 9.02 dev, can confirm, still not working

PR #512 (2024-11-04): Various fixes for HTTP, MQTT, PSRAM and Ring detection + Add Home assistant enable/disable checkbox
## Description:

**Related issue:** fixes #481 

## Checklist:
  - [x] The pull request is done against the latest master branch
  - [x] Only relevant files were touched
  - [x] The code change is tested and works
  - [x] I accept the [CLA](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md#contributor-license-agreement-cla).
closed 2024-11-04 by iranl