technyon/nuki_hub #483: [Feature Request] Check if keypad code and codeId combination is valid

iranl, 2024-10-11:
### Discussed in https://github.com/technyon/nuki_hub/discussions/479

<div type='discussions-op-text'>

<sup>Originally posted by **jkrenzer** October  3, 2024</sup>
Hi!

After setting a key-code it would be helpful if there was an endpoint for checking a given 6-digit code against a codeId. This way an external interface updating the code could afterwards verify, that the new code is now set without publishing all codes in the mqtt permanently.

Something like:

Request:

````json
{
  "action": "check",
  "codeId": 1234,
  "code": 654321
}
```` 

Response:

````json
{
  "valid": true
}
````

Kind regards,
Jörn</div>

technyon, 2024-10-14:
I'm really thinking if we shouldn't just add that option to publish the keypad codes. It should be clearly marked it in the web interface as safety relevant. If someone takes the necessary precautions to secure their system like using SSL encryption, it maybe is not that big of a safety issue, but we should communicate that clearly. Having such an option to just verify a code is a good thing to have too though. @iranl Are you looking into this already?

technyon, 2024-10-14:
Nevermind, I just realized we have the "Also publish keypad codes (Disadvised for security reasons)" already.

iranl, 2024-10-14:
I'll work on this for 9.02 as an optional (disabled by default) feature

iranl, 2024-10-15:
I've added the code in #488 with some basic brute force prevention and a toggle in de Web configurator that needs to be set to allow checking codes at all.

I have not been able to test as I'm somewhat limited by my home remoddeling in what I can test on my main Nuki 4.0 pro + keypad 2.

Format should be as suggested to the `nukihub/keypad/actionJson` topic:
```
{
  "action": "check",
  "codeId": 1234,
  "code": 654321
}
```

Result will be in the `keypad/commandResultJson` topic (`checkingKeypadCodesDisabled`, `codeValid`, `codeInvalid`, `checkingCodesBlockedTooManyInvalid`)

PR still needs a README update and testing

PR #488 (2024-11-02): Check keypad codes + unicode check for user/password
## Description:

**Related issue (if applicable):** fixes #483, fixes #487, fixes #505

To-Do
- [x] Test
- [x] README update (toggle in webcfg, mqtt topics, keypad section)

## Checklist:
  - [x] The pull request is done against the latest master branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works
  - [x] I accept the [CLA](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md#contributor-license-agre [... cut]