technyon/nuki_hub #487: Are there limitations to username / password for http basic authentication?

Ra72xx, 2024-10-15:
I wanted to set a strong password for http basic authentication and therefore generated one with a generator. However, I repeatedly locked myself out this way, as I was not able to log in with the password. I had to re-flash everything.

The username was "nuki_master" and the password "ESMBiet3tZäh1enP0l3nSturm" (it's no longer relevant).

Is it possible that the 'ä' character messes up the authentication, or is the password to long, or whatever? Everything went fine when using a simpler, shorter password.

Probably there should be a hint in the documentation concerning those restrictions, if they exist.

mundschenk-at, 2024-10-15:
There is a length limit.

technyon, 2024-10-15:
Hi. Not sure I'd need to check the code, but at some level it'll use plain old C char arrays. Avoiding such special characters as Umlaute is probably good, maybe we should add some checks.

iranl, 2024-10-15:
Unicode is not supported currently, so yes the 'ä' character is the problem.
Max length of 30 for both user and password is mentioned:
[image]

I've added a check for unicode chars in #488

PR #488 (2024-11-02): Check keypad codes + unicode check for user/password
## Description:

**Related issue (if applicable):** fixes #483, fixes #487, fixes #505

To-Do
- [x] Test
- [x] README update (toggle in webcfg, mqtt topics, keypad section)

## Checklist:
  - [x] The pull request is done against the latest master branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works
  - [x] I accept the [CLA](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md#contributor-license-agre [... cut]