technyon/nuki_hub #489: NukiOpener open action does not always trigger successfully

kotylo, 2024-10-17:
### PROBLEM DESCRIPTION
Not sure if that's a 9.0 problem or it was couple of releases before, but I've noticed my door won't open as fast as before.
When i trigger open action on the opener, first one is successful every time. My buzzer duration is usually about 6 seconds. So if i try to open again, also after 6 seconds, the request will fail with 4 retries. Then in about 30 seconds i can try again and it will work again. But not sooner.
I've tried updating opener to latest firmware, nuki hub to 9.01, but same behavior. Haven't tried to roll back to 8 though, but if that helps with debugging, i can do it.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [x] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [ ] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [x] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [x] Device used (e.g., ESP32-S3): ESP32 Dev Kit
- [x] Nuki Hub binary firmware version number used: 9.01
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Provide the output of http://nukihubIP/info:
```
System Information
------------ NUKI HUB ------------
Version: 9.01
Build: 11345224811.2.1
Build type: Release
Build date: 2024-10-15
Updater version: 9.01
Updater build: 11345224811.2.1
Updater build date: 2024-10-13
Uptime (min): 963
Config version: 901
Last restart reason FW: RequestedViaWebServer
Last restart reason ESP: ESP_RST_SW: Software reset via esp_restart.
Free internal heap: 68104
Total internal heap: 260664
PSRAM Available: No
Network task stack high watermark: 8464
Nuki task stack high watermark: 5044

------------ GENERAL SETTINGS ------------
Network task stack size: 12288
Nuki task stack size: 8192
Check for updates: No
Latest version: 
Allow update from MQTT: No
Web configurator username: ***
Web configurator password: ***
Web configurator enabled: Yes
Publish debug information enabled: No
MQTT log enabled: Yes
Webserial enabled: No
Bootloop protection enabled: No

------------ NETWORK ------------
Network device: Built-in Wi-Fi
Network connected: Yes
IP Address: 192.168.86.42
SSID: home
BSSID of AP: CC:F4:11:xx:xx:xx
ESP32 MAC address: A8:42:E3:xx:xx:xx

------------ NETWORK SETTINGS ------------
Nuki Hub hostname: nukihub
DHCP enabled: Yes
Fallback to Wi-Fi / Wi-Fi config portal disabled: No
Connect to AP with the best signal enabled: No
RSSI Publish interval (s): 60
Restart ESP32 on network disconnect enabled: Yes
Reconnect network on MQTT connection failure enabled: No
MQTT Timeout until restart (s): 30

------------ MQTT ------------
MQTT connected: Yes
MQTT broker address: 192.168.86.150
MQTT broker port: 1883
MQTT username: ***
MQTT password: ***
MQTT lock base topic: nuki
MQTT opener base topic: nuki
MQTT SSL CA: Not set
MQTT SSL CRT: Not set
MQTT SSL Key: Not set

------------ BLUETOOTH ------------
Bluetooth TX power (dB): 9
Bluetooth command nr of retries: 3
Bluetooth command retry delay (ms): 100
Seconds until reboot when no BLE beacons recieved: 60

------------ QUERY / PUBLISH SETTINGS ------------
Lock/Opener state query interval (s): 1800
Publish Nuki device authorization log: No
Max authorization log entries to retrieve: 5
Battery state query interval (s): 1800
Most non-JSON MQTT topics disabled: No
Publish Nuki device config: Yes
Config query interval (s): 3600
Publish Keypad info: No
Keypad query interval (s): 1800
Enable Keypad control: Yes
Publish Keypad topic per entry: No
Publish Keypad codes: No
Max keypad entries to retrieve: 10
Publish timecontrol info: No
Keypad query interval (s): 1800
Enable timecontrol control: No
Publish timecontrol topic per entry: No
Max timecontrol entries to retrieve: 10

------------ HOME ASSISTANT ------------
Home Assistant auto discovery enabled: Yes
Home Assistant auto discovery topic: homeassistant/
Nuki Hub configuration URL for HA: http://192.168.86.42

------------ NUKI LOCK ------------
Lock enabled: Yes
Paired: Yes
Nuki Hub device ID: 4021158070
Nuki device ID: ***
Firmware version: 3.9.5
Hardware version: 2.10
Valid PIN set: No
Has door sensor: No
Has keypad: No
Timecontrol highest entries count: 0
Register as: Bridge

------------ HYBRID MODE ------------
Hybrid mode enabled: No

------------ NUKI LOCK ACL ------------
Lock: Allowed
Unlock: Allowed
Unlatch: Allowed
Lock N Go: Allowed
Lock N Go Unlatch: Allowed
Full Lock: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed

------------ NUKI LOCK CONFIG ACL ------------
Name: Disallowed
Latitude: Disallowed
Longitude: Disallowed
Auto Unlatch: Disallowed
Pairing enabled: Allowed
Button enabled: Allowed
LED flash enabled: Allowed
LED brightness: Disallowed
Timezone offset: Disallowed
DST mode: Disallowed
Fob Action 1: Disallowed
Fob Action 2: Disallowed
Fob Action 3: Disallowed
Single Lock: Allowed
Advertising Mode: Disallowed
Timezone ID: Disallowed
Unlocked Position Offset Degrees: Disallowed
Locked Position Offset Degrees: Disallowed
Single Locked Position Offset Degrees: Disallowed
Unlocked To Locked Transition Offset Degrees: Disallowed
Lock n Go timeout: Disallowed
Single button press action: Disallowed
Double button press action: Disallowed
Detached cylinder: Disallowed
Battery type: Disallowed
Automatic battery type detection: Disallowed
Unlatch duration: Disallowed
Auto lock timeout: Disallowed
Auto unlock disabled: Allowed
Nightmode enabled: Disallowed
Nightmode start time: Disallowed
Nightmode end time: Disallowed
Nightmode auto lock enabled: Disallowed
Nightmode auto unlock disabled: Disallowed
Nightmode immediate lock on start: Disallowed
Auto lock enabled: Allowed
Immediate auto lock enabled: Disallowed
Auto update enabled: Disallowed

------------ NUKI OPENER ------------
Opener enabled: Yes
Paired: Yes
Nuki Hub device ID: 4021158070
Nuki device ID: ***
Firmware version: 1.11.0
Hardware version: 4.17
Opener valid PIN set: Yes
Opener has keypad: Yes
Timecontrol highest entries count: 0
Register as: Bridge
Nuki Opener Lock/Unlock action set to Continuous mode in Home Assistant: No

------------ NUKI OPENER ACL ------------
Activate Ring-to-Open: Allowed
Deactivate Ring-to-Open: Allowed
Electric Strike Actuation: Allowed
Activate Continuous Mode: Allowed
Deactivate Continuous Mode: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed

------------ NUKI OPENER CONFIG ACL ------------
Name: Disallowed
Latitude: Disallowed
Longitude: Disallowed
Pairing enabled: Disallowed
Button enabled: Allowed
LED flash enabled: Allowed
Timezone offset: Disallowed
DST mode: Disallowed
Fob Action 1: Disallowed
Fob Action 2: Disallowed
Fob Action 3: Disallowed
Operating Mode: Disallowed
Advertising Mode: Disallowed
Timezone ID: Disallowed
Intercom ID: Disallowed
BUS mode Switch: Disallowed
Short Circuit Duration: Disallowed
Eletric Strike Delay: Allowed
Random Electric Strike Delay: Disallowed
Electric Strike Duration: Allowed
Disable RTO after ring: Disallowed
RTO timeout: Disallowed
Doorbell suppression: Disallowed
Doorbell suppression duration: Disallowed
Sound Ring: Allowed
Sound Open: Disallowed
Sound RTO: Disallowed
Sound CM: Disallowed
Sound confirmation: Disallowed
Sound level: Allowed
Single button press action: Disallowed
Double button press action: Disallowed
Battery type: Disallowed
Automatic battery type detection: Disallowed

------------ GPIO ------------
```

### TO REPRODUCE
Just open lock, wait for it to close. Open again. Try multiple times.

### EXPECTED BEHAVIOUR
Lock should be opened second+ time and not fail.

### SCREENSHOTS
[image]

### ADDITIONAL CONTEXT
First call (starts at 4:28:16) is ok, second one fails (starts at 4:28:52).
Logs:
````
4:28:53 Opener: Maximum number of retries exceeded, aborting.
4:28:52 Opener: Last command failed, retrying after 100 milliseconds. Retry 4 of 3 4:28:52 Opener action result: error
4:28:52 Opener: Last command failed, retrying after 100 milliseconds. Retry 3 of 3 4:28:52 Opener action result: error
4:28:52 Opener: Last command failed, retrying after 100 milliseconds. Retry 2 of 3 4:28:52 Opener action result: error
4:28:52 Opener: Last command failed, retrying after 100 milliseconds. Retry 1 of 3 4:28:52 Result (attempt 2): Result (attempt 3): Result (attempt 4): Opener action result: error
4:28:52 Result (attempt 1): Opener action received: electricStrikeActuation
4:28:22 Opener action result: success
4:28:16 Opener action received: electricStrikeActuation
````

**(Please, remember to close the issue when the problem has been addressed)**

iranl, 2024-11-02:
PR #508 and https://github.com/I-Connect/NukiBleEsp32/pull/77 have some changes that might be able to fix this. Preliminary testing shows commands to no issues sending multiple consecutive commands in short order.

iranl, 2024-11-02:
PR's have been merged. 
I can't reproduce this issue anymore in 9.02-master9 on a Nuki Lock 4.0 (Pro) and Nuki Opener

Please try 9.02-master9 or 9.02-beta2 or the final release of 9.02 (beta and final not released yet)

kotylo, 2024-11-03:
Thanks, that works now blazing fast and if i don't make it in 6 seconds to the door, i can trigger it after 4 seconds again for another electric strike.

p.s. I was building via docker for about an hour... and then realized the master**9** must mean something and remembered that updater allows development updations now. Oh well, at least some tips for someone building on windows via docker, use the // for the paths in `build_with_docker_pio.sh` and winpty:

````
winpty docker run --rm -it -v //d/Projects/nuki_hub/Docker/..://src -w //src ${IMAGE_NAME} //bin/bash -c "make deps && make updater && make release"
````

closed 2024-11-03 by kotylo

Seinuh, 2024-11-03:
Confirmed great work @iranl