technyon/nuki_hub #497: GPIO not UP or Down as General Input

Alexander0815, 2024-10-19:
Hello
The GPIO pin in General Input (Pull UP and Pull Down) no longer works reliably in the current version 9.1. I'm currently using version 8.33 again, the problem doesn't exist there.
Usually, the status changes between 0 and 1 only once.
Greetings Alex

iranl, 2024-10-19:
Following the issue template is mandatory. Please edit your issue with the fully completed template.

Alexander0815, 2024-10-19:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [ ] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [ ] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [ ] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [ x] Device used (e.g., ESP32-S3): wt32-eth01
- [ x] Nuki Hub binary firmware version number used: 9.1
  - [ x] Pre-compiled
  - [ ] Self-compiled
- [ ] Provide the output of http://nukihubIP/info:
```

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

technyon, 2024-10-20:
Should be fixed in the latest build:

https://github.com/technyon/nuki_hub/actions/runs/11424601420

Alexander0815, 2024-10-20:
Super Dankeschön

closed 2024-10-20 by technyon