technyon/nuki_hub #498: Page load issues on Safari browser (v9 firmware)

bcutter, 2024-10-19:
### PROBLEM DESCRIPTION
Safari on a iOS 18 device can not load the web interface. "Error: parsing of the answer not possible" is shown (translated to English). Firefox on the same iOS 18 device loads and shows the page as expected. Username and password for web authentication are set of course. Blocking addons/extensions are disabled in Safari.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [X] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [X] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [ ] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [X] Device used (e.g., ESP32-S3): ESP32 (LOLIN32 Lite)
- [X] Nuki Hub binary firmware version number used: v9.01 (build 11345224811.2.1)
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Provide the output of http://nukihubIP/info:
```
System Information

------------ NUKI HUB ------------
Version: 9.01
Build: 11345224811.2.1
Build type: Release
Build date: 2024-10-15
Updater version: 
Updater build: 
Updater build date: 
Uptime (min): 46
Config version: 901
Last restart reason FW: NotApplicable
Last restart reason ESP: ESP_RST_PANIC: Software reset due to exception/panic.
Free internal heap: 68700
Total internal heap: 260664
PSRAM Available: No
Network task stack high watermark: 8332
Nuki task stack high watermark: 5072

------------ GENERAL SETTINGS ------------
Network task stack size: 12288
Nuki task stack size: 8192
Check for updates: Yes
Latest version: 9.01
Allow update from MQTT: No
Web configurator username: ***
Web configurator password: ***
Web configurator enabled: Yes
Publish debug information enabled: No
MQTT log enabled: No
Webserial enabled: No
Bootloop protection enabled: No

------------ NETWORK ------------
Network device: Built-in Wi-Fi
Network connected: Yes
IP Address: xxx.xxx.xxx.xxx
SSID: XXX
BSSID of AP: XX:XX:XX:XX:XX:XX
ESP32 MAC address: XX:XX:XX:XX:XX:XX

------------ NETWORK SETTINGS ------------
Nuki Hub hostname: XXX
DHCP enabled: Yes
Fallback to Wi-Fi / Wi-Fi config portal disabled: No
Connect to AP with the best signal enabled: Yes
RSSI Publish interval (s): Disabled
Restart ESP32 on network disconnect enabled: Yes
Reconnect network on MQTT connection failure enabled: No
MQTT Timeout until restart (s): Disabled

------------ MQTT ------------
MQTT connected: Yes
MQTT broker address: xxx.xxx.xxx.xxx
MQTT broker port: 1883
MQTT username: ***
MQTT password: ***
MQTT lock base topic: xxx
MQTT SSL CA: Not set
MQTT SSL CRT: Not set
MQTT SSL Key: Not set

------------ BLUETOOTH ------------
Bluetooth TX power (dB): 9
Bluetooth command nr of retries: 5
Bluetooth command retry delay (ms): 100
Seconds until reboot when no BLE beacons recieved: 60

------------ QUERY / PUBLISH SETTINGS ------------
Lock/Opener state query interval (s): 3600
Publish Nuki device authorization log: Yes
Max authorization log entries to retrieve: 5
Battery state query interval (s): 7200
Most non-JSON MQTT topics disabled: No
Publish Nuki device config: Yes
Config query interval (s): 3600
Publish Keypad info: No
Keypad query interval (s): 1800
Enable Keypad control: Yes
Publish Keypad topic per entry: No
Publish Keypad codes: No
Max keypad entries to retrieve: 10
Publish timecontrol info: No
Keypad query interval (s): 1800
Enable timecontrol control: No
Publish timecontrol topic per entry: No
Max timecontrol entries to retrieve: 10

------------ HOME ASSISTANT ------------
Home Assistant auto discovery enabled: Yes
Home Assistant auto discovery topic: homeassistant/
Nuki Hub configuration URL for HA: http://xxx.xxx.xxx.xxx

------------ NUKI LOCK ------------
Lock enabled: Yes
Paired: Yes
Nuki Hub device ID: XXXXXXXXX
Nuki device ID: ***
Firmware version: 3.10.7
Hardware version: 4.10
Valid PIN set: Yes
Has door sensor: No
Has keypad: No
Timecontrol highest entries count: 0
Register as: Bridge

------------ HYBRID MODE ------------
Hybrid mode enabled: No

------------ NUKI LOCK ACL ------------
Lock: Allowed
Unlock: Allowed
Unlatch: Allowed
Lock N Go: Allowed
Lock N Go Unlatch: Allowed
Full Lock: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed

------------ NUKI LOCK CONFIG ACL ------------
Name: Disallowed
Latitude: Disallowed
Longitude: Disallowed
Auto Unlatch: Disallowed
Pairing enabled: Allowed
Button enabled: Allowed
LED flash enabled: Allowed
LED brightness: Allowed
Timezone offset: Disallowed
DST mode: Disallowed
Fob Action 1: Disallowed
Fob Action 2: Disallowed
Fob Action 3: Disallowed
Single Lock: Allowed
Advertising Mode: Disallowed
Timezone ID: Disallowed
Unlocked Position Offset Degrees: Disallowed
Locked Position Offset Degrees: Disallowed
Single Locked Position Offset Degrees: Disallowed
Unlocked To Locked Transition Offset Degrees: Disallowed
Lock n Go timeout: Allowed
Single button press action: Allowed
Double button press action: Allowed
Detached cylinder: Disallowed
Battery type: Disallowed
Automatic battery type detection: Disallowed
Unlatch duration: Allowed
Auto lock timeout: Disallowed
Auto unlock disabled: Disallowed
Nightmode enabled: Disallowed
Nightmode start time: Disallowed
Nightmode end time: Disallowed
Nightmode auto lock enabled: Disallowed
Nightmode auto unlock disabled: Disallowed
Nightmode immediate lock on start: Disallowed
Auto lock enabled: Disallowed
Immediate auto lock enabled: Disallowed
Auto update enabled: Disallowed

------------ NUKI OPENER ------------
Opener enabled: No

------------ GPIO ------------

```

### TO REPRODUCE
Use v9.01 and try to load the Nuki Hub webinterface on an iDevice running iOS 18 (not tested with another iOS version).

### EXPECTED BEHAVIOUR
Page loads fine as with other browsers.

### SCREENSHOTS
[image]

### ADDITIONAL CONTEXT
Did only try with one specific iOS device.

**(Please, remember to close the issue when the problem has been addressed)**

iranl, 2024-10-19:
Release 9.01 has been pulled because of instabilities.

This has already been fixed in master.

closed 2024-10-19 by iranl

bcutter, 2024-10-19:
> Release 9.01 has been pulled because of instabilities.
> 
> This has already been fixed in master.

Pardon, did not find anything in the closed issues section. So 9.02 will likely fix this I guess.

iranl, 2024-10-19:
9.01 was a pulled (from GH, not yet from the ota server) because it was too buggy and current state of master for 9.02 also probably is not there yet for using it on anything else than a dev device.

9.01 switched to a different webserver and 9.02 switched webserver again and also multiple other libraries were removed or replaced. New webserver in 9.02 should address this issue.

bcutter, 2024-10-19:
Oh I thought v9 is stable meanwhile...

iranl, 2024-10-19:
9,00 should be stable enough, which is why it is marked as the latest stable release on GH.

bcutter, 2024-10-19:
Ah now I see. I was wondering to see the 9.01 and even 9.02 releases (RSS feed of NH releases by GitHub yesterday) but not seeing them on the release page https://github.com/technyon/nuki_hub/releases (anymore).

Well I think I stay on 9.01 (upgraded from 8.XX using webflash) as everything else seems to work. Except one Lock/NH staying in "locking/unlocking" state (according to Home Assistant) for a veeeeery long time, sometimes for more than 40 seconds to report the new state (locked/unlocked) back. With 8.xx it took ~ 15 seconds at max.