technyon/nuki_hub #505: authorization/action noExistingAuthIdSet

L338ar, 2024-10-28:
### PROBLEM DESCRIPTION
It seems that I can't update authorizations using json.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [x] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [x] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [x] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [x] Device used (e.g., ESP32-S3): ESP32-S3
- [x] Nuki Hub binary firmware version number used: 9.02-master6 (11510549745.117.1) (also on  9.01 11345224811.2.1)
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Provide the output of http://nukihubIP/info: For some strange reason, /info  301-redirect me to /
```

```

### TO REPRODUCE
1. Query ```nuki/authorization/json```. get back the following:
```
[{"authId":123456,"idType":0,"enabled":1,"name":"Lee","remoteAllowed":1,"dateCreated":"2023-10-11 00:00:05","lockCount":91,"dateLastActive":"2024-10-28 17:32:53","timeLimited":0,"allowedFrom":"0000-00-00 00:00:00","allowedUntil":"0000-00-00 00:00:00","allowedWeekdays":[],"allowedFromTime":"00:00","allowedUntilTime":"00:00"},{"authId":654321,"idType":1,"enabled":1,"name":"NukiHub","remoteAllowed":1,"dateCreated":"2024-10-08 20:21:43","lockCount":15,"dateLastActive":"2024-10-27 13:48:35","timeLimited":0,"allowedFrom":"0000-00-00 00:00:00","allowedUntil":"0000-00-00 00:00:00","allowedWeekdays":[],"allowedFromTime":"00:00","allowedUntilTime":"00:00"}]
```
2. Trying to update the name using ```authorization/action``` with the following json: ```{"action": "update", "authId": "123456", "name": "test"}``` and getting back the following error ```authorization/commandResult noExistingAuthIdSet```

### EXPECTED BEHAVIOUR
The name of authId 123456 should be changed to "test"

I've test it on the latest released version 9.01 and the latest development version.

Thank you!

sjors-lemniscap, 2024-10-29:
Having similar issues on `11474751836.1501.1` - https://github.com/technyon/nuki_hub/actions/runs/11474751836

Other JSON actions (add / delete) seem to be working fine tho.

L338ar, 2024-11-03:
Hi,
Latest version (11652369316.129.1) is still having issues regarding authorization/action.
Now it no longer gives "noExistingAuthIdSet", it just does nothing. I've also noticed that authorization/json only return 4 results (instead of 4 users + nuki_hub in my case).
Thanks,
Lee.

iranl, 2024-11-04:
Pretty sure this is fixed.

Note that a couple of MQTT topics have moved in the latest 9.02 build.
The authorization action topic is now under `nukihub/lock/authorization/action` (where it previously was under `nukihub/authorization/action`) or under `nukihub/opener/authorization/action`

L338ar, 2024-11-04:
Thank you for clearing the issue around the topics.

However, I've noticed that the user list for nuki_hub only updates upon resetting my MQTT server and waiting for the hub to reconnect.
I can confirm that sending an update works properly for the 'name' and 'enabled' fields.

Additionally, I'm experiencing issues with the /lock/authorization/json topic. I have 4 users, but while the JSON lists 4 users, one of them is named "NukiHub," which I assume is the hub itself. I am uncertain about the missing user. The 'Max authorization entries' under '/advanced' is set to 5. Let me know if I can provide any information to help resolve it.

Thank you!

iranl, 2024-11-04:
> However, I've noticed that the user list for nuki_hub only updates upon resetting my MQTT server and waiting for the hub to reconnect.

Not sure what you mean by this. The `/lock/authorization/json` and `/lock/authorization/entries` (if you enabled `Publish a topic per authorization entry and create HA sensor`) topics update together with the Nuki configuration based on the `Query interval configuration` or manually by setting `/lock/query/config = 1` or by sending an action to `/lock/authorization/action`

> Additionally, I'm experiencing issues with the /lock/authorization/json topic. I have 4 users, but while the JSON lists 4 users, one of them is named "NukiHub," which I assume is the hub itself. I [...]

Can't reproduce this. I have mine set to 10 and get 10 authorizations (this will also include NukiHub itself, keypads, fobs etc.)
Are you sure you get 4 and not 5? Note that the index goes from 0 to 4 in this case and not 1 to 5.

L338ar, 2024-11-04:
There was probably a configuration error on my side regarding the topic update.

I'm getting 4 users, index 0 to 3.

For testing I've added another user via the app, and now I am getting 5 users on the topic. I can't see the new user I've added, but the missing user from before is there. Very strange...

Could you try maximizing the authorizations and verify if you're receiving all of them or -1?

iranl, 2024-11-04:
I have 10 "users" (1 app user, 4x NukiHub, 1x keypad, 4x fob) connected to my lock and have max auth users in Nuki Hub set to 10.
And I get 10 users as expected in my auth json and 10 separate entries when enabling this option in NukiHub.

You could just set `Max authorization entries` to 10 or 15. Do take note of the recommended stack sizes when upping this number and adjust accordingly.

L338ar, 2024-11-05:
`Max authorization entries` already set to 10.
For debugging I've created another user, and now I see 6 (should be 7 with NukiHub). It seems that I am always missing the last user on the list. strange.

iranl, 2024-11-05:
I am going to close this as the main issue here is fixed. If you have another problem please open a new issue and provide (USB) serial logs using a program like hterm while running a debug build (OTA update using `http://nukihubip/otadebug?`) of 9.02-beta2 or up or 9.02-master14 or up

PR #488 (2024-11-02): Check keypad codes + unicode check for user/password
## Description:

**Related issue (if applicable):** fixes #483, fixes #487, fixes #505

To-Do
- [x] Test
- [x] README update (toggle in webcfg, mqtt topics, keypad section)

## Checklist:
  - [x] The pull request is done against the latest master branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works
  - [x] I accept the [CLA](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md#contributor-license-agre [... cut]
closed 2024-11-05 by iranl