technyon/nuki_hub #519: [Request] Looking for Nuki Smart Lock Ultra owners

iranl, 2024-11-12:
We are looking for owners of the new [Nuki Smart Lock Ultra
](https://nuki.io/smart-lock-ultra/) that can confirm if NukiHub works with the new device as expected.

Testing will include confirming:
- NukiHub pairs with the device correctly
- NukiHub can query the correct device lock state
- NukiHub can query additional parameters of the device state (auth log, etc.)
- NukiHub can change the lock state as expected
- NukiHub can query and change all supported device settings
- NukiHub can query and change keypad settings and codes for keypads connected through the device
- NukiHub can query and change authorization settings
- NukiHub can query and change time control settings

If you use NukiHub with the Nuki Smart Lock Ultra please let us know if any or all of the above works as expected or not in this issue.

HolgerHees, 2024-12-16:
I ordered it right now. 

And my M5Stack Atom POE (ESP32-S3 Version) is already flashed with nuki hub and ready to be used :-)

Mincka, 2024-12-16:
@HolgerHees Did you order it from their website? It says "This product variant is currently not available" for me.
Do we know how many keys are provided with the universal cylinder? I did not find this information.

HolgerHees, 2024-12-16:
> @HolgerHees Did you order it from their website? It says "This product variant is currently not available" for me. Do we know how many keys are provided with the universal cylinder? I did not find this information.

yes, its available since today. And you get 3 keys, but you can order additional keys later with the security card.

iranl, 2024-12-19:
@reznikov: If I am correct the Nuki Ultra doesn't support the Nuki bridge. This might mean it also will not support using NukiHub in bridge mode with the Ultra (which is the default). 

Have you enabled `Hybrid mode` under `Advanced MQTT Configuration` (recommended if official MQTT is setup) and/or enabled `Lock: Nuki Bridge is running alongside Nuki Hub` under `Advanced Nuki Configuration`?

reznikov, 2024-12-20:
I'm not running a Nuki Bridge.

`Lock: Nuki Bridge is running alongside Nuki Hub` was disabled
`Enable hybrid official MQTT and Nuki Hub setup` was also disabled.

tried enabling them both and holding the button on the lock - no luck, didn't pair.

iranl, 2024-12-20:
If you have enabled these options correctly (and rebooted NukiHub) you should see the message `Pairing as app` in the MQTT logs instead of `Pairing as bridge`

reznikov, 2024-12-23:
I’ve tried both, and had both messages appear. Still doesn’t pair

technyon, 2024-12-23:
@iranl There's a chance that that UUID used for the services are different for the ultra locks, so that the app can detect that an ultra lock is being paired. If that's the case I'd expect changes to the API. With some luck it's still compatible and they just added functionality. It's a shame they don't document anything anymore.

mallorca2288, 2024-12-23:
I've got mine today and I can confirm it's not possible to pair it with Nuki Hub. Also tried enabling and disabling hybrid mode but it didn't work.

Please let me know if there is any kind of logs that I can send to help debug.

Thanks for your work!

iranl, 2024-12-23:
I am pretty certain the new UUID's are:

`keyturnerPairingServiceUUID  = "A92EE300-5501-11E4-916C-0800200C9A66"`
`keyturnerGdioUUID  = "A92EE301-5501-11E4-916C-0800200C9A66"`
`keyturnerServiceUUID  = "A92EE200-5501-11E4-916C-0800200C9A66"` (Looks like it is unchanged from regular lock)
`keyturnerUserDataUUID  = "A92EE202-5501-11E4-916C-0800200C9A66"` (Looks like it is unchanged from regular lock)

It also seems to implement secure pairing, but I am not sure if this is mandatory.

This combination probably makes this to hard to work on without:
- Having an updated Bluetooth API by Nuki
AND/OR
- Owning the device myself

Having the keyturnerServiceUUID unchanged does probably make the process pretty straightforward after the pairing is sorted.

I might be able to get my hands on one of these devices when it goes back on sale again.

I have updated the README to show the Ultra is unsupported at this time.

I am doing a **one-time only (without owning the device myself)** test build which you can find at: https://github.com/iranl/nuki_hub/actions/runs/12473517959.

You will need to manually flash the binary using esptool. This test build will not work with the regular lock.

If pairing starts but fails and usefull information is provided using serial logging I might consider a follow-up build.
Before logging make sure to enable all Nuki debug logging functions from the `/get?page=advanced` page and rebooting.
And make sure `Lock: Nuki Bridge is running alongside Nuki Hub` is enabled on the Nuki configuration page

mallorca2288, 2024-12-23:
I've flashed the binary you provided. Still unable to pair with nuki hub.

<details>
<summary>System Information</summary>
------------ NUKI HUB ------------
Version: 9.03
Build: 12473517959.1202.1
Build type: Debug
Build date: 2024-12-23
Updater version: 
Updater build: 
Updater build date: 
Uptime (min): 0
Config version: 1061194471
Last restart reason FW: NotApplicable
Last restart reason ESP: ESP_RST_POWERON: Reset due to power-on event.
Free internal heap: 44476
Total internal heap: 253620
PSRAM Available: No
Network task stack high watermark: 11056
Nuki task stack high watermark: 7644

------------ GENERAL SETTINGS ------------
Network task stack size: 12288
Nuki task stack size: 8192
Check for updates: Yes
Latest version: 
Allow update from MQTT: No
Web configurator username: Not set
Web configurator password: Not set
Web configurator enabled: Yes
Publish debug information enabled: No
MQTT log enabled: No
Webserial enabled: No
Bootloop protection enabled: No

------------ NETWORK ------------
Network device: Built-in Wi-Fi
Network connected: No

------------ NETWORK SETTINGS ------------
Nuki Hub hostname: nukihub
DHCP enabled: No
Static IP address: 192.168.8.56
Static IP subnet: 255.255.255.0
Static IP gateway: 192.168.8.1
Static IP DNS server: 1.0.0.1
RSSI Publish interval (s): 60
Find WiFi AP with strongest signal: Yes
Restart ESP32 on network disconnect enabled: No
MQTT Timeout until restart (s): 60

------------ MQTT ------------
MQTT connected: No
MQTT broker address: 
MQTT broker port: 1883
MQTT username: Not set
MQTT password: Not set
MQTT base topic: nukihub
MQTT SSL CA: Not set
MQTT SSL CRT: Not set
MQTT SSL Key: Not set

------------ BLUETOOTH ------------
Bluetooth TX power (dB): 9
Bluetooth command nr of retries: 3
Bluetooth command retry delay (ms): 100
Seconds until reboot when no BLE beacons received: 60

------------ QUERY / PUBLISH SETTINGS ------------
[... 2077 lines cut ...]
[00:10:51]Error getting data from NVSNo nuki in pairing mode foundpairing result 0Nuki lock start pairing
[00:10:51]Pairing as app
[00:10:51]E (209175) ARDUINO: nvs_get_blob len fail: bleAddress NOT_FOUND
[00:10:52]Error getting data from NVSNo nuki in pairing mode foundpairing result 0MQTT Broker not configured, aborting connection attempt.
[00:10:54]Nuki lock start pairing
[00:10:54]Pairing as app
[00:10:54]E (211909) ARDUINO: nvs_get_blob len fail: bleAddress NOT_FOUND
[00:10:57]Error getting data from NVSNo nuki in pairing mode foundpairing result 0Nuki lock start pairing
[00:10:57]Pairing as app
[00:10:57]E (214633) ARDUINO: nvs_get_blob len fail: bleAddress NOT_FOUND

</details>

> I am pretty certain the new UUID's are:
>
> keyturnerPairingServiceUUID  = "A92EE300-5501-11E4-916C-0800200C9A66"
> keyturnerGdioUUID  = "A92EE301-5501-11E4-916C-0800200C9A66"
> keyturnerServiceUUID  = [...]

I used [this app](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp) to scan for bluetooth devices and my nuki appears to have the following UUID: "a92ee200-5501-11e4-916c-0800200c9a66". I can also see UUIDs 0x1801, 0x1800, 0x180A, 0xFFF6 listed but I don't really know what that means.

> If pairing starts but fails and usefull information is provided using serial logging I might consider a follow-up build. Before logging make sure to enable all Nuki debug logging functions from the [...]

I didn't find any logging options in the advanced settings, this is what the page looks like:
[image]

iranl, 2024-12-24:
@mallorca2288: Test build was made not against the latest master by my mistake. Sorry for that. That is why you didn't find the Nuki debug logging functions at the bottom of the advanced page.

Your logs did provide some usefull insights. It seems the pairing process starts and gets pretty far along.

I also believe I have found how Nuki has made pairing secure on https://help.nuki.io/hc/en-001/articles/4407702320017-Security-code

The Ultra requires you to set a 6-digit PIN code which is needed as part of any bluetooth pairing operation.

Can you try with this build: https://github.com/iranl/nuki_hub/actions/runs/12482870076

Steps:
-Flash a release build
-Enable all Nuki debug logging functions from the /get?page=advanced page and Save
-Enter the Nuki Smart Lock Ultra 6-digit PIN on the /get?page=advanced page and Save
-Make sure `Lock: Nuki Bridge is running alongside Nuki Hub` is enabled on the Nuki configuration page and Save
-Reboot NukiHub
-Put Smart Lock Ultra in pairing mode
-Log result over serial
-Please check your logs for data you might not want to share online (BLE address, IP address etc.) before posting it.

Note that it is possible that your ESP was banned (temporarily) from connecting/pairing looking at your second log (for trying repeatedly with the default PIN of 123456). Usually easiest fix would be to restart the Nuki device by removing the batteries from the Nuki device, but I am not sure if that is even possible with the Ultra.

mallorca2288, 2024-12-24:
I've tried pairing it two times with the new build, here are the logs:

<details>

<summary>First try</summary>

```
Using 'COM7' as serial port.
Unexpected error: could not open port 'COM7': PermissionError(13, 'Acceso denegado.', None, 5)
[15:46:27]Nuki lock start pairing
[15:46:27]Pairing as app
[15:46:27]Error getting data from NVS
[15:46:27]No nuki in pairing mode found
[15:46:27]pairing result 0
[15:46:30]Nuki lock start pairing
[15:46:30]Pairing as app
[15:46:30]Error getting data from NVS
[15:46:30]No nuki in pairing mode found
[15:46:30]pairing result 0
[15:46:32]MQTT Broker not configured, aborting connection attempt.
[15:46:33]Nuki lock start pairing
[15:46:33]Pairing as app
[15:46:33]Error getting data from NVS
[15:46:33]No nuki in pairing mode found
[15:46:33]pairing result 0
[15:46:34]Found nuki in pairing state:  addr: 54:d2:72:0a:b5:7f
[15:46:35]Found nuki in pairing state:  addr: 54:d2:72:0a:b5:7f
[15:46:36]Found nuki in pairing state:  addr: 54:d2:72:0a:b5:7f
[15:46:36]Nuki lock start pairing
[15:46:36]Pairing as app
[15:46:36]Error getting data from NVS
[15:46:36]Nuki in pairing mode found
[15:46:36]connecting within: nuki
[15:46:36][NukiHub] Connect timeout 3000 ms
[15:46:36]BLE connected
[15:46:36][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: -38
[15:46:36]GDIO characteristic registered
[15:46:36]##################### REQUEST REMOTE PUBLIC KEY #########################
[15:46:36]Sending plain message: 01 00 03 00 27 a7 
[15:46:36]Command identifier: 01, CRC: a727
[15:46:36]connecting within: nuki
[15:46:36][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: -39
[15:46:36]GDIO characteristic registered
[15:46:38]MQTT Broker not configured, aborting connection attempt.
[15:46:38]Notify callback for characteristic: a92ee301-5501-11e4-916c-0800200c9a66 of length: 36
[15:46:38]Received data: 03 00 7d 8f 2f 4d 37 80 9f a8 58 23 c0 2d f6 75 0d 65 44 a3 ee 34 ed d0 a6 92 3e 57 d6 8d f8 97 4b 54 d0 fe 
[15:46:38]CRC CHECK OK
[15:46:38]Remote public key: 7d 8f 2f 4d 37 80 9f a8 58 23 c0 2d f6 75 0d 65 44 a3 ee 34 ed d0 a6 92 3e 57 d6 8d f8 97 4b 54 
[15:46:38]##################### SEND CLIENT PUBLIC KEY #########################
[15:46:38]Sending plain message: 03 00 53 32 61 a0 d3 c5 8b dd 00 92 30 7b 48 2f 42 ed 9d 0f ec f5 ba 55 23 96 b1 ae 78 9f f3 cf c1 49 53 9f 
[15:46:38]Command identifier: 03, CRC: 9f53
[15:46:38]connecting within: nuki
[15:46:38][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: -49
[15:46:38]GDIO characteristic registered
[15:46:39]Notify callback for characteristic: a92ee301-5501-11e4-916c-0800200c9a66 of length: 36
[15:46:39]##################### CALCULATE DH SHARED KEY s ############################################## CALCULATE DH SHARED KEY s #########################Received data: 04 00 1c a9 50 2b b8 09 f0 21 27 d4 30 ae fb 1e b4 b4 4d 42 40 cd 82 2b 2f 9a 1b e9 a8 86 1f 81 4a 75 3b 65 
[15:46:39]CRC CHECK OK
[15:46:39]
[15:46:39]Challenge: 1c a9 50 2b b8 09 f0 21 27 d4 30 ae fb 1e b4 b4 4d 42 40 cd 82 2b 2f 9a 1b e9 a8 86 1f 81 4a 75 
[15:46:39]Shared key s: d3 86 a7 7e cf 3b 09 d4 47 98 04 f9 17 e7 e7 ab a7 54 ea 2d e4 b4 59 3d f3 ee d1 2f b9 cf d2 40 
[15:46:39]##################### DERIVE LONG TERM SHARED SECRET KEY k #########################
[15:46:39]Secret key k: fd 81 50 52 83 9c 91 f3 f1 fd e4 51 b5 2d b1 a0 b7 e7 5f bf 87 b5 a7 d1 03 07 01 1c 51 c0 6e 2e 
[15:46:39]##################### CALCULATE/VERIFY AUTHENTICATOR #########################
[15:46:39]Concatenated data r: 53 32 61 a0 d3 c5 8b dd 00 92 30 7b 48 2f 42 ed 9d 0f ec f5 ba 55 23 96 b1 ae 78 9f f3 cf c1 49 7d 8f 2f 4d 37 80 9f a8 58 23 c0 2d f6 75 0d 65 44 a3 ee 34 ed d0 a6 92 3e 57 d6 8d f8 97 4b 54 1c a9 50 2b b8 09 f0 21 27 d4 30 ae fb 1e b4 b4 4d 42 40 cd 82 2b 2f 9a 1b e9 a8 86 1f 81 4a 75 
[15:46:39]HMAC 256 result: df 2f 35 dd e9 01 90 e4 8c 54 a4 1c 26 57 f0 b3 fb e3 68 d1 50 b7 e7 8d 66 d3 31 e5 b7 f7 b7 22 
[15:46:39]##################### SEND AUTHENTICATOR #########################
[15:46:39]Sending plain message: 05 00 df 2f 35 dd e9 01 90 e4 8c 54 a4 1c 26 57 f0 b3 fb e3 68 d1 50 b7 e7 8d 66 d3 31 e5 b7 f7 b7 22 63 7b 
[15:46:39]Command identifier: 05, CRC: 7b63
[15:46:39]connecting within: nuki
[15:46:39][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: -45
[... 165 lines cut ...]
[15:48:15]CRC CHECK OK
[15:48:15]UNKNOWN RETURN COMMAND: 004c
[15:48:19]MQTT Broker not configured, aborting connection attempt.
[15:48:25]MQTT Broker not configured, aborting connection attempt.
[15:48:31]MQTT Broker not configured, aborting connection attempt.
[15:48:37]MQTT Broker not configured, aborting connection attempt.
[15:48:40]Pairing timeout
[15:48:41]pairing result 2
[15:48:43]MQTT Broker not configured, aborting connection attempt.
[15:48:43]Nuki lock start pairing
[15:48:43]Pairing as app
[15:48:43]Error getting data from NVS
[15:48:43]No nuki in pairing mode found
[15:48:43]pairing result 0
[15:48:46]Nuki lock start pairing
[15:48:46]Pairing as app
[15:48:46]Error getting data from NVS
[15:48:46]No nuki in pairing mode found
[15:48:46]pairing result 0

```

</details>

Thanks for your time!

iranl, 2024-12-24:
Try number 3: https://github.com/iranl/nuki_hub/actions/runs/12483995128 (still building, should be up by 16:30 UTC)

Same steps.

This version will try to force a secure connection from the client as your logs show the passkey is never used.
It will also publish hex data of the unknown return command (which I guess is an error log of the failed pairing)

mallorca2288, 2024-12-24:
Logs with last build:

```
Using 'COM7' as serial port.
[17:42:12]Wi-Fi connected
[17:42:13]IP: 192.168.8.56
[17:42:13]MQTT Broker not configured, aborting connection attempt.
[17:42:14]Found nuki in pairing state:  addr: 54:d2:72:0a:b5:7f
[17:42:15]Found nuki in pairing state:  addr: 54:d2:72:0a:b5:7f
[17:42:15]Found nuki in pairing state:  addr: 54:d2:72:0a:b5:7f
[17:42:15]Found nuki in pairing state:  addr: 54:d2:72:0a:b5:7f
[17:42:15]Found nuki in pairing state:  addr: 54:d2:72:0a:b5:7f
[17:42:16]Found nuki in pairing state:  addr: 54:d2:72:0a:b5:7f
[17:42:16]Nuki lock start pairing
[17:42:16]Pairing as app
[17:42:16]Error getting data from NVS
[17:42:16]Nuki in pairing mode found
[17:42:16]connecting within: nuki
[17:42:16][NukiHub] Connect timeout 3000 ms
[17:42:16]BLE connected
[17:42:16][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: -66
[17:42:17]GDIO characteristic registered
[17:42:17]##################### REQUEST REMOTE PUBLIC KEY #########################
[17:42:17]Sending plain message: 01 00 03 00 27 a7 
[17:42:17]Command identifier: 01, CRC: a727
[17:42:17]connecting within: nuki
[17:42:17][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: -50
[17:42:17]GDIO characteristic registered
[17:42:19]Notify callback for characteristic: a92ee301-5501-11e4-916c-0800200c9a66 of length: 36
[17:42:19]Received data: 03 00 82 d2 98 7d e7 50 94 a9 95 63 fc 10 77 fe 5f 28 9a ad c1 44 d1 c5 51 be 2b 83 b9 0e dc 85 e3 48 2a a9 
[17:42:19]CRC CHECK OK
[17:42:19]Remote public key: 82 d2 98 7d e7 50 94 a9 95 63 fc 10 77 fe 5f 28 9a ad c1 44 d1 c5 51 be 2b 83 b9 0e dc 85 e3 48 
[17:42:19]##################### SEND CLIENT PUBLIC KEY #########################
[17:42:19]Sending plain message: 03 00 ed 21 e1 01 69 d6 50 6c b6 8c bc 2d 19 53 59 95 b8 f9 14 97 f7 c5 a7 69 01 6f fb 70 20 c9 fb 28 4f c8 
[17:42:19]Command identifier: 03, CRC: c84f
[17:42:19]connecting within: nuki
[17:42:19][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: -48
[17:42:19]GDIO characteristic registered
[17:42:19]Notify callback for characteristic: a92ee301-5501-11e4-916c-0800200c9a66 of length: 36
[17:42:19]##################### CALCULATE DH SHARED KEY s ############################################## CALCULATE DH SHARED KEY s #########################Received data: 04 00 69 39 30 34 71 8b 95 f7 96 6e ef af 83 b9 72 07 13 4d d7 02 4d de 68 29 66 61 f8 5d 83 16 a0 be 24 c2 
[17:42:19]CRC CHECK OK
[17:42:19]
[17:42:19]Challenge: 69 39 30 34 71 8b 95 f7 96 6e ef af 83 b9 72 07 13 4d d7 02 4d de 68 29 66 61 f8 5d 83 16 a0 be 
[17:42:19]Shared key s: 9c 23 c9 f8 23 13 a3 96 d5 30 8f 85 db 0a 24 b5 95 0d c7 8f 49 3c 33 a2 94 3d 92 5a d2 ee 56 39 
[17:42:19]##################### DERIVE LONG TERM SHARED SECRET KEY k #########################
[17:42:19]Secret key k: 3c f4 10 69 cb ed 58 75 da a4 a1 17 9a 62 e4 08 d8 ca 2e 46 ca 15 ef 45 1b c3 91 e8 d1 59 ea e3 
[17:42:19]##################### CALCULATE/VERIFY AUTHENTICATOR #########################
[17:42:19]Concatenated data r: ed 21 e1 01 69 d6 50 6c b6 8c bc 2d 19 53 59 95 b8 f9 14 97 f7 c5 a7 69 01 6f fb 70 20 c9 fb 28 82 d2 98 7d e7 50 94 a9 95 63 fc 10 77 fe 5f 28 9a ad c1 44 d1 c5 51 be 2b 83 b9 0e dc 85 e3 48 69 39 30 34 71 8b 95 f7 96 6e ef af 83 b9 72 07 13 4d d7 02 4d de 68 29 66 61 f8 5d 83 16 a0 be 
[17:42:19]HMAC 256 result: 79 1b a2 0f 67 75 69 ac cf a3 c2 67 45 ec b9 52 dd cf ed 1d e7 8b 0b 70 1c 9c 29 8a ff 21 33 9a 
[17:42:19]##################### SEND AUTHENTICATOR #########################
[17:42:19]Sending plain message: 05 00 79 1b a2 0f 67 75 69 ac cf a3 c2 67 45 ec b9 52 dd cf ed 1d e7 8b 0b 70 1c 9c 29 8a ff 21 33 9a 5b 42 
[17:42:19]Command identifier: 05, CRC: 425b
[17:42:19]connecting within: nuki
[17:42:19][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: -51
[17:42:19]GDIO characteristic registered
[17:42:19]MQTT Broker not configured, aborting connection attempt.
[17:42:21]Notify callback for characteristic: a92ee301-5501-11e4-916c-0800200c9a66 of length: 5
[17:42:21]Received data: 4c 00 01 71 b4 
[17:42:21]CRC CHECK OK
[17:42:21]UNKNOWN RETURN COMMAND: 004c
[17:42:21]Unknown command: 01 
[17:42:25]MQTT Broker not configured, aborting connection attempt.
[17:42:31]MQTT Broker not configured, aborting connection attempt.
[17:42:37]MQTT Broker not configured, aborting connection attempt.
[17:42:43]MQTT Broker not configured, aborting connection attempt.
[17:42:47]Pairing timeout
[17:42:47]pairing result 2
[17:42:49]MQTT Broker not configured, aborting connection attempt.
[17:42:49]Nuki lock start pairing
[17:42:49]Pairing as app
[17:42:49]Error getting data from NVS
[17:42:49]No nuki in pairing mode found
[17:42:49]pairing result 0
[17:42:52]Nuki lock start pairing
[17:42:52]Pairing as app
[17:42:52]Error getting data from NVS
[17:42:52]No nuki in pairing mode found
[17:42:52]pairing result 0

```

I'm using a temporary pin code so no problem if it shows in the logs, I'll change it later.

iranl, 2024-12-24:
Try 3 still didn't secure the connection using the PIN. Turns out I didn't add it in the right place.

Try 4: https://github.com/iranl/nuki_hub/actions/runs/12484546616

We are looking for the messages `Securing Connection`, `Passkey Entry` and `Done securing Connection`

mallorca2288, 2024-12-25:
Logs with [this build](https://github.com/iranl/nuki_hub/actions/runs/12484546616):

<details>

```

Using 'COM7' as serial port.
Showing logs:
[01:45:59]IP: 192.168.8.56
[01:45:59]MQTT Broker not configured, aborting connection attempt.
[01:46:01]Nuki lock start pairing
[01:46:01]Pairing as app
[01:46:01]Error getting data from NVS
[01:46:01]No nuki in pairing mode found
[01:46:01]pairing result 0
[01:46:04]Nuki lock start pairing
[01:46:04]Pairing as app
[01:46:04]Error getting data from NVS
[01:46:04]No nuki in pairing mode found
[01:46:04]pairing result 0
[01:46:05]MQTT Broker not configured, aborting connection attempt.
[01:46:06]Nuki lock start pairing
[01:46:06]Pairing as app
[01:46:06]Error getting data from NVS
[01:46:06]No nuki in pairing mode found
[01:46:06]pairing result 0
[01:46:09]Nuki lock start pairing
[01:46:09]Pairing as app
[01:46:09]Error getting data from NVS
[01:46:09]No nuki in pairing mode found
[01:46:09]pairing result 0
[01:46:11]MQTT Broker not configured, aborting connection attempt.
[01:46:11]Found nuki in pairing state:  addr: 54:d2:72:0a:b5:7f
[01:46:11]Found nuki in pairing state:  addr: 54:d2:72:0a:b5:7f
[01:46:12]Found nuki in pairing state:  addr: 54:d2:72:0a:b5:7f
[01:46:12]Found nuki in pairing state:  addr: 54:d2:72:0a:b5:7f
[01:46:12]Nuki lock start pairing
[01:46:12]Pairing as app
[01:46:12]Error getting data from NVS
[01:46:12]Nuki in pairing mode found
[01:46:12]connecting within: nuki
[01:46:12][NukiHub] Connect timeout 3000 ms
[01:46:13]BLE connected
[01:46:13][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: -42
[01:46:13]Securing Connection
[01:46:13]Done securing Connection
[01:46:14]GDIO characteristic registered
[01:46:14]##################### REQUEST REMOTE PUBLIC KEY #########################
[01:46:14]Sending plain message: 01 00 03 00 27 a7 
[01:46:14]Command identifier: 01, CRC: a727
[01:46:14]connecting within: nuki
[01:46:14][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: -49
[01:46:14]Securing Connection
[01:46:14]Done securing Connection
[01:46:14]GDIO characteristic registered
[01:46:16]Notify callback for characteristic: a92ee301-5501-11e4-916c-0800200c9a66 of length: 36
[01:46:16]Received data: 03 00 5b 0c 2d eb cd 29 ad f6 b0 80 94 a5 a3 32 24 85 c0 a4 54 fe 08 da e6 b4 09 17 e3 cc b1 e2 be 61 95 07 
[01:46:16]CRC CHECK OK
[01:46:16]Remote public key: 5b 0c 2d eb cd 29 ad f6 b0 80 94 a5 a3 32 24 85 c0 a4 54 fe 08 da e6 b4 09 17 e3 cc b1 e2 be 61 
[01:46:16]##################### SEND CLIENT PUBLIC KEY #########################
[01:46:16]Sending plain message: 03 00 b8 09 a5 c5 63 a4 47 c5 f5 79 ec 07 6d df 4c da 14 14 1f c2 cf d6 1a bf 12 f1 f3 65 f6 b4 7a 25 17 e7 
[01:46:16]Command identifier: 03, CRC: e717
[01:46:16]connecting within: nuki
[01:46:16][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: -47
[01:46:16]Securing Connection
[01:46:17]MQTT Broker not configured, aborting connection attempt.
[01:46:17]Done securing Connection
[01:46:17]GDIO characteristic registered
[01:46:17]Notify callback for characteristic: a92ee301-5501-11e4-916c-0800200c9a66 of length: 36
[01:46:17]##################### CALCULATE DH SHARED KEY s ############################################## CALCULATE DH SHARED KEY s #########################Received data: 04 00 43 d3 e0 66 bb fc 0c 7e e2 97 79 59 fb 17 53 af 0d f2 42 48 43 71 f0 d4 12 f4 40 92 6a 7b 7c 4c 82 25 
[... 586 lines cut ...]
[01:55:10]Securing Connection
[01:55:10]BLE disconnected
[01:55:10]Done securing Connection
[01:55:10]Unable to get keyturner pairing service
[01:55:10][NukiHub] Failed to connect on registering GDIO
[01:55:10]BLE connected
[01:55:10][NukiHub] Reconnect failed
[01:55:10][NukiHub] Reconnect failed
[01:55:10][NukiHub] Reconnect failed
[01:55:10]pairing result 0
[01:55:13]Nuki lock start pairing
[01:55:13]Pairing as app
[01:55:13]Error getting data from NVS
[01:55:13]No nuki in pairing mode found
[01:55:13]pairing result 0
[01:55:15]MQTT Broker not configured, aborting connection attempt.
[01:55:16]Nuki lock start pairing
[01:55:16]Pairing as app
[01:55:16]Error getting data from NVS
[01:55:16]No nuki in pairing mode found
[01:55:16]pairing result 0

```

</details>

iranl, 2024-12-25:
Ok. So now it tries to create a secureConnection but the correct PIN/passkey is still not used/requested. 

I'll think on this over the holidays and maybe try testing this with a custom BLE server mimicking a SmartLock Ultra before creating a new build.

@mallorca2288 Thanks for all the work up to this point. Your input makes this work possible at all without help from Nuki and/or owning a device myself.

Happy holidays

iranl, 2024-12-26:
I have another build up at: https://github.com/iranl/nuki_hub/actions/runs/12507302612

I changed a few things that might (emphasis on might) help.

And especially the debug builds have logging turned up to the max, which might help in further narrowing the remaining issues down.

mallorca2288, 2024-12-26:
Logs with the latest build :wink:

<details>

```
Using 'COM7' as serial port.
Showing logs:
[00:46:16]Wi-Fi connected
[00:46:17]IP: 192.168.8.56
[00:46:17]MQTT Broker not configured, aborting connection attempt.
[00:46:17]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[00:46:17]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[00:46:17]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[00:46:17]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[00:46:18]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[00:46:18]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[00:46:19]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[00:46:19]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[00:46:19]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[00:46:19]Nuki lock start pairing
[00:46:19]Pairing as app
[00:46:19]Error getting data from NVS
[00:46:19]Nuki in pairing mode found
[00:46:19]connecting within: nuki
[00:46:19][NukiHub] Connect timeout 3000 ms
[00:46:20]BLE connected
[00:46:20][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: -45
[00:46:20]Securing Connection
[00:46:20]Done securing Connection
[00:46:20]Unable to get keyturner pairing service
[00:46:20]Disconnecting BLE
[00:46:20][NukiHub] Failed to connect on registering GDIO
[00:46:20][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: 0
[00:46:20]Securing Connection
[00:46:20]BLE disconnected
[00:46:20]Done securing Connection
[00:46:20]Unable to get keyturner pairing service
[00:46:20][NukiHub] Failed to connect on registering GDIO
[00:46:21]BLE connected
[00:46:21][NukiHub] Reconnect success
[00:46:21][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: -37
[00:46:21]Securing Connection
[00:46:22]Done securing Connection
[00:46:22]Unable to get keyturner pairing service
[00:46:22]Disconnecting BLE
[00:46:22][NukiHub] Failed to connect on registering GDIO
[00:46:22][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: 0
[00:46:22]Securing Connection
[00:46:22]BLE disconnected
[00:46:22]Done securing Connection
[00:46:22]Unable to get keyturner pairing service
[00:46:22][NukiHub] Failed to connect on registering GDIO
[00:46:22]BLE connected
[00:46:22][NukiHub] Reconnect success
[00:46:22][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: -36
[00:46:22]Securing Connection
[00:46:23]Done securing Connection
[00:46:23]Unable to get keyturner pairing service
[00:46:23]Disconnecting BLE
[00:46:23][NukiHub] Failed to connect on registering GDIO
[00:46:23]MQTT Broker not configured, aborting connection attempt.
[00:46:23]pairing result 0
[00:46:23]BLE disconnected
[00:46:23]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[00:46:23]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[00:46:23]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[00:46:24]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[00:46:24]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[00:46:25]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[00:46:25]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[... 445 lines cut ...]
[00:47:08][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: 0
[00:47:08]Securing Connection
[00:47:08]BLE disconnected
[00:47:08]Done securing Connection
[00:47:08]Unable to get keyturner pairing service
[00:47:08][NukiHub] Failed to connect on registering GDIO
[00:47:08]BLE connected
[00:47:08][NukiHub] Reconnect success
[00:47:08][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: -40
[00:47:08]Securing Connection
[00:47:09]Done securing Connection
[00:47:09]Unable to get keyturner pairing service
[00:47:09]Disconnecting BLE
[00:47:09][NukiHub] Failed to connect on registering GDIO
[00:47:09]pairing result 0
[00:47:09]BLE disconnected
[00:47:11]MQTT Broker not configured, aborting connection attempt.
[00:47:12]Nuki lock start pairing
[00:47:12]Pairing as app
[00:47:12]Error getting data from NVS
[00:47:12]No nuki in pairing mode found
[00:47:12]pairing result 0
```

</details>

iranl, 2024-12-28:
So trying to force the secure connection probably fails and it actually gets less far than on the unsecured connection.
I'll try to make a build with much more bluetooth logging to really see what is going on on a lower level

iranl, 2025-01-12:
Nuki released a new version of the Bluetooth API documentation, where they describe the changes for the Ultra, a couple of days ago.

It turns out the direction I was working towards in the last builds was completely wrong.

Although the documentation is pretty bare bones it should be possible to get the Ultra up and running in a few days. Will update here when a new test build is available.

iranl, 2025-01-12:
New build for testing: https://github.com/iranl/nuki_hub/actions/runs/12737209941

Steps as before

mallorca2288, 2025-01-12:
Logs with the new build:

<details>

```
Using 'COM7' as serial port.
Showing logs:
[22:45:41]Nuki lock start pairing
[22:45:41]Pairing as app
[22:45:41]Error getting data from NVS
[22:45:41]No nuki in pairing mode found
[22:45:41]pairing result 0
[22:45:43]MQTT Broker not configured, aborting connection attempt.
[22:45:44][WiFi-event] event: 113
[22:45:44]Disconnected from WiFi access point
[22:45:44]Wi-Fi disconnected
[22:45:44]ets Jun  8 2016 00:22:57
[22:45:44]
[22:45:44]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[22:45:44]configsip: 0, SPIWP:0xee
[22:45:44]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[22:45:44]mode:DIO, clock div:2
[22:45:44]load:0x3fff0030,len:4644
[22:45:44]load:0x40078000,len:15312
[22:45:44]ho 0 tail 12 room 4
[22:45:44]load:0x40080400,len:4
[22:45:44]load:0x40080404,len:3220
[22:45:44]entry 0x400805a4
[22:45:44]Last config version: 908
[22:45:44]Current config version: 908
[22:45:44]Partition size: 2555904
[22:45:44]Partition subtype: 16
[22:45:44]1
[22:45:44]Nuki Hub version 9.08
[22:45:44]Nuki Hub build 12737209941.1318.1
[22:45:44]Load GPIO configuration
[22:45:45]IP configuration: IP address: 192.168.8.56, Subnet: 255.255.255.0, Gateway: 192.168.8.1, DNS: 1.0.0.1
[22:45:45]Hardware detect: 1
[22:45:45]MQTT without TLS.
[22:45:45]Network device: Built-in Wi-Fi
[22:45:45]Attempting to connect to saved SSID Foquitas
[22:45:45][WiFi-event] event: 101
[22:45:45]WiFi interface ready
[22:45:45]Wi-Fi async scan started
[22:45:45]Host name: nukihub
[22:45:45]MQTT Broker: :1883
[22:45:45][WiFi-event] event: 110
[22:45:45]WiFi client started
[22:45:45]Nuki Lock enabled
[22:45:45]Device id lock: 2939628044
[22:45:45]Error getting data from NVS
[22:45:45]Lock state interval: 1800 | Battery interval: 1800 | Publish auth data: yes
[22:45:45]Nuki Opener disabled
[22:45:45]Not running on PSRAM enabled device
[22:45:47][WiFi-event] event: 102
[22:45:47]Completed scan for access points

[22:45:50]Wi-Fi connected
[22:45:50][WiFi-event] event: 115
[22:45:50]Obtained IP address: 192.168.8.56
[22:45:50]Wi-Fi connected
[22:45:51]IP: 192.168.8.56
[22:45:51]MQTT Broker not configured, aborting connection attempt.
[22:45:52]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[22:45:52][http] connection #43 connected from 192.168.8.56
[22:45:52]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[22:45:53]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[22:45:53]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[22:45:53]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[22:45:53]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[... 585 lines cut ...]
[22:46:55]Error getting data from NVS
[22:46:55]No nuki in pairing mode found
[22:46:55]pairing result 0
[22:46:57]MQTT Broker not configured, aborting connection attempt.
[22:46:58]Nuki lock start pairing
[22:46:58]Pairing as app
[22:46:58]Error getting data from NVS
[22:46:58]No nuki in pairing mode found
[22:46:58]pairing result 0
[22:47:01]Nuki lock start pairing
[22:47:01]Pairing as app
[22:47:01]Error getting data from NVS
[22:47:01]No nuki in pairing mode found
[22:47:01]pairing result 0
[22:47:03]MQTT Broker not configured, aborting connection attempt.
[22:47:04]Nuki lock start pairing
[22:47:04]Pairing as app
[22:47:04]Error getting data from NVS
[22:47:04]No nuki in pairing mode found
[22:47:04]pairing result 0
```

</details>

Thanks for all your work iranl!

iranl, 2025-01-12:
@mallorca2288: Thank you for all the debugging. Release build is fine, no need to use the debug ones.

Sorry, made a dumb mistake in the previous build again. As we should start approching a working build I tried to create a build that will work on any lock (1-4 and Ultra) but missed something that was needed to make this work.

New build: https://github.com/iranl/nuki_hub/actions/runs/12737467338

mallorca2288, 2025-01-12:
New logs 😉 

<details>

```
Using 'COM7' as serial port.
Showing logs:
[23:15:54]8 2016 00:22:57
[23:15:54]
[23:15:54]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[23:15:54]configsip: 0, SPIWP:0xee
[23:15:54]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[23:15:54]mode:DIO, clock div:2
[23:15:54]load:0x3fff0030,len:4644
[23:15:54]load:0x40078000,len:15312
[23:15:54]ho 0 tail 12 room 4
[23:15:54]load:0x40080400,len:4
[23:15:54]load:0x40080404,len:3220
[23:15:54]entry 0x400805a4
[23:15:55]Last config version: 908
[23:15:55]Current config version: 908
[23:15:55]Partition size: 2555904
[23:15:55]Partition subtype: 16
[23:15:55]0
[23:15:55]Nuki Hub version 9.08
[23:15:55]Nuki Hub build 12737467338.1321.1
[23:15:55]Load GPIO configuration
[23:15:55]IP configuration: IP address: 192.168.8.56, Subnet: 255.255.255.0, Gateway: 192.168.8.1, DNS: 1.0.0.1
[23:15:55]Hardware detect: 1
[23:15:55]MQTT without TLS.
[23:15:55]Network device: Built-in Wi-Fi
[23:15:55]Attempting to connect to saved SSID Foquitas
[23:15:55][WiFi-event] event: 101
[23:15:55]WiFi interface ready
[23:15:55]Wi-Fi async scan started
[23:15:55]Host name: nukihub
[23:15:55]MQTT Broker: :1883
[23:15:55][WiFi-event] event: 110
[23:15:55]WiFi client started
[23:15:55]Nuki Lock enabled
[23:15:55]Device id lock: 2939628044
[23:15:55]Error getting data from NVS
[23:15:55]Lock state interval: 1800 | Battery interval: 1800 | Publish auth data: yes
[23:15:55]Nuki Opener disabled
[23:15:55]Not running on PSRAM enabled device
[23:15:58][WiFi-event] event: 102
[23:15:58]Completed scan for access points

[23:16:01]WiFi connecting.........................
[23:16:01][WiFi-event] event: 112
[23:16:01]Connected to access point
[23:16:01]Wi-Fi connected
[23:16:01][WiFi-event] event: 115
[23:16:01]Obtained IP address: 192.168.8.56
[23:16:01]Wi-Fi connected
[23:16:01]IP: 192.168.8.56
[23:16:01]MQTT Broker not configured, aborting connection attempt.
[23:16:02][http] connection #43 connected from 192.168.8.56
[23:16:03]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[23:16:03]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[23:16:04]Nuki lock start pairing
[23:16:04]Pairing as app
[23:16:04]Error getting data from NVS
[23:16:04]Nuki in pairing mode found
[23:16:04]connecting within: nuki
[23:16:04][NukiHub] Connect timeout 3000 ms
[23:16:05]BLE connected
[23:16:05][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: -64
[23:16:05]Unable to get GDIO characteristic
[23:16:05]Disconnecting BLE
[... 437 lines cut ...]
[23:17:00]Disconnecting BLE
[23:17:00].
[23:17:00]BLE disconnected
[23:17:01][NukiHub] Failed to connect on registering GDIO
[23:17:01]pairing result 0
[23:17:01]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[23:17:01]MQTT Broker not configured, aborting connection attempt.
[23:17:01]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[23:17:02]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[23:17:02]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[23:17:03]Nuki lock start pairing
[23:17:03]Pairing as app
[23:17:03]Error getting data from NVS
[23:17:03]Nuki in pairing mode found
[23:17:03]connecting within: nuki
[23:17:03][NukiHub] Reconnect failed
[23:17:04][NukiHub] Reconnect failed
[23:17:05][NukiHub] Reconnect failed
[23:17:06][NukiHub] Reconnect failed
[23:17:07][NukiHub] Reconnect failed
[23:17:07]pairing result 0
[23:17:07]MQTT Broker not configured, aborting connection attempt.
```

</details>

iranl, 2025-01-12:
Note to self: Remember to not assume Nuki is capable of creating API documentation that works.

Next build: https://github.com/iranl/nuki_hub/actions/runs/12737644894 (Last build for today)

mallorca2288, 2025-01-12:
<details>

```
Using 'COM7' as serial port.
Showing logs:
[23:40:25]WiFi connecting..........................
[23:40:25][WiFi-event] event: 112
[23:40:25]Connected to access point
[23:40:25]Wi-Fi connected
[23:40:25][WiFi-event] event: 115
[23:40:25]Obtained IP address: 192.168.8.56
[23:40:25]Wi-Fi connected
[23:40:25]IP: 192.168.8.56
[23:40:26]MQTT Broker not configured, aborting connection attempt.
[23:40:26][http] connection #43 connected from 192.168.8.56
[23:40:26][http] connection #44 connected from 192.168.8.56
[23:40:27][http] connection #45 connected from 192.168.8.56
[23:40:28]Nuki lock start pairing
[23:40:28]Pairing as app
[23:40:28]Error getting data from NVS
[23:40:28]No nuki in pairing mode found
[23:40:28]pairing result 0
[23:40:30]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[23:40:31]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[23:40:31]Nuki lock start pairing
[23:40:31]Pairing as app
[23:40:31]Error getting data from NVS
[23:40:31]Nuki in pairing mode found
[23:40:31]connecting within: nuki
[23:40:31][NukiHub] Connect timeout 3000 ms
[23:40:31]BLE connected
[23:40:31][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: -68
[23:40:31]Unable to get GDIO characteristic
[23:40:31]Disconnecting BLE
[23:40:31].
[23:40:31]BLE disconnected
[23:40:31][NukiHub] Failed to connect on registering GDIO
[23:40:31]MQTT Broker not configured, aborting connection attempt.
[23:40:32]BLE connected
[23:40:32][NukiHub] Reconnect success
[23:40:32][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: -72
[23:40:32]Unable to get GDIO characteristic
[23:40:32]Disconnecting BLE
[23:40:32].
[23:40:32]BLE disconnected
[23:40:32][http] connection #45 closed from 192.168.8.56
[23:40:32][NukiHub] Failed to connect on registering GDIO
[23:40:32][NukiHub] Reconnect failed
[23:40:34][NukiHub] Reconnect failed
[23:40:34][NukiHub] Reconnect failed
[23:40:34]pairing result 0
[23:40:34]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[23:40:34]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[23:40:35]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[23:40:35]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[23:40:35]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[23:40:36]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[23:40:36]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[23:40:36]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[23:40:37]Nuki lock start pairing
[23:40:37]Pairing as app
[23:40:37]Error getting data from NVS
[23:40:37]Nuki in pairing mode found
[23:40:37]connecting within: nuki
[23:40:37]BLE connected
[23:40:37][NukiHub] Reconnect success
[23:40:37][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: -69
[23:40:37]Unable to get GDIO characteristic
[23:40:37]Disconnecting BLE
[23:40:37].
[... 394 lines cut ...]
[23:41:30]Unable to get GDIO characteristic
[23:41:30]Disconnecting BLE
[23:41:30].
[23:41:30]BLE disconnected
[23:41:30][NukiHub] Failed to connect on registering GDIO
[23:41:30]BLE connected
[23:41:30][NukiHub] Reconnect success
[23:41:30][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: -48
[23:41:30]Unable to get GDIO characteristic
[23:41:30]Disconnecting BLE
[23:41:30].
[23:41:30]BLE disconnected
[23:41:30][NukiHub] Failed to connect on registering GDIO
[23:41:30]pairing result 0
[23:41:32]MQTT Broker not configured, aborting connection attempt.
[23:41:33]Nuki lock start pairing
[23:41:33]Pairing as app
[23:41:33]Error getting data from NVS
[23:41:33]No nuki in pairing mode found
[23:41:33]pairing result 0
```

</details>

Appreciate your work! I'll continue to try the new versions that you release whenever you have free time.

iranl, 2025-01-12:
Ok last one today ;)

https://github.com/iranl/nuki_hub/actions/runs/12737764572 (still building, takes about 7 mins)

mallorca2288, 2025-01-12:
I don't know what it all means, but I think the pairing process is moving forward now.

<details>

```
Using 'COM7' as serial port.
Showing logs:
[23:58:09] Jun  8 2016 00:22:57
[23:58:09]
[23:58:09]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[23:58:09]configsip: 0, SPIWP:0xee
[23:58:09]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[23:58:09]mode:DIO, clock div:2
[23:58:09]load:0x3fff0030,len:4644
[23:58:09]load:0x40078000,len:15312
[23:58:09]ho 0 tail 12 room 4
[23:58:09]load:0x40080400,len:4
[23:58:09]load:0x40080404,len:3220
[23:58:09]entry 0x400805a4
[23:58:09]Last config version: 908
[23:58:09]Current config version: 908
[23:58:09]Partition size: 2555904
[23:58:09]Partition subtype: 16
[23:58:09]0
[23:58:09]Nuki Hub version 9.08
[23:58:09]Nuki Hub build 12737764572.1323.1
[23:58:09]Load GPIO configuration
[23:58:09]IP configuration: IP address: 192.168.8.56, Subnet: 255.255.255.0, Gateway: 192.168.8.1, DNS: 1.0.0.1
[23:58:09]Hardware detect: 1
[23:58:09]MQTT without TLS.
[23:58:09]Network device: Built-in Wi-Fi
[23:58:09]Attempting to connect to saved SSID Foquitas
[23:58:09][WiFi-event] event: 101
[23:58:09]WiFi interface ready
[23:58:09]Wi-Fi async scan started
[23:58:09]Host name: nukihub
[23:58:09]MQTT Broker: :1883
[23:58:09][WiFi-event] event: 110
[23:58:09]WiFi client started
[23:58:10]Nuki Lock enabled
[23:58:10]Device id lock: 2939628044
[23:58:10]Error getting data from NVS
[23:58:10]Lock state interval: 1800 | Battery interval: 1800 | Publish auth data: yes
[23:58:10]Nuki Opener disabled
[23:58:10]Not running on PSRAM enabled device
[23:58:12][WiFi-event] event: 102
[23:58:12]Completed scan for access points

[23:58:15]WiFi connecting.........................
[23:58:15][WiFi-event] event: 112
[23:58:15]Connected to access point
[23:58:15]Wi-Fi connected
[23:58:15][WiFi-event] event: 115
[23:58:15]Obtained IP address: 192.168.8.56
[23:58:15]Wi-Fi connected
[23:58:16][http] connection #43 connected from 192.168.8.56
[23:58:16]IP: 192.168.8.56
[23:58:16]MQTT Broker not configured, aborting connection attempt.
[23:58:18]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[23:58:18]Nuki lock start pairing
[23:58:18]Pairing as app
[23:58:18]Error getting data from NVS
[23:58:18]Nuki in pairing mode found
[23:58:18]connecting within: nuki
[23:58:18][NukiHub] Connect timeout 3000 ms
[23:58:19]BLE connected
[23:58:19][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: -69
[23:58:19]GDIO characteristic registered
[23:58:19]##################### REQUEST REMOTE PUBLIC KEY #########################
[23:58:19]Sending plain message: 01 00 03 00 27 a7 
[... 60 lines cut ...]
[23:58:52]pairing result 0
[23:58:55]Nuki lock start pairing
[23:58:55]Pairing as app
[23:58:55]Error getting data from NVS
[23:58:55]No nuki in pairing mode found
[23:58:55]pairing result 0
[23:58:57]Nuki lock start pairing
[23:58:57]Pairing as app
[23:58:57]Error getting data from NVS
[23:58:57]No nuki in pairing mode found
[23:58:57]pairing result 0
[23:58:58]MQTT Broker not configured, aborting connection attempt.
[23:59:00]Nuki lock start pairing
[23:59:00]Pairing as app
[23:59:00]Error getting data from NVS
[23:59:00]No nuki in pairing mode found
[23:59:00]pairing result 0
[23:59:03]Nuki lock start pairing
[23:59:03]Pairing as app
[23:59:03]Error getting data from NVS
[23:59:03]No nuki in pairing mode found
[23:59:03]pairing result 0
```

</details>

iranl, 2025-01-13:
Next try: https://github.com/iranl/nuki_hub/actions/runs/12754114877

mallorca2288, 2025-01-13:
Logs

<details>

```
Using 'COM7' as serial port.
Showing logs:
[20:18:35]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[20:18:35]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[20:18:35]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[20:18:35]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[20:18:35]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[20:18:36]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[20:18:36]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[20:18:36]Nuki lock start pairing
[20:18:36]Pairing as app
[20:18:36]Error getting data from NVS
[20:18:36]Nuki in pairing mode found
[20:18:36]connecting within: nuki
[20:18:36][NukiHub] Connect timeout 3000 ms
[20:18:36][http] connection #44 connected from 192.168.8.56
[20:18:37]BLE connected
[20:18:37][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: -60
[20:18:37]GDIO characteristic registered
[20:18:37]##################### REQUEST REMOTE PUBLIC KEY #########################
[20:18:37]Sending plain message: 01 00 03 00 27 a7 
[20:18:37]Command identifier: 01, CRC: a727
[20:18:37]connecting within: nuki
[20:18:37][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: -52
[20:18:37]GDIO characteristic registered
[20:18:39][http] connection #44 closed from 192.168.8.56Notify callback for characteristic: a92ee301-5501-11e4-916c-0800200c9a66 of length: 36[http] connection #44 closed from 192.168.8.56Notify callback for characteristic: a92ee301-5501-11e4-916c-0800200c9a66 of length: 36
[20:18:39]
[20:18:39][http] connection #44 connected from 192.168.8.56Received data: 03 00 44 a7 fe e7 5e 4c 0c 04 0b bf 75 c7 7c 3f fe 83 d8 86 ee fa cc 0b 71 b4 58 c5 cb fe 0d de c2 16 0d 89 MQTT recently connected, ignoring lock action.[http] connection #44 connected from 192.168.8.56Received data: 03 00 44 a7 fe e7 5e 4c 0c 04 0b bf 75 c7 7c 3f fe 83 d8 86 ee fa cc 0b 71 b4 58 c5 cb fe 0d de c2 16 0d 89 MQTT recently connected, ignoring lock action.
[20:18:39]
[20:18:39]CRC CHECK OK
[20:18:39]CRC CHECK OKtion #44 connected from 192.160b bf 75 c7 7c 3f fe 83 d8 86 ee fa cc 0b 71 b4 58 c5 cb fe 0d de c2 16 3 d8 86 ee fa cc 0b 71 b4 58 c5 cb fe 0d de c2 16 0d 89 MQTT recently connected, ignoring lock action.Remote public key: 44 a7 fe e7 5e 4c 0c 04 0b bf 75 c7 7c 3f fe 83 d8 86 ee fa cc 0b 71 b4 58 c5 cb fe 0d de c2 16 
[20:18:39]
[20:18:39]##################### SEND CLIENT PUBLIC KEY #########################
[20:18:39]MQTT recently connected, ignoring lock action.
[20:18:39]Sending plain message: 03 00 64 b0 99 6c 7d 95 0e ca 0f 8b 4b 63 0b 60 fa 4b 91 33 1d 5f cd 86 fb 9b 06 f9 ee 9e fe 31 5a 5b a2 fa 
[20:18:40]Command identifier: 03, CRC: faa2
[20:18:40]connecting within: nuki
[20:18:40][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: -54
[20:18:40]GDIO characteristic registered
[20:18:40]Notify callback for characteristic: a92ee301-5501-11e4-916c-0800200c9a66 of length: 36
[20:18:40]##################### CALCULATE DH SHARED KEY s #########################
[20:18:40]Received data: 04 00 bc 23 c7 8d 45 26 4d 52 cf 95 8b a1 86 c6 77 32 ad 65 a6 40 5f 4d 7b 2b 89 29 45 e1 48 57 17 d6 5a f2 
[20:18:40]CRC CHECK OK
[20:18:40]Challenge: bc 23 c7 8d 45 26 4d 52 cf 95 8b a1 86 c6 77 32 ad 65 a6 40 5f 4d 7b 2b 89 29 45 e1 48 57 17 d6 
[20:18:40]Shared key s: 0d 54 ba af b5 39 fc 82 94 c8 47 c7 f2 db cf b8 80 4c fe f6 94 56 13 25 45 0b d2 c8 9c 8b 2a 5d 
[20:18:40]##################### DERIVE LONG TERM SHARED SECRET KEY k #########################
[20:18:40]Secret key k: ff 41 8d 70 5c 6b 08 6a 46 bc 5d 3b c6 e2 6e 81 86 cd 6e a3 ed 30 6f 96 01 b2 68 9a ac d0 ab 18 
[20:18:40]##################### CALCULATE/VERIFY AUTHENTICATOR #########################
[20:18:40]Concatenated data r: 64 b0 99 6c 7d 95 0e ca 0f 8b 4b 63 0b 60 fa 4b 91 33 1d 5f cd 86 fb 9b 06 f9 ee 9e fe 31 5a 5b 44 a7 fe e7 5e 4c 0c 04 0b bf 75 c7 7c 3f fe 83 d8 86 ee fa cc 0b 71 b4 58 c5 cb fe 0d de c2 16 bc 23 c7 8d 45 26 4d 52 cf 95 8b a1 86 c6 77 32 ad 65 a6 40 5f 4d 7b 2b 89 29 45 e1 48 57 17 d6 
[20:18:40]HMAC 256 result: 45 2b 4c fb 8c 36 ea 7c 5f 9e 61 25 15 cc 9c ef 4f 0f 89 63 08 5d 6e c2 6a f0 65 48 78 be 78 7c 
[20:18:40]##################### SEND AUTHENTICATOR #########################
[20:18:40]Sending plain message: 05 00 45 2b 4c fb 8c 36 ea 7c 5f 9e 61 25 15 cc 9c ef 4f 0f 89 63 08 5d 6e c2 6a f0 65 48 78 be 78 7c 08 92 
[20:18:40]Command identifier: 05, CRC: 9208
[20:18:40]connecting within: nuki
[20:18:40][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: -59
[20:18:41]GDIO characteristic registered
[20:18:42]Notify callback for characteristic: a92ee301-5501-11e4-916c-0800200c9a66 of length: 5
[20:18:42]Received data: 4c 00 01 71 b4 
[20:18:42]CRC CHECK OK
[20:18:42]authorizationInfo: 01 
[20:18:42]##################### SEND AUTHORIZATION DATA (ULTRA) #########################
[20:18:42]payloadlen: 40
[20:18:42]sizeof(plainData): 46
[20:18:42]CRC: 2437
[20:18:42]Plain data with CRC: : 00 00 00 00 06 00 0c 2a 37 af 4e 75 6b 69 48 75 62 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 de 14 04 00 37 24 
[... 97 lines cut ...]
[20:18:56]MQTT recently connected, ignoring lock action.
[20:18:56]MQTT recently connected, ignoring lock action.
[20:18:56]Nuki lock start pairing
[20:18:56]Pairing as app
[20:18:56]Error getting data from NVS
[20:18:56]No nuki in pairing mode found
[20:18:56]pairing result 0
[20:18:59]Nuki lock start pairing
[20:18:59]Pairing as app
[20:18:59]Error getting data from NVS
[20:18:59]No nuki in pairing mode found
[20:18:59]pairing result 0
[20:19:02]Nuki lock start pairing
[20:19:02]Pairing as app
[20:19:02]Error getting data from NVS
[20:19:02]No nuki in pairing mode found
[20:19:02]pairing result 0
[20:19:05]Nuki lock start pairing
[20:19:05]Pairing as app
[20:19:05]Error getting data from NVS
[20:19:05]No nuki in pairing mode found
[20:19:05]pairing result 0
```

</details>

iranl, 2025-01-13:
Next one: https://github.com/iranl/nuki_hub/actions/runs/12756243693

mallorca2288, 2025-01-13:
<details>

```
Using 'COM7' as serial port.
Showing logs:
[22:26:42]IP: 192.168.8.56
[22:26:42]MQTT Broker not configured, aborting connection attempt.
[22:26:43]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[22:26:44]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[22:26:44]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[22:26:44]Nuki lock start pairing
[22:26:44]Pairing as app
[22:26:44]Error getting data from NVS
[22:26:44]Nuki in pairing mode found
[22:26:44]connecting within: nuki
[22:26:44][NukiHub] Connect timeout 3000 ms
[22:26:45]BLE connected
[22:26:45][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: -51
[22:26:45]GDIO characteristic registered
[22:26:46]##################### REQUEST REMOTE PUBLIC KEY #########################
[22:26:46]Sending plain message: 01 00 03 00 27 a7 
[22:26:46]Command identifier: 01, CRC: a727
[22:26:46]connecting within: nuki
[22:26:46][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: -53
[22:26:46]GDIO characteristic registered
[22:26:47]Notify callback for characteristic: a92ee301-5501-11e4-916c-0800200c9a66 of length: 36
[22:26:47]Received data: 03 00 13 a1 26 b6 41 1e 1c 0c cd 61 a1 f9 78 60 33 68 c1 3f 89 17 c3 ae 8e 3a 7e 6e a5 0a da c9 c0 5b a1 d4 
[22:26:47]CRC CHECK OK
[22:26:47]Remote public key: 13 a1 26 b6 41 1e 1c 0c cd 61 a1 f9 78 60 33 68 c1 3f 89 17 c3 ae 8e 3a 7e 6e a5 0a da c9 c0 5b 
[22:26:48]##################### SEND CLIENT PUBLIC KEY #########################
[22:26:48]Sending plain message: 03 00 65 de 7d 09 16 91 5d d9 d4 69 55 e0 c0 0d d8 36 94 0d 54 4a 89 cb f2 7e dd a7 40 c4 6f 94 ee 26 23 8a 
[22:26:48]Command identifier: 03, CRC: 8a23
[22:26:48]connecting within: nuki
[22:26:48][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: -51
[22:26:48]GDIO characteristic registered
[22:26:48]Notify callback for characteristic: a92ee301-5501-11e4-916c-0800200c9a66 of length: 36
[22:26:48]##################### CALCULATE DH SHARED KEY s ############################################## CALCULATE DH SHARED KEY s #########################Received data: 04 00 7a c2 98 eb 00 8c 67 60 87 64 dc 79 98 05 42 ca c5 51 d7 9a 06 16 e5 e9 3a 03 2a 2e 8b c8 3d 2a c3 82 
[22:26:48]CRC CHECK OK
[22:26:48]
[22:26:48]Challenge: 7a c2 98 eb 00 8c 67 60 87 64 dc 79 98 05 42 ca c5 51 d7 9a 06 16 e5 e9 3a 03 2a 2e 8b c8 3d 2a 
[22:26:48]Shared key s: 85 10 12 b3 d3 81 85 c2 ae a7 ac 15 68 31 15 e5 40 40 0f 42 63 d5 a4 63 6b aa 58 25 af b2 98 0a 
[22:26:48]##################### DERIVE LONG TERM SHARED SECRET KEY k #########################
[22:26:48]Secret key k: 1a e4 86 8e 7b 53 16 0e 1c cb de 25 9a 9c 9f 98 80 23 31 74 54 d4 fc 9c 46 53 66 17 09 44 0b ae 
[22:26:48]##################### CALCULATE/VERIFY AUTHENTICATOR #########################
[22:26:48]Concatenated data r: 65 de 7d 09 16 91 5d d9 d4 69 55 e0 c0 0d d8 36 94 0d 54 4a 89 cb f2 7e dd a7 40 c4 6f 94 ee 26 13 a1 26 b6 41 1e 1c 0c cd 61 a1 f9 78 60 33 68 c1 3f 89 17 c3 ae 8e 3a 7e 6e a5 0a da c9 c0 5b 7a c2 98 eb 00 8c 67 60 87 64 dc 79 98 05 42 ca c5 51 d7 9a 06 16 e5 e9 3a 03 2a 2e 8b c8 3d 2a 
[22:26:48]HMAC 256 result: 09 ed 10 9f cd 32 de f3 33 b9 86 db 1c ed ee 95 c3 87 f8 6c 3b 42 75 be 6c 70 4e 2b f8 2d d9 c8 
[22:26:48]##################### SEND AUTHENTICATOR #########################
[22:26:48]Sending plain message: 05 00 09 ed 10 9f cd 32 de f3 33 b9 86 db 1c ed ee 95 c3 87 f8 6c 3b 42 75 be 6c 70 4e 2b f8 2d d9 c8 43 2b 
[22:26:48]Command identifier: 05, CRC: 2b43
[22:26:48]connecting within: nuki
[22:26:48][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: -55
[22:26:48]MQTT Broker not configured, aborting connection attempt.
[22:26:48]GDIO characteristic registered
[22:26:50]Notify callback for characteristic: a92ee301-5501-11e4-916c-0800200c9a66 of length: 5
[22:26:50]Received data: 4c 00 01 71 b4 
[22:26:50]CRC CHECK OK
[22:26:50]authorizationInfo: 01 
[22:26:50]##################### SEND AUTHORIZATION DATA (ULTRA) #########################
[22:26:50]payloadlen: 40
[22:26:50]sizeof(plainData): 46
[22:26:50]CRC: d001
[22:26:50]Plain data with CRC: : 7f ff ff ff 06 00 0c 2a 37 af 4e 75 6b 69 48 75 62 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 de 14 04 00 01 d0 
[22:26:50]Nonce: 41 7c f2 af 7b 52 8a 9d 9a 8c 92 e4 41 c3 52 e3 06 fc d5 b6 4a 72 5d fe 
[22:26:50]Additional data: : 41 7c f2 af 7b 52 8a 9d 9a 8c 92 e4 41 c3 52 e3 06 fc d5 b6 4a 72 5d fe 00 00 00 00 40 00 
[22:26:50]Encryption key (secretKey): : 1a e4 86 8e 7b 53 16 0e 1c cb de 25 9a 9c 9f 98 80 23 31 74 54 d4 fc 9c 46 53 66 17 09 44 0b ae 
[22:26:50]Plain data encrypted: : cd 10 b6 9e c7 b0 1e 7d c0 e5 91 b4 be 4d c6 c0 da 23 a1 6f 2a 53 19 a9 38 ff 63 37 a8 32 cd 1d b1 45 4b a2 80 23 1c 95 1f 49 11 09 99 98 81 16 55 9c f0 02 c6 c7 59 28 fc ca 86 ec a5 94 0e e2 
[22:26:50]connecting within: nuki
[22:26:50][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: -54
[22:26:50]GDIO characteristic registered
[22:26:50]Sending encrypted pairing message: 41 7c f2 af 7b 52 8a 9d 9a 8c 92 e4 41 c3 52 e3 06 fc d5 b6 4a 72 5d fe 00 00 00 00 40 00 cd 10 b6 9e c7 b0 1e 7d c0 e5 91 b4 be 4d c6 c0 da 23 a1 6f 2a 53 19 a9 38 ff 63 37 a8 32 cd 1d b1 45 4b a2 80 23 1c 95 1f 49 11 09 99 98 81 16 55 9c f0 02 c6 c7 59 28 fc ca 86 ec a5 94 0e e2 
[22:26:50]Notify callback for characteristic: a92ee301-5501-11e4-916c-0800200c9a66 of length: 7
[22:26:50]Received data: 12 00 20 00 00 13 d7 
[22:26:50]CRC CHECK OK
[22:26:50]Error: 20 for command: 00:00
[22:26:50]K_ERROR_NOT_AUTHORIZED
[... cut ...]
[22:27:00]MQTT Broker not configured, aborting connection attempt.
[22:27:06]MQTT Broker not configured, aborting connection attempt.
[22:27:12]MQTT Broker not configured, aborting connection attempt.
[22:27:16]Pairing timeout
[22:27:16]pairing result 2
[22:27:18]MQTT Broker not configured, aborting connection attempt.
[22:27:18]Nuki lock start pairing
[22:27:18]Pairing as app
[22:27:18]Error getting data from NVS
[22:27:18]No nuki in pairing mode found
[22:27:18]pairing result 0
```

</details>

iranl, 2025-01-13:
Test 11: https://github.com/iranl/nuki_hub/actions/runs/12756542328

and

Test 12: https://github.com/iranl/nuki_hub/actions/runs/12756562753

Both building atm

mallorca2288, 2025-01-13:
Test 11. It did pair! 🎉 👏 

<details>

```
Using 'COM7' as serial port.
Showing logs:
[22:46:11]IP: 192.168.8.56
[22:46:11]MQTT Broker not configured, aborting connection attempt.
[22:46:13]Nuki lock start pairing
[22:46:13]Pairing as app
[22:46:13]Error getting data from NVS
[22:46:13]No nuki in pairing mode found
[22:46:13]pairing result 0
[22:46:16]Nuki lock start pairing
[22:46:16]Pairing as app
[22:46:16]Error getting data from NVS
[22:46:16]No nuki in pairing mode found
[22:46:16]pairing result 0
[22:46:17]MQTT Broker not configured, aborting connection attempt.
[22:46:18]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[22:46:18]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[22:46:19]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[22:46:19]Nuki lock start pairing
[22:46:19]Pairing as app
[22:46:19]Error getting data from NVS
[22:46:19]Nuki in pairing mode found
[22:46:19]connecting within: nuki
[22:46:19][NukiHub] Connect timeout 3000 ms
[22:46:19]BLE connected
[22:46:19][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: -51
[22:46:19]GDIO characteristic registered
[22:46:19]##################### REQUEST REMOTE PUBLIC KEY #########################
[22:46:19]Sending plain message: 01 00 03 00 27 a7 
[22:46:19]Command identifier: 01, CRC: a727
[22:46:19]connecting within: nuki
[22:46:19][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: -56
[22:46:19]GDIO characteristic registered
[22:46:21]Notify callback for characteristic: a92ee301-5501-11e4-916c-0800200c9a66 of length: 36
[22:46:21]Received data: 03 00 77 0c ef 95 b7 37 d3 d3 1e c2 22 ac 76 51 7a 8a 5f b4 e3 ef 96 0a de a4 d5 5d d8 07 9c a9 f2 02 13 ac 
[22:46:21]CRC CHECK OK
[22:46:21]Remote public key: 77 0c ef 95 b7 37 d3 d3 1e c2 22 ac 76 51 7a 8a 5f b4 e3 ef 96 0a de a4 d5 5d d8 07 9c a9 f2 02 
[22:46:21]##################### SEND CLIENT PUBLIC KEY #########################
[22:46:21]Sending plain message: 03 00 58 aa 70 31 1b 54 b2 9a f7 b6 1d dd 53 b3 cd 4f 2e dc 5d 3a 12 b7 7b f8 d9 03 8d 27 cb f3 84 09 92 70 
[22:46:21]Command identifier: 03, CRC: 7092
[22:46:21]connecting within: nuki
[22:46:21][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: -55
[22:46:21]GDIO characteristic registered
[22:46:22]Notify callback for characteristic: a92ee301-5501-11e4-916c-0800200c9a66 of length: 36
[22:46:22]##################### CALCULATE DH SHARED KEY s ############################################## CALCULATE DH SHARED KEY s #########################Received data: 04 00 72 8f d5 2b 7b be 54 64 b0 d3 37 a1 2b bd ec c0 ac 02 e9 72 cf 93 30 bc 26 47 c1 1d e5 0f e9 92 37 31 
[22:46:22]CRC CHECK OK
[22:46:22]
[22:46:22]Challenge: 72 8f d5 2b 7b be 54 64 b0 d3 37 a1 2b bd ec c0 ac 02 e9 72 cf 93 30 bc 26 47 c1 1d e5 0f e9 92 
[22:46:22]Shared key s: 8d 0c a1 21 c0 33 79 45 47 b3 89 b7 51 d2 80 e0 c4 b0 33 24 c0 e2 f1 14 66 a0 85 8c 9c 99 0b 62 
[22:46:22]##################### DERIVE LONG TERM SHARED SECRET KEY k #########################
[22:46:22]Secret key k: 9d 00 39 03 50 06 74 f7 bd 3b a8 7c c9 93 64 f6 6a 80 c2 a6 23 e0 af 13 55 a1 cb ac a2 e8 14 81 
[22:46:22]##################### CALCULATE/VERIFY AUTHENTICATOR #########################
[22:46:22]Concatenated data r: 58 aa 70 31 1b 54 b2 9a f7 b6 1d dd 53 b3 cd 4f 2e dc 5d 3a 12 b7 7b f8 d9 03 8d 27 cb f3 84 09 77 0c ef 95 b7 37 d3 d3 1e c2 22 ac 76 51 7a 8a 5f b4 e3 ef 96 0a de a4 d5 5d d8 07 9c a9 f2 02 72 8f d5 2b 7b be 54 64 b0 d3 37 a1 2b bd ec c0 ac 02 e9 72 cf 93 30 bc 26 47 c1 1d e5 0f e9 92 
[22:46:22]HMAC 256 result: 71 6d 78 90 82 34 01 a4 77 24 56 f5 e8 67 9f 36 8a c2 7f 57 45 e6 af ef dd dd 15 3a 38 8b 90 f0 
[22:46:22]##################### SEND AUTHENTICATOR #########################
[22:46:22]Sending plain message: 05 00 71 6d 78 90 82 34 01 a4 77 24 56 f5 e8 67 9f 36 8a c2 7f 57 45 e6 af ef dd dd 15 3a 38 8b 90 f0 a3 3f 
[22:46:22]Command identifier: 05, CRC: 3fa3
[22:46:22]connecting within: nuki
[22:46:22][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: -55
[22:46:22]GDIO characteristic registered
[22:46:23]MQTT Broker not configured, aborting connection attempt.
[22:46:24]Notify callback for characteristic: a92ee301-5501-11e4-916c-0800200c9a66 of length: 5
[22:46:24]Received data: 4c 00 01 71 b4 
[22:46:24]CRC CHECK OK
[22:46:24]authorizationInfo: 01 
[... 113 lines cut ...]
[22:46:57]authorizationId: 90 7e fe 3f 
[22:46:57]Credentials retrieved from preferences, ready for commands
[22:46:57]Start executing: 01 
[22:46:57]************************ SENDING COMMAND [1] ************************
[22:46:57]payloadlen: 2
[22:46:57]sizeof(plainData): 8
[22:46:57]CRC: c33
[22:46:57]Plain data with CRC: : 90 7e fe 3f 01 00 0c 00 33 0c 
[22:46:57]Nonce: 5a 4a e7 c1 36 3b 30 e9 ec b1 37 9c 01 b9 3f 5e db 74 bb 6b 75 9a 69 75 
[22:46:57]Additional data: : 5a 4a e7 c1 36 3b 30 e9 ec b1 37 9c 01 b9 3f 5e db 74 bb 6b 75 9a 69 75 90 7e fe 3f 1a 00 
[22:46:57]Encryption key (secretKey): : 9d 00 39 03 50 06 74 f7 bd 3b a8 7c c9 93 64 f6 6a 80 c2 a6 23 e0 af 13 55 a1 cb ac a2 e8 14 81 
[22:46:57]Plain data encrypted: : 85 87 f2 32 f1 29 32 dd a7 d8 49 d3 bd 0b a3 c6 23 50 f3 59 6e 7d 0f 5f eb b6 
[22:46:57]connecting within: nuki
[22:46:58]BLE connected
[22:46:58][NukiHub] Reconnect success
[22:46:58][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: -44
[22:46:58]USDIO characteristic registered
[22:46:58]Sending encrypted message: 5a 4a e7 c1 36 3b 30 e9 ec b1 37 9c 01 b9 3f 5e db 74 bb 6b 75 9a 69 75 90 7e fe 3f 1a 00 85 87 f2 32 f1 29 32 dd a7 d8 49 d3 bd 0b a3 c6 23 50 f3 59 6e 7d 0f 5f eb b6 
[22:46:59]MQTT Broker not configured, aborting connection attempt.

```

</details>

[image]

mallorca2288, 2025-01-13:
Unpaired, then tested build 12. It's also pairing!

<details>

```
[22:52:51]MQTT Broker not configured, aborting connection attempt.
[22:52:53]Nuki lock start pairing
[22:52:53]Pairing as app
[22:52:53]Error getting data from NVS
[22:52:53]No nuki in pairing mode found
[22:52:53]pairing result 0
[22:52:56]Nuki lock start pairing
[22:52:56]Pairing as app
[22:52:56]Error getting data from NVS
[22:52:56]No nuki in pairing mode found
[22:52:56]pairing result 0
[22:52:57]MQTT Broker not configured, aborting connection attempt.
[22:52:57]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[22:52:57]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[22:52:57]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[22:52:58]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[22:52:58]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[22:52:59]Nuki lock start pairing
[22:52:59]Pairing as app
[22:52:59]Error getting data from NVS
[22:52:59]Nuki in pairing mode found
[22:52:59]connecting within: nuki
[22:52:59][NukiHub] Connect timeout 3000 ms
[22:52:59]BLE connected
[22:52:59][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: -61
[22:52:59]GDIO characteristic registered
[22:53:00]##################### REQUEST REMOTE PUBLIC KEY #########################
[22:53:00]Sending plain message: 01 00 03 00 27 a7 
[22:53:00]Command identifier: 01, CRC: a727
[22:53:00]connecting within: nuki
[22:53:00][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: -68
[22:53:00]GDIO characteristic registered
[22:53:01]Notify callback for characteristic: a92ee301-5501-11e4-916c-0800200c9a66 of length: 36
[22:53:01]Received data: 03 00 92 48 13 68 8b 57 d5 6b d6 9e e4 1b c0 d5 3b 41 f8 83 31 89 88 ec 04 10 72 80 ea 96 53 ec 93 25 b1 ae 
[22:53:01]CRC CHECK OK
[22:53:01]Remote public key: 92 48 13 68 8b 57 d5 6b d6 9e e4 1b c0 d5 3b 41 f8 83 31 89 88 ec 04 10 72 80 ea 96 53 ec 93 25 
[22:53:02]##################### SEND CLIENT PUBLIC KEY #########################
[22:53:02]Sending plain message: 03 00 5b 1d f6 1f 73 c1 85 77 30 fe 23 9e 0d 08 c1 39 0a 07 2e 30 98 08 d7 fb fc ca dd 4a 3b 12 c1 36 e9 10 
[22:53:02]Command identifier: 03, CRC: 10e9
[22:53:02]connecting within: nuki
[22:53:02][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: -56
[22:53:02]GDIO characteristic registered
[22:53:02]Notify callback for characteristic: a92ee301-5501-11e4-916c-0800200c9a66 of length: 36
[22:53:02]##################### CALCULATE DH SHARED KEY s ############################################## CALCULATE DH SHARED KEY s #########################Received data: 04 00 31 06 86 99 ba b8 48 a6 63 39 b1 3c 26 6a f9 90 64 77 d0 b1 12 b8 ff 6a db 3c ff 72 60 f4 a7 89 d0 9e 
[22:53:02]CRC CHECK OK
[22:53:02]
[22:53:02]Challenge: 31 06 86 99 ba b8 48 a6 63 39 b1 3c 26 6a f9 90 64 77 d0 b1 12 b8 ff 6a db 3c ff 72 60 f4 a7 89 
[22:53:02]Shared key s: 6b e7 2d eb 71 5a aa a5 74 0b ff 49 bf 7b f1 15 90 33 5f 82 83 4b 03 11 21 bd ea e3 92 da 99 72 
[22:53:02]##################### DERIVE LONG TERM SHARED SECRET KEY k #########################
[22:53:02]Secret key k: 25 65 8c 60 42 19 81 8e 9d be 06 f3 66 88 7d 61 cb 69 5d db 59 c0 ec 8d 90 d7 4e aa 8c e4 e1 f3 
[22:53:02]##################### CALCULATE/VERIFY AUTHENTICATOR #########################
[22:53:02]Concatenated data r: 5b 1d f6 1f 73 c1 85 77 30 fe 23 9e 0d 08 c1 39 0a 07 2e 30 98 08 d7 fb fc ca dd 4a 3b 12 c1 36 92 48 13 68 8b 57 d5 6b d6 9e e4 1b c0 d5 3b 41 f8 83 31 89 88 ec 04 10 72 80 ea 96 53 ec 93 25 31 06 86 99 ba b8 48 a6 63 39 b1 3c 26 6a f9 90 64 77 d0 b1 12 b8 ff 6a db 3c ff 72 60 f4 a7 89 
[22:53:02]HMAC 256 result: 56 55 37 08 1a 56 df 8e fb 68 c7 0e b8 c1 e6 49 db 3d 6b 87 ab 86 b2 fc bf c5 26 11 71 9e c2 75 
[22:53:02]##################### SEND AUTHENTICATOR #########################
[22:53:02]Sending plain message: 05 00 56 55 37 08 1a 56 df 8e fb 68 c7 0e b8 c1 e6 49 db 3d 6b 87 ab 86 b2 fc bf c5 26 11 71 9e c2 75 bf 79 
[22:53:02]Command identifier: 05, CRC: 79bf
[22:53:02]connecting within: nuki
[22:53:02][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: -75
[22:53:02]GDIO characteristic registered
[22:53:03]MQTT Broker not configured, aborting connection attempt.
[22:53:04]Notify callback for characteristic: a92ee301-5501-11e4-916c-0800200c9a66 of length: 5
[22:53:04]Received data: 4c 00 01 71 b4 
[22:53:04]CRC CHECK OK
[22:53:04]authorizationInfo: 01 
[22:53:04]##################### SEND AUTHORIZATION DATA (ULTRA) #########################
[... 159 lines cut ...]
[22:53:44]disconnecting BLE on timeout
[22:53:44]disconnecting BLE on timeout
[22:53:44]disconnecting BLE on timeout
[22:53:44]Nuki Advertising: Name: , Address: 54:d2:72:0a:b5:7f, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66430ab57fc8
[22:53:44]iBeacon ID: 004C Major: 17162 Minor: 46463 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -56
[22:53:45]disconnecting BLE on timeout
[22:53:45]MQTT Broker not configured, aborting connection attempt.
[22:53:45]disconnecting BLE on timeout
[22:53:45]Nuki Advertising: Name: , Address: 54:d2:72:0a:b5:7f, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66430ab57fc8
[22:53:45]iBeacon ID: 004C Major: 17162 Minor: 46463 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -56
[22:53:45]disconnecting BLE on timeout
[22:53:45]disconnecting BLE on timeout
[22:53:45]Nuki Advertising: Name: , Address: 54:d2:72:0a:b5:7f, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66430ab57fc8
[22:53:45]iBeacon ID: 004C Major: 17162 Minor: 46463 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -56
[22:53:45]disconnecting BLE on timeout
[22:53:46]disconnecting BLE on timeout
[22:53:46]disconnecting BLE on timeout
[22:53:46]disconnecting BLE on timeout
[22:53:46]Nuki Advertising: Name: , Address: 54:d2:72:0a:b5:7f, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66430ab57fc8
[22:53:46]iBeacon ID: 004C Major: 17162 Minor: 46463 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -56
```

</details>

[image]

mallorca2288, 2025-01-13:
Also tried setting the PIN code of the lock in the Credentials page but it doesn't work with my password.

iranl, 2025-01-13:
Slowly getting there. 

Please unpair again as there is still an issue with saving the authorization ID (this is what probably causes the failure of all commands after pairing).

Test 13: https://github.com/iranl/nuki_hub/actions/runs/12756923186

> Also tried setting the PIN code of the lock in the Credentials page but it doesn't work with my password.

The credentials page is not used for the Ultra yet.

mallorca2288, 2025-01-13:
Test 13 logs after unpairing:

<details>

```
[23:13:04]WiFi connecting.........................
[23:13:04][WiFi-event] event: 112
[23:13:04]Connected to access point
[23:13:04]Wi-Fi connected
[23:13:04][WiFi-event] event: 115
[23:13:04]Obtained IP address: 192.168.8.56
[23:13:04]Wi-Fi connected
[23:13:04][http] connection #43 connected from 192.168.8.56
[23:13:04]IP: 192.168.8.56
[23:13:04]MQTT Broker not configured, aborting connection attempt.
[23:13:04][http] connection #44 connected from 192.168.8.56
[23:13:06]Nuki lock start pairing
[23:13:06]Pairing as app
[23:13:06]Error getting data from NVS
[23:13:06]No nuki in pairing mode found
[23:13:06]pairing result 0
[23:13:09]Nuki lock start pairing
[23:13:09]Pairing as app
[23:13:09]Error getting data from NVS
[23:13:09]No nuki in pairing mode found
[23:13:09]pairing result 0
[23:13:10]MQTT Broker not configured, aborting connection attempt.
[23:13:12]Nuki lock start pairing
[23:13:12]Pairing as app
[23:13:12]Error getting data from NVS
[23:13:12]No nuki in pairing mode found
[23:13:12]pairing result 0
[23:13:15]Nuki lock start pairing
[23:13:15]Pairing as app
[23:13:15]Error getting data from NVS
[23:13:15]No nuki in pairing mode found
[23:13:15]pairing result 0
[23:13:16]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[23:13:16]MQTT Broker not configured, aborting connection attempt.
[23:13:16]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[23:13:17]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[23:13:17]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[23:13:17]Nuki lock start pairing
[23:13:17]Pairing as app
[23:13:17]Error getting data from NVS
[23:13:17]Nuki in pairing mode found
[23:13:17]connecting within: nuki
[23:13:17][NukiHub] Connect timeout 3000 ms
[23:13:18]BLE connected
[23:13:18][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: -51
[23:13:18]GDIO characteristic registered
[23:13:18]##################### REQUEST REMOTE PUBLIC KEY #########################
[23:13:18]Sending plain message: 01 00 03 00 27 a7 
[23:13:18]Command identifier: 01, CRC: a727
[23:13:18]connecting within: nuki
[23:13:18][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: -54
[23:13:18]GDIO characteristic registered
[23:13:20]Notify callback for characteristic: a92ee301-5501-11e4-916c-0800200c9a66 of length: 36
[23:13:20]Received data: 03 00 d5 12 06 c0 75 67 3c 31 78 b9 f9 b7 d0 d5 1a 84 30 13 ca f5 cc 5b af db 69 dc 50 80 3f e2 df 50 40 11 
[23:13:20]CRC CHECK OK
[23:13:20]Remote public key: d5 12 06 c0 75 67 3c 31 78 b9 f9 b7 d0 d5 1a 84 30 13 ca f5 cc 5b af db 69 dc 50 80 3f e2 df 50 
[23:13:20]##################### SEND CLIENT PUBLIC KEY #########################
[23:13:20]Sending plain message: 03 00 b3 30 33 c6 b4 b4 b6 96 23 8e fc c2 20 d0 32 51 3d 79 59 d3 aa e3 cd 6a ce 16 5f 39 76 9b 02 2a a3 15 
[23:13:20]Command identifier: 03, CRC: 15a3
[23:13:20]connecting within: nuki
[23:13:20][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: -47
[23:13:20]GDIO characteristic registered
[23:13:20]Notify callback for characteristic: a92ee301-5501-11e4-916c-0800200c9a66 of length: 36
[23:13:20]##################### CALCULATE DH SHARED KEY s ############################################## CALCULATE DH SHARED KEY s #########################Received data: 04 00 dd 7c e2 a1 c5 3b 74 ea a7 99 59 b7 e6 39 24 7c c1 ca 41 8b 96 4c 32 c4 4f 55 ec 6c cc 12 c8 98 d9 f4 
[23:13:20]CRC CHECK OK
[... 184 lines cut ...]
[23:13:51]Network device: Built-in Wi-Fi

[23:13:51][WiFi-event] event: 101
[23:13:51]WiFi interface ready
[23:13:51]Wi-Fi async scan started
[23:13:51]Host name: nukihub
[23:13:51]MQTT Broker: :1883
[23:13:51][WiFi-event] event: 110
[23:13:51]WiFi client started
[23:13:51]Nuki Lock enabled
[23:13:51]Device id lock: 2755546827
[23:13:51][NukiHub] Credentials retrieved :
[23:13:51]secretKeyK: ac 3b 28 2e 86 67 c6 85 cc 5c a6 14 da 6d 2a d3 0c ee 69 23 27 3d 7e 3c ec 4e cf de 9e bd a8 da 
[23:13:51]bleAddress: 54:d2:72:0a:b5:7f
[23:13:51]authorizationId: 90 7e fe 3f 
[23:13:51]Lock state interval: 1800 | Battery interval: 1800 | Publish auth data: yes
[23:13:51]Nuki Opener disabled
[23:13:51]Not running on PSRAM enabled device
[23:13:54][WiFi-event] event: 102
[23:13:54]Completed scan for access points
```

</details>

Showing as paired on the info page.

iranl, 2025-01-13:
The last log is not from test 13.

Build ID is 12756923186 for test 13.
Your logs show Build ID 12756562753

mallorca2288, 2025-01-13:
Sorry, I forgot to unzip the latest build. Now it pairs and it also shows the state of the lock!

<details>

```
Using 'COM7' as serial port.
Showing logs:
[23:24:44]ets Jun  8 2016 00:22:57
[23:24:44]
[23:24:44]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[23:24:44]configsip: 0, SPIWP:0xee
[23:24:44]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[23:24:44]mode:DIO, clock div:2
[23:24:44]load:0x3fff0030,len:4644
[23:24:44]load:0x40078000,len:15312
[23:24:44]ho 0 tail 12 room 4
[23:24:44]load:0x40080400,len:4
[23:24:44]load:0x40080404,len:3220
[23:24:44]entry 0x400805a4
[23:24:45]Last config version: 908
[23:24:45]Current config version: 908
[23:24:45]Partition size: 2555904
[23:24:45]Partition subtype: 16
[23:24:45]0
[23:24:45]Nuki Hub version 9.08
[23:24:45]Nuki Hub build 12756923186.1330.1
[23:24:45]Load GPIO configuration
[23:24:45]IP configuration: IP address: 192.168.8.56, Subnet: 255.255.255.0, Gateway: 192.168.8.1, DNS: 1.0.0.1
[23:24:45]Hardware detect: 1
[23:24:45]MQTT without TLS.
[23:24:45]Network device: Built-in Wi-Fi

[23:24:45][WiFi-event] event: 101
[23:24:45]WiFi interface ready
[23:24:45]Wi-Fi async scan started
[23:24:45]Host name: nukihub
[23:24:45]MQTT Broker: :1883
[23:24:45][WiFi-event] event: 110
[23:24:45]WiFi client started
[23:24:45]Nuki Lock enabled
[23:24:45]Device id lock: 2755546827
[23:24:45]Error getting data from NVS
[23:24:45]Lock state interval: 1800 | Battery interval: 1800 | Publish auth data: yes
[23:24:45]Nuki Opener disabled
[23:24:45]Not running on PSRAM enabled device
[23:24:48][WiFi-event] event: 102
[23:24:48]Completed scan for access points

[23:24:51]WiFi connecting.........................
[23:24:51][WiFi-event] event: 112
[23:24:51]Connected to access point
[23:24:51]Wi-Fi connected
[23:24:51][WiFi-event] event: 115
[23:24:51]Obtained IP address: 192.168.8.56
[23:24:51]Wi-Fi connected
[23:24:51]IP: 192.168.8.56
[23:24:51]MQTT Broker not configured, aborting connection attempt.
[23:24:53]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[23:24:53]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[23:24:53]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[23:24:53]Found nuki ultra in pairing state:  addr: 54:d2:72:0a:b5:7f
[23:24:54]Nuki lock start pairing
[23:24:54]Pairing as app
[23:24:54]Error getting data from NVS
[23:24:54]Nuki in pairing mode found
[23:24:54]connecting within: nuki
[23:24:54][NukiHub] Connect timeout 3000 ms
[23:24:54]BLE connected
[23:24:54][NukiHub] Connected to: 54:d2:72:0a:b5:7f RSSI: -59
[23:24:54]GDIO characteristic registered
[... 295 lines cut ...]
[23:25:20]iBeacon ID: 004C Major: 17162 Minor: 46463 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -56
[23:25:20]disconnecting BLE on timeout
[23:25:20]disconnecting BLE on timeout
[23:25:20]Nuki Advertising: Name: , Address: 54:d2:72:0a:b5:7f, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66430ab57fc8
[23:25:20]iBeacon ID: 004C Major: 17162 Minor: 46463 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -56
[23:25:21]disconnecting BLE on timeout
[23:25:21]disconnecting BLE on timeout
[23:25:21]MQTT Broker not configured, aborting connection attempt.
[23:25:21]disconnecting BLE on timeout
[23:25:21]disconnecting BLE on timeout
[23:25:22]disconnecting BLE on timeout
[23:25:22]Nuki Advertising: Name: , Address: 54:d2:72:0a:b5:7f, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66430ab57fc8
[23:25:22]iBeacon ID: 004C Major: 17162 Minor: 46463 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -56
[23:25:22]disconnecting BLE on timeout
[23:25:22]disconnecting BLE on timeout
[23:25:22]Nuki Advertising: Name: , Address: 54:d2:72:0a:b5:7f, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66430ab57fc8
[23:25:22]iBeacon ID: 004C Major: 17162 Minor: 46463 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -56
[23:25:22]disconnecting BLE on timeout
[23:25:22]disconnecting BLE on timeout
[23:25:23]disconnecting BLE on timeout
```

</details>

[image]

iranl, 2025-01-13:
Awesome. This should basically be it for changes to the BLE communication library in regards to the Ultra. 

Will need to make a couple(/many) changes to NukiHub to add proper support before this can be merged for 9.08, but this was the hard part.

@mallorca2288: Your help has been invaluable. Couldn't have been done without your excellent support. 

Shall I add you as co-author on the final PR?

mallorca2288, 2025-01-13:
Many thanks to you @iranl ! You did all the work.
I didn't really do much, I just copied and pasted what appeared in the console.

I'll be happy to test all the changes in the code and will report back on things that don't work as expected.

Thanks again, I appreciate you taking the time to do this.

iranl, 2025-01-14:
@mallorca2288: Can you test: https://github.com/iranl/nuki_hub/actions/runs/12776839680

And report on each of the following:

- Understandability of the README: https://github.com/iranl/nuki_hub/tree/api-2.3#pairing-with-a-nuki-lock-ultra
- NukiHub pairs with the device correctly **(Please follow the steps in the README, these have changed from before)**
- NukiHub can query the correct device lock state
- NukiHub can query additional parameters of the device state (auth log, etc.)
- NukiHub can change the lock state as expected
- NukiHub can query and change all supported device settings (including motor speed)

Optional:
- NukiHub can query and change keypad settings and codes for keypads connected through the device
- NukiHub can query and change authorization settings
- NukiHub can query and change time control settings

mallorca2288, 2025-01-14:
>     * Understandability of the README: https://github.com/iranl/nuki_hub/tree/api-2.3#pairing-with-a-nuki-lock-ultra
I think it's very clear!
>     * NukiHub pairs with the device correctly **(Please follow the steps in the README, these have changed from before)**
It paired without any issue following the instructions in the README section.
>     * NukiHub can query the correct device lock state
It shows the corrrect state of the lock.
Manually locking/unlocking the lock didn't update the state. 
Locking/unlocking through the Nuki App, state was correctly updated.
>     * NukiHub can query additional parameters of the device state (auth log, etc.)
All parameters appear to be correct. Auth log is updating. Keypad entries information is correct.

Keypad status is blank. On the last build it switched between success and blank status. Don't know the meaning of this.
>     * NukiHub can change the lock state as expected
It does lock and unlock.

>     * NukiHub can query and change all supported device settings (including motor speed)
Didn't try changing all of the settings, but the ones I tried seem to be working.

Tried changing LED state, LED brightness, auto-update enabled, Single button press action and a few more. All worked.

On the MQTT log I can see this message updated every second or less:
> disconnecting BLE on timeout

Nuki hub shows that hybrid mode is not connected:
[image]
[image]
However I can see on the nuki app that nuki ultra is correctly connected to MQTT server. I've enabled all the hybrid settings on nuki hub.

I will continue to test everything and report back on any bugs I might find. Will also test changing keypad/authorizations/time controls.

mallorca2288, 2025-01-14:
Forgot to say, unfortunately the motor speed setting is not showing in my Home Assistant. Also I can't find the state of the speed setting using MQTT explorer.

iranl, 2025-01-14:
> On the MQTT log I can see this message updated every second or less:
> 
> > disconnecting BLE on timeout

Thats fine, it's supposed to do that.

> Forgot to say, unfortunately the motor speed setting is not showing in my Home Assistant. Also I can't find the state of the speed setting using MQTT explorer.

The motor speed state should be in the MQTT topic `nukihub/lock/configuration/advancedJson`

> Nuki hub shows that hybrid mode is not connected

Can you connect to serial logging and then reboot NukiHub (using reboot button on the webconfigurator) and get me the logs from boot all the way to it getting all the configuration settings.

Note that you need to refresh the WebConfigurator to get the current Hybrid connected state (it doesn't update on the background as the other settings)

mallorca2288, 2025-01-14:
- Motor speed status is correct and I'm able to change it.
- Battery type is blank.
- Serial logs while restarting through web:
<details>

```
Using 'COM7' as serial port.
Showing logs:
[23:36:31]ets Jun  8 2016 00:22:57
[23:36:31]
[23:36:31]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[23:36:31]configsip: 0, SPIWP:0xee
[23:36:31]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[23:36:31]mode:DIO, clock div:2
[23:36:31]load:0x3fff0030,len:4644
[23:36:31]load:0x40078000,len:15312
[23:36:31]ho 0 tail 12 room 4
[23:36:31]load:0x40080400,len:4
[23:36:31]load:0x40080404,len:3220
[23:36:31]entry 0x400805a4
[23:36:32]Last config version: 908
[23:36:32]Current config version: 908
[23:36:32]Partition size: 2555904
[23:36:32]Partition subtype: 16
[23:36:32]0
[23:36:32]Nuki Hub version 9.08
[23:36:32]Nuki Hub build 12776839680.1336.1
[23:36:32]Load GPIO configuration
[23:36:32]IP configuration: IP address: 192.168.8.56, Subnet: 255.255.255.0, Gateway: 192.168.8.1, DNS: 1.0.0.1
[23:36:32]Hardware detect: 1
[23:36:32]MQTT without TLS.
[23:36:32]Network device: Built-in Wi-Fi
[23:36:32]Attempting to connect to saved SSID Foquitas
[23:36:32][WiFi-event] event: 101
[23:36:32]WiFi interface ready
[23:36:32]Wi-Fi async scan started
[23:36:32]Host name: nukihub
[23:36:32]MQTT Broker: 192.168.8.10:1883
[23:36:32][WiFi-event] event: 110
[23:36:32]WiFi client started
[23:36:32]Nuki Lock enabled
[23:36:32]Device id lock: 2755546827
[23:36:32][NukiHub] Credentials retrieved :
[23:36:32]secretKeyK: c1 25 6d 42 5f 99 57 4e 87 a8 fc d6 25 5a 77 5f 1e e3 9e 91 35 6a f7 1e c6 d9 dd 0f d2 05 65 e9 
[23:36:32]bleAddress: 54:d2:72:0a:b5:7f
[23:36:32]authorizationId: fc 8e 06 00 
[23:36:32]Lock state interval: 1800 | Battery interval: 1800 | Publish auth data: yes
[23:36:32]Nuki Opener disabled
[23:36:32]Not running on PSRAM enabled device
[23:36:35][WiFi-event] event: 102
[23:36:35]Completed scan for access points

[23:36:38]WiFi connecting.........................
[23:36:38][WiFi-event] event: 112
[23:36:38]Connected to access point
[23:36:38]Wi-Fi connected
[23:36:38][WiFi-event] event: 115
[23:36:38]Obtained IP address: 192.168.8.56
[23:36:38]Wi-Fi connected
[23:36:38]IP: 192.168.8.56
[23:36:38]Attempting MQTT connection
[23:36:38]MQTT: Connecting with user: ----
[23:36:39]MQTT connected
[23:36:39]HASS setup for NukiHub completed.
[23:36:39]Subscribing to MQTT topic: nukihub/lock/action
[23:36:39]Subscribing to MQTT topic: nukihub/lock/configuration/action
[23:36:39]Subscribing to MQTT topic: nukihub/lock/query/config
[23:36:39]Subscribing to MQTT topic: nukihub/lock/query/lockstate
[23:36:39]Subscribing to MQTT topic: nukihub/lock/query/battery
[23:36:39]Subscribing to MQTT topic: nukihub/lock/keypad/command/action
[... 1103 lines cut ...]
[23:37:38]disconnecting BLE on timeout
[23:37:38]Nuki Advertising: Name: , Address: 54:d2:72:0a:b5:7f, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66430ab57fc8
[23:37:38]iBeacon ID: 004C Major: 17162 Minor: 46463 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -56
[23:37:38]disconnecting BLE on timeout
[23:37:38]disconnecting BLE on timeout
[23:37:39]Nuki Advertising: Name: , Address: 54:d2:72:0a:b5:7f, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66430ab57fc8
[23:37:39]iBeacon ID: 004C Major: 17162 Minor: 46463 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -56
[23:37:39]disconnecting BLE on timeout
[23:37:39]disconnecting BLE on timeout
[23:37:39]Nuki Advertising: Name: , Address: 54:d2:72:0a:b5:7f, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66430ab57fc8
[23:37:39]iBeacon ID: 004C Major: 17162 Minor: 46463 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -56
[23:37:39]disconnecting BLE on timeout
[23:37:39]disconnecting BLE on timeout
[23:37:40]disconnecting BLE on timeout
[23:37:40]disconnecting BLE on timeout
[23:37:40]disconnecting BLE on timeout
[23:37:40]disconnecting BLE on timeout
[23:37:40]disconnecting BLE on timeout
[23:37:41]disconnecting BLE on timeout
[23:37:41]disconnecting BLE on timeout
```

</details>

After this it keeps flooding the console with the disconnecting BLE on timeout message a couple of times per second.

mallorca2288, 2025-01-15:
Just FYI to anyone interested, the Ultra is back in stock in Nuki's website.

iranl, 2025-01-15:
Ultra support has been merged in master and is available in 9.08-master2.
Anyone with a Ultra lock can flash 9.08-master2 using OTA (Flash 9.07 using Webflash/esptool and than update to 9.08-master2 from the firmware update page by selection development version)

I will keep this topic open for now to get further feedback from @mallorca2288 and other users with an Ultra.

B-Hartley, 2025-01-16:
Hi,
As a new user.
I've installed the latest dev version from webupdate.
Connected Ultra lock sucessfully.
Seems to work well.

Will experiement with it over the next few days.
I have it in hybrid mode, so just need to fully understand what that means, but I'll get there.

Thanks for your work on this.  I'm unsure how stable the matter (thread) mqtt comms are from the lock, so having a backup plan is great.

A curiosity question....

I don't have a nuki door sensor, but I do have a door sensor as part of my alarm system that is available in home assistant.
   Could I somehow use the updates from that sensor to update the state of the door within the nuki system, so the lock would know that the door was open ?

mallorca2288, 2025-01-16:
I've been testing the latest binary and everything seems to be working ok.

Tried enabling and disabling keypad entries again and it worked as expected.

Hybrid connection to the lock has been stable without any issues.

The "BLE disconnecting" message flooding the logs has disappeared but now a new message keeps flooding the mqtt log (updated multiple times per second):
`iBeacon ID: 004C Major: 17162 Minor: 46463 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -56`

I'll keep testing and report back if I find any issues. Right now I feel like this build is very stable for releasing.

Thanks again!

iranl, 2025-01-16:
Disable all Nuki debug logging to stop logging the iBeacons

iranl, 2025-01-16:
> A curiosity question....
>
> I don't have a nuki door sensor, but I do have a door sensor as part of my alarm system that is available in home assistant. Could I somehow use the updates from that [...]

You are asking for emulation of Nuki accessories. The communication between a Nuki lock/opener and it's accessories is completely undocumented by Nuki and thus emulation is no easy task.
Altough trying to implement a function like this has been on my backlog for a long time, this will not be worked on any time soon.

B-Hartley, 2025-01-16:
I noticed you mention an advanced setting of force look door sensor connected?
What does that do?
Could I combine that with writing to the doorsensor topic.
Or will that have no effect ?

I think there is a typo in the below....

Advanced Configuration
The advanced configuration menu is not reachable from the main menu of the web configurator by default.
You can reach the menu directly by browsing to http://nukihubip/?get=advanced or enable showing it in the main menu by browsing to http://nukihubip/?get=debugon once (http://nukihubip/?get=debugoff to disable).

should say get?page=advanced  ?

Thanks,
Bruce.

iranl, 2025-01-17:
> should say get?page=advanced

Correct, tnx for the report, fixed.

> I noticed you mention an advanced setting of force look door sensor connected?
> What does that do?
> Could I combine that with writing to the doorsensor topic.
> Or will that have no effect ?

This only has an effect when you have a Nuki door sensor

iranl, 2025-01-17:
Closing this issue for now, please open a separate issue if any specific issues with the Ultra pop up.

closed 2025-01-17 by iranl