technyon/nuki_hub #525: Connection problems with 9.01

Ra72xx, 2024-11-17:
I realize that release 9.01 has been pulled from GitHub. Could you elaborate on the reasons? I currently run 9.01, at first without problems, but suddenly I've got massive connectivity problems. Might this be connected to the issues because of which you pulled 9.01?

- Nukihub looses connection to MQTT broker all the time
- Homeassistant seems not to pick up the MQTT connection again even if Nukihub is connected, so the Nuki devices are offline most of the time.
- Flashing from scratch with the web flasher always gives me 9.01 and not 9.0
- Updating from the integrated updater to the latest beta or some development does not work at all.
- Using the Wifi configuration menu to connect to another Wifi network does not work, the AP is reachable, but the configuration page does not show.

Some of the symptoms seem to be related to constant Wifi connections losses, because issues 1 and 2 can be cured by re-positioning NukiHub away (!) from the router, (however, it has worked in its original position literally for years without flaws).

Ra72xx, 2024-11-17:
I have to correct myself: Trying to flash to a beta from 9.01 does not brick the device. After some relatively long wait the hub returns online, but still with 9.01.

technyon, 2024-11-17:
9.02 is on track to be released soon, maybe it's a good idea to just wait for it. Also, it's hard to tell what's going on with flashing, be it OTA or web flasher. After that much trying to flash back and forth, the easiest thing to do maybe is to erase your ESP and do a factory reset to start with a clean slate, and then flash to 9.02 when it's released.

regevbr, 2024-11-20:
I have the same issues.
Upgraded to latest beta - 9.02-beta7 but still get those restarts due to NetworkTimeoutWatchdog. Would love to help resolve this

regevbr, 2024-11-20:
This didn't happen before 9.01 ofcourse

iranl, 2024-11-20:
We will not be debugging issues with 9.01, it was pulled for being to unstable.
Flash 9.0 using the release on GitHub and esptool or wait for 9.02 or flash a 9.02 beta.
Retrying OTA update to 9.02 beta should also work (there was an issue with converting WiFi settings going from 9.0/9.01 to 9.02 which required you to trigger the OTA update twice).

@regevbr: If you are having issues with 9.02 beta's create your own issue and provide USB/serial logs right before the NetworkTimeoutWatchdog triggers and follow the issue template to share your current settings.

Ra72xx, 2024-11-20:
I was not able to upgrade to a newer beta version from a configured 9.01. Which was a bit awkward, as the web flasher still always flashes 9.01 and not 9.0 (which means that currently everybody following the installation instructions will end up with an problematic firmware version).

However, I was able to upgrade from 9.01 to a newer revision as long as the 9.01 installation was not yet configured.

pilot1981, 2024-11-20:
> We will not be debugging issues with 9.01, it was pulled for being to unstable. Flash 9.0 using the release on GitHub and esptool or wait for 9.02 or flash a 9.02 beta. Retrying OTA update to 9.02 [...]

@technyon web updater of 9.0 firmware propose to update to 9.0.1, so many people colud update to this version and come in problems of 9.01...there is a way to stop it? you already removed 9.01 from stable version, but why web updater propose again 9.0.1 as the lastest stable version?

ohadbenita, 2024-11-24:
Since updating to 9.0.2 I have 4 devices unable to connect to MQTT, what's the needed information to be able to resolve this?

Ra72xx, 2024-11-24:
Seems to be a different problem than mine. Once I was able to get rid of 9.01 (which was not so easy),  I was able to use NukiHub again with more recent betas and also  9.02.
But the configuration has changed a bit, the common hub options are now separate from locks and openers, and I have the feeling that something got messed up here. 
But as I currently have a working 9.02(beta) setup (from scratch),  I'm closing my report now.

closed 2024-11-24 by iranl