technyon/nuki_hub #526: Battery information incorrect

isibizi, 2024-11-17:
Hello,
In Home Assistant, my battery shows as 87% full, but when I connect to my lock with the Nuki app, it shows 20%. I have already deleted the MQTT lock and cleared all entries in my MQTT database using MQTT Explorer, but it still shows the battery as 87% full.

technyon, 2024-11-18:
Please check using an MQTT client like MQTT explorer what value you receive for battery level after executing a lock action (nuki/lock/battery/level).

technyon, 2024-11-19:
Hi. I had the same problem until I noticed the path has changed and the old value was there from a previous build:

Old: Old: mqtt.0.nuki.battery.level
New: mqtt.0.nuki.lock.battery.level

isibizi, 2024-11-19:
What does it mean? Do you must change that in the next release? I have already cleared all previous entries with mqtt explorer.

technyon, 2024-11-20:
I'm using iobroker and had a UI element tied to that path. It was still showing the old value which was not updated anymore. For home assistant no idea, I don't use it. I guess auto discovery should point to the correct path. Maybe try clearing all data under under /nuki in MQTT and restart Nuki Hub. Maybe @iranl has an idea if the home assistant integration can cause some issue.

isibizi, 2024-11-20:
I have a suspicion: when the battery has the status “charging,” the hub might receive an incorrect value from the lock. Could that be possible?

iranl, 2024-11-20:
Definitely not a NukiHub issue based on the information provided.

Could be the NukiBLE library interpreting the raw data from the device incorrectly (because of recent firmware changes by Nuki?)
But seeing as firmware 3.10.7 is not really new I would be amazed if this is a general issue with this firmware that no one else has reported this earlier.

To debug this issue you would have to provide a USB/serial log generated with a debug build of NukiHub (switch to a debug build by browsing to http://NUKIHUBIP/otadebug and selecting a release/beta/master build)

isibizi, 2024-11-20:
I have now replaced my batteries, and the battery level is showing 100%. The MQTT path should be correct. I believe the issue occurs when the battery status is 'charged,' as it seems to send an incorrect level to the hub.

iranl, 2024-11-20:
How would the battery status be "charging" when not using a Power pack?
Or do you mean by "I have now replaced my batteries" that you have replaced your power pack?

isibizi, 2024-11-20:
I don't know, i don't use a powerpack, but the app had shows me "charging"

iranl, 2024-11-20:
Definitely sounds like a Nuki Lock 3 firmware bug in that case as it should never be in "charging" state when not using a Power Pack.

Closing as not NH related.

closed 2024-11-20 by iranl

technyon, 2024-11-21:
@isibizi I've created a debug build:

https://github.com/technyon/nuki_hub/actions/runs/11952124998

After flashing, you'll get an additional node next to "level" called "debug". This node will publish the raw "critical battery state" information (it's a bit of an odd name). This byte contains the battery level, and two more flags. From the API:

Bit 0	Smart Lock Battery State Critical
Bit 1	Charging Flag
Bit 2-7	Smart Lock Battery in %

It's published as decimal, so it has to be converted to binary manually. Please post the value here together with the reading in your app (level, charging, battery critical).

isibizi, 2024-11-21:
I think the problem is, the zip file contains only the esp32 firmware, i have a c3

technyon, 2024-11-22:
Hi. You're right there must have been some problem with the build job. I rebuilt the branch, this one has all assets:

https://github.com/technyon/nuki_hub/actions/runs/11971198737

isibizi, 2024-11-22:
Hello Jan-Ole,
I replaced my batteries and reinserted the old ones. In the app, I changed the automatic detection setting to manual and selected alkaline. Currently, the battery level appears to be correct. I can flash my hub with the debug firmware, but I don’t think debugging is possible with the current data. What do you think?

technyon, 2024-11-22:
Hi,

if the bug doesn't occur anymore, there's no point in testing.

isibizi, 2024-11-22:
Okay, I’ll keep an eye on it. If the error occurs again, I’ll post here. In the app, I no longer see the “charging” status; it currently shows the status “critical.”

technyon, 2024-11-22:
Critical means your batteries are almost empty, is that correct?

isibizi, 2024-11-22:
Yes, right! My old batteries had 18%, i think everything under 20% have the state „critical“ I replaced it with new ones again.