technyon/nuki_hub #527: Nuki Opener Ring event fired on reboot

kotylo, 2024-11-18:
### PROBLEM DESCRIPTION
When the Nuki Hub reboots, it changes the state of `events.ring` to new time, which triggers my automation to flash lights.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [x] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [x] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [x] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [X] Device used (e.g., ESP32-S3): ESP32 Devkit
- [x] Nuki Hub binary firmware version number used: 9.02-master21
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Provide the output of http://nukihubIP/info:
```
System Information
------------ NUKI HUB ------------
Version: 9.02-master21
Build: 11863591817.147.1
Build type: Release
Build date: 2024-11-15
Updater version: 9.02-master21
Updater build: 11863591817.147.1
Updater build date: 2024-11-15
Uptime (min): 8
Config version: 902
Last restart reason FW: NotApplicable
Last restart reason ESP: ESP_RST_PANIC: Software reset due to exception/panic.
Free internal heap: 59512
Total internal heap: 258504
PSRAM Available: No
Network task stack high watermark: 9196
Nuki task stack high watermark: 5088

------------ GENERAL SETTINGS ------------
Network task stack size: 12288
Nuki task stack size: 8192
Check for updates: No
Latest version: 9.02-master21
Allow update from MQTT: No
Web configurator username: ***
Web configurator password: ***
Web configurator enabled: Yes
Publish debug information enabled: No
MQTT log enabled: Yes
Webserial enabled: No
Bootloop protection enabled: No

------------ NETWORK ------------
Network device: Built-in Wi-Fi
Network connected: Yes
IP Address: 192.168.x.x
SSID: s16net
BSSID of AP: CC:F4:x
ESP32 MAC address: A8:42:x

------------ NETWORK SETTINGS ------------
Nuki Hub hostname: nukihub
DHCP enabled: Yes
RSSI Publish interval (s): 60
Find WiFi AP with strongest signal: No
Restart ESP32 on network disconnect enabled: Yes
MQTT Timeout until restart (s): 30

------------ MQTT ------------
MQTT connected: Yes
MQTT broker address: 192.168.x.x
MQTT broker port: 1883
MQTT username: ***
MQTT password: ***
MQTT base topic: nuki
MQTT SSL CA: Not set
MQTT SSL CRT: Not set
MQTT SSL Key: Not set

------------ BLUETOOTH ------------
Bluetooth TX power (dB): 9
Bluetooth command nr of retries: 3
Bluetooth command retry delay (ms): 100
Seconds until reboot when no BLE beacons received: 60

------------ QUERY / PUBLISH SETTINGS ------------
Lock/Opener state query interval (s): 1800
Publish Nuki device authorization log: No
Max authorization log entries to retrieve: 5
Battery state query interval (s): 1800
Most non-JSON MQTT topics disabled: No
Publish Nuki device config: Yes
Config query interval (s): 3600
Publish Keypad info: No
Keypad query interval (s): 1800
Enable Keypad control: Yes
Publish Keypad topic per entry: No
Publish Keypad codes: No
Allow checking Keypad codes: No
Max keypad entries to retrieve: 10
Publish timecontrol info: No
Keypad query interval (s): 1800
Enable timecontrol control: No
Publish timecontrol topic per entry: No
Max timecontrol entries to retrieve: 10

------------ HOME ASSISTANT ------------
Home Assistant auto discovery enabled: Yes
Home Assistant auto discovery topic: homeassistant/
Nuki Hub configuration URL for HA: http://192.168.x.x

------------ NUKI LOCK ------------
Lock enabled: Yes
Paired: Yes
Nuki Hub device ID: 4021158070
Nuki device ID: ***
Firmware version: 3.9.5
Hardware version: 2.10
Valid PIN set: No
Has door sensor: No
Has keypad: No
Timecontrol highest entries count: 0
Register as: Bridge

------------ HYBRID MODE ------------
Hybrid mode enabled: No

------------ NUKI LOCK ACL ------------
Lock: Allowed
Unlock: Allowed
Unlatch: Allowed
Lock N Go: Allowed
Lock N Go Unlatch: Allowed
Full Lock: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed

------------ NUKI LOCK CONFIG ACL ------------
Name: Disallowed
Latitude: Disallowed
Longitude: Disallowed
Auto Unlatch: Disallowed
Pairing enabled: Allowed
Button enabled: Allowed
LED flash enabled: Allowed
LED brightness: Disallowed
Timezone offset: Disallowed
DST mode: Disallowed
Fob Action 1: Disallowed
Fob Action 2: Disallowed
Fob Action 3: Disallowed
Single Lock: Allowed
Advertising Mode: Disallowed
Timezone ID: Disallowed
Unlocked Position Offset Degrees: Disallowed
Locked Position Offset Degrees: Disallowed
Single Locked Position Offset Degrees: Disallowed
Unlocked To Locked Transition Offset Degrees: Disallowed
Lock n Go timeout: Disallowed
Single button press action: Disallowed
Double button press action: Disallowed
Detached cylinder: Disallowed
Battery type: Disallowed
Automatic battery type detection: Disallowed
Unlatch duration: Disallowed
Auto lock timeout: Disallowed
Auto unlock disabled: Allowed
Nightmode enabled: Disallowed
Nightmode start time: Disallowed
Nightmode end time: Disallowed
Nightmode auto lock enabled: Disallowed
Nightmode auto unlock disabled: Disallowed
Nightmode immediate lock on start: Disallowed
Auto lock enabled: Allowed
Immediate auto lock enabled: Disallowed
Auto update enabled: Disallowed

------------ NUKI OPENER ------------
Opener enabled: Yes
Paired: Yes
Nuki Hub device ID: 4021158070
Nuki device ID: ***
Firmware version: 1.11.0
Hardware version: 4.17
Opener valid PIN set: Yes
Opener has keypad: No
Timecontrol highest entries count: 0
Register as: Bridge
Nuki Opener Lock/Unlock action set to Continuous mode in Home Assistant: No

------------ NUKI OPENER ACL ------------
Activate Ring-to-Open: Allowed
Deactivate Ring-to-Open: Allowed
Electric Strike Actuation: Allowed
Activate Continuous Mode: Allowed
Deactivate Continuous Mode: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed

------------ NUKI OPENER CONFIG ACL ------------
Name: Disallowed
Latitude: Disallowed
Longitude: Disallowed
Pairing enabled: Disallowed
Button enabled: Allowed
LED flash enabled: Allowed
Timezone offset: Disallowed
DST mode: Disallowed
Fob Action 1: Disallowed
Fob Action 2: Disallowed
Fob Action 3: Disallowed
Operating Mode: Disallowed
Advertising Mode: Disallowed
Timezone ID: Disallowed
Intercom ID: Disallowed
BUS mode Switch: Disallowed
Short Circuit Duration: Disallowed
Eletric Strike Delay: Allowed
Random Electric Strike Delay: Disallowed
Electric Strike Duration: Allowed
Disable RTO after ring: Disallowed
RTO timeout: Disallowed
Doorbell suppression: Disallowed
Doorbell suppression duration: Disallowed
Sound Ring: Allowed
Sound Open: Disallowed
Sound RTO: Disallowed
Sound CM: Disallowed
Sound confirmation: Disallowed
Sound level: Allowed
Single button press action: Disallowed
Double button press action: Disallowed
Battery type: Disallowed
Automatic battery type detection: Disallowed

------------ GPIO ------------
```

### TO REPRODUCE
Wait for some time for Nuki Hub to crash with `ESP_RST_PANIC: Software reset due to exception/panic.`. Probably rebooting hub via Web UI will also trigger this.

### EXPECTED BEHAVIOUR
Ring event should maintain previous value

### SCREENSHOTS
[image]

### ADDITIONAL CONTEXT
Buzzer MQTT Log changed to Subscribing to MQTT topic: nuki/lock/action
9:29:14 AM - 3 minutes ago
Buzzer MQTT Log changed to HASS setup for NukiHub completed.
9:29:14 AM - 3 minutes ago
Buzzer **Ring detected an event**
9:29:14 AM - 3 minutes ago
Buzzer MQTT Log changed to MQTT connected

**(Please, remember to close the issue when the problem has been addressed)**

iranl, 2024-11-18:
As you can see in your image the event sensor has a additional status. This status can be "ring", "ringlocked" and "standby". You can filter the "standby" state from your automation.

kotylo, 2024-11-18:
Ok, thanks, changed the when condition to trace **ringlocked**. Seems like ring is never happening by me.
```
platform: state
entity_id:
  - event.buzzer_ring
attribute: event_type
to: ringlocked
```
Although it would be nice to see the cleaner history in the graph. But i can live with extending automation traces to 200 and going back through them manually if needed...