technyon/nuki_hub #533: Having trouble with Keypad control using JSON

DaveGalbraith, 2024-11-22:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

I'm trying to set the time limited access for keypad permissions, however, I get the following results:

v9.0 - it never seems to work at all, commandResultJson reports configNotReady
v9.01 - I can change all the parameters I try apart from allowedUntil, commandResultJson reports invalidAllowedUntil

Example JSON used :
`{ "action": "add", "code": "589472", "name": "Test", "timeLimited": "1", "allowedFrom": "2024-04-12 10:00:00", "allowedUntil": "2034-04-12 10:00:00", "allowedWeekdays": [ "wed", "thu", "fri" ], "allowedFromTime": "08:00", "allowedUntilTime": "16:00" }`

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [ x] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [ x] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [ ] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [ ] Device used (e.g., ESP32-S3): ___ESP32__
- [ ] Nuki Hub binary firmware version number used: ___9.0 and 9.01__
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [ x ] Provide the output of http://nukihubIP/info:
```
System Information
Nuki Hub version: 9.00
Nuki Hub build: 10235422676.1.1
Nuki Hub build type: Release
run: true
confVersion: 901
deviceId: 3597738940
deviceIdOp: 3597738940
nukiId: ***
nukidOp: 
mqttbroker: 192.168.86.22
mqttport: 1883
mqttuser: ***
mqttpass: ***
mqttlog: false
checkupdates: true
websrvena: false
lockena: true
lockpin: 1
mqttpath: nuki
openerena: false
openerpin: 
openercont: false
mqttoppath: 
maxkpad: 7
opmaxkpad: 
maxtc: 
opmaxtc: 
enabtlprst: false
mqttca: 
mqttcrt: 
mqttkey: 
hassdiscovery: homeassistant
hassConfigUrl: 
buffsize: 
dhcpena: true
ipaddr: 
ipsub: 
ipgtw: 
dnssrv: 
nwhw: 1
nwwififb: false
rssipb: 60
hostname: nukihub
nwbestrssi: true
nettmout: 60
restdisc: true
rstbcn: 60
lockStInterval: 1800
tcPerEntry: true
kpPerEntry: true
configInterval: 3600
batInterval: 1800
kpInterval: 1800
kpCntrlEnabled: true
kpInfoEnabled: true
kpPubCode: false
tcCntrlEnabled: true
tcInfoEnabled: true
cnfInfoEnabled: false
regAsApp: true
regOpnAsApp: false
nrRetry: 3
rtryDelay: 100
crdusr: ***
crdpass: ***
disnonjson: false
pubAuth: true
pubdbg: false
prdtimeout: 
offHybrid: true
hybridTimer: 600
hybridAct: false
hybridRtry: false
hasmac: false
macb0: 
macb1: 
macb2: 
latest: 9.01
tsksznetw: 
tsksznuki: 
authmaxentry: 
kpmaxentry: 
tcmaxentry: 
updMqtt: true
showSecr: false
bleTxPwr: 
recNtwMqttDis: true
MQTT connected: Yes
Lock firmware version: 
Lock hardware version: 
Lock paired: Yes
Lock valid PIN set: Yes
Lock has door sensor: No
Lock has keypad: No
Lock ACL (Lock): Allowed
Lock ACL (Unlock): Allowed
Lock ACL (Unlatch): Allowed
Lock ACL (Lock N Go): Allowed
Lock ACL (Lock N Go Unlatch): Allowed
Lock ACL (Full Lock): Allowed
Lock ACL (Fob Action 1): Allowed
Lock ACL (Fob Action 2): Allowed
Lock ACL (Fob Action 3): Allowed
Lock config ACL (Name): Allowed
Lock config ACL (Latitude): Allowed
Lock config ACL (Longitude): Allowed
Lock config ACL (Auto Unlatch): Allowed
Lock config ACL (Pairing enabled): Allowed
Lock config ACL (Button enabled): Allowed
Lock config ACL (LED flash enabled): Allowed
Lock config ACL (LED brightness): Allowed
Lock config ACL (Timezone offset): Allowed
Lock config ACL (DST mode): Allowed
Lock config ACL (Fob Action 1): Allowed
Lock config ACL (Fob Action 2): Allowed
Lock config ACL (Fob Action 3): Allowed
Lock config ACL (Single Lock): Allowed
Lock config ACL (Advertising Mode): Allowed
Lock config ACL (Timezone ID): Allowed
Lock config ACL (Unlocked Position Offset Degrees): Allowed
Lock config ACL (Locked Position Offset Degrees): Allowed
Lock config ACL (Single Locked Position Offset Degrees): Allowed
Lock config ACL (Unlocked To Locked Transition Offset Degrees): Allowed
Lock config ACL (Lock n Go timeout): Allowed
Lock config ACL (Single button press action): Allowed
Lock config ACL (Double button press action): Allowed
Lock config ACL (Detached cylinder): Allowed
Lock config ACL (Battery type): Allowed
Lock config ACL (Automatic battery type detection): Allowed
Lock config ACL (Unlatch duration): Allowed
Lock config ACL (Auto lock timeout): Allowed
Lock config ACL (Auto unlock disabled): Allowed
Lock config ACL (Nightmode enabled): Allowed
Lock config ACL (Nightmode start time): Allowed
Lock config ACL (Nightmode end time): Allowed
Lock config ACL (Nightmode auto lock enabled): Allowed
Lock config ACL (Nightmode auto unlock disabled): Allowed
Lock config ACL (Nightmode immediate lock on start): Allowed
Lock config ACL (Auto lock enabled): Allowed
Lock config ACL (Immediate auto lock enabled): Allowed
Lock config ACL (Auto update enabled): Allowed
Network device: Built-in Wi-Fi
BSSID of AP: F0:72:EA:61:A0:CA
Uptime: 16 minutes
Heap: 72388
Stack watermarks: nw: 8204, nuki: 5040
Restart reason FW: OTACompleted
Restart reason ESP: ESP_RST_SW: Software reset via esp_restart.
```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Set json topics as above example and check the time limited access settings on the lock

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

allowedUntil time is correctly set 

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

iranl, 2024-11-22:
Already fixed in master and discussed earlier (among others) in https://github.com/technyon/nuki_hub/issues/469 and https://github.com/technyon/nuki_hub/pull/502.

Update to 9.02 beta or wait for release.

closed 2024-11-22 by iranl

DaveGalbraith, 2024-11-22:
Apologies, my search criteria must have been too narrowly defined. Thanks for the update