technyon/nuki_hub #537: v9.02 update mess - redundant information in MQTT

bcutter, 2024-11-24:
### PROBLEM DESCRIPTION
The update process from v9.01 to v9.02 was very tricky (update to latest version using the web interface, in the first attempt the ping timed out and a power cycle was needed, then switching from a strange partition "reboot to Nuki Hub" and initiating the update once again).

Now after successfully managing to upgrade to v9.02, there's a lot of things mixed up in MQTT looking at what is available in Home Assistant now:

Existing device, fine:
[image]

New device, unexpected and with redundant entities:
[image]

[image]
**Is this a desired outcome or the result of a faulty update?**
I don't get why one would need to have a sub set of entities redundantly available, one named by the lock (as it was for years) and in addition now one named by the Nuki Hub device.

If it's intentional (looks like as somehow the actual Nuki device is referenced as "linked device" at the bottom), I guess it's safe to disable all redundant entities or even delete the whole device, isn't it?

**Quite confusing.

Please clarify.**

Aside from this, in general, I'd really love to get back to Nuki Hub updates which a) don't create difficulties on really every single update and b) as a result don't waste a huge amount of time (for the first Nuki Hub now already one hour including this issure report). Stable updates are a long time ago unfortunately. Maybe we shouldn't pack too much changes in a minor release, looking at the quantity of 9.02 seems a bit oversized, just my 2 cents.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [X] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [X] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [ ] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [ ] Device used (e.g., ESP32-S3): ESP32 (LOLIN32 Lite)
- [X] Nuki Hub binary firmware version number used: v9.02 (build 11995573722.4.1)
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Provide the output of http://nukihubIP/info:
```
System Information

------------ NUKI HUB ------------
Version: 9.02
Build: 11995573722.4.1
Build type: Release
Build date: 2024-11-24
Updater version: 9.02
Updater build: 11995573722.4.1
Updater build date: 2024-11-24
Uptime (min): 12
Config version: 902
Last restart reason FW: RequestedViaWebServer
Last restart reason ESP: ESP_RST_SW: Software reset via esp_restart.
Free internal heap: 66976
Total internal heap: 260580
PSRAM Available: No
Network task stack high watermark: 8316
Nuki task stack high watermark: 5088

------------ GENERAL SETTINGS ------------
Network task stack size: 12288
Nuki task stack size: 8192
Check for updates: Yes
Latest version: 9.02
Allow update from MQTT: No
Web configurator username: ***
Web configurator password: ***
Web configurator enabled: Yes
Publish debug information enabled: No
MQTT log enabled: No
Webserial enabled: No
Bootloop protection enabled: No

------------ NETWORK ------------
Network device: Built-in Wi-Fi
Network connected: Yes
IP Address: xxx.xxx.xxx.xxx
SSID: XXX
BSSID of AP: XX:XX:XX:XX:XX:XX
ESP32 MAC address: XX:XX:XX:XX:XX:XX

------------ NETWORK SETTINGS ------------
Nuki Hub hostname: ESP-Nuki-Hub-XXX-XXX
DHCP enabled: Yes
RSSI Publish interval (s): Disabled
Find WiFi AP with strongest signal: Yes
Restart ESP32 on network disconnect enabled: Yes
MQTT Timeout until restart (s): Disabled

------------ MQTT ------------
MQTT connected: Yes
MQTT broker address: xxx.xxx.xxx.xxx
MQTT broker port: 1883
MQTT username: ***
MQTT password: ***
MQTT base topic: xxx
MQTT SSL CA: Not set
MQTT SSL CRT: Not set
MQTT SSL Key: Not set

------------ BLUETOOTH ------------
Bluetooth TX power (dB): 9
Bluetooth command nr of retries: 5
Bluetooth command retry delay (ms): 100
Seconds until reboot when no BLE beacons received: 60

------------ QUERY / PUBLISH SETTINGS ------------
Lock/Opener state query interval (s): 3600
Publish Nuki device authorization log: Yes
Max authorization log entries to retrieve: 5
Battery state query interval (s): 7200
Most non-JSON MQTT topics disabled: No
Publish Nuki device config: Yes
Config query interval (s): 3600
Publish Keypad info: No
Keypad query interval (s): 1800
Enable Keypad control: Yes
Publish Keypad topic per entry: No
Publish Keypad codes: No
Allow checking Keypad codes: No
Max keypad entries to retrieve: 10
Publish timecontrol info: No
Keypad query interval (s): 1800
Enable timecontrol control: No
Publish timecontrol topic per entry: No
Max timecontrol entries to retrieve: 10

------------ HOME ASSISTANT ------------
Home Assistant auto discovery enabled: Yes
Home Assistant auto discovery topic: homeassistant/
Nuki Hub configuration URL for HA: http://xxx.xxx.xxx.xxx

------------ NUKI LOCK ------------
Lock enabled: Yes
Paired: Yes
Nuki Hub device ID: XXXXXXXXX
Nuki device ID: ***
Firmware version: 3.10.7
Hardware version: 4.10
Valid PIN set: Yes
Has door sensor: No
Has keypad: No
Timecontrol highest entries count: 0
Register as: Bridge

------------ HYBRID MODE ------------
Hybrid mode enabled: No

------------ NUKI LOCK ACL ------------
Lock: Allowed
Unlock: Allowed
Unlatch: Allowed
Lock N Go: Allowed
Lock N Go Unlatch: Allowed
Full Lock: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed

------------ NUKI LOCK CONFIG ACL ------------
Name: Disallowed
Latitude: Disallowed
Longitude: Disallowed
Auto Unlatch: Disallowed
Pairing enabled: Allowed
Button enabled: Allowed
LED flash enabled: Allowed
LED brightness: Allowed
Timezone offset: Disallowed
DST mode: Disallowed
Fob Action 1: Disallowed
Fob Action 2: Disallowed
Fob Action 3: Disallowed
Single Lock: Allowed
Advertising Mode: Disallowed
Timezone ID: Disallowed
Unlocked Position Offset Degrees: Disallowed
Locked Position Offset Degrees: Disallowed
Single Locked Position Offset Degrees: Disallowed
Unlocked To Locked Transition Offset Degrees: Disallowed
Lock n Go timeout: Allowed
Single button press action: Allowed
Double button press action: Allowed
Detached cylinder: Disallowed
Battery type: Disallowed
Automatic battery type detection: Disallowed
Unlatch duration: Allowed
Auto lock timeout: Disallowed
Auto unlock disabled: Disallowed
Nightmode enabled: Disallowed
Nightmode start time: Disallowed
Nightmode end time: Disallowed
Nightmode auto lock enabled: Disallowed
Nightmode auto unlock disabled: Disallowed
Nightmode immediate lock on start: Disallowed
Auto lock enabled: Disallowed
Immediate auto lock enabled: Disallowed
Auto update enabled: Disallowed
Reboot Nuki: Allowed

------------ NUKI OPENER ------------
Opener enabled: No

------------ GPIO ------------

```

### TO REPRODUCE
Update NH v9.01 to NH v9.02

### EXPECTED BEHAVIOUR
No new device in MQTT, no redundant set of entities

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

bcutter, 2024-11-24:
Note:

Same update experience with a 2nd Nuki Hub from 9.01 to 9.02.

After ping times out, a physical power cycle of the ESP is needed.

Then, the web interface presents this (after the initial update tab times out long time ago, like http://esp-nuki-hub-door.domain/autoupdate?release=1&release=1&token=5028):

[image]

Reboot to Nuki Hub:
[image]

Then: nothing has happened, still on 9.01
[image]

So second attempt with the green button:
[image]

For half a second a two-liner is visible before the page times out (http://esp-nuki-hub-door.domain/autoupdate?release=1&release=1&token=1737). Reloading the default page after a few minutes now gives the 9.02 update.
[image]

Now I had to reboot the Nuki Hub once again to make HA entities show up again at all (all unavailable). Besides that, the same mess in MQTT (HA to be precise) of course.

I decided to simply disable all entities (because all of them are redundant end therefore completely useless) of the new additional Nuki Hub MQTT device.

BUT: the second Nuki Hub MQTT device doesn't even has the "linked device" entry: seems another thing is broken here... I'm pretty done with NH updates for today.

[image]

alexdelprete, 2024-11-25:
I confirm this new "nukihub" device creation with duplicate diagnostic entities created.

I wouldn't call this a "mess", we have a different view of that word. I have experienced real messes in HA, and this not even close to those. I feel your frustration, but let's not exaggerate, you can simply disable those duplicate entities for now, no real production issues.

bcutter, 2024-11-25:
Been there done that, see

> I decided to simply disable all entities (because all of them are redundant end therefore completely useless) of the new additional Nuki Hub MQTT device.

Maybe a bit lost in translation. Redundant trash data is what I call a mess. Nevermind, "solved" (worked around) that for the moment.

tmakowka, 2024-11-26:
Hi everyone,

i'm not sure whether it's the same issue, but i updated to 9.02 from 9.0 today. I also had to do the ota update process twice, but it worked afterwards.

The problem started after restart of Homeassistant. Now HA is telling me that 'This entity is no longer being provided by the mqtt integration. If the entity is no longer in use, delete it in settings.'

Neither nukihub nor the lock.* entities are active any more. I don't see a duplicate nukihub entity in my setup.

Anyone got an idea how to fix this?

bcutter, 2024-11-26:
Luckily I don't see that behavior which would be a real issue. But: I did not restart HA yet.

I guess you should at first check in your MQTT broker if the Nuki Hub topics exist and if - once you restart a Nuki Hub - the corresponding topics in the homeassistant domain are created.

Besides the overall update issues I tend to say this is a separate topic.

We'll see how the maintainers will handle this.

technyon, 2024-11-27:
So can this be closed then, or does it need further investigation?

alexdelprete, 2024-11-27:
> Of course it requires further investigation.

I had only 4 entities duplicated. BTW: they're not really duplicates, after the upgrade the ones on the lock were orphaned, I simply deleted them.

Can you show which entities are "duplicated"?

bcutter, 2024-11-27:
For me all entities remain on the lock, and many have been duplicated to the NH.

Those are the 13 duplicates (another entity ID, same function). Same for another NH:

[image]

alexdelprete, 2024-11-28:
Those are not duplicate entities, they are orphaned (they're not being updated), since the "live" entities are now on the nukihub device. You simply need to delete them.

Difference is I had 4 of them.

bcutter, 2024-11-28:
No, lock entities are not orphaned. Continue to work just fine afaics. One lock for example:

[image]

[image]

[image]

alexdelprete, 2024-11-28:
IP and FW Update sensors, for example, should only be available on the nukihub device, not on the lock device. There are other 2-3 diag. sensors like that. You should see those orphaned on the lock, and active on the nukihub device.

If that's not the case, @iranl will probably need more info to understand what's happening in your case.

In my case it was very simple, I just needed to delete 4 orphaned sensors from the lock.

bcutter, 2024-11-28:
That's what it is for all my NHs.

Tell me what is needed in terms of information/logs (or: safely purge all MQTT topics once and restart NHs?).

iranl, 2024-11-28:
The new NukiHub device separate from the lock and opener devices are there by design.

This is the proper way of providing hub/coordinator etc. devices in Home Assistant and is for example the same as how it is done by Zigbee2MQTT.
As the Nuki Lock, Nuki Opener and NukiHub are separate devices they should be presented in HA as separate devices,

This actually prevents duplicate entities going forward as previously NukiHub IP/Firmware etc. was provided to HA in both lock and opener devices, causing duplicates for users that used both these devices through one NukiHub device.

This was introduced based on a request in #494 in PR #515

It does cause some orphaned and/or duplicate entities in HA for users upgrading from earlier versions.

To fully remove these you need to search for MQTT topics containing `"nuki_` using a program like MQTT Explorer and delete all found topics. Reboot NukiHub afterwards and than remove any orphaned/unavailable entities from the Nuki lock and/or opener device from inside Home Assistant.

[image]

bcutter, 2024-11-28:
I'll have a look at MQTT topics at the weekend. You mean looking for the nuki_ like topics in the homeassistant or the nuki hub root topic?

Curious to see if really only 4 entities are affected. With a bit of entity renaming everything should be back to normal in theory...

---

In general I would like to be made aware of such changes in the release notes. As I read them. Sometimes twice. And then decide if the amount of changes makes this a 2 minutes stressless update or a one hour pain or something in between.

Better a note than discovering strange behavior (not even consistent: one nuki hub has a link to the lock in HA, the other one does not) afterwards with the need of creating issues.

Fundamental changes usually require such changes to be flagged as breaking change or similar. Better safe than sorry (a little extra note is overall probably less struggle than handling issues I guess).

bcutter, 2024-12-01:
OK folks @iranl I need a bit more precise instruction on how to check/clean and hopefully finally fix this.

> You mean looking for the nuki_ like topics in the homeassistant or the nuki hub root topic?

Otherwise it's literally _impossible_ to detect which entities should be orphaned on the lock device. As mentioned I have potentially 13 new entities on the NH device. Confusing.

bcutter, 2024-12-01:
To sum up my adventure of the last 3 hours which included fixing a endless reboot issue (https://github.com/technyon/nuki_hub/issues/538#issuecomment-2510231276):

- "switch.esp_nuki_hub_door_b_restart_nuki_hub" is still "unknown",
- same for "switch.esp_nuki_hub_door_b_nuki_hub_webserver_enabled"

there's no reset or webserver topic in the mqtt "nuki_door_b/maintenance" section which might be the reason for this. The Nuki Hub simply does not create it. It did not no matter which version of NH was running: 9.02 or 9.03-master6.

Done with NH (for today), gosh it totally broke my sunday evening. for nothing.

bcutter, 2024-12-24:
I purged all topics - both the nuki hub ones as well as the homeassistant autodiscovery ones and restarted the Nuki Hubs.

Afterwards I could somehow solve this. Unfortunately, from time to time redundant device entries of the new Nuki Hub device appear, names like "Unnamed device" or "Wifi signal strength", always with the same set of entities but no connection to the locks, also no manufacturer or model is given. No idea what's the reason for those redundant devices. The corresponding entities are named e. g. `sensor.none_nuki_hub_restart_reason` where "none" should usually be "esp_nuki_hub_door_b". Maybe it's a MQTT broker mess and outside of NH control, as a matter of fact all this got introduced with the dividing of lock/opener and NH topics/entities into a separate device.

Word. Happy holidays.

mundschenk-at, 2024-12-24:
Have you checked your YAML for manual MQTT configurations? I'm not sure "restart reason" was ever in the discovery payloads.

bcutter, 2024-12-24:
No idea what's the point with that. But my only custom sensor is a uptime sensor which was not available out of the box in the past. So those restart ones are stock.

alexdelprete, 2024-12-25:
The point is that you have a mess in your configuration and/or MQTT broker.

NH autodiscovery entities are ONLY the ones in the device page (see below): please show the entire device page with ALL entities, and highlight the entities there that are unavailable.

[image]

alexdelprete, 2024-12-25:
These are the manual entities I configured for NH:

```yaml
mqtt:
  sensor:
    # sensors for Nuki Hub
    - name: "Portoncino Log"
      state_topic: "nuki_hub/maintenance/log"
    - name: "Portoncino Uptime"
      state_topic: "nuki_hub/maintenance/uptime"
      device_class: duration
      unit_of_measurement: min
    - name: "Portoncino Free Heap"
      state_topic: "nuki_hub/maintenance/freeHeap"
    - name: "Portoncino Restart Reason FW"
      state_topic: "nuki_hub/maintenance/restartReasonNukiHub"
    - name: "Portoncino Restart Reason ESP"
      state_topic: "nuki_hub/maintenance/restartReasonNukiEsp"
```

mundschenk-at, 2024-12-25:
@alexdelprete The MQTT log & uptime are now included in auto-discovery. Available heap and the two restart reason entities still have to be configured entirely manually.

@bcutter Please check your configuration again. The restart reason has to be configured in your YAML somewhere (maybe a package?).

bcutter, 2024-12-25:
I don't get what we are talking about here. I just mentioned I managed to solve the initial redundant information by doing a full MQTT topics purge of all NH related HA autodiscovery topics as well as the NHs original topics, then restarted the NHs. I then removed the former unavailable entities on the lock devices.

Still on the HA side from time to time there pops up a new MQTT device in the MQTT integration with the very same set of entities like the new NH device introduced with 9.02.

So the issue's current name for me is "9.02 update mess - redundant information in MQTT integration in Home Assistant" and therefore very likely outside of this project's scope. MQTT/HA related, not NH.

Anyway it's a bit funny that 3 weeks after my comprehensive post with all the MQTT topics someone's now picking a detail and yelling "ESP reset is a custom thing, it's a custom thing!" :-D It made me smile :-)

As checked twice and proven already, my only YAML based custom MQTT entities related to locks are about uptime - a thing from the past. Everything else comes from NH via MQTT, so do the ESP reset topics and entities (`_restart_reason` and `_restart_reason_esp`). I have no idea where this would have been configured, maybe directly in NHs web interface. But as I'm discussing with project members here you would be aware of this I think.

locks.yaml
```
# Für Schlösser

# Am 2023-02-12 angelegt als Zusatz-Sensor um die Laufzeit der Nuki Hub-ESPs zu überwachen
# Gem. https://community.home-assistant.io/t/nuki-hub-firmware-for-the-esp32/417572
# Gem. https://www.home-assistant.io/integrations/sensor.mqtt/
mqtt:
  sensor:
    - name: "schloss_door_a_uptime_nukihub"
      state_topic: "nuki_door_a/maintenance/uptime"
      value_template: "{{ value }}"
      unit_of_measurement: Minuten
      unique_id: xyz
      state_class: total
      icon: mdi:clock
      device:
        identifiers: nuki_XXXXXXXX
      entity_category: diagnostic
    - name: "schloss_door_b_uptime_nukihub"
      state_topic: "nuki_door_b/maintenance/uptime"
      value_template: "{{ value }}"
      unit_of_measurement: Minuten
      unique_id: xyz
      state_class: total
      icon: mdi:clock
      device:
        identifiers: nuki_YYYYYYYY
      entity_category: diagnostic
```

So hopefully we won't waste more time here for irrelevant side discussions.

In case the redundant (yes they are and as long as they do I'll call them _redundant_) ghost devices pop up again I can publish the screenshots here so you might believe me. For the last times I instantly deleted those devices.

mundschenk-at, 2024-12-25:
So you don't think there is bug in NH anymore but there is a problem with your HA install/the MQTT integration? Then we should close the issue.

bcutter, 2024-12-25:
Well, just tell me. Are the redundant/ghost devices a symptom created by NH 9.02 update (first seen when updating to 9.02) and the issue is within MQTT or even HA or is it an NH issue? I have indicators for both, not sure.

alexdelprete, 2024-12-25:
> The MQTT log & uptime are now included in auto-discovery. Available heap and the two restart reason entities still have to be configured entirely manually.

That's exactly what I wrote in my post my friend. :)

@bcutter frankly speaking, with all due respect, your posts are really confusing and misleading, you talk about confusing issues, messing up everything, not giving any valuable info to allow us to help. And on top of this, the tone of your posts are really disturbing, you criticize the project like it was a commercial one that you paid for, suggesting there are bugs that are non-existent, while the problem is your HA configuration most probably.

So let me be very clear: this absurd and confusing thread will be closed, and if you still have issues, open a new one, but you'll have to be VERY SPECIFIC on the issue, you'll need to provide all the information needed to devs in order to REPRODUCE the issue, otherwise it will be impossible to help and it will be closed again.

Merry Christmas.

closed 2024-12-25 by alexdelprete

bcutter, 2024-12-25:
I think people are not reading things precisely (which happened to you too, see the start of your post). For example you were talking about ghost entities while I see ghost devices etc.

Then you come up with "these specific two topics (I had posted 3 weeks ago...) are not auto generated", pointing a few meters next to what this issue is about. That's confusing to me.

But well I think everyone is doing his best and I'm glad this great project exists. I'm lucky I finally have a stable release with 9.05 again after a lot of trouble in the past weeks. Hopefully my beta testing times are over for quite a while now ;-)

---

As mentioned I will provide screenshots once the ghost devices re-appear again. Precisely as requested: lock device, Nuki hub device, Nuki hub ghost device(s).

alexdelprete, 2024-12-25:
> I think people are not reading things precisely

Ever thought that the cause is that you write in a very confusing way?
But let's agree to disagree and move forward.

bcutter, 2024-12-26:
I don't see writing in a confusing way. For example I provided all information regarding custom MQTT entities - extensively - after I was blamed "you must have configured it manually!". After I provided proof no one even responds to that information.

Ah well. Just be patient and wait for my ghost device screenshots.

As this is now a meta discussion on how people read and interpret information here, every additional word until then is a waste of time for the one writing and the ones reading them. I'm out, finger's crossed the redundant devices will never appear again.

alexdelprete, 2024-12-26:
> For example I provided all information regarding custom MQTT entities - extensively - after I was blamed "you must have configured it manually!". After I provided proof no one even responds to that information.

You have no idea (or weird idea) of what a "proof" is probably. You said you had unavailable entities and we told you that those were NOT created by NH autodiscovery (check source code before replying), so logically we replied telling you that the ONLY rational reason is that you configured manual MQTT entities, but you said "no I didn't" and you think that's a PROOF. Then you're here telling us to "wait" for ghost devices to appear...ROTFL. We're not ghostbusters. :)

But I agree with you on last stament: this whole thread was/is just a waste of time, we live in two different multiverses from a technical standpoint. Keep chasing your ghost devices, but don't expect us to be able to help you.

bcutter, 2024-12-26:
Doing a system wide (VS Code takes care perfectly) search for "mqtt:" and checking all results for any lock/Nuki hub related sections and providing the only one found is the best I can do.

But as you are one of the "I need to have the last word in every discussion" character you will surely have another idea on this. Welcome on my friend's list.

mundschenk-at, 2024-12-26:
@alexdelprete Regarding the YAML: I answered based on the initial post before the edits (got it via e-mail).