technyon/nuki_hub #538: Reset flag not unset after successful reboot (stays 1)

bcutter, 2024-11-24:
### PROBLEM DESCRIPTION
With v9.02 when performing a reboot initiated by setting the `switch.lock_restart_nuki_hub` switch to on, after the reboot the switch stays an and so does the flag in the MQTT broker:
[image]

[image]

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [X] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [X] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [X] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [X] Device used (e.g., ESP32-S3): ESP32 (LOLIN32 Lite)
- [X] Nuki Hub binary firmware version number used: v9.02 (build 11995573722.4.1) 
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Provide the output of http://nukihubIP/info:
```
System Information

------------ NUKI HUB ------------
Version: 9.02
Build: 11995573722.4.1
Build type: Release
Build date: 2024-11-24
Updater version: 9.02
Updater build: 11995573722.4.1
Updater build date: 2024-11-24
Uptime (min): 12
Config version: 902
Last restart reason FW: RequestedViaWebServer
Last restart reason ESP: ESP_RST_SW: Software reset via esp_restart.
Free internal heap: 66976
Total internal heap: 260580
PSRAM Available: No
Network task stack high watermark: 8316
Nuki task stack high watermark: 5088

------------ GENERAL SETTINGS ------------
Network task stack size: 12288
Nuki task stack size: 8192
Check for updates: Yes
Latest version: 9.02
Allow update from MQTT: No
Web configurator username: ***
Web configurator password: ***
Web configurator enabled: Yes
Publish debug information enabled: No
MQTT log enabled: No
Webserial enabled: No
Bootloop protection enabled: No

------------ NETWORK ------------
Network device: Built-in Wi-Fi
Network connected: Yes
IP Address: xxx.xxx.xxx.xxx
SSID: XXX
BSSID of AP: XX:XX:XX:XX:XX:XX
ESP32 MAC address: XX:XX:XX:XX:XX:XX

------------ NETWORK SETTINGS ------------
Nuki Hub hostname: ESP-Nuki-Hub-XXX-XXX
DHCP enabled: Yes
RSSI Publish interval (s): Disabled
Find WiFi AP with strongest signal: Yes
Restart ESP32 on network disconnect enabled: Yes
MQTT Timeout until restart (s): Disabled

------------ MQTT ------------
MQTT connected: Yes
MQTT broker address: xxx.xxx.xxx.xxx
MQTT broker port: 1883
MQTT username: ***
MQTT password: ***
MQTT base topic: xxx
MQTT SSL CA: Not set
MQTT SSL CRT: Not set
MQTT SSL Key: Not set

------------ BLUETOOTH ------------
Bluetooth TX power (dB): 9
Bluetooth command nr of retries: 5
Bluetooth command retry delay (ms): 100
Seconds until reboot when no BLE beacons received: 60

------------ QUERY / PUBLISH SETTINGS ------------
Lock/Opener state query interval (s): 3600
Publish Nuki device authorization log: Yes
Max authorization log entries to retrieve: 5
Battery state query interval (s): 7200
Most non-JSON MQTT topics disabled: No
Publish Nuki device config: Yes
Config query interval (s): 3600
Publish Keypad info: No
Keypad query interval (s): 1800
Enable Keypad control: Yes
Publish Keypad topic per entry: No
Publish Keypad codes: No
Allow checking Keypad codes: No
Max keypad entries to retrieve: 10
Publish timecontrol info: No
Keypad query interval (s): 1800
Enable timecontrol control: No
Publish timecontrol topic per entry: No
Max timecontrol entries to retrieve: 10

------------ HOME ASSISTANT ------------
Home Assistant auto discovery enabled: Yes
Home Assistant auto discovery topic: homeassistant/
Nuki Hub configuration URL for HA: http://xxx.xxx.xxx.xxx

------------ NUKI LOCK ------------
Lock enabled: Yes
Paired: Yes
Nuki Hub device ID: XXXXXXXXX
Nuki device ID: ***
Firmware version: 3.10.7
Hardware version: 4.10
Valid PIN set: Yes
Has door sensor: No
Has keypad: No
Timecontrol highest entries count: 0
Register as: Bridge

------------ HYBRID MODE ------------
Hybrid mode enabled: No

------------ NUKI LOCK ACL ------------
Lock: Allowed
Unlock: Allowed
Unlatch: Allowed
Lock N Go: Allowed
Lock N Go Unlatch: Allowed
Full Lock: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed

------------ NUKI LOCK CONFIG ACL ------------
Name: Disallowed
Latitude: Disallowed
Longitude: Disallowed
Auto Unlatch: Disallowed
Pairing enabled: Allowed
Button enabled: Allowed
LED flash enabled: Allowed
LED brightness: Allowed
Timezone offset: Disallowed
DST mode: Disallowed
Fob Action 1: Disallowed
Fob Action 2: Disallowed
Fob Action 3: Disallowed
Single Lock: Allowed
Advertising Mode: Disallowed
Timezone ID: Disallowed
Unlocked Position Offset Degrees: Disallowed
Locked Position Offset Degrees: Disallowed
Single Locked Position Offset Degrees: Disallowed
Unlocked To Locked Transition Offset Degrees: Disallowed
Lock n Go timeout: Allowed
Single button press action: Allowed
Double button press action: Allowed
Detached cylinder: Disallowed
Battery type: Disallowed
Automatic battery type detection: Disallowed
Unlatch duration: Allowed
Auto lock timeout: Disallowed
Auto unlock disabled: Disallowed
Nightmode enabled: Disallowed
Nightmode start time: Disallowed
Nightmode end time: Disallowed
Nightmode auto lock enabled: Disallowed
Nightmode auto unlock disabled: Disallowed
Nightmode immediate lock on start: Disallowed
Auto lock enabled: Disallowed
Immediate auto lock enabled: Disallowed
Auto update enabled: Disallowed
Reboot Nuki: Allowed

------------ NUKI OPENER ------------
Opener enabled: No

------------ GPIO ------------

```

### TO REPRODUCE
See problem description

### EXPECTED BEHAVIOUR
Flag is set to 0. Therefore the HA entity will reflect the finished update.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
Just another thing broken after an update :-(

**(Please, remember to close the issue when the problem has been addressed)**

bcutter, 2024-11-24:
Same behaviour for a second NH ESP. One has to turn off the switch (using HA for example) manually, otherwise "reset = 1" will be stored in MQTT forever.

technyon, 2024-11-25:
Yes I can confirm the reset node isn't set back to 0

jolichter, 2024-11-27:
Interesting, I don’t actually need the reset switch, but I tested it in ioBroker anyway (nuki/maintenance/reset).

The value is `null`, and after entering a `1` (true/false doesn't work), the value persists, causing the Nuki Hub to repeatedly restart in a loop. As a workaround, I would integrate this datapoint as a button and not a switch or use a JS (Blockly) script to reset the value to `null` after setting it to `1`.

bcutter, 2024-11-27:
HA/ioBroker/... implementation doesn't matter.

The issue is in the NH firmware so it does not reset the MQTT topic. Everything afterwards is smart home system specific and irrelevant here I guess.

jolichter, 2024-11-27:
Sure, I just wanted to confirm the matter and share how I would solve it with ioBroker as a workaround. This should also work with HA?

technyon, 2024-11-28:
Yes. Actually there is code to reset the node, I'm not sure why it doesn't work, it needs further investigation.

iranl, 2024-11-28:
This is already fixed by @technyon in commit https://github.com/technyon/nuki_hub/commit/dab9baf9cad95165b246063b0622ce7def7fc381

Please update to 9.03-master2 or later if you need this fixed right away.

closed 2024-11-28 by iranl

bcutter, 2024-11-28:
Thx, I'll wait for "stable" 9.03 release.

bcutter, 2024-12-01:
Somehow one of my NHs ran into the endless restart loop. And there's no reset topic in MQTT broker. How to upgrade to the 9.03-master2 @iranl ?

technyon, 2024-12-02:
This should be fixed in the master branch.

bcutter, 2024-12-16:
So the fix is part of the new 9.03 update?