technyon/nuki_hub #542: LAN8720A unable to initialise ethernet connection fails back to wifi

Bucky-22, 2024-11-27:
### PROBLEM DESCRIPTION
Using a ESP32 PoE devkit i am unable to set the required pins in custom ethernet, cauing it to fall back to wifi.
settings should be as follows
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO17_OUT
  phy_addr: 0
  power_pin: GPIO12

However im unbale to set addr to 0 it always changes back to 1 on reboot.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [x] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [x] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [x] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [x] Device used (e.g., ESP32-S3): ESP32 PoE Development Kit 
- [ ] Nuki Hub binary firmware version number used: 9.02
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Provide the output of http://nukihubIP/info:
```
System Information

------------ NUKI HUB ------------
Version: 9.02
Build: 11995573722.4.1
Build type: Release
Build date: 2024-11-24
Updater version: 
Updater build: 
Updater build date: 
Uptime (min): 17
Config version: 902
Last restart reason FW: NetworkTimeoutWatchdog
Last restart reason ESP: ESP_RST_SW: Software reset via esp_restart.
Free internal heap: 39884
Total internal heap: 260580
PSRAM Available: No
Network task stack high watermark: 10160
Nuki task stack high watermark: 6676

------------ GENERAL SETTINGS ------------
Network task stack size: 12288
Nuki task stack size: 8192
Check for updates: Yes
Latest version: 
Allow update from MQTT: No
Web configurator username: Not set
Web configurator password: Not set
Web configurator enabled: Yes
Publish debug information enabled: No
MQTT log enabled: No
Webserial enabled: No
Bootloop protection enabled: No

------------ NETWORK ------------
Network device: Built-in Wi-Fi
Network connected: Yes
IP Address: 192.168.3.252
SSID: iot
BSSID of AP: B2:8B:A9:DB:D4:75
ESP32 MAC address: 48:E7:29:20:19:80

------------ NETWORK SETTINGS ------------
Nuki Hub hostname: nukihub
DHCP enabled: Yes
RSSI Publish interval (s): 60
Find WiFi AP with strongest signal: Yes
Restart ESP32 on network disconnect enabled: No
MQTT Timeout until restart (s): 60

------------ MQTT ------------
MQTT connected: No
MQTT broker address: 
MQTT broker port: 1883
MQTT username: Not set
MQTT password: Not set
MQTT base topic: nukihub
MQTT SSL CA: Not set
MQTT SSL CRT: Not set
MQTT SSL Key: Not set

------------ BLUETOOTH ------------
Bluetooth TX power (dB): 9
Bluetooth command nr of retries: 3
Bluetooth command retry delay (ms): 100
Seconds until reboot when no BLE beacons received: 60

------------ QUERY / PUBLISH SETTINGS ------------
Lock/Opener state query interval (s): 1800
Publish Nuki device authorization log: No
Max authorization log entries to retrieve: 5
Battery state query interval (s): 1800
Most non-JSON MQTT topics disabled: No
Publish Nuki device config: Yes
Config query interval (s): 3600
Publish Keypad info: No
Keypad query interval (s): 1800
Enable Keypad control: Yes
Publish Keypad topic per entry: No
Publish Keypad codes: No
Allow checking Keypad codes: No
Max keypad entries to retrieve: 10
Publish timecontrol info: No
Keypad query interval (s): 1800
Enable timecontrol control: No
Publish timecontrol topic per entry: No
Max timecontrol entries to retrieve: 10

------------ HOME ASSISTANT ------------
Home Assistant auto discovery enabled: No

------------ NUKI LOCK ------------
Lock enabled: Yes
Paired: No
Nuki Hub device ID: 2456822488
Nuki device ID: Not set
Firmware version: 
Hardware version: 
Valid PIN set: -
Has door sensor: No
Has keypad: No
Timecontrol highest entries count: 0
Register as: Bridge

------------ HYBRID MODE ------------
Hybrid mode enabled: No

------------ NUKI LOCK ACL ------------
Lock: Allowed
Unlock: Allowed
Unlatch: Allowed
Lock N Go: Allowed
Lock N Go Unlatch: Allowed
Full Lock: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed

------------ NUKI LOCK CONFIG ACL ------------
Name: Allowed
Latitude: Allowed
Longitude: Allowed
Auto Unlatch: Allowed
Pairing enabled: Allowed
Button enabled: Allowed
LED flash enabled: Allowed
LED brightness: Allowed
Timezone offset: Allowed
DST mode: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed
Single Lock: Allowed
Advertising Mode: Allowed
Timezone ID: Allowed
Unlocked Position Offset Degrees: Allowed
Locked Position Offset Degrees: Allowed
Single Locked Position Offset Degrees: Allowed
Unlocked To Locked Transition Offset Degrees: Allowed
Lock n Go timeout: Allowed
Single button press action: Allowed
Double button press action: Allowed
Detached cylinder: Allowed
Battery type: Allowed
Automatic battery type detection: Allowed
Unlatch duration: Allowed
Auto lock timeout: Allowed
Auto unlock disabled: Allowed
Nightmode enabled: Allowed
Nightmode start time: Allowed
Nightmode end time: Allowed
Nightmode auto lock enabled: Allowed
Nightmode auto unlock disabled: Allowed
Nightmode immediate lock on start: Allowed
Auto lock enabled: Allowed
Immediate auto lock enabled: Allowed
Auto update enabled: Allowed
Reboot Nuki: Allowed

------------ NUKI OPENER ------------
Opener enabled: No

------------ GPIO ------------

```

### TO REPRODUCE
Flash with firmware, input require pin settings for ethernet, reboot

### EXPECTED BEHAVIOUR
Ethernet to initalise 

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

technyon, 2024-11-28:
Hi,

I'm using a LAN8720 chip on an Olimex board without problems. You're saying you use a devkit, does this mean you did the wiring yourself?

Bucky-22, 2024-11-28:
Hi Jan,

Good to know it does work, no I purchased it prebuilt

https://shop.everythingsmart.io/products/est-poe-32-esp32-poe-development-board?srsltid=AfmBOorJKo4Qxfm1gymdnry-PdzUJSiststcsZjxxbb0vZAXdSN6DBgH

A basic script to run it up as a BT proxy works so I know the hardware is good, as I say i think I need to be able to set the physical address to 0 but on custom ethernet am forced to 1.

Cheers

iranl, 2024-11-28:
Should be fixed by #543.

Please update to 9.03-master3 (will be available 2024-11-29 at around 02:00 GMT+2) to confirm.

closed 2024-11-28 by iranl

Bucky-22, 2024-11-30:
That looks to have resolved the issue, many thanks.

PR #543 (2024-11-28): Fix custom ethernet PHY addr save
## Description:

**Related issue (if applicable):** fixes #542 

## Checklist:
  - [x] The pull request is done against the latest master branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works
  - [x] I accept the [CLA](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md#contributor-license-agreement-cla).