technyon/nuki_hub #547: v9.02 won't reboot by itself (if it hangs)

kotylo, 2024-11-28:
### PROBLEM DESCRIPTION
Nuki Hub stopped rebooting by itself. In older versions i didn't have to go near it forever. It just silently restarted in case of crash (?), but not anymore. At least not always as seen on screenshot below.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [x] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [x] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [x] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [x] Device used (e.g., ESP32-S3): ESP32 Dev Kit
- [x] Nuki Hub binary firmware version number used: 9.02
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Provide the output of http://nukihubIP/info:
```
System Information
------------ NUKI HUB ------------
Version: 9.02
Build: 11995573722.4.1
Build type: Release
Build date: 2024-11-24
Updater version: 9.02
Updater build: 11995573722.4.1
Updater build date: 2024-11-24
Uptime (min): 0
Config version: 902
Last restart reason FW: NotApplicable
Last restart reason ESP: ESP_RST_POWERON: Reset due to power-on event.
Free internal heap: 61920
Total internal heap: 260580
PSRAM Available: No
Network task stack high watermark: 8652
Nuki task stack high watermark: 5024

------------ GENERAL SETTINGS ------------
Network task stack size: 12288
Nuki task stack size: 8192
Check for updates: No
Latest version: 9.02
Allow update from MQTT: No
Web configurator username: ***
Web configurator password: ***
Web configurator enabled: Yes
Publish debug information enabled: No
MQTT log enabled: Yes
Webserial enabled: No
Bootloop protection enabled: No

------------ NETWORK ------------
Network device: Built-in Wi-Fi
Network connected: Yes
IP Address: 192.168.x.40
SSID: s16net
BSSID of AP: CC:F4:11:xx
ESP32 MAC address: A8:42:E3:xx

------------ NETWORK SETTINGS ------------
Nuki Hub hostname: nukihub
DHCP enabled: Yes
RSSI Publish interval (s): 60
Find WiFi AP with strongest signal: No
Restart ESP32 on network disconnect enabled: Yes
MQTT Timeout until restart (s): 30

------------ MQTT ------------
MQTT connected: Yes
MQTT broker address: 192.168.86.150
MQTT broker port: 1883
MQTT username: ***
MQTT password: ***
MQTT base topic: nuki
MQTT SSL CA: Not set
MQTT SSL CRT: Not set
MQTT SSL Key: Not set

------------ BLUETOOTH ------------
Bluetooth TX power (dB): 9
Bluetooth command nr of retries: 3
Bluetooth command retry delay (ms): 100
Seconds until reboot when no BLE beacons received: 60

------------ QUERY / PUBLISH SETTINGS ------------
Lock/Opener state query interval (s): 1800
Publish Nuki device authorization log: No
Max authorization log entries to retrieve: 5
Battery state query interval (s): 1800
Most non-JSON MQTT topics disabled: No
Publish Nuki device config: Yes
Config query interval (s): 3600
Publish Keypad info: No
Keypad query interval (s): 1800
Enable Keypad control: Yes
Publish Keypad topic per entry: No
Publish Keypad codes: No
Allow checking Keypad codes: No
Max keypad entries to retrieve: 10
Publish timecontrol info: No
Keypad query interval (s): 1800
Enable timecontrol control: No
Publish timecontrol topic per entry: No
Max timecontrol entries to retrieve: 10

------------ HOME ASSISTANT ------------
Home Assistant auto discovery enabled: Yes
Home Assistant auto discovery topic: homeassistant/
Nuki Hub configuration URL for HA: http://192.168.x.40

------------ NUKI LOCK ------------
Lock enabled: Yes
Paired: Yes
Nuki Hub device ID: 4021158070
Nuki device ID: ***
Firmware version: 3.9.5
Hardware version: 2.10
Valid PIN set: No
Has door sensor: No
Has keypad: No
Timecontrol highest entries count: 0
Register as: Bridge

------------ HYBRID MODE ------------
Hybrid mode enabled: No

------------ NUKI LOCK ACL ------------
Lock: Allowed
Unlock: Allowed
Unlatch: Allowed
Lock N Go: Allowed
Lock N Go Unlatch: Allowed
Full Lock: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed

------------ NUKI LOCK CONFIG ACL ------------
Name: Disallowed
Latitude: Disallowed
Longitude: Disallowed
Auto Unlatch: Disallowed
Pairing enabled: Allowed
Button enabled: Allowed
LED flash enabled: Allowed
LED brightness: Disallowed
Timezone offset: Disallowed
DST mode: Disallowed
Fob Action 1: Disallowed
Fob Action 2: Disallowed
Fob Action 3: Disallowed
Single Lock: Allowed
Advertising Mode: Disallowed
Timezone ID: Disallowed
Unlocked Position Offset Degrees: Disallowed
Locked Position Offset Degrees: Disallowed
Single Locked Position Offset Degrees: Disallowed
Unlocked To Locked Transition Offset Degrees: Disallowed
Lock n Go timeout: Disallowed
Single button press action: Disallowed
Double button press action: Disallowed
Detached cylinder: Disallowed
Battery type: Disallowed
Automatic battery type detection: Disallowed
Unlatch duration: Disallowed
Auto lock timeout: Disallowed
Auto unlock disabled: Allowed
Nightmode enabled: Disallowed
Nightmode start time: Disallowed
Nightmode end time: Disallowed
Nightmode auto lock enabled: Disallowed
Nightmode auto unlock disabled: Disallowed
Nightmode immediate lock on start: Disallowed
Auto lock enabled: Allowed
Immediate auto lock enabled: Disallowed
Auto update enabled: Disallowed
Reboot Nuki: Disallowed

------------ NUKI OPENER ------------
Opener enabled: Yes
Paired: Yes
Nuki Hub device ID: 4021158070
Nuki device ID: ***
Firmware version: 1.11.0
Hardware version: 4.17
Opener valid PIN set: Yes
Opener has keypad: No
Timecontrol highest entries count: 0
Register as: Bridge
Nuki Opener Lock/Unlock action set to Continuous mode in Home Assistant: No

------------ NUKI OPENER ACL ------------
Activate Ring-to-Open: Allowed
Deactivate Ring-to-Open: Allowed
Electric Strike Actuation: Allowed
Activate Continuous Mode: Allowed
Deactivate Continuous Mode: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed

------------ NUKI OPENER CONFIG ACL ------------
Name: Disallowed
Latitude: Disallowed
Longitude: Disallowed
Pairing enabled: Disallowed
Button enabled: Allowed
LED flash enabled: Allowed
Timezone offset: Disallowed
DST mode: Disallowed
Fob Action 1: Disallowed
Fob Action 2: Disallowed
Fob Action 3: Disallowed
Operating Mode: Disallowed
Advertising Mode: Disallowed
Timezone ID: Disallowed
Intercom ID: Disallowed
BUS mode Switch: Disallowed
Short Circuit Duration: Disallowed
Eletric Strike Delay: Allowed
Random Electric Strike Delay: Disallowed
Electric Strike Duration: Allowed
Disable RTO after ring: Disallowed
RTO timeout: Disallowed
Doorbell suppression: Disallowed
Doorbell suppression duration: Disallowed
Sound Ring: Allowed
Sound Open: Disallowed
Sound RTO: Disallowed
Sound CM: Disallowed
Sound confirmation: Disallowed
Sound level: Allowed
Single button press action: Disallowed
Double button press action: Disallowed
Battery type: Disallowed
Automatic battery type detection: Disallowed
Reboot Nuki: Disallowed

------------ GPIO ------------
```

### TO REPRODUCE
Leave running for some time, eventually it dies and won't recover.
The red onboard LED is still on when it "hangs". Manually plugging it out and in boots it up again:
[image]

### EXPECTED BEHAVIOUR
Reboot automatically if it "hangs", as it was with pre-9 version.

### SCREENSHOTS

### ADDITIONAL CONTEXT
Some logs when it becomes unavailable. But i don't see any pattern there:
```
Home Door Nuki Hub restart reason ESP became unavailable
7:42:08 PM
Home Door MQTT Log changed to networkTask is running
7:40:16 PM
Home Door MQTT Log changed to Done querying lock state
7:39:20 PM
```

**(Please, remember to close the issue when the problem has been addressed)**

combine-space, 2024-11-29:
I also get connection loss with 9.02 after some hours of uptime. I don't know if its because it wont reboot, it's kind of hard to debug. At least I can report that the ring detection works again with this version 🎉.

Device: `M5Stack Atom Lite`

Info:
```
System Information
------------ NUKI HUB ------------
Version: 9.02
Build: 11995573722.4.1
Build type: Release
Build date: 2024-11-24
Updater version: 
Updater build: 
Updater build date: 
Uptime (min): 2
Config version: 902
Last restart reason FW: NotApplicable
Last restart reason ESP: ESP_RST_POWERON: Reset due to power-on event.
Free internal heap: 66396
Total internal heap: 260580
PSRAM Available: No
Network task stack high watermark: 8320
Nuki task stack high watermark: 5092

------------ GENERAL SETTINGS ------------
Network task stack size: 12288
Nuki task stack size: 8192
Check for updates: Yes
Latest version: 9.02
Allow update from MQTT: No
Web configurator username: ***
Web configurator password: ***
Web configurator enabled: Yes
Publish debug information enabled: No
MQTT log enabled: Yes
Webserial enabled: No
Bootloop protection enabled: No

------------ NETWORK ------------
Network device: Built-in Wi-Fi
Network connected: Yes
IP Address: 10.0.x.10
SSID: ***
BSSID of AP: ***
ESP32 MAC address: ***

------------ NETWORK SETTINGS ------------
Nuki Hub hostname: nukihub
DHCP enabled: Yes
RSSI Publish interval (s): 60
Find WiFi AP with strongest signal: No
Restart ESP32 on network disconnect enabled: Yes
MQTT Timeout until restart (s): 60

------------ MQTT ------------
MQTT connected: Yes
MQTT broker address: 192.168.x.13
MQTT broker port: 1883
MQTT username: ***
MQTT password: ***
MQTT base topic: nukihub
MQTT SSL CA: Not set
MQTT SSL CRT: Not set
MQTT SSL Key: Not set

------------ BLUETOOTH ------------
Bluetooth TX power (dB): 9
Bluetooth command nr of retries: 3
Bluetooth command retry delay (ms): 100
Seconds until reboot when no BLE beacons received: 60

------------ QUERY / PUBLISH SETTINGS ------------
Lock/Opener state query interval (s): 1800
Publish Nuki device authorization log: No
Max authorization log entries to retrieve: 5
Battery state query interval (s): 1800
Most non-JSON MQTT topics disabled: No
Publish Nuki device config: Yes
Config query interval (s): 300
Publish Keypad info: No
Keypad query interval (s): 1800
Enable Keypad control: Yes
Publish Keypad topic per entry: No
Publish Keypad codes: No
Allow checking Keypad codes: No
Max keypad entries to retrieve: 10
Publish timecontrol info: No
Keypad query interval (s): 1800
Enable timecontrol control: No
Publish timecontrol topic per entry: No
Max timecontrol entries to retrieve: 10

------------ HOME ASSISTANT ------------
Home Assistant auto discovery enabled: Yes
Home Assistant auto discovery topic: homeassistant/
Nuki Hub configuration URL for HA: http://10.0.x.10

------------ NUKI LOCK ------------
Lock enabled: No

------------ NUKI OPENER ------------
Opener enabled: Yes
Paired: Yes
Nuki Hub device ID: 2328545398
Nuki device ID: ***
Firmware version: 1.10.1
Hardware version: 4.17
Opener valid PIN set: Yes
Opener has keypad: Yes
Keypad highest entries count: 0
Timecontrol highest entries count: 0
Register as: Bridge
Nuki Opener Lock/Unlock action set to Continuous mode in Home Assistant: No

------------ NUKI OPENER ACL ------------
Activate Ring-to-Open: Allowed
Deactivate Ring-to-Open: Allowed
Electric Strike Actuation: Allowed
Activate Continuous Mode: Allowed
Deactivate Continuous Mode: Allowed
Fob Action 1: Disallowed
Fob Action 2: Disallowed
Fob Action 3: Disallowed

------------ NUKI OPENER CONFIG ACL ------------
Name: Allowed
Latitude: Allowed
Longitude: Allowed
Pairing enabled: Allowed
Button enabled: Allowed
LED flash enabled: Allowed
Timezone offset: Allowed
DST mode: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed
Operating Mode: Allowed
Advertising Mode: Allowed
Timezone ID: Allowed
Intercom ID: Allowed
BUS mode Switch: Allowed
Short Circuit Duration: Allowed
Eletric Strike Delay: Allowed
Random Electric Strike Delay: Allowed
Electric Strike Duration: Allowed
Disable RTO after ring: Allowed
RTO timeout: Allowed
Doorbell suppression: Allowed
Doorbell suppression duration: Allowed
Sound Ring: Allowed
Sound Open: Allowed
Sound RTO: Allowed
Sound CM: Allowed
Sound confirmation: Allowed
Sound level: Allowed
Single button press action: Allowed
Double button press action: Allowed
Battery type: Allowed
Automatic battery type detection: Allowed
Reboot Nuki: Allowed

------------ GPIO ------------
```

[image]

If I can provide more info or do some debugging, just let me know.

ohadbenita, 2024-11-29:
Happens to me as well, didn't happen in 9.01 which, for some reason, I can't downgrade to.

ohadbenita, 2024-11-29:
> ### PROBLEM DESCRIPTION
> Nuki Hub stopped rebooting by itself. In older versions i didn't have to go near it forever. It just silently restarted in case of crash (?), but not anymore. At least not [...]

In my case I was able to connect to the lock, but it dropped the MQTT connection to home assistant.
Restarting the Nuki hub from the web interface regained the MQTT connection between the Nuki hub and HA.

combine-space, 2024-12-07:
I gathered the serial logs. In my case as soon the nuki hub gets unresponsive, logging also stops (it returns some empty lines before it dies). I can't see any hints that a problem was detected prior to the connection loss.

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4560
load:0x40078000,len:15336
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3176
entry 0x400805b8
Partition size: 2555904
Partition subtype: 16
Nuki Hub version 9.02
Nuki Hub build 11995573722.4.1
Load GPIO configuration
IP configuration: DHCP
Hardware detect: 1
MQTT without TLS.
Network device: Built-in Wi-Fi
Attempting to connect to saved SSID [REDACTED]
Wi-Fi async scan started
Host name: nukihub
MQTT Broker: 192.168.X.13:1883
Nuki Lock disabled
Nuki Opener enabled
Device id opener: 2328545398
Opener state interval: 1800 | Battery interval: 1800 | Publish auth data: no
Wi-Fi scan done
SSID [REDACTED]
SSID [REDACTED]
SSID [REDACTED]
SSID [REDACTED]
SSID [REDACTED]
SSID [REDACTED]
SSID [REDACTED]
SSID [REDACTED]
SSID [REDACTED]
SSID [REDACTED]
WiFi disconnected
Retrying
Wi-Fi connected
IP: 10.0.X.10
Attempting MQTT connection
MQTT: Connecting with user: nukihub
MQTT connected
HASS setup for NukiHub completed.
Subscribing to MQTT topic: nukihub/opener/action
Subscribing to MQTT topic: nukihub/opener/configuration/action
Subscribing to MQTT topic: nukihub/opener/query/config
Subscribing to MQTT topic: nukihub/opener/query/lockstate
Subscribing to MQTT topic: nukihub/opener/query/battery
Subscribing to MQTT topic: nukihub/opener/keypad/command/action
Subscribing to MQTT topic: nukihub/opener/keypad/command/id
Subscribing to MQTT topic: nukihub/opener/keypad/command/name
Subscribing to MQTT topic: nukihub/opener/keypad/command/code
Subscribing to MQTT topic: nukihub/opener/keypad/command/enabled
Subscribing to MQTT topic: nukihub/opener/query/keypad
Subscribing to MQTT topic: nukihub/opener/keypad/actionJson
Subscribing to MQTT topic: nukihub/maintenance/reset
Subscribing to MQTT topic: nukihub/maintenance/webserver/enable
Nuki opener start pairing
Nuki opener paired
Result (attempt 1): MQTT recently connected, ignoring opener action.
MQTT recently connected, ignoring opener action.
locked
Done querying opener state
Querying opener battery state: success
[... 98 lines cut ...]
nukiTask is running
networkTask is running
nukiTask is running
networkTask is running
nukiTask is running
Opener config result: success(1)
Nuki opener PIN is set
Nuki opener PIN is valid
Opener advanced config result: success
Done retrieving opener config and advanced config
[TRUNCATED repeating messages]
networkTask is running
nukiTask is running
networkTask is running
nukiTask is running
Opener config result: success(1)
Nuki opener PIN is set
Nuki opener PIN is valid
Opener advanced config result: success
Done retrieving opener config and advanced config
networkTask is running
nukiTask is running

EOF
```

iranl, 2024-12-18:
Please report if this is fixed in 9.05-master4

kotylo, 2024-12-19:
just updated, thanks. Let the timer begin :)

iranl, 2024-12-19:
BLE issues should be fixed in 9.05-master5.
If so please report back here if the unrecoverable hangups are also fixed.

combine-space, 2024-12-21:
`9.05-master5` has been running for 2 days without any hangups 👍

iranl, 2024-12-21:
I'll close this as fixed by 9.05, specifically the fixes for both BLE and addition of the RTC watchdog. If this issue reoccurs please re-open

closed 2024-12-21 by iranl

leo-b, 2024-12-27:
I tested 9.05 on one device now for several days. (Since 2024-12-21)
I confirm that it doesn't freeze anymore.
However, it reboots after 19 to 44 hours:

[image]

- at 2023-12-23 09:39 at uptime=2640 min
- at 2023-12-25 06:35 at uptime=2694 min
- at 2023-12-26 10:22 at uptime=1666 min
- at 2023-12-27 05:25 at uptime=1141 min

It looks like the restart reason is always:

```
Last restart reason FW: RestartOnDisconnectWatchdog
Last restart reason ESP: ESP_RST_SW: Software reset via esp_restart.
```

I have a second device with FW 9.00 that doesn't show regular reboots.

kotylo, 2024-12-27:
Would be nice to fix, but I personally can live with that. The chance that it reboots while i try to open the door is low, i think. My stats:

[image]

leo-b, 2024-12-27:
> Would be nice to fix, but I personally can live with that.

Did you experience similar reboots with firmware <= 9.00?

kotylo, 2024-12-27:
I think i had the same as you with 9.0, that worked really stable (i don't have old stats anymore). Before 9.0 though, i had reboots and the BLE watchdog helped with autoreboots.

iranl, 2024-12-27:
I can confirm that 9.01 and up have issues with Network (mainly WiFi) reconnection.
This will be worked on for release 9.06 or 9.07.

It is generally not advisable to enable "Restart on disconnect" as this will almost immediatly trigger a reboot on WiFI/Ethernet loss, without giving the device much change of recovering the connection.

Usually it is better to set "MQTT Timeout until restart (seconds; -1 to disable)" with a appropriate value (between 60-120 seconds)

leo-b, 2025-01-05:
I have disabled "Restart on disconnect" and updated to 9.06.

Now the resets are `due to task watchdog`:

[image]

iranl, 2025-01-05:
I'm working on better debug logging to the esp32 filesystem so we can do postmortem analysis of panics. 

Right now the only true way of doing this is having the device connected to serial all the time. 

The MQTT log (when enabled) might shed some light on the reboot reason, especially when enabling extra debug logging of the ble library from the advanced page.

leo-b, 2025-01-09:
> The MQTT log (when enabled) might shed some light on the reboot reason, especially when enabling extra debug logging of the ble library from the advanced page.

Thanks for the hint!

Here is the MQTT log around the latest _reset due to task watchdog_ with bluetooth debug options enabled:

```
# grep '^2025-01-07T21:1[345]' /tmp/nuki-garten.log | perl -ple 's|((?: [a-f0-9]{2}){3,})|" xx"x(length($1)/3)|e'
2025-01-07T21:13:10+0100 Nuki Advertising: Name: , Address: 54:d2:72:93:57:f5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66179357f5c4
2025-01-07T21:13:10+0100 iBeacon ID: 004C Major: 6035 Minor: 22517 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -60
2025-01-07T21:13:24+0100 Nuki Advertising: Name: , Address: 54:d2:72:93:57:f5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66179357f5c4
2025-01-07T21:13:24+0100 iBeacon ID: 004C Major: 6035 Minor: 22517 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -60
2025-01-07T21:13:27+0100 networkTask is running
2025-01-07T21:13:31+0100 Nuki Advertising: Name: , Address: 54:d2:72:93:57:f5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66179357f5c4
2025-01-07T21:13:31+0100 iBeacon ID: 004C Major: 6035 Minor: 22517 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -60
2025-01-07T21:13:40+0100 Nuki Advertising: Name: , Address: 54:d2:72:93:57:f5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66179357f5c4
2025-01-07T21:13:40+0100 iBeacon ID: 004C Major: 6035 Minor: 22517 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -60
2025-01-07T21:13:46+0100 Nuki Advertising: Name: , Address: 54:d2:72:93:57:f5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66179357f5c4
2025-01-07T21:13:46+0100 iBeacon ID: 004C Major: 6035 Minor: 22517 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -60
2025-01-07T21:13:48+0100 Nuki Advertising: Name: , Address: 54:d2:72:93:57:f5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66179357f5c4
2025-01-07T21:13:48+0100 iBeacon ID: 004C Major: 6035 Minor: 22517 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -60
2025-01-07T21:13:49+0100 nukiTask is running
2025-01-07T21:13:54+0100 Nuki Advertising: Name: , Address: 54:d2:72:93:57:f5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66179357f5c4
2025-01-07T21:13:54+0100 iBeacon ID: 004C Major: 6035 Minor: 22517 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -60
2025-01-07T21:14:01+0100 Nuki Advertising: Name: , Address: 54:d2:72:93:57:f5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66179357f5c4
2025-01-07T21:14:01+0100 iBeacon ID: 004C Major: 6035 Minor: 22517 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -60
2025-01-07T21:14:17+0100 Nuki Advertising: Name: , Address: 54:d2:72:93:57:f5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66179357f5c4
2025-01-07T21:14:17+0100 iBeacon ID: 004C Major: 6035 Minor: 22517 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -60
2025-01-07T21:14:32+0100 Nuki Advertising: Name: , Address: 54:d2:72:93:57:f5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66179357f5c4
2025-01-07T21:14:32+0100 iBeacon ID: 004C Major: 6035 Minor: 22517 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -60
2025-01-07T21:14:34+0100 Nuki Advertising: Name: , Address: 54:d2:72:93:57:f5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66179357f5c4
2025-01-07T21:14:34+0100 iBeacon ID: 004C Major: 6035 Minor: 22517 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -60
2025-01-07T21:14:42+0100 Nuki Advertising: Name: , Address: 54:d2:72:93:57:f5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66179357f5c4
2025-01-07T21:14:42+0100 iBeacon ID: 004C Major: 6035 Minor: 22517 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -60
2025-01-07T21:14:50+0100 Nuki Advertising: Name: , Address: 54:d2:72:93:57:f5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66179357f5c4
2025-01-07T21:14:50+0100 iBeacon ID: 004C Major: 6035 Minor: 22517 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -60
2025-01-07T21:14:55+0100 Nuki Advertising: Name: , Address: 54:d2:72:93:57:f5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66179357f5c4
2025-01-07T21:14:55+0100 iBeacon ID: 004C Major: 6035 Minor: 22517 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -60
2025-01-07T21:14:57+0100 Nuki Advertising: Name: , Address: 54:d2:72:93:57:f5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66179357f5c4
2025-01-07T21:14:57+0100 iBeacon ID: 004C Major: 6035 Minor: 22517 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -60
2025-01-07T21:15:11+0100 Nuki Advertising: Name: , Address: 54:d2:72:93:57:f5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66179357f5c4
2025-01-07T21:15:11+0100 iBeacon ID: 004C Major: 6035 Minor: 22517 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -60
2025-01-07T21:15:27+0100 networkTask is running
2025-01-07T21:15:37+0100 Updating Lock config based on timer or query
2025-01-07T21:15:37+0100 ************************ CHECK PAIRED ************************
2025-01-07T21:15:37+0100 [NukiHub] Credentials retrieved :
2025-01-07T21:15:37+0100 secretKeyK: xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx 
2025-01-07T21:15:37+0100 bleAddress: 54:d2:72:93:57:f5
2025-01-07T21:15:37+0100 authorizationId: xx xx xx xx 
2025-01-07T21:15:37+0100 Credentials retrieved from preferences, ready for commands
2025-01-07T21:15:37+0100 Start executing: 14 
[... cut ...]
2025-01-07T21:15:54+0100 Subscribing to MQTT topic: nuki-garten/lock/query/battery
2025-01-07T21:15:54+0100 Subscribing to MQTT topic: nuki-garten/lock/query/keypad
2025-01-07T21:15:54+0100 Subscribing to MQTT topic: nuki-garten/lock/keypad/actionJson
2025-01-07T21:15:54+0100 Subscribing to MQTT topic: nuki-garten/maintenance/reset
2025-01-07T21:15:54+0100 Subscribing to MQTT topic: nuki-garten/maintenance/webserver/enable
2025-01-07T21:15:54+0100 Nuki lock start pairing
2025-01-07T21:15:54+0100 Pairing as bridge
MQTT recently connected, ignoring lock action.
```

leo-b, 2025-01-11:
Here is a current example of a WDT reboot because of `ESP_RST_PANIC: Software reset due to exception/panic`:

```
2025-01-11T03:13:08+0100 Nuki Advertising: Name: , Address: 54:d2:72:93:57:f5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66179357f5c4
2025-01-11T03:13:08+0100 iBeacon ID: 004C Major: 6035 Minor: 22517 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -60
2025-01-11T03:13:09+0100 Nuki Advertising: Name: , Address: 54:d2:72:93:57:f5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66179357f5c4
2025-01-11T03:13:09+0100 iBeacon ID: 004C Major: 6035 Minor: 22517 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -60
2025-01-11T03:13:10+0100 Nuki Advertising: Name: , Address: 54:d2:72:93:57:f5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66179357f5c4
2025-01-11T03:13:10+0100 iBeacon ID: 004C Major: 6035 Minor: 22517 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -60
2025-01-11T03:13:15+0100 Nuki Advertising: Name: , Address: 54:d2:72:93:57:f5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66179357f5c4
2025-01-11T03:13:15+0100 iBeacon ID: 004C Major: 6035 Minor: 22517 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -60
2025-01-11T03:13:18+0100 Nuki Advertising: Name: , Address: 54:d2:72:93:57:f5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66179357f5c4
2025-01-11T03:13:18+0100 iBeacon ID: 004C Major: 6035 Minor: 22517 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -60
2025-01-11T03:13:37+0100 networkTask is running
2025-01-11T03:13:39+0100 Nuki Advertising: Name: , Address: 54:d2:72:93:57:f5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66179357f5c4
2025-01-11T03:13:39+0100 iBeacon ID: 004C Major: 6035 Minor: 22517 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -60
2025-01-11T03:13:40+0100 Nuki Advertising: Name: , Address: 54:d2:72:93:57:f5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66179357f5c4
2025-01-11T03:13:40+0100 iBeacon ID: 004C Major: 6035 Minor: 22517 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -60
2025-01-11T03:13:41+0100 Nuki Advertising: Name: , Address: 54:d2:72:93:57:f5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66179357f5c4
2025-01-11T03:13:41+0100 iBeacon ID: 004C Major: 6035 Minor: 22517 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -60
2025-01-11T03:13:46+0100 nukiTask is running
2025-01-11T03:14:04+0100 Nuki Advertising: Name: , Address: 54:d2:72:93:57:f5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66179357f5c4
2025-01-11T03:14:04+0100 iBeacon ID: 004C Major: 6035 Minor: 22517 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -60
2025-01-11T03:14:05+0100 Nuki Advertising: Name: , Address: 54:d2:72:93:57:f5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66179357f5c4
2025-01-11T03:14:05+0100 iBeacon ID: 004C Major: 6035 Minor: 22517 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -60
2025-01-11T03:14:11+0100 Nuki Advertising: Name: , Address: 54:d2:72:93:57:f5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66179357f5c4
2025-01-11T03:14:11+0100 iBeacon ID: 004C Major: 6035 Minor: 22517 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -60
2025-01-11T03:14:17+0100 Nuki Advertising: Name: , Address: 54:d2:72:93:57:f5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66179357f5c4
2025-01-11T03:14:17+0100 iBeacon ID: 004C Major: 6035 Minor: 22517 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -60
2025-01-11T03:14:18+0100 Nuki Advertising: Name: , Address: 54:d2:72:93:57:f5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66179357f5c4
2025-01-11T03:14:18+0100 iBeacon ID: 004C Major: 6035 Minor: 22517 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -60
2025-01-11T03:14:25+0100 Nuki Advertising: Name: , Address: 54:d2:72:93:57:f5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66179357f5c4
2025-01-11T03:14:25+0100 iBeacon ID: 004C Major: 6035 Minor: 22517 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -60
2025-01-11T03:14:34+0100 Nuki Advertising: Name: , Address: 54:d2:72:93:57:f5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66179357f5c4
2025-01-11T03:14:34+0100 iBeacon ID: 004C Major: 6035 Minor: 22517 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -60
2025-01-11T03:14:40+0100 Nuki Advertising: Name: , Address: 54:d2:72:93:57:f5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66179357f5c4
2025-01-11T03:14:40+0100 iBeacon ID: 004C Major: 6035 Minor: 22517 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -60
2025-01-11T03:14:41+0100 Nuki Advertising: Name: , Address: 54:d2:72:93:57:f5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66179357f5c4
2025-01-11T03:14:41+0100 iBeacon ID: 004C Major: 6035 Minor: 22517 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -60
2025-01-11T03:14:42+0100 Nuki Advertising: Name: , Address: 54:d2:72:93:57:f5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66179357f5c4
2025-01-11T03:14:42+0100 iBeacon ID: 004C Major: 6035 Minor: 22517 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -60
2025-01-11T03:14:50+0100 Nuki Advertising: Name: , Address: 54:d2:72:93:57:f5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66179357f5c4
2025-01-11T03:14:50+0100 iBeacon ID: 004C Major: 6035 Minor: 22517 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -60
2025-01-11T03:14:59+0100 Nuki Advertising: Name: , Address: 54:d2:72:93:57:f5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a66179357f5c4
[... cut ...]
2025-01-11T03:16:08+0100 iBeacon ID: 004C Major: 6035 Minor: 22517 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -60
2025-01-11T03:16:09+0100 Nuki lock start pairing
2025-01-11T03:16:09+0100 Pairing as bridge
2025-01-11T03:16:09+0100 [NukiHub] Credentials retrieved :
2025-01-11T03:16:09+0100 secretKeyK: xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx 
2025-01-11T03:16:09+0100 bleAddress: 54:d2:72:93:57:f5
2025-01-11T03:16:09+0100 authorizationId: xx xx xx xx 
2025-01-11T03:16:09+0100 Already paired
2025-01-11T03:16:09+0100 Nuki paired
```

PR #565 (2024-12-18): RTC Watchdog
## Description:

**Related issue (if applicable): Might fix #547 

## Checklist:
  - [x] The pull request is done against the latest master branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works
  - [x] I accept the [CLA](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md#contributor-license-agreement-cla).