technyon/nuki_hub #548: Unstable HTTP header response for pages

duhow, 2024-11-29:
### PROBLEM DESCRIPTION

**After upgrading from 9.01 to 9.02** and enable Nuki Lock device in settings, first reaction in mobile browser Chrome is to "download" the website as a file.
🐞  There are **two** issues I noticed:

- Header `Content-Disposition: inline` is present, where this :x: **should not** be the case, as we don't want to download the page we are viewing.
- Header `Content-Type` returns as corrupted.

```
curl -IXGET http://192.168.10.21/ota
HTTP/1.1 200 OK
Content-Type: ���?o
Transfer-Encoding: chunked
Content-Disposition: inline

curl -IXGET http://192.168.10.21/
HTTP/1.1 200 OK
Content-Type: ���?#
Transfer-Encoding: chunked
Content-Disposition: inline
```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [x] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [x] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [x] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [x] Device used (e.g., ESP32-S3): ESP32
- [x] Nuki Hub binary firmware version number used: 9.02
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Provide the output of http://nukihubIP/info:

<details>

```
------------ NUKI HUB ------------
Version: 9.02
Build: 11995573722.4.1
Build type: Release
Build date: 2024-11-24
Updater version: 9.02
Updater build: 11995573722.4.1
Updater build date: 2024-11-24
Uptime (min): 1
Config version: 902
Last restart reason FW: NotApplicable
Last restart reason ESP: ESP_RST_POWERON: Reset due to power-on event.
Free internal heap: 58848
Total internal heap: 260580
PSRAM Available: No
Network task stack high watermark: 7156
Nuki task stack high watermark: 5080

------------ GENERAL SETTINGS ------------
Network task stack size: 12288
Nuki task stack size: 8192
Check for updates: No
Latest version: 9.02
Allow update from MQTT: Yes
Web configurator username: Not set
Web configurator password: Not set
Web configurator enabled: Yes
Publish debug information enabled: No
MQTT log enabled: No
Webserial enabled: No
Bootloop protection enabled: Yes

------------ NETWORK ------------
Network device: Built-in Wi-Fi
Network connected: Yes
IP Address: 192.168.10.21
SSID: MOVISTAR_BD8C
BSSID of AP: 94:83:C4:4F:3D:8E
ESP32 MAC address: 24:6F:28:7C:6C:F8

------------ NETWORK SETTINGS ------------
Nuki Hub hostname: nukihub
DHCP enabled: Yes
RSSI Publish interval (s): 60
Find WiFi AP with strongest signal: No
Restart ESP32 on network disconnect enabled: Yes
MQTT Timeout until restart (s): Disabled

------------ MQTT ------------
MQTT connected: Yes
MQTT broker address: 192.168.10.4
MQTT broker port: 1883
MQTT username: ***
MQTT password: ***
MQTT base topic: nuki
MQTT SSL CA: Not set
MQTT SSL CRT: Not set
MQTT SSL Key: Not set

------------ BLUETOOTH ------------
Bluetooth TX power (dB): 9
Bluetooth command nr of retries: 3
Bluetooth command retry delay (ms): 100
Seconds until reboot when no BLE beacons received: -1

------------ QUERY / PUBLISH SETTINGS ------------
Lock/Opener state query interval (s): 1800
Publish Nuki device authorization log: Yes
Max authorization log entries to retrieve: 5
Battery state query interval (s): 1800
Most non-JSON MQTT topics disabled: No
Publish Nuki device config: No
Config query interval (s): 3600
Publish Keypad info: Yes
Keypad query interval (s): 1800
Enable Keypad control: Yes
Publish Keypad topic per entry: No
Publish Keypad codes: Yes
Allow checking Keypad codes: No
Max keypad entries to retrieve: 10
Publish timecontrol info: Yes
Keypad query interval (s): 1800
Enable timecontrol control: No
Publish timecontrol topic per entry: Yes
Max timecontrol entries to retrieve: 10

------------ HOME ASSISTANT ------------
Home Assistant auto discovery enabled: Yes
Home Assistant auto discovery topic: homeassistant/
Nuki Hub configuration URL for HA: http://192.168.10.21

------------ NUKI LOCK ------------
Lock enabled: Yes
Paired: Yes
Nuki Hub device ID: 4268661439
Nuki device ID: ***
Firmware version: 4.4.9
Hardware version: 7.0
Valid PIN set: Yes
Has door sensor: No
Has keypad: No
Timecontrol highest entries count: 0
Register as: App

------------ HYBRID MODE ------------
Hybrid mode enabled: Yes
Hybrid mode connected: Yes
Sending actions through official MQTT enabled: No
Time between status updates when official MQTT is offline (s): 600

------------ NUKI LOCK ACL ------------
Lock: Allowed
Unlock: Allowed
Unlatch: Allowed
Lock N Go: Disallowed
Lock N Go Unlatch: Disallowed
Full Lock: Disallowed
Fob Action 1: Disallowed
Fob Action 2: Disallowed
Fob Action 3: Disallowed

------------ NUKI LOCK CONFIG ACL ------------
Name: Disallowed
Latitude: Disallowed
Longitude: Disallowed
Auto Unlatch: Disallowed
Pairing enabled: Disallowed
Button enabled: Disallowed
LED flash enabled: Disallowed
LED brightness: Disallowed
Timezone offset: Disallowed
DST mode: Disallowed
Fob Action 1: Disallowed
Fob Action 2: Disallowed
Fob Action 3: Disallowed
Single Lock: Disallowed
Advertising Mode: Disallowed
Timezone ID: Disallowed
Unlocked Position Offset Degrees: Disallowed
Locked Position Offset Degrees: Disallowed
Single Locked Position Offset Degrees: Disallowed
Unlocked To Locked Transition Offset Degrees: Disallowed
Lock n Go timeout: Disallowed
Single button press action: Disallowed
Double button press action: Disallowed
Detached cylinder: Disallowed
Battery type: Disallowed
Automatic battery type detection: Disallowed
Unlatch duration: Disallowed
Auto lock timeout: Disallowed
Auto unlock disabled: Disallowed
Nightmode enabled: Disallowed
Nightmode start time: Disallowed
Nightmode end time: Disallowed
Nightmode auto lock enabled: Disallowed
Nightmode auto unlock disabled: Disallowed
Nightmode immediate lock on start: Disallowed
Auto lock enabled: Disallowed
Immediate auto lock enabled: Disallowed
Auto update enabled: Disallowed
Reboot Nuki: Allowed

------------ NUKI OPENER ------------
Opener enabled: Yes
Paired: Yes
Nuki Hub device ID: 4268661439
Nuki device ID: ***
Firmware version: 1.10.1
Hardware version: 4.17
Opener valid PIN set: Yes
Opener has keypad: No
Timecontrol highest entries count: 0
Register as: Bridge
Nuki Opener Lock/Unlock action set to Continuous mode in Home Assistant: No

------------ NUKI OPENER ACL ------------
Activate Ring-to-Open: Allowed
Deactivate Ring-to-Open: Allowed
Electric Strike Actuation: Allowed
Activate Continuous Mode: Allowed
Deactivate Continuous Mode: Allowed
Fob Action 1: Disallowed
Fob Action 2: Disallowed
Fob Action 3: Disallowed

------------ NUKI OPENER CONFIG ACL ------------
Name: Allowed
Latitude: Disallowed
Longitude: Disallowed
Pairing enabled: Allowed
Button enabled: Allowed
LED flash enabled: Allowed
Timezone offset: Disallowed
DST mode: Disallowed
Fob Action 1: Disallowed
Fob Action 2: Disallowed
Fob Action 3: Disallowed
Operating Mode: Allowed
Advertising Mode: Disallowed
Timezone ID: Disallowed
Intercom ID: Disallowed
BUS mode Switch: Allowed
Short Circuit Duration: Allowed
Eletric Strike Delay: Allowed
Random Electric Strike Delay: Disallowed
Electric Strike Duration: Allowed
Disable RTO after ring: Disallowed
RTO timeout: Disallowed
Doorbell suppression: Allowed
Doorbell suppression duration: Disallowed
Sound Ring: Disallowed
Sound Open: Disallowed
Sound RTO: Disallowed
Sound CM: Disallowed
Sound confirmation: Disallowed
Sound level: Disallowed
Single button press action: Disallowed
Double button press action: Disallowed
Battery type: Allowed
Automatic battery type detection: Allowed
Reboot Nuki: Disallowed

------------ GPIO ------------
```
</details>

**(Please, remember to close the issue when the problem has been addressed)**

technyon, 2024-12-01:
Looking at the code, this seems to be automatically set by the web server library PsychicHttp. Not sure if it's a bug on their side, but for Content-Type it sure seems so. There's no direct way to override this.

iranl, 2024-12-03:
Please try the next nightly.
Might be fixed with #554

duhow, 2024-12-03:
Fixed! ✅