technyon/nuki_hub #550: Opener: "open" state not reported to Home Assistant

duhow, 2024-11-30:
### PROBLEM DESCRIPTION

Nuki Opener `unlatch` (**open**) state does not report to Home Assistant (MQTT).
I would like to receive the event, so I can add automations after unlatching.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [x] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [x] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [x] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [x] Device used (e.g., ESP32-S3): ESP32
- [x] Nuki Hub binary firmware version number used: 9.02
  - [x] Pre-compiled
  - [ ] Self-compiled
- [X] Provide the output of http://nukihubIP/info:
```
N/A
```

### TO REPRODUCE

MQTT received events:

```
nuki/opener/action electricStrikeActuation
nuki/opener/action ack
nuki/opener/commandResult success
nuki/opener/retry --
nuki/opener/query/lockstateCommandResult success
```

### EXPECTED BEHAVIOUR

```mqtt
nuki/opener/hastate open
# wait few seconds and return to previous state
nuki/opener/hastate locked
```

### SCREENSHOTS

Goal: 
[image]

### ADDITIONAL CONTEXT

The `state` is supported: https://github.com/technyon/nuki_hub/blob/3c933546582d1e7e5879823c4dd510f91223a0dd/src/HomeAssistantDiscovery.cpp#L498-L506

**(Please, remember to close the issue when the problem has been addressed)**

iranl, 2024-12-03:
Have you confirmed that you can get an `open` or `opening` state in the Nuki app when publishing `nuki/opener/action electricStrikeActuation`?

NukiOpener API does seem to support the states and NukiHub also supports the states and publishes these to HA if it recieves them.

I expect that NH and the opener simply communicate to slowly to catch an intermediate state. The lock had the same issue untill recently.

The lock has gotten some extra code in NH to enter a kind of high-performance mode when expecting intermediate states (such as directly after publishing an action using MQTT).

I am porting this code to the opener in https://github.com/iranl/nuki_hub/tree/update-libs

duhow, 2024-12-03:
Interesting... opening via Nuki App I do get the values for `state` and `hastate`. ✅ 

<details>

```
nuki/opener/log [{"index":351,"authorizationId":0000000,"authorizationName":"NukiHub","timeYear":2024,"timeMonth":12,"timeDay":2,"timeHour":18,"timeMinute":23,"timeSecond":16,"type":"LockAction","action":"ElectricStrikeActuation","trigger":"system","completionStatus":"success"},{"index":352,"authorizationId":0000000,"authorizationName":"NukiHub","timeYear":2024,"timeMonth":12,"timeDay":3,"timeHour":18,"timeMinute":12,"timeSecond":35,"type":"LockAction","action":"ElectricStrikeActuation","trigger":"system","completionStatus":"success"},{"index":353,"authorizationId":0000000,"authorizationName":"NukiHub","timeYear":2024,"timeMonth":12,"timeDay":3,"timeHour":18,"timeMinute":14,"timeSecond":29,"type":"LockAction","action":"ElectricStrikeActuation","trigger":"system","completionStatus":"success"},{"index":354,"authorizationId":0000001,"authorizationName":"duhow","timeYear":2024,"timeMonth":12,"timeDay":3,"timeHour":18,"timeMinute":14,"timeSecond":46,"type":"LockAction","action":"ElectricStrikeActuation","trigger":"system","completionStatus":"success"},{"index":355,"authorizationId":0000000,"authorizationName":"NukiHub","timeYear":2024,"timeMonth":12,"timeDay":3,"timeHour":18,"timeMinute":15,"timeSecond":14,"type":"LockAction","action":"ElectricStrikeActuation","trigger":"system","completionStatus":"success"}]
nuki/opener/authorizationId 0000000
nuki/opener/authorizationName NukiHub
nuki/opener/rssi -45
nuki/opener/statusUpdated 1
nuki/opener/query/lockstateCommandResult success
nuki/maintenance/wifiRssi -59
nuki/opener/shortLog [{"index":353,"authorizationId":0000000,"authorizationName":"NukiHub","timeYear":2024,"timeMonth":12,"timeDay":3,"timeHour":18,"timeMinute":14,"timeSecond":29,"type":"LockAction","action":"ElectricStrikeActuation","trigger":"system","completionStatus":"success"},{"index":354,"authorizationId":0000001,"authorizationName":"duhow","timeYear":2024,"timeMonth":12,"timeDay":3,"timeHour":18,"timeMinute":14,"timeSecond":46,"type":"LockAction","action":"ElectricStrikeActuation","trigger":"system","completionStatus":"success"},{"index":355,"authorizationId":0000000,"authorizationName":"NukiHub","timeYear":2024,"timeMonth":12,"timeDay":3,"timeHour":18,"timeMinute":15,"timeSecond":14,"type":"LockAction","action":"ElectricStrikeActuation","trigger":"system","completionStatus":"success"}]
nuki/opener/authorizationId 0000000
nuki/opener/authorizationName NukiHub
nuki/opener/state opening
nuki/opener/hastate opening
nuki/opener/binaryState unlocked
nuki/opener/continuousMode off
nuki/opener/json {"lock_state":"opening","continuous_mode":0,"trigger":"system","ringToOpenTimer":0,"currentTime":"2024-12-03 18:15:32","timeZoneOffset":0,"last_lock_action":"Unknown","last_lock_action_trigger":"automatic","lock_completion_status":"success","door_sensor_state":"unavailable","auth_id":0000000,"auth_name":"NukiHub"}
nuki/opener/battery/basicJson {"critical":"0"}
nuki/opener/statusUpdated 0
nuki/opener/log [{"index":351,"authorizationId":0000000,"authorizationName":"NukiHub","timeYear":2024,"timeMonth":12,"timeDay":2,"timeHour":18,"timeMinute":23,"timeSecond":16,"type":"LockAction","action":"ElectricStrikeActuation","trigger":"system","completionStatus":"success"},{"index":352,"authorizationId":0000000,"authorizationName":"NukiHub","timeYear":2024,"timeMonth":12,"timeDay":3,"timeHour":18,"timeMinute":12,"timeSecond":35,"type":"LockAction","action":"ElectricStrikeActuation","trigger":"system","completionStatus":"success"},{"index":353,"authorizationId":0000000,"authorizationName":"NukiHub","timeYear":2024,"timeMonth":12,"timeDay":3,"timeHour":18,"timeMinute":14,"timeSecond":29,"type":"LockAction","action":"ElectricStrikeActuation","trigger":"system","completionStatus":"success"},{"index":354,"authorizationId":0000001,"authorizationName":"duhow","timeYear":2024,"timeMonth":12,"timeDay":3,"timeHour":18,"timeMinute":14,"timeSecond":46,"type":"LockAction","action":"ElectricStrikeActuation","trigger":"system","completionStatus":"success"},{"index":355,"authorizationId":0000000,"authorizationName":"NukiHub","timeYear":2024,"timeMonth":12,"timeDay":3,"timeHour":18,"timeMinute":15,"timeSecond":14,"type":"LockAction","action":"ElectricStrikeActuation","trigger":"system","completionStatus":"success"}]
nuki/opener/authorizationId 0000000
nuki/opener/authorizationName NukiHub
nuki/opener/statusUpdated 1
nuki/opener/query/lockstateCommandResult success
nuki/opener/rollingLog {"index":356,"authorizationId":0000001,"authorizationName":"duhow","timeYear":2024,"timeMonth":12,"timeDay":3,"timeHour":18,"timeMinute":15,"timeSecond":35,"type":"LockAction","action":"ElectricStrikeActuation","trigger":"system","completionStatus":"success"}
nuki/opener/lastRollingLog 356
nuki/opener/shortLog [{"index":354,"authorizationId":0000001,"authorizationName":"duhow","timeYear":2024,"timeMonth":12,"timeDay":3,"timeHour":18,"timeMinute":14,"timeSecond":46,"type":"LockAction","action":"ElectricStrikeActuation","trigger":"system","completionStatus":"success"},{"index":355,"authorizationId":0000000,"authorizationName":"NukiHub","timeYear":2024,"timeMonth":12,"timeDay":3,"timeHour":18,"timeMinute":15,"timeSecond":14,"type":"LockAction","action":"ElectricStrikeActuation","trigger":"system","completionStatus":"success"},{"index":356,"authorizationId":0000001,"authorizationName":"duhow","timeYear":2024,"timeMonth":12,"timeDay":3,"timeHour":18,"tim [... cut ...] timeDay":3,"timeHour":18,"timeMinute":15,"timeSecond":35,"type":"LockAction","action":"ElectricStrikeActuation","trigger":"system","completionStatus":"success"},{"index":357,"authorizationId":0000050,"authorizationName":"NukiHub","timeYear":2024,"timeMonth":12,"timeDay":3,"timeHour":18,"timeMinute":18,"timeSecond":2,"type":"LockAction","action":"ElectricStrikeActuation","trigger":"system","completionStatus":"success"}]
nuki/opener/authorizationId 0000050
nuki/opener/authorizationName NukiHub
```

iranl, 2024-12-03:
Actually not very surprising considering how BLE works in the Nuki BLE libary.

BLE scanning is paused while a BLE connection is open.

To send the `nuki/opener/action` a BLE connection is made with the opener and thus as long as the connection is open (couple of seconds) there is no scanning. After the connection is closed scanning will resume and NukiHub will notice that the opener is broadcasting the signal that a state change has occured and again connect to request the new state from the opener. At this time the intermediate state has probably already occured and the opener is back in an end state (locked/RTOactive).

When an action is performed using another method (app, fob, ring etc.) NukiHub will be in scanning mode and immediately start querying the new state. This has a much higher chance of still receiving the intermediate state.

We already changed this behaviour for the lock to always start querying for new states after NukiHub performs an action and keep querying when NukiHub sees an intermediate state, as it expects an end state in a couple of seconds.

I've added the same behaviour to the opener in #554. Please try the next nightly.

duhow, 2024-12-03:
Works! ✅ 
The state takes around 7 seconds to update to `opening` after Nuki has already opened the door, but yeah.

closed 2024-12-03 by duhow

duhow, 2024-12-03:
Indeed, disabling the auth log makes the status to appear faster. 👍🏻 
Since I'll be using Home Assistant to trigger the remote open, I don't think i need those for the moment.