technyon/nuki_hub #552: MQTT SSL Error

Unkn0wn-MDCLXIV, 2024-12-02:
### PROBLEM DESCRIPTION
After updating to 9.02 I cannot connect to my mosquitto server with ssl anymore.
The mosquitto log says
`OpenSSL Error[0]: error:0A000126:SSL routines::unexpected eof while reading
Client <unknown> disconnected: Protocol error`

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [X] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [X] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [X] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [X] Device used (e.g., ESP32-S3): ESP32
- [X] Nuki Hub binary firmware version number used: 9.02
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Provide the output of http://nukihubIP/info:
```
------------ NUKI HUB ------------
Version: 9.02
Build: 11995573722.4.1
Build type: Release
Build date: 2024-11-24
Updater version: 9.02
Updater build: 11995573722.4.1
Updater build date: 2024-11-24
Uptime (min): 1
Config version: 902
Last restart reason FW: NotApplicable
Last restart reason ESP: ESP_RST_PANIC: Software reset due to exception/panic.
Free internal heap: 55588
Total internal heap: 260580
PSRAM Available: No
Network task stack high watermark: 7896
Nuki task stack high watermark: 5468

------------ GENERAL SETTINGS ------------
Network task stack size: 12288
Nuki task stack size: 8192
Check for updates: No
Latest version: 
Allow update from MQTT: No
Web configurator username: ***
Web configurator password: ***
Web configurator enabled: Yes
Publish debug information enabled: No
MQTT log enabled: No
Webserial enabled: No
Bootloop protection enabled: No

------------ NETWORK ------------
Network device: Built-in Wi-Fi
Network connected: Yes
IP Address: 192.168.178.110
SSID: FRITZ!Wlan
BSSID of AP: 2D:71:AC:84:53:DD
ESP32 MAC address: DA:D3:DA:59:EF:34

------------ NETWORK SETTINGS ------------
Nuki Hub hostname: nukihub
DHCP enabled: Yes
RSSI Publish interval (s): Disabled
Find WiFi AP with strongest signal: No
Restart ESP32 on network disconnect enabled: Yes
MQTT Timeout until restart (s): 300

------------ MQTT ------------
MQTT connected: No
MQTT broker address: mqtt.lan
MQTT broker port: 8883
MQTT username: ***
MQTT password: ***
MQTT base topic: nuki/smartlock
MQTT SSL CA: ***
MQTT SSL CRT: Not set
MQTT SSL Key: Not set

------------ BLUETOOTH ------------
Bluetooth TX power (dB): 9
Bluetooth command nr of retries: 3
Bluetooth command retry delay (ms): 100
Seconds until reboot when no BLE beacons received: -1

------------ QUERY / PUBLISH SETTINGS ------------
Lock/Opener state query interval (s): 1800
Publish Nuki device authorization log: No
Max authorization log entries to retrieve: 5
Battery state query interval (s): 1800
Most non-JSON MQTT topics disabled: Yes
Publish Nuki device config: Yes
Config query interval (s): 3600
Publish Keypad info: No
Keypad query interval (s): 1800
Enable Keypad control: No
Publish Keypad topic per entry: No
Publish Keypad codes: No
Allow checking Keypad codes: No
Max keypad entries to retrieve: 10
Publish timecontrol info: No
Keypad query interval (s): 1800
Enable timecontrol control: No
Publish timecontrol topic per entry: No
Max timecontrol entries to retrieve: 10

------------ HOME ASSISTANT ------------
Home Assistant auto discovery enabled: Yes
Home Assistant auto discovery topic: homeassistant/
Nuki Hub configuration URL for HA: http://192.168.178.110

------------ NUKI LOCK ------------
Lock enabled: Yes
Paired: Yes
Nuki Hub device ID: 1169263314
Nuki device ID: ***
Firmware version: 
Hardware version: 
Valid PIN set: Yes
Has door sensor: Yes
Has keypad: No
Timecontrol highest entries count: 0
Register as: Bridge

------------ HYBRID MODE ------------
Hybrid mode enabled: No

------------ NUKI LOCK ACL ------------
Lock: Allowed
Unlock: Allowed
Unlatch: Allowed
Lock N Go: Disallowed
Lock N Go Unlatch: Disallowed
Full Lock: Disallowed
Fob Action 1: Disallowed
Fob Action 2: Disallowed
Fob Action 3: Disallowed

------------ NUKI LOCK CONFIG ACL ------------
Name: Disallowed
Latitude: Disallowed
Longitude: Disallowed
Auto Unlatch: Disallowed
Pairing enabled: Disallowed
Button enabled: Disallowed
LED flash enabled: Disallowed
LED brightness: Disallowed
Timezone offset: Disallowed
DST mode: Disallowed
Fob Action 1: Disallowed
Fob Action 2: Disallowed
Fob Action 3: Disallowed
Single Lock: Disallowed
Advertising Mode: Disallowed
Timezone ID: Disallowed
Unlocked Position Offset Degrees: Disallowed
Locked Position Offset Degrees: Disallowed
Single Locked Position Offset Degrees: Disallowed
Unlocked To Locked Transition Offset Degrees: Disallowed
Lock n Go timeout: Disallowed
Single button press action: Disallowed
Double button press action: Disallowed
Detached cylinder: Disallowed
Battery type: Disallowed
Automatic battery type detection: Disallowed
Unlatch duration: Disallowed
Auto lock timeout: Disallowed
Auto unlock disabled: Disallowed
Nightmode enabled: Disallowed
Nightmode start time: Disallowed
Nightmode end time: Disallowed
Nightmode auto lock enabled: Disallowed
Nightmode auto unlock disabled: Disallowed
Nightmode immediate lock on start: Disallowed
Auto lock enabled: Disallowed
Immediate auto lock enabled: Disallowed
Auto update enabled: Disallowed
Reboot Nuki: Disallowed

------------ NUKI OPENER ------------
Opener enabled: Yes
Paired: Yes
Nuki Hub device ID: 1169263314
Nuki device ID: ***
Firmware version: 
Hardware version: 
Opener valid PIN set: Yes
Opener has keypad: No
Timecontrol highest entries count: 0
Register as: Bridge
Nuki Opener Lock/Unlock action set to Continuous mode in Home Assistant: No

------------ NUKI OPENER ACL ------------
Activate Ring-to-Open: Allowed
Deactivate Ring-to-Open: Allowed
Electric Strike Actuation: Allowed
Activate Continuous Mode: Disallowed
Deactivate Continuous Mode: Disallowed
Fob Action 1: Disallowed
Fob Action 2: Disallowed
Fob Action 3: Disallowed

------------ NUKI OPENER CONFIG ACL ------------
Name: Disallowed
Latitude: Disallowed
Longitude: Disallowed
Pairing enabled: Disallowed
Button enabled: Disallowed
LED flash enabled: Disallowed
Timezone offset: Disallowed
DST mode: Disallowed
Fob Action 1: Disallowed
Fob Action 2: Disallowed
Fob Action 3: Disallowed
Operating Mode: Disallowed
Advertising Mode: Disallowed
Timezone ID: Disallowed
Intercom ID: Disallowed
BUS mode Switch: Disallowed
Short Circuit Duration: Disallowed
Eletric Strike Delay: Disallowed
Random Electric Strike Delay: Disallowed
Electric Strike Duration: Disallowed
Disable RTO after ring: Disallowed
RTO timeout: Disallowed
Doorbell suppression: Disallowed
Doorbell suppression duration: Disallowed
Sound Ring: Disallowed
Sound Open: Disallowed
Sound RTO: Disallowed
Sound CM: Disallowed
Sound confirmation: Disallowed
Sound level: Disallowed
Single button press action: Disallowed
Double button press action: Disallowed
Battery type: Disallowed
Automatic battery type detection: Disallowed
Reboot Nuki: Disallowed

------------ GPIO ------------
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Update to 9.02, connect to mqtt via ssl

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
A mqtt connection

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
It was working fine using nuki 9.0

**(Please, remember to close the issue when the problem has been addressed)**

iranl, 2024-12-03:
MQTT SSL has no general issue in 9.02 as it is tested and confirmed working.

There might be a RAM/stack limitation in your setup.
You can expect errors/undefined behaviour when getting up to the available stack.
Combining SSL and a lock and an opener will probably get you there in 9.02
NukiHub has become more RAM hungry in the last year because of added functionality and updating to the latest Arduino and ESP-IDF.

Try disabling your lock and opener in NukiHub first to test if the MQTT connection will work then. If mqtt works in that case you can try disabling SSL, changing your SSL cert to a smaller size, use 2 esps (one for the lock and one for the opener) or get an esp with psram.

iranl, 2024-12-03:
Yes. New webserver has multiple benefits but also seems to eat more RAM.

technyon, 2024-12-04:
@Unkn0wn-MDCLXIV You're already using an ESP32-S3. I think the best solution for you would be to get one with PSRAM.

iranl, 2024-12-22:
9.06-master2 includes some optimizations that will same about 20k RAM. Might be that this will allow MQTT SSL to start functioning again in your setup.

Otherwise consider switching to an ESP32-S3 with PSRAM.

iranl, 2024-12-26:
Closing due to lack of activity.

Please re-open if you observe the same issues on 9.06-master4 or up and for testing purposes with the suggestions I made regarding disabling lock/opener.

closed 2024-12-26 by iranl

Unkn0wn-MDCLXIV, 2024-12-30:
> 9.06-master2 includes some optimizations that will same about 20k RAM. Might be that this will allow MQTT SSL to start functioning again in your setup.

Thanks for letting me know. I switched to a ESP32-S3 with PSRAM just before I read your comment, it's working fine with it so far.