technyon/nuki_hub #557: Nuki Opener timeout after upgrade to 9.03-master8

duhow, 2024-12-06:
### PROBLEM DESCRIPTION

Timeouts when attempting to open Nuki Opener. 
Can't open the intercom door.
All 4 retry attempts appear within 1 second, expecting to delay a bit.

Noticed these errors couple of days after upgrading to #554 ...

```
nuki/opener/action electricStrikeActuation
nuki/opener/action ack
nuki/opener/commandResult error
nuki/opener/retry 1
nuki/opener/commandResult error
nuki/opener/retry 2
nuki/opener/commandResult error
nuki/opener/retry 3
nuki/opener/commandResult error
nuki/opener/retry 4
nuki/opener/retry failed
```

### REQUESTED INFORMATION

- [x] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [x] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [x] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [x] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [x] Device used (e.g., ESP32-S3): ESP32
- [x] Nuki Hub binary firmware version number used: `9.03-master8`
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Provide the output of http://nukihubIP/info:

<details>

```
------------ NUKI HUB ------------
Version: 9.03-master8
Build: 12148007889.169.1
Build type: Release
Build date: 2024-12-03
Updater version: 9.03-master8
Updater build: 12148007889.169.1
Updater build date: 2024-12-03
Uptime (min): 84
Config version: 1061194719
Last restart reason FW: NotApplicable
Last restart reason ESP: ESP_RST_BROWNOUT: Brownout reset (software or hardware)
Free internal heap: 61096
Total internal heap: 260360
PSRAM Available: No
Network task stack high watermark: 7148
Nuki task stack high watermark: 5072

------------ GENERAL SETTINGS ------------
Network task stack size: 12288
Nuki task stack size: 8192
Check for updates: No
Latest version: 9.03-master8
Allow update from MQTT: Yes
Web configurator username: Not set
Web configurator password: Not set
Web configurator enabled: Yes
Publish debug information enabled: No
MQTT log enabled: No
Webserial enabled: No
Bootloop protection enabled: Yes

------------ NETWORK ------------
Network device: Built-in Wi-Fi
Network connected: Yes
IP Address: 192.168.10.21
SSID: MOVISTAR_BD8C
BSSID of AP: 94:83:C4:4F:3D:8E
ESP32 MAC address: 24:6F:28:7C:6C:F8

------------ NETWORK SETTINGS ------------
Nuki Hub hostname: nukihub
DHCP enabled: Yes
RSSI Publish interval (s): 60
Find WiFi AP with strongest signal: No
Restart ESP32 on network disconnect enabled: Yes
MQTT Timeout until restart (s): Disabled

------------ MQTT ------------
MQTT connected: Yes
MQTT broker address: 192.168.10.4
MQTT broker port: 1883
MQTT username: ***
MQTT password: ***
MQTT base topic: nuki
MQTT SSL CA: Not set
MQTT SSL CRT: Not set
MQTT SSL Key: Not set

------------ BLUETOOTH ------------
Bluetooth TX power (dB): 9
Bluetooth command nr of retries: 3
Bluetooth command retry delay (ms): 100
Seconds until reboot when no BLE beacons received: -1

------------ QUERY / PUBLISH SETTINGS ------------
Lock/Opener state query interval (s): 1800
Publish Nuki device authorization log: No
Max authorization log entries to retrieve: 5
Battery state query interval (s): 1800
Most non-JSON MQTT topics disabled: No
Publish Nuki device config: No
Config query interval (s): 3600
Publish Keypad info: Yes
Keypad query interval (s): 1800
Enable Keypad control: Yes
Publish Keypad topic per entry: No
Publish Keypad codes: Yes
Allow checking Keypad codes: No
Max keypad entries to retrieve: 10
Publish timecontrol info: Yes
Keypad query interval (s): 1800
Enable timecontrol control: No
Publish timecontrol topic per entry: Yes
Max timecontrol entries to retrieve: 10

------------ HOME ASSISTANT ------------
Home Assistant auto discovery enabled: Yes
Home Assistant auto discovery topic: homeassistant/
Nuki Hub configuration URL for HA: http://192.168.10.21

------------ NUKI LOCK ------------
Lock enabled: Yes
Paired: Yes
Nuki Hub device ID: 4268661439
Nuki device ID: ***
Firmware version: 4.4.9
Hardware version: 7.0
Valid PIN set: Yes
Has door sensor: No
Has keypad: No
Timecontrol highest entries count: 0
Register as: App

------------ HYBRID MODE ------------
Hybrid mode enabled: Yes
Hybrid mode connected: Yes
Sending actions through official MQTT enabled: Yes
Time between status updates when official MQTT is offline (s): 600

------------ NUKI LOCK ACL ------------
Lock: Allowed
Unlock: Allowed
Unlatch: Allowed
Lock N Go: Disallowed
Lock N Go Unlatch: Disallowed
Full Lock: Disallowed
Fob Action 1: Disallowed
Fob Action 2: Disallowed
Fob Action 3: Disallowed

------------ NUKI LOCK CONFIG ACL ------------
Name: Disallowed
Latitude: Disallowed
Longitude: Disallowed
Auto Unlatch: Disallowed
Pairing enabled: Disallowed
Button enabled: Disallowed
LED flash enabled: Disallowed
LED brightness: Disallowed
Timezone offset: Disallowed
DST mode: Disallowed
Fob Action 1: Disallowed
Fob Action 2: Disallowed
Fob Action 3: Disallowed
Single Lock: Disallowed
Advertising Mode: Disallowed
Timezone ID: Disallowed
Unlocked Position Offset Degrees: Disallowed
Locked Position Offset Degrees: Disallowed
Single Locked Position Offset Degrees: Disallowed
Unlocked To Locked Transition Offset Degrees: Disallowed
Lock n Go timeout: Disallowed
Single button press action: Disallowed
Double button press action: Disallowed
Detached cylinder: Disallowed
Battery type: Disallowed
Automatic battery type detection: Disallowed
Unlatch duration: Disallowed
Auto lock timeout: Disallowed
Auto unlock disabled: Disallowed
Nightmode enabled: Disallowed
Nightmode start time: Disallowed
Nightmode end time: Disallowed
Nightmode auto lock enabled: Disallowed
Nightmode auto unlock disabled: Disallowed
Nightmode immediate lock on start: Disallowed
Auto lock enabled: Disallowed
Immediate auto lock enabled: Disallowed
Auto update enabled: Disallowed
Reboot Nuki: Allowed

------------ NUKI OPENER ------------
Opener enabled: Yes
Paired: Yes
Nuki Hub device ID: 4268661439
Nuki device ID: ***
Firmware version: 1.10.1
Hardware version: 4.17
Opener valid PIN set: Yes
Opener has keypad: No
Timecontrol highest entries count: 0
Register as: Bridge
Nuki Opener Lock/Unlock action set to Continuous mode in Home Assistant: No

------------ NUKI OPENER ACL ------------
Activate Ring-to-Open: Allowed
Deactivate Ring-to-Open: Allowed
Electric Strike Actuation: Allowed
Activate Continuous Mode: Allowed
Deactivate Continuous Mode: Allowed
Fob Action 1: Disallowed
Fob Action 2: Disallowed
Fob Action 3: Disallowed

------------ NUKI OPENER CONFIG ACL ------------
Name: Allowed
Latitude: Disallowed
Longitude: Disallowed
Pairing enabled: Allowed
Button enabled: Allowed
LED flash enabled: Allowed
Timezone offset: Disallowed
DST mode: Disallowed
Fob Action 1: Disallowed
Fob Action 2: Disallowed
Fob Action 3: Disallowed
Operating Mode: Allowed
Advertising Mode: Disallowed
Timezone ID: Disallowed
Intercom ID: Disallowed
BUS mode Switch: Allowed
Short Circuit Duration: Allowed
Eletric Strike Delay: Allowed
Random Electric Strike Delay: Disallowed
Electric Strike Duration: Allowed
Disable RTO after ring: Disallowed
RTO timeout: Disallowed
Doorbell suppression: Allowed
Doorbell suppression duration: Disallowed
Sound Ring: Disallowed
Sound Open: Disallowed
Sound RTO: Disallowed
Sound CM: Disallowed
Sound confirmation: Disallowed
Sound level: Disallowed
Single button press action: Disallowed
Double button press action: Disallowed
Battery type: Allowed
Automatic battery type detection: Allowed
Reboot Nuki: Disallowed

------------ GPIO ------------
```
</details>

duhow, 2024-12-06:
I have opened via mobile app and now seems responsive.
Unless there's anything else to check, give me few days to see if this problem reocurrs again.

iranl, 2024-12-06:
With the updated libs I also changed the way nuki ble library reconnects with the lock/opener. It tries to do this faster than before by not querying services each time it connects. Might give some issues though when something in the connection goes wrong and then does not recover. 

Maybe connecting to the opener through the app disconnected the esp32 and the opener in some way that allows the esp32 to recover the connection afterwards. 

Will see about forcing a full reconnect on errors.

duhow, 2024-12-12:
The problem repeats after a while, despite I can open with the App and later with Nuki Hub, the same disconnect issue happens again (eg. the next day).

technyon, 2024-12-17:
@duhow Is there any change with the 9.04 release?

duhow, 2024-12-18:
Still having issues. Even after opening with the app, 2h later I would have issues with the device.
Right now I have unpaired and paired again the device, will report later how it goes...

iranl, 2024-12-18:
Please report if this is fixed in 9.05-master4

kotylo, 2024-12-19:
Not sure if related, but i've updated to 9.05-master4 and after 4+ hours could not get in to the Opener or Lock:

```
Buzzer MQTT Log changed to nukiTask is running
8:50:43 AM - 1 minute ago
Buzzer MQTT Log changed to networkTask is running
8:50:17 AM - 1 minute ago
Buzzer MQTT Log changed to Result (attempt 1): Result (attempt 2): Result (attempt 3): Result (attempt 4): Query opener state failed
8:50:01 AM - 2 minutes ago
Buzzer MQTT Log changed to nukiTask is running
8:48:43 AM - 3 minutes ago
Buzzer MQTT Log changed to networkTask is running
8:48:17 AM - 3 minutes ago
Buzzer MQTT Log changed to Lock: Maximum number of retries exceeded, aborting.
8:47:24 AM - 4 minutes ago
Buzzer MQTT Log changed to Lock: Last command failed, retrying after 100 milliseconds. Retry 4 of 3
8:47:24 AM - 4 minutes ago
Buzzer MQTT Log changed to Lock action result: error
8:47:24 AM - 4 minutes ago
Buzzer MQTT Log changed to Lock: Last command failed, retrying after 100 milliseconds. Retry 3 of 3
8:47:24 AM - 4 minutes ago
Buzzer MQTT Log changed to Lock action result: error
8:47:24 AM - 4 minutes ago
Buzzer MQTT Log changed to Lock: Last command failed, retrying after 100 milliseconds. Retry 2 of 3
8:47:23 AM - 4 minutes ago
Buzzer MQTT Log changed to Lock action result: error
8:47:23 AM - 4 minutes ago
Buzzer MQTT Log changed to Lock: Last command failed, retrying after 100 milliseconds. Retry 1 of 3
8:47:23 AM - 4 minutes ago
Buzzer MQTT Log changed to Lock action result: error
8:47:23 AM - 4 minutes ago

Home Door MQTT Log changed to nukiTask is running
8:50:43 AM - 1 minute ago
Home Door MQTT Log changed to networkTask is running
8:50:17 AM - 2 minutes ago
Home Door MQTT Log changed to Result (attempt 1): Result (attempt 2): Result (attempt 3): Result (attempt 4): Query opener state failed
8:50:01 AM - 2 minutes ago
Home Door MQTT Log changed to nukiTask is running
8:48:43 AM - 3 minutes ago
Home Door MQTT Log changed to networkTask is running
8:48:17 AM - 4 minutes ago
Home Door MQTT Log changed to Lock: Maximum number of retries exceeded, aborting.
8:47:24 AM - 5 minutes ago
Home Door MQTT Log changed to Lock: Last command failed, retrying after 100 milliseconds. Retry 4 of 3
8:47:24 AM - 5 minutes ago
Home Door MQTT Log changed to Lock action result: error
8:47:24 AM - 5 minutes ago
Home Door MQTT Log changed to Lock: Last command failed, retrying after 100 milliseconds. Retry 3 of 3
8:47:24 AM - 5 minutes ago
Home Door MQTT Log changed to Lock action result: error
8:47:24 AM - 5 minutes ago
Home Door MQTT Log changed to Lock: Last command failed, retrying after 100 milliseconds. Retry 2 of 3
8:47:23 AM - 5 minutes ago
Home Door MQTT Log changed to Lock action result: error
8:47:23 AM - 5 minutes ago
Home Door MQTT Log changed to Lock: Last command failed, retrying after 100 milliseconds. Retry 1 of 3
8:47:23 AM - 5 minutes ago
Home Door MQTT Log changed to Lock action result: error
8:47:23 AM - 5 minutes ago
Home Door MQTT Log changed to Lock action received: lock
8:47:23 AM - 5 minutes ago

```

iranl, 2024-12-19:
Please report if this is fixed in 9.05-master5. I'm optimistic the issue has been found and addressed.

kotylo, 2024-12-20:
No retries for me at least, so seems to work for now for both Opener and Lock.

iranl, 2024-12-21:
This seems fixed. Please re-open if issues persist on 9.05-master5/9.05-beta1 and up.
In that case please add serial logs

closed 2024-12-21 by iranl