technyon/nuki_hub #561: Nuki_ble starting with NH v9.03 breaks communication

bcutter, 2024-12-16:
### PROBLEM DESCRIPTION
After updating from NH 9.02 to 9.03 or 9.04, the communication between NH and the Nuki lock seems to get stuck. Lock/unlock attempts sent by HA via NH are only performed after a few seconds after rebooting NH, like for two to three quick lock/unlock attempts. After that, it's broken. Just another unreliable update. Happened for all my NHs and all my locks. Downgraded to NH 9.02, everything is working normally.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [X] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [X] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [ ] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [X] Device used (e.g., ESP32-S3): LOLIN32-Lite
- [X] Nuki Hub binary firmware version number used: 9.03 and 9.04
  - [X] Pre-compiled
  - [ ] Self-compiled
- [ ] Provide the output of http://nukihubIP/info:
```
Not available as downgraded to 9.02 again because I need a reliable NH
```

### TO REPRODUCE
Update NH to 9.03 or 9.04 with the updated nuki_ble (seems to not be well tested...).

### EXPECTED BEHAVIOUR
Lock/unlock commands are reliably working, even more than 3 times or 2 minutes after rebooting NH.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

technyon, 2024-12-17:
@bcutter When this happens, is your ESP reachable over the network? Can you reach the configuration portal or ping the device?

VDRainer, 2024-12-17:
> Can you reach the configuration portal or ping the device?

For me the ESP is reachable over the network.
After a restart, the lock works, but after a little time, it fails again.

andonevris, 2024-12-17:
I'm seeing the same thing, nuki hub is reporting incorrect status for locks, after a reboot it works fine for a bit.

How easy is it to downgrade firmware?

mundschenk-at, 2024-12-17:
I have noticed these BLE issues as well after upgrading to 9.04, most noticeably with the Opener (the Lock is connected via Hybrid mode). It's not a network issue. I see the failed BLE attempts in the MQTT log.

regevbr, 2024-12-17:
sorry for spamming but same for me :-(

thebadboyltz, 2024-12-18:
Hello, same problem at home.
Can you tell me where to find the .bin 9.0.3 file to downgrade.
I find the ZIP file but not the BIN.
thanks

bcutter, 2024-12-18:
The BIN is in the ZIP file...

And as I had these issues already with 9.03 I recommend to downgrade to 9.02.

thebadboyltz, 2024-12-18:
Sorry to be annoying, but in the ZIP I only see 3 BIN files and none of them work.

[image]

bcutter, 2024-12-18:
I have no idea which one you choose. Better the maintainers fix this and push a 9.05 as downgrading is not for everyone...

iranl, 2024-12-18:
Please report if this is fixed in 9.05-master4

tmakowka, 2024-12-19:
@iranl I'm experiencing the same issue as described. I just installed 9.05-master4 and after approx 5 minutes it still stops working. In MQTT i can see that 3 retries (as configured) are made, but the "commandResult" is "error".

ohadbenita, 2024-12-19:
Same here, updating to 9.05-master4 the lock stops responding, shows incorrect state in HA.

andonevris, 2024-12-19:
> Sorry to be annoying, but in the ZIP I only see 3 BIN files and none of them work.
> 
> ![unnamed](https://private-user-images.githubusercontent.com/153684851/397102522-e65eb162-c7f6-4f35-a40b-05b5c818449d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ2MDE5MjIsIm5iZiI6MTczNDYwMTYyMiwicGF0aCI6Ii8xNTM2ODQ4NTEvMzk3MTAyNTIyLWU2NWViMTYyLWM3ZjYtNGYzNS1hNDBiLTA1YjVjODE4NDQ5ZC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjE5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIxOVQwOTQ3MDJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zZmNjNGE5ODEwNzhlZTg4NTgwZDc5ZGQ1YjYwNTU1NDI2OTRlMDc3NzVmZGE2MGM1NzhmZGVjMWQ2ODkzMDA2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.9Uy-1oIN5GHmYZN4e0U5YPVCtx-rGYzHGZ31c3_xaog)

That doesn't look like the correct zip file.  Make sure you download from the releases page
https://github.com/technyon/nuki_hub/releases

Contents should look like this

[image]

tmakowka, 2024-12-19:
> @iranl I'm experiencing the same issue as described. I just installed 9.05-master4 and after approx 5 minutes it still stops working. In MQTT i can see that 3 retries (as configured) are made, but the "commandResult" is "error".

Quick update, i'm not using the Opener, just the Lock Pro 4th Gen.

StopAndG0, 2024-12-19:
Sorry for spamming, but same for me.

iranl, 2024-12-19:
Please report if this is fixed in 9.05-master5. I'm optimistic the issue has been found and addressed.

Seinuh, 2024-12-19:
For some reason OTA update from my olimex esp32-poe doesn't work anymore, yesterday  after update to master4 it was completely dead. Now after update to master5 it is also completely dead. Have to reinstall via serial when I'm at home

VDRainer, 2024-12-20:
> Please report if this is fixed in 9.05-master5. I'm optimistic the issue has been found and addressed.

You're right. Up 1 hour now and everything works for me. :hand_with_index_finger_and_thumb_crossed:

ohadbenita, 2024-12-20:
> Please report if this is fixed in 9.05-master5. I'm optimistic the issue has been found and addressed.

2 hours up with 9.05-master5 with no issues so far, will report back in a few hours.

iranl, 2024-12-21:
This seems fixed. Please re-open if issues persist on 9.05-master5/9.05-beta1 and up.
In that case please add serial logs

PR #564 (2024-12-18): Update to esp-nimble-cpp 2.0.1 + Allow runtime choice of BLE connect mode and Nuki BLE debug logging
## Description:

**Related issue (if applicable):** Might fix #561 and #557

## Checklist:
  - [x] The pull request is done against the latest master branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works
  - [x] I accept the [CLA](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md#contributor-license-agreement-cla).
closed 2024-12-21 by iranl