technyon/nuki_hub #568: Nuki Opener: Home Assistant Commands not working

h1ghrise, 2024-12-20:
### PROBLEM DESCRIPTION
I'm running **9.0.4.** and have a Nuki Opener connected.
Home Assistant is running on an RPI5 (HA OS) and has the build in MQTT Broker Add-On running (and appropriately setup in nukibridge.

I noticed, that in HA the commands (i.e. changing the `lock` entity from `open` to `closed` (`activateRTO`) will be successful after the ESP is rebooted.

After a couple of minutes, the command fails (as Lock Entity is not updated anymore).
Also i do not receive ring notifications anymore.

MQTT Logging is enabled, with the following output (`activateRTO` as an example)
```
Opener action received: activateRTO
Opener action result: error
Opener: Last command failed, retrying after 100 milliseconds. Retry 1 of 3
...
Opener: Maximum number of retries exceeded, aborting.
```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [x] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [x] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [ ] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [x] Device used (e.g., ESP32-S3): ESP32-S3
- [x] Nuki Hub binary firmware version number used: 9.0.4.
  - [x] Pre-compiled (build 12361409996.6.1)
  - [ ] Self-compiled
- [ ] Provide the output of http://nukihubIP/info:
```
System Information
------------ NUKI HUB ------------
Version: 9.04
Build: 12361409996.6.1
Build type: Release
Build date: 2024-12-16
Updater version: 9.04
Updater build: 12361409996.6.1
Updater build date: 2024-12-03
Uptime (min): 55
Config version: 904
Last restart reason FW: RequestedViaWebServer
Last restart reason ESP: ESP_RST_SW: Software reset via esp_restart.
Free internal heap: 67312
Total internal heap: 260360
PSRAM Available: No
Network task stack high watermark: 8192
Nuki task stack high watermark: 5092

------------ GENERAL SETTINGS ------------
Network task stack size: 12288
Nuki task stack size: 8192
Check for updates: Yes
Latest version: 9.04
Allow update from MQTT: Yes
Web configurator username: Not set
Web configurator password: Not set
Web configurator enabled: Yes
Publish debug information enabled: No
MQTT log enabled: Yes
Webserial enabled: No
Bootloop protection enabled: No

------------ NETWORK ------------
Network device: Built-in Wi-Fi
Network connected: Yes
IP Address: 10.10.20.6
SSID: IoT
BSSID of AP: D4:01:C3:FA:36:6A
ESP32 MAC address: B8:F0:09:CC:92:9C

------------ NETWORK SETTINGS ------------
Nuki Hub hostname: nukihub
DHCP enabled: Yes
RSSI Publish interval (s): 60
Find WiFi AP with strongest signal: Yes
Restart ESP32 on network disconnect enabled: No
MQTT Timeout until restart (s): 60

------------ MQTT ------------
MQTT connected: Yes
MQTT broker address: 10.10.20.2
MQTT broker port: 1883
MQTT username: ***
MQTT password: ***
MQTT base topic: nukihub
MQTT SSL CA: Not set
MQTT SSL CRT: Not set
MQTT SSL Key: Not set

------------ BLUETOOTH ------------
Bluetooth TX power (dB): 9
Bluetooth command nr of retries: 3
Bluetooth command retry delay (ms): 100
Seconds until reboot when no BLE beacons received: 60

------------ QUERY / PUBLISH SETTINGS ------------
Lock/Opener state query interval (s): 1800
Publish Nuki device authorization log: No
Max authorization log entries to retrieve: 5
Battery state query interval (s): 1800
Most non-JSON MQTT topics disabled: Yes
Publish Nuki device config: Yes
Config query interval (s): 3600
Publish Keypad info: No
Keypad query interval (s): 1800
Enable Keypad control: Yes
Publish Keypad topic per entry: No
Publish Keypad codes: No
Allow checking Keypad codes: No
Max keypad entries to retrieve: 10
Publish timecontrol info: No
Keypad query interval (s): 1800
Enable timecontrol control: No
Publish timecontrol topic per entry: No
Max timecontrol entries to retrieve: 10

------------ HOME ASSISTANT ------------
Home Assistant auto discovery enabled: Yes
Home Assistant auto discovery topic: homeassistant/
Nuki Hub configuration URL for HA: http://10.10.20.6

------------ NUKI LOCK ------------
Lock enabled: No

------------ NUKI OPENER ------------
Opener enabled: Yes
Paired: Yes
Nuki Hub device ID: 595935146
Nuki device ID: ***
Firmware version: 1.11.0
Hardware version: 4.17
Opener valid PIN set: No
Opener has keypad: No
Timecontrol highest entries count: 0
Register as: Bridge
Nuki Opener Lock/Unlock action set to Continuous mode in Home Assistant: No

------------ NUKI OPENER ACL ------------
Activate Ring-to-Open: Allowed
Deactivate Ring-to-Open: Allowed
Electric Strike Actuation: Allowed
Activate Continuous Mode: Allowed
Deactivate Continuous Mode: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed

------------ NUKI OPENER CONFIG ACL ------------
Name: Allowed
Latitude: Allowed
Longitude: Allowed
Pairing enabled: Allowed
Button enabled: Allowed
LED flash enabled: Allowed
Timezone offset: Allowed
DST mode: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed
Operating Mode: Allowed
Advertising Mode: Allowed
Timezone ID: Allowed
Intercom ID: Allowed
BUS mode Switch: Allowed
Short Circuit Duration: Allowed
Eletric Strike Delay: Allowed
Random Electric Strike Delay: Allowed
Electric Strike Duration: Allowed
Disable RTO after ring: Allowed
RTO timeout: Allowed
Doorbell suppression: Allowed
Doorbell suppression duration: Allowed
Sound Ring: Allowed
Sound Open: Allowed
Sound RTO: Allowed
Sound CM: Allowed
Sound confirmation: Allowed
Sound level: Allowed
Single button press action: Allowed
Double button press action: Allowed
Battery type: Allowed
Automatic battery type detection: Allowed
Reboot Nuki: Allowed

------------ GPIO ------------
```

### TO REPRODUCE
1. Restart NukiHub
2. Lock/unlock HA Lock entity
3. Wait 5 Minutes
4. Lock/Unlock HA entity

### EXPECTED BEHAVIOUR
ER: Entity State should update according to set lock state
AR: Enity does not update, command is not carried out (`error` state, see above)
ER != AR

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

iranl, 2024-12-20:
Duplicate of #561

Update to 9.05-master5

closed 2024-12-20 by iranl

h1ghrise, 2024-12-20:
wow that was fast :D thanks.
Sorry for duplicate post