technyon/nuki_hub #573: unable to get battery status

kujrv1, 2024-12-24:
### PROBLEM DESCRIPTION
Unable to get battery status from nuki 4 bluethoot version

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [ ] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [ ] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [ ] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [ ] Device used (e.g., ESP32-S3): _____
- [ ] Nuki Hub binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Provide the output of http://nukihubIP/info:
```

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

jolichter, 2024-12-24:
Hi, I can't reproduce the issue. My ESP32-S3 with Nuki Hub 9.05 works without any problems. Check your MQTT DP -> lock.battery.level.

If you encounter any issues, always share your system information.

iranl, 2024-12-24:
Closing for failure to follow the issue template.
You can reopen after providing all required information from the template.

closed 2024-12-24 by iranl

kujrv1, 2024-12-24:
factory reset and reconfigured... now.. all data receiving