technyon/nuki_hub #581: Ghost Nuki Hub devices in MQTT/HA since NH v9.02

bcutter, 2024-12-29:
### PROBLEM DESCRIPTION
Following up on https://github.com/technyon/nuki_hub/issues/537 where originally a few entities were redundant (sorted meanwhile), this issue is about a **redundant set of _devices_ of the separate nuki hub device**.

Ghost devices pop up randomly. A pattern _might_ be: once a NH update gets detected. This was the case today.
Ghost devices have strange names like "Unnamed device" or "WIFI signal strength" depending on the GUI view. Entities are named like `binary_sensor.none_mqtt_connected` or `binary_sensor.none_mqtt_connected_2`

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [X] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [X] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [ ] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [X] Device used (e.g., ESP32-S3): ESP32 (LOLIN32 Lite)
- [X] Nuki Hub binary firmware version number used: v9.05
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Provide the output of http://nukihubIP/info:
```
System Information

------------ NUKI HUB ------------
Version: 9.05
Build: 12444947845.7.1
Build type: Release
Build date: 2024-12-21
Updater version: 
Updater build: 
Updater build date: 
Uptime (min): 128
Config version: 905
Last restart reason FW: NotApplicable
Last restart reason ESP: ESP_RST_PANIC: Software reset due to exception/panic.
Free internal heap: 58272
Total internal heap: 252684
PSRAM Available: No
Network task stack high watermark: 8140
Nuki task stack high watermark: 5120

------------ GENERAL SETTINGS ------------
Network task stack size: 12288
Nuki task stack size: 8192
Check for updates: Yes
Latest version: 9.06
Allow update from MQTT: No
Web configurator username: ***
Web configurator password: ***
Web configurator enabled: Yes
Publish debug information enabled: No
MQTT log enabled: No
Webserial enabled: No
Bootloop protection enabled: No

------------ NETWORK ------------
Network device: Built-in Wi-Fi
Network connected: Yes
IP Address: xxx.xxx.xxx.xxx
SSID: XXX
BSSID of AP: XX:XX:XX:XX:XX:XX
ESP32 MAC address: XX:XX:XX:XX:XX:XX

------------ NETWORK SETTINGS ------------
Nuki Hub hostname: ESP-Nuki-Hub-Door-B
DHCP enabled: Yes
RSSI Publish interval (s): Disabled
Find WiFi AP with strongest signal: Yes
Restart ESP32 on network disconnect enabled: Yes
MQTT Timeout until restart (s): Disabled

------------ MQTT ------------
MQTT connected: Yes
MQTT broker address: xxx.xxx.xxx.xxx
MQTT broker port: 1883
MQTT username: ***
MQTT password: ***
MQTT base topic: nuki_door_b
MQTT SSL CA: Not set
MQTT SSL CRT: Not set
MQTT SSL Key: Not set

------------ BLUETOOTH ------------
Bluetooth TX power (dB): 9
Bluetooth command nr of retries: 5
Bluetooth command retry delay (ms): 100
Seconds until reboot when no BLE beacons received: 60

------------ QUERY / PUBLISH SETTINGS ------------
Lock/Opener state query interval (s): 3600
Publish Nuki device authorization log: Yes
Max authorization log entries to retrieve: 5
Battery state query interval (s): 7200
Most non-JSON MQTT topics disabled: No
Publish Nuki device config: Yes
Config query interval (s): 3600
Publish Keypad info: No
Keypad query interval (s): 1800
Enable Keypad control: Yes
Publish Keypad topic per entry: No
Publish Keypad codes: No
Allow checking Keypad codes: No
Max keypad entries to retrieve: 10
Publish timecontrol info: No
Keypad query interval (s): 1800
Enable timecontrol control: No
Publish timecontrol topic per entry: No
Max timecontrol entries to retrieve: 10

------------ HOME ASSISTANT ------------
Home Assistant auto discovery enabled: Yes
Home Assistant auto discovery topic: homeassistant/
Nuki Hub configuration URL for HA: http://xxx.xxx.xxx.xxx

------------ NUKI LOCK ------------
Lock enabled: Yes
Paired: Yes
Nuki Hub device ID: XXXXXXXXXX
Nuki device ID: ***
Firmware version: 3.10.7
Hardware version: 4.10
Valid PIN set: Yes
Has door sensor: No
Has keypad: No
Timecontrol highest entries count: 0
Register as: Bridge

------------ HYBRID MODE ------------
Hybrid mode enabled: No

------------ NUKI LOCK ACL ------------
Lock: Allowed
Unlock: Allowed
Unlatch: Allowed
Lock N Go: Allowed
Lock N Go Unlatch: Allowed
Full Lock: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed

------------ NUKI LOCK CONFIG ACL ------------
Name: Disallowed
Latitude: Disallowed
Longitude: Disallowed
Auto Unlatch: Disallowed
Pairing enabled: Allowed
Button enabled: Allowed
LED flash enabled: Allowed
LED brightness: Allowed
Timezone offset: Disallowed
DST mode: Disallowed
Fob Action 1: Disallowed
Fob Action 2: Disallowed
Fob Action 3: Disallowed
Single Lock: Allowed
Advertising Mode: Disallowed
Timezone ID: Disallowed
Unlocked Position Offset Degrees: Disallowed
Locked Position Offset Degrees: Disallowed
Single Locked Position Offset Degrees: Disallowed
Unlocked To Locked Transition Offset Degrees: Disallowed
Lock n Go timeout: Allowed
Single button press action: Allowed
Double button press action: Allowed
Detached cylinder: Disallowed
Battery type: Disallowed
Automatic battery type detection: Disallowed
Unlatch duration: Allowed
Auto lock timeout: Disallowed
Auto unlock disabled: Disallowed
Nightmode enabled: Disallowed
Nightmode start time: Disallowed
Nightmode end time: Disallowed
Nightmode auto lock enabled: Disallowed
Nightmode auto unlock disabled: Disallowed
Nightmode immediate lock on start: Disallowed
Auto lock enabled: Disallowed
Immediate auto lock enabled: Disallowed
Auto update enabled: Disallowed
Reboot Nuki: Allowed

------------ NUKI OPENER ------------
Opener enabled: No

------------ GPIO ------------
```

### TO REPRODUCE
1. Use NH
2. Wait for NH discovering a new update
3. Immediately see the ghost devices popping up in MQTT integration in HA

### EXPECTED BEHAVIOUR
The NH device introduced with NH v9.02 is the one and only of its kind, connected with the lock/opener.

### SCREENSHOTS
**1. ✅ Lock device**
[image]

**2. ✅ Nuki Hub device (e. g. `binary_sensor.esp_nuki_hub_door_b_mqtt_connected`)**
[image]

[image]

**3. ❌ Ghost Nuki Hub device 1 (e. g. `binary_sensor.none_mqtt_connected`)**
[image]

[image]

**4. ❌ Ghost Nuki Hub device 2 (e. g. `binary_sensor.none_mqtt_connected_2`)**
[image]

[image]

**5. Overview of plausible (lock and nuki hub) and redundant/ghost nuki hub devices**
[image]

_(also note: clearly a reference is missing here, indicators: entity_id naming (`*none*`), no reference in columns "manufacturer" and "model".)_

### ADDITIONAL CONTEXT
- Root cause:
  - Still not sure what triggers the creation of the redundant NH devices. For the affected NH device today it seems to have happened within one hour after the ESP has been restarted:
     - 12:55 NH ESP restarted
     - 13:40 discovery of 9.06 update
     - 13:40 creation of two NH ghost devices
  - Therefore I manually restarted my other NH (for `door_a`) now, aiming to provoke the same behavior.
- A bit of (painful as one project member hardly didn't believe me) discussion can be found at https://github.com/technyon/nuki_hub/issues/537, probably adding nothing worthful to the above description.
- I tend to say it's either a MQTT broker (but: all!!! topics both HA autodiscovery as well as the original NH topics have been fully purged once already) or MQTT integration issue. I am filing this here as the split of the original lock/opener/NH device into separate MQTT devices (lock, opener, NH) has been introduced in 9.02 and this was the very beginning I saw this issue.
- Usually I just deleted those ghost devices from the MQTT integration immediately. Today I renamed and disabled them.

**(Please, remember to close the issue when the problem has been addressed)**

bcutter, 2024-12-29:
(added MQTT topics screenshot number 6)

iranl, 2024-12-29:
I'm assuming that the `XXXXXXXXXX` in

```
------------ NUKI LOCK ------------
Lock enabled: Yes
Paired: Yes
Nuki Hub device ID: XXXXXXXXXX
```

corresponds to the number you made green on screenshot 6.
(Note that this is a randomly generated number by NukiHub on first boot and not really private information)

If you have erased the device/done factory resets this number will change (or imported config from a different device or device that was later factory reset/erased). Looking at the issues you have had between 9.00 and 9.05 I wouldn't find it odd if you have done this once or multiple times.

In that case my guess would be that the homeassistant discovery topics with other numbers (4266...., 1090....., 3247.....) were once valid Nuki Hub device ID's used by your device. As long as these topics are retained on your broker Home Assistant will keep on creating entities and a device based on these topics.

A change in the topic that is connected to these entries (such as the update topic, which is still a valid mqtt topic) might trigger Home Assistant to recreate the entities/device.

Steps to complete purge your NukiHub HomeAssistant topics:

- Enter `ids":["nuki_` in the MQTT explorer search bar
- Click on the top `homeassistant` topic

[image]

- Press delete
- Click `Yes`
- Reboot NukiHub to recreate the correct topics for the current Nuki Hub device ID

If these devices/entities and entries on your broker return after this please let me know and post the contents of the new incorrect mqtt topics.

bcutter, 2024-12-29:
Oh wow, you pointed me into an interesting direction with:

> I'm assuming that the `XXXXXXXXXX` in
>
> ```
> ------------ NUKI LOCK ------------
> Lock enabled: Yes
> Paired: Yes
> Nuki Hub device ID: XXXXXXXXXX
> ```
>
> corresponds to the number you made green on [...]

No, indeed it does not! It actually is:
```
------------ NUKI LOCK ------------
Lock enabled: Yes
Paired: Yes
Nuki Hub device ID: 3247846687
Nuki device ID: ***
```

So one of the ghost devices.

Yes I had to re-flash especially this ESP (not so important door compared to the other one, so acting as some kind of productive test ESP) a few times in the last weeks.

It's strange that this current device (`3247846687`) is NOT linked to the lock and shown with the corresponding meta data in the colums (screenshot 5).

I think I need to purge everything once again. But even if I manage to finally get rid of those duplicates: **how to prevent this from happening again in future?** I mean it could happen quite quickly you have to re-flash NH (like everyone had to when upgrading from v8 to v9 release).
- Can we improve something technically here?
- Or at least (!) note this in the docs or update part of the web interface etc.?
A short sentence and your brief "how-to purge NH MQTT entities (for various reasons)" would be nice - even I'm not sure I personally would recall this if this happens to me in a few months again.

Enough theory/ideas, I'm taking care of the topics purge once again, turning back. **Looking at your guide:**

Using `ids":["nuki_` is dangerous, as I have another NH with another lock in place. I don't want to delete anything working fine. I think I need to manually purge all entities which are definitely not associated to the other lock/NH.

iranl, 2024-12-29:
> I mean it could happen quite quickly you have to re-flash NH (like everyone had to when upgrading from v8 to v9 release).

Reflashing NH (with erase) is not needed often. 8 -> 9 was an outlier because of the changed partition scheme. The NukiHub HomeAssistant discovery only changed in 9.02 so this is not an issue as part of the 8 -> 9 upgrade.

> Can we improve something technically here?

Pretty hard to prevent completely. We might be able to exclude the ID from a factory reset from inside NukiHub (although then it won't really be a factory reset) and it will still not survive a true erase using esptool/webflasher. We could use the unique mac instead of a random number (`esp_efuse_mac_get_default`) which would always be the same on a device and thus survive factory resets, but what do you do with import/export to a new/different device? Best option might be to use the efuse if the ID is empty and afterwards allow it to be exported/imported to a different device. But than if you reuse the old device it will collide with the new device that uses its ID. Or we should detect that the efuse != deviceD and at that time remove the topics of the deviceID and create new topics at the efuse ID.

> Or at least (!) note this in the docs or update part of the web interface etc.?
A short sentence and your brief "how-to purge NH MQTT entities (for various reasons)" would be nice - even I'm not sure I personally would recall this if this happens to me in a few months again.

Good suggestion

> Using `ids":["nuki_` is dangerous, as I have another NH with another lock in place. I don't want to delete anything working fine. I think I need to manually purge all entities which are definitely not associated to the other lock/NH.

It is safe as long as you reboot any NukiHub device on this broker afterwards. Probably quicker and ensures a complete purge compared to removing topics by hand

closed 2024-12-30 by iranl

mundschenk-at, 2024-12-30:
Did you reload the MQTT integration after purging? The "deactivated" orphaned entities should go away then.

bcutter, 2024-12-30:
> Did you reload the MQTT integration after purging? The "deactivated" orphaned entities should go away then.

Yes I did now. Still exist. Same in other categories like
[image]

But: I absolutely don't care about those. It's a HA thing.
--> NH fine, MQTT fine, no ghost devices at the moment in HA. Solved for me, for now.

bcutter, 2025-01-13:
Please note: after updating NH from 9.06 to 9.07 there are new MQTT ghost devices - again. One for every NH.

In MQTT broker there are ones with an ID consisting of 10 numbers and ones with an ID length of 15 numbers.

The old device rendered HA entity ids without respect of German "Umlaut" (e.g. "tur"), the new devices respect them for the first time ever (e.g. "tuer").

Are those new redundant devices to be expected because of the changes in NH 9.07 (like #583)? I guess it is recommended to purge everything in the MQTT broker - once again - right?

Hopefully that
- is not necessary on every update...
- does not affect every NH and HA user...

Thanks for a short clarification on what I experienced with the NH 9.07 update. @iranl

iranl, 2025-01-13:
> Please note: after updating NH from 9.06 to 9.07 there are new MQTT ghost devices - again. One for every NH.
> 
> In MQTT broker there are ones with an ID consisting of 10 numbers and ones with an ID length of 15 numbers.

> Can we improve something technically here?

That's what you get for asking me to improve the HA ID handling ;)
The 15 digit HA ID is the new one and is based on the ESP MAC. This should be unique to every device and prevent collisions.
When moving from one ESP to another one it should also auto correct the ID after import/export and remove old topics.

> The old device rendered HA entity ids without respect of German "Umlaut" (e.g. "tur"), the new devices respect them for the first time ever (e.g. "tuer").

Can't take credit for that. Probably a HA improvement.

> Are those new redundant devices to be expected because of the changes in NH 9.07 (like #583)?

Yes

> **I guess it is recommended to purge everything in the MQTT broker - once again (and maybe for the very last time) - right?**

No promises for forever, but yes.

> Hopefully that
> 
> * is not necessary on every update...
> * does not affect every NH and HA user...

No, this shouldn't be necessary on every update. There is a mechanism for removing old ID's but this will probably not work in any and all cases, so could affect multiple NH and HA users.

> Thanks for a short clarification on what I experienced with the NH 9.07 update. @iranl

np

bcutter, 2025-01-13:
Well... I'm pretty done with that MQTT topics purging. Somehow accidentally deleted ALL homeassistant topics as the Nuki filter in MQTT Explorer somehow seems to not have worked. Alle devices and hundreds of entities gone. In the end I had to restore a Mosquitto broker backup and re-configure every single device and almost every entity manually.

Personal luck I guess, but a short NH ended up in ruining my whole MQTT setup and rendering 1/2 of my smart home dumb for hours. I really hope that ID mess is finished forever, I'm not going to purge MQTT topics again, only selectively. Very dangerous, lesson learned today.

PR #583 (2024-12-30): Improve device ID for HA discovery
## Description:

**Related issue (if applicable):** fixes #581 

## Checklist:
  - [x] The pull request is done against the latest master branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works
  - [x] I accept the [CLA](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md#contributor-license-agreement-cla).