technyon/nuki_hub #584: MQTT connections drops

FBlass, 2024-12-30:
### PROBLEM DESCRIPTION

Nuki Hub keeps disconnecting from MQTT Broker when connected to mains power. 

I have the Nuki Hub version 8.x version running on my Shelly 1 Plus (dual core, ESP32-S3) for approx. 1 year without any issues. I decided to upgrade to 9.x today. The flash worked fine (tried via WebFlash and esptool.py). When powered with 3.3v via the UART adapter the MQTT connection seems stable. When connected directly via mains it occasionally connects, but drops just a few seconds after. WIFI connection is strong at about -40 dBm.

Interestingly, when connected, it still reports a Nuki Hub version of 8.35 in the nuki topic. In the nukihub topic the version 9.06 is reported correctly. Maybe there are leftovers from the old Nuki Hub version? I did however check the Clear Flash button in the WebFlasher and `erase_flash` in the esptool.py. 

Is there any way I can find logs on why the MQTT connection drops?
   

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [x] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [x] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [x] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [x] Device used (e.g., ESP32-S3): ESP32-S3
- [x] Nuki Hub binary firmware version number used: 9.06
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Provide the output of http://nukihubIP/info:
```
System Information
------------ NUKI HUB ------------
Version: 9.06
Build: 12534715018.8.1
Build type: Release
Build date: 2024-12-29
Updater version: 
Updater build: 
Updater build date: 
Uptime (min): 10
Config version: 906
Last restart reason FW: NotApplicable
Last restart reason ESP: ESP_RST_POWERON: Reset due to power-on event.
Free internal heap: 100280
Total internal heap: 308228
PSRAM Available: No
Network task stack high watermark: 8164
Nuki task stack high watermark: 6504

------------ GENERAL SETTINGS ------------
Network task stack size: 12288
Nuki task stack size: 8192
Check for updates: Yes
Latest version: 9.06
Allow update from MQTT: No
Web configurator username: Not set
Web configurator password: Not set
Web configurator enabled: Yes
Publish debug information enabled: No
MQTT log enabled: No
Webserial enabled: No
Bootloop protection enabled: No

------------ NETWORK ------------
Network device: Built-in Wi-Fi
Network connected: Yes
IP Address: 192.168.178.102
SSID: XXX
BSSID of AP: XXX
ESP32 MAC address: XXX

------------ NETWORK SETTINGS ------------
Nuki Hub hostname: nukihub
DHCP enabled: Yes
RSSI Publish interval (s): 60
Find WiFi AP with strongest signal: No
Restart ESP32 on network disconnect enabled: No
MQTT Timeout until restart (s): 60

------------ MQTT ------------
MQTT connected: Yes
MQTT broker address: 192.168.178.10
MQTT broker port: 42883
MQTT username: ***
MQTT password: ***
MQTT base topic: nukihub
MQTT SSL CA: Not set
MQTT SSL CRT: Not set
MQTT SSL Key: Not set

------------ BLUETOOTH ------------
Bluetooth TX power (dB): 9
Bluetooth command nr of retries: 3
Bluetooth command retry delay (ms): 100
Seconds until reboot when no BLE beacons received: 60

------------ QUERY / PUBLISH SETTINGS ------------
Lock/Opener state query interval (s): 1800
Publish Nuki device authorization log: No
Max authorization log entries to retrieve: 5
Battery state query interval (s): 1800
Most non-JSON MQTT topics disabled: No
Publish Nuki device config: Yes
Config query interval (s): 3600
Publish Keypad info: No
Keypad query interval (s): 1800
Enable Keypad control: Yes
Publish Keypad topic per entry: No
Publish Keypad codes: No
Allow checking Keypad codes: No
Max keypad entries to retrieve: 10
Publish timecontrol info: No
Keypad query interval (s): 1800
Enable timecontrol control: No
Publish timecontrol topic per entry: No
Max timecontrol entries to retrieve: 10

------------ HOME ASSISTANT ------------
Home Assistant auto discovery enabled: No

------------ NUKI LOCK ------------
Lock enabled: Yes
Paired: No
Nuki Hub device ID: 3186046867
Nuki device ID: Not set
Firmware version: 
Hardware version: 
Valid PIN set: -
Has door sensor: No
Has keypad: No
Timecontrol highest entries count: 0
Register as: Bridge

------------ HYBRID MODE ------------
Hybrid mode enabled: No

------------ NUKI LOCK ACL ------------
Lock: Allowed
Unlock: Allowed
Unlatch: Allowed
Lock N Go: Allowed
Lock N Go Unlatch: Allowed
Full Lock: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed

------------ NUKI LOCK CONFIG ACL ------------
Name: Allowed
Latitude: Allowed
Longitude: Allowed
Auto Unlatch: Allowed
Pairing enabled: Allowed
Button enabled: Allowed
LED flash enabled: Allowed
LED brightness: Allowed
Timezone offset: Allowed
DST mode: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed
Single Lock: Allowed
Advertising Mode: Allowed
Timezone ID: Allowed
Unlocked Position Offset Degrees: Allowed
Locked Position Offset Degrees: Allowed
Single Locked Position Offset Degrees: Allowed
Unlocked To Locked Transition Offset Degrees: Allowed
Lock n Go timeout: Allowed
Single button press action: Allowed
Double button press action: Allowed
Detached cylinder: Allowed
Battery type: Allowed
Automatic battery type detection: Allowed
Unlatch duration: Allowed
Auto lock timeout: Allowed
Auto unlock disabled: Allowed
Nightmode enabled: Allowed
Nightmode start time: Allowed
Nightmode end time: Allowed
Nightmode auto lock enabled: Allowed
Nightmode auto unlock disabled: Allowed
Nightmode immediate lock on start: Allowed
Auto lock enabled: Allowed
Immediate auto lock enabled: Allowed
Auto update enabled: Allowed
Reboot Nuki: Allowed

------------ NUKI OPENER ------------
Opener enabled: Yes
Paired: No
Nuki Hub device ID: 3186046867
Nuki device ID: Not set
Firmware version: 
Hardware version: 
Opener valid PIN set: -
Opener has keypad: No
Timecontrol highest entries count: 0
Register as: Bridge
Nuki Opener Lock/Unlock action set to Continuous mode in Home Assistant: No

------------ NUKI OPENER ACL ------------
Activate Ring-to-Open: Allowed
Deactivate Ring-to-Open: Allowed
Electric Strike Actuation: Allowed
Activate Continuous Mode: Allowed
Deactivate Continuous Mode: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed

------------ NUKI OPENER CONFIG ACL ------------
Name: Allowed
Latitude: Allowed
Longitude: Allowed
Pairing enabled: Allowed
Button enabled: Allowed
LED flash enabled: Allowed
Timezone offset: Allowed
DST mode: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed
Operating Mode: Allowed
Advertising Mode: Allowed
Timezone ID: Allowed
Intercom ID: Allowed
BUS mode Switch: Allowed
Short Circuit Duration: Allowed
Eletric Strike Delay: Allowed
Random Electric Strike Delay: Allowed
Electric Strike Duration: Allowed
Disable RTO after ring: Allowed
RTO timeout: Allowed
Doorbell suppression: Allowed
Doorbell suppression duration: Allowed
Sound Ring: Allowed
Sound Open: Allowed
Sound RTO: Allowed
Sound CM: Allowed
Sound confirmation: Allowed
Sound level: Allowed
Single button press action: Allowed
Double button press action: Allowed
Battery type: Allowed
Automatic battery type detection: Allowed
Reboot Nuki: Allowed

------------ GPIO ------------
```

### TO REPRODUCE

Unknown. For me it happens everytime. I was only able to test with the Shelly Plus 1.

### EXPECTED BEHAVIOUR

Stable MQTT connection.

### SCREENSHOTS

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

iranl, 2024-12-30:
`nukihub` is your current topic for NukiHub as you can see on your infopage (`MQTT base topic: nukihub`). 
The `nuki` topic is not used by your device when running 9.06.

> When powered with 3.3v via the UART adapter the MQTT connection seems stable. When connected directly to mains voltage it occasionally connects, but drops just a few seconds after. WIFI connection is strong at about -40 dBm.

You clearly have a power supply issue. Multiple users have previously reported this here and elsewhere (e.g. https://github.com/espressif/arduino-esp32/issues/9799) after upgrading to versions built on more recent Arduino and ESP-IDF versions. Power requirements have gone up in more recent Arduino/ESP-IDF versions. NukiHub upgraded to Arduino 3.0/ESP-IDF 5.1 in 9.0 and Arduino 3.1/ESP-IDF 5.3 in 9.05.

Switching to a more stable power supply is the solution, but this is probably kind of hard to do on a Shelly.

I would recommend using the Shelly for what it is intended to do (e.g. mains connected relay) and getting a [ESP32-S3 N16R8](aliexpress.com/w/wholesale-esp32s3-n16r8.html) for NukiHub and powering this with any leftover phone charger.

closed 2024-12-30 by iranl

FBlass, 2024-12-31:
Thanks for the answer and sorry for the confusion. I figured out the issue.

The nuki (and nukiopener) topics always popped up again, along with nukihub. I was able to successfully flash the Shelly on Windows, where it was detected as an ESP32-S3. This was however not correct, it is a standard ESP32. For that reason, apparently the old and the new version ran next to each other at the same time. 

I did a clean flash on Ubuntu with the esptool, where the chip was recognised correctly. Now, only the nukihub topic appears in MQTT and the MQTT connection is stable.