technyon/nuki_hub #595: Nuki Hub does not connect to WiFi because of wrong secret key (which is not wrong)

madCase1988, 2025-01-06:
### PROBLEM DESCRIPTION

Problem is that when i flash the latest version of nuki hub to an esp32 I did not manage to connect the device to wifi. I always get the error "Failed to connect to the given SSID with the given secret key, credentials not saved". The secret key is definitely correct. When I flash the e.g. the ESP Home Bluetooth Proxy Firmware to the same devices I do not have any problem connecting them to the wifi. It only occurs, when I flash the NukiHub Firmware

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ X] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [ X] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [ ] Device used (e.g., ESP32-S3): ESP32 WROOM 32 (‎AM03DE-GXFB0014-005) and M5Stack ATOM Lite ESP32
- [ ] Nuki Hub binary firmware version number used: 9.06

### TO REPRODUCE
_Steps to reproduce the behavior:_
- install Nuki Hub 9.06 to ESP32 WROOM 32 or M5Stack ATOM Lite ESP32
- connect to the Nuki Hub Wifi
- choose your WiFi
- enter the correct secret key 

### EXPECTED BEHAVIOUR
Nuki Hub should connect to Wifi with the entered secret key

### SCREENSHOTS
[image]

iranl, 2025-01-06:
Or are you using a Hidden SSID?

madCase1988, 2025-01-06:
> Is there anything special about your key? Special characters, length?

To my mind not. The key consists only of digits and has 16 digits in total. Shouldn't be a problem, should it?

madCase1988, 2025-01-06:
> Or are you using a Hidden SSID?

No. I also tried several router configuration concerning Encryption: tried WPA2-PSK (AES) and WPA3 Personal.

madCase1988, 2025-01-06:
What I forgot: when i connect the Nuki Hub to  Mobile Hotspot of my mobile it works. So I guess its something with the router configuration but i don't have a clue...

iranl, 2025-01-06:
Anything special in your SSID (length, characters)?

Have you tried changing your SSID and pass to the same as your mobile hotspot to test?

DHCP working?

madCase1988, 2025-01-06:
> Anything special in your SSID (length, characters)?
> 
> Have you tried changing your SSID and pass to the same as your mobile hotspot to test?
> 
> DHCP working?

To my mind nothing special. SSID consists of 23 (normal) letters and one number. No special characters. I named my mobile hotspot excatly like my home wifi and also set the Enryption exatly the same and mobile hotspot works but home wifi not. I also deactivated all integrated antivirus filters temporarily. 

When I flash tzhe same esp32 with bluetooth proxy software wifi connection is no problem!

technyon, 2025-01-06:
To ask again, DHCP is activated and working?
Your home Wifi works on the 2.4GHz band? ESPs don't work on 5GHz. If your network appears in the list when configuring the ESP, it's 2.4GHz.
Sure no MAC filter is active?
What kind model is your router?

madCase1988, 2025-01-06:
> Sure no MAC filter is active?

Sure no MAC filter is active? Yes
Your home Wifi works on the 2.4GHz band? Yes

I am also sure, because conneting same ESP with Proxy Firmware works...

iranl, 2025-01-06:
Note that is doesn't say anywhere that you have given an incorrect secret key. It just says it failed to connect to the ssid with the given key. Doesn't mean it's the pass

madCase1988, 2025-01-06:
> Have you disabled the option to search for the best AP and/or set a static IP on NukiHub while selecting your ssid/inputing password?

Yes, tried both.
> 
> If all that doesn't work you can try getting a serial log over usb while setting up the wifi
Will do that

madCase1988, 2025-01-06:
> Note that is doesn't say anywhere that you have given an incorrect secret key. It just says it failed to connect to the ssid with the given key. Doesn't mean it's the pass

Got you. Its correct! :)

iranl, 2025-01-15:
Closing for inactivity. Please re-open if the issue persists and you have provided the requested information (Serial logs)

closed 2025-01-15 by iranl