technyon/nuki_hub #598: No door Sensor

gatd, 2025-01-07:
wont see the nuki door sensor in nukihub mqtt is shows up but is unavailable
so i cant see if the door is open
the nuki lock work when i close the door it locks

nuki hub v 9.0.6
nuki lock 2

iranl, 2025-01-07:
Closing for failure to follow the issue template.
You can reopen after providing all required information from the template.

closed 2025-01-07 by iranl

gatd, 2025-01-07:
{
  "run": "1",
  "confVersion": "906",
  "mqttbroker": "192.168.0.11",
  "mqttport": "1883",
  "mqttlog": "",
  "checkupdates": "1",
  "websrvena": "",
  "lockena": "1",
  "lockpin": "1",
  "mqttpath": "nukihub",
  "openerena": "0",
  "openerpin": "",
  "openercont": "0",
  "maxkpad": "",
  "opmaxkpad": "",
  "maxtc": "",
  "opmaxtc": "",
  "enabtlprst": "0",
  "hassdiscovery": "homeassistant",
  "hassConfigUrl": "",
  "buffsize": "4096",
  "dhcpena": "0",
  "ipaddr": "192.168.0.48",
  "ipsub": "255.255.255.0",
  "ipgtw": "192.168.0.1",
  "dnssrv": "",
  "nwhw": "1",
  "rssipb": "60",
  "hostname": "nukihub",
  "nettmout": "60",
  "restdisc": "",
  "rstbcn": "60",
  "lockStInterval": "1800",
  "tcPerEntry": "0",
  "kpPerEntry": "0",
  "configInterval": "3600",
  "batInterval": "1800",
  "kpInterval": "1800",
  "kpCntrlEnabled": "1",
  "kpInfoEnabled": "0",
  "kpPubCode": "0",
  "tcCntrlEnabled": "0",
  "tcInfoEnabled": "0",
  "cnfInfoEnabled": "1",
  "regAsApp": "0",
  "regOpnAsApp": "0",
  "nrRetry": "3",
  "rtryDelay": "100",
  "disnonjson": "0",
  "pubAuth": "0",
  "pubdbg": "",
  "offHybrid": "0",
  "hybridTimer": "600",
  "hybridAct": "0",
  "hybridRtry": "0",
  "tsksznetw": "12288",
  "tsksznuki": "8192",
  "authmaxentry": "5",
  "kpmaxentry": "10",
  "tcmaxentry": "10",
  "updMqtt": "0",
  "bleTxPwr": "",
  "weblog": "",
  "nwbestrssi": "1",
  "ntwMDC": "",
  "ntwCLK": "",
  "ntwPHY": "",
  "ntwADDR": "",
  "ntwIRQ": "",
  "ntwRST": "",
  "ntwCS": "",
  "ntwSCK": "",
  "ntwMISO": "",
  "ntwMOSI": "",
  "ntwPWR": "",
  "ntwMDIO": "",
  "ntwRECONF": "",
  "maxauth": "",
  "opmaxauth": "",
  "authCtrlEna": "",
  "authPerEntry": "",
  "authInfoEna": "",
  "wifiSSID": "SBS",
  "kpChkEna": "",
  "disNtwNoCon": "",
  "hassena": "1",
  "hassdevdisc": "",
  "retGpio": "0",
  "dbgConnect": "0",
  "dbgCommu": "0",
  "dbgReadData": "0",
  "dbgHexData": "0",
  "dbgCommand": "0",
  "nukiConnMode": "1",
  "lckForceId": "",
  "lckForceDrsns": "",
  "lckForceKp": "",
  "opForceId": "",
  "opForceKp": "",
  "aclLckOpn": "0100000001000000010000000100000001000000010000000100000001000000010000000000000000000000000000000000000000000000000000000000000000000000",
  "confLckBasAcl": "01000000010000000100000001000000010000000100000001000000010000000100000001000000010000000100000001000000010000000100000001000000",
  "confLckAdvAcl": "0100000001000000010000000100000001000000010000000100000001000000010000000100000001000000010000000100000001000000010000000100000001000000010000000100000001000000010000000100000001000000",
  "confOpnBasAcl": "0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000",   "confOpnAdvAcl": "000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000" Sent from my iPhone