technyon/nuki_hub #599: nuki door sensor problem

gatd, 2025-01-07:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
the door sensor shows as unavailable
the nuki lock is working fine with the sensor and locks when door is closed
is just nukihub mqtt shows as unavailable

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [ ] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [ ] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [ ] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [ ] Device used (e.g., ESP32-S3): _____esp32 developer
- [ ] Nuki Hub binary firmware version number used: _____9.0.6
  - [ ] Pre-compiled yes installed via web
  - [ ] Self-compiled
- [ ] Provide the output of http://nukihubIP/info:
`
System Information
------------ NUKI HUB ------------
Version: 9.06
Build: 12534715018.8.1
Build type: Release
Build date: 2024-12-29
Updater version: 
Updater build: 
Updater build date: 
Uptime (min): 0
Config version: 906
Last restart reason FW: NotApplicable
Last restart reason ESP: ESP_RST_PANIC: Software reset due to exception/panic.
Free internal heap: 70420
Total internal heap: 252840
PSRAM Available: No
Network task stack high watermark: 11088
Nuki task stack high watermark: 7652

------------ GENERAL SETTINGS ------------
Network task stack size: 12288
Nuki task stack size: 8192
Check for updates: Yes
Latest version: 
Allow update from MQTT: No
Web configurator username: Not set
Web configurator password: Not set
Web configurator enabled: Yes
Publish debug information enabled: No
MQTT log enabled: No
Webserial enabled: No
Bootloop protection enabled: No

------------ NETWORK ------------
Network device: Built-in Wi-Fi
Network connected: Yes
IP Address: 192.168.0.48
SSID: SBS
BSSID of AP: 18:35:D1:FB:58:99
ESP32 MAC address: C0:49:EF:F9:9E:18

------------ NETWORK SETTINGS ------------
Nuki Hub hostname: nukihub
DHCP enabled: No
Static IP address: 192.168.0.48
Static IP subnet: 255.255.255.0
Static IP gateway: 192.168.0.1
Static IP DNS server: 
RSSI Publish interval (s): 60
Find WiFi AP with strongest signal: Yes
Restart ESP32 on network disconnect enabled: No
MQTT Timeout until restart (s): 60

------------ MQTT ------------
MQTT connected: No
MQTT broker address: 192.168.0.11
MQTT broker port: 1883
MQTT username: ***
MQTT password: ***
MQTT base topic: nukihub
MQTT SSL CA: Not set
MQTT SSL CRT: Not set
MQTT SSL Key: Not set

------------ BLUETOOTH ------------
Bluetooth TX power (dB): 9
Bluetooth command nr of retries: 3
Bluetooth command retry delay (ms): 100
Seconds until reboot when no BLE beacons received: 60

------------ QUERY / PUBLISH SETTINGS ------------
Lock/Opener state query interval (s): 1800
Publish Nuki device authorization log: No
Max authorization log entries to retrieve: 5
Battery state query interval (s): 1800
Most non-JSON MQTT topics disabled: No
Publish Nuki device config: Yes
Config query interval (s): 3600
Publish Keypad info: No
Keypad query interval (s): 1800
Enable Keypad control: Yes
Publish Keypad topic per entry: No
Publish Keypad codes: No
Allow checking Keypad codes: No
Max keypad entries to retrieve: 10
Publish timecontrol info: No
Keypad query interval (s): 1800
Enable timecontrol control: No
Publish timecontrol topic per entry: No
Max timecontrol entries to retrieve: 10

------------ HOME ASSISTANT ------------
Home Assistant auto discovery enabled: Yes
Home Assistant auto discovery topic: homeassistant/
Nuki Hub configuration URL for HA: http://192.168.0.48

------------ NUKI LOCK ------------
Lock enabled: Yes
Paired: No
Nuki Hub device ID: 1753238776
Nuki device ID: ***
Firmware version: 
Hardware version: 
Valid PIN set: -
Has door sensor: No
Has keypad: No
Timecontrol highest entries count: 0
Register as: Bridge

------------ HYBRID MODE ------------
Hybrid mode enabled: No

------------ NUKI LOCK ACL ------------
Lock: Allowed
Unlock: Allowed
Unlatch: Allowed
Lock N Go: Allowed
Lock N Go Unlatch: Allowed
Full Lock: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed

------------ NUKI LOCK CONFIG ACL ------------
Name: Allowed
Latitude: Allowed
Longitude: Allowed
Auto Unlatch: Allowed
Pairing enabled: Allowed
Button enabled: Allowed
LED flash enabled: Allowed
LED brightness: Allowed
Timezone offset: Allowed
DST mode: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed
Single Lock: Allowed
Advertising Mode: Allowed
Timezone ID: Allowed
Unlocked Position Offset Degrees: Allowed
Locked Position Offset Degrees: Allowed
Single Locked Position Offset Degrees: Allowed
Unlocked To Locked Transition Offset Degrees: Allowed
Lock n Go timeout: Allowed
Single button press action: Allowed
Double button press action: Allowed
Detached cylinder: Allowed
Battery type: Allowed
Automatic battery type detection: Allowed
Unlatch duration: Allowed
Auto lock timeout: Allowed
Auto unlock disabled: Allowed
Nightmode enabled: Allowed
Nightmode start time: Allowed
Nightmode end time: Allowed
Nightmode auto lock enabled: Allowed
Nightmode auto unlock disabled: Allowed
Nightmode immediate lock on start: Allowed
Auto lock enabled: Allowed
Immediate auto lock enabled: Allowed
Auto update enabled: Allowed
Reboot Nuki: Allowed

------------ NUKI OPENER ------------
Opener enabled: No

------------ GPIO ------------``
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
rebooted still the same
### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

razorbac91, 2025-01-07:
same here, it disappeared. Opening the Nuki app i found that it was in error, i redid the calibration process, but still unavailable on nukihub mqtt

iranl, 2025-01-07:
Your info page shows your ESP has just rebooted and thus the info page shows no usefull information.

The page shows you are not connected to MQTT and no device is paired.

Does your NukiHub even stay up for more than a couple of minutes? (Uptime on info page > 5 minutes?)
If your NukiHub is constantly crashing first reflash NukiHub using the Webflash and do a full erase using the webflasher.
Setup your device from scratch from there and pair your device again.
Check if it stays up in that case and post a new info page or get serial logs if your device is unstable.

gatd, 2025-01-07:
ASent from my iPhone

gatd, 2025-01-07:
I have re flashed 3x just sent the info after thatAnother thing is is I unlock the door with Nuki hub when I close the door and it auto locks Nuki hub still shows as unlocked But main problem is the door Sent from my iPhone

iranl, 2025-01-07:
We can't do anything with the information you have provided.
Provide the information as requested.

gatd, 2025-01-07:
Is this what you needSent from my iPhone

gatd, 2025-01-08:
ok i have updated to 9.07-master11 and the door sensor is now back! and working
only problem is mqtt shows 2 nukihubs 9.0.7-master11 and 9.0.6

iranl, 2025-01-08:
I'm assuming you mean in Home assistant.

Clear Home Assistant discovery as outlined jn the README (Faq section)

closed 2025-01-09 by iranl