technyon/nuki_hub #604: Issues with WT32-ETH01, ESP32-Solo1, not Booting with 9.05/06

dutzi88, 2025-01-10:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

After 9.05 i cant get the ESP32-S1 to boot.
First try via OTA, afterwards direct flash over Wire.
9.04 works, but the functionality on the Opener is quiet bad.

Also, after 9.02 the Opener gets more and more erratic. RTO does only work every third or fourth time and if i trigger it directli to unlock i have long delays (have not yet measured the time, but if its cold out it and i have to pee and standing in front of my locked door it feels veeery long ;-))
But maybe this is gone with the latest improvements in the firmware, but this version wont run...

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [x ] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [ x] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [ x] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [ x] Device used (e.g., ESP32-S3): ESP32-S1, WT32-ETH01
- [x ] Nuki Hub binary firmware version number used: 9.05/06
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [ x] Provide the output of http://nukihubIP/info:
```
System Information

------------ NUKI HUB ------------
Version: 9.02
Build: 11995573722.4.1
Build type: Release
Build date: 2024-11-24
Updater version: 9.04
Updater build: 12361409996.6.1
Updater build date: 2024-12-03
Uptime (min): 11171
Config version: 904
Last restart reason FW: RequestedViaWebServer
Last restart reason ESP: ESP_RST_SW: Software reset via esp_restart.
Free internal heap: 138932
Total internal heap: 309912
PSRAM Available: No
Network task stack high watermark: 8240
Nuki task stack high watermark: 5124

------------ GENERAL SETTINGS ------------
Network task stack size: 12288
Nuki task stack size: 8192
Check for updates: Yes
Latest version: 9.06
Allow update from MQTT: Yes
Web configurator username: Not set
Web configurator password: Not set
Web configurator enabled: Yes
Publish debug information enabled: No
MQTT log enabled: No
Webserial enabled: No
Bootloop protection enabled: No

------------ NETWORK ------------
Network device: WT32-ETH01
Network connected: Yes
IP Address: 192.168.5.181

------------ NETWORK SETTINGS ------------
Nuki Hub hostname: NukiOpener
DHCP enabled: No
Static IP address: 192.168.5.181
Static IP subnet: 255.255.255.0
Static IP gateway: 192.168.5.1
Static IP DNS server: 192.168.5.1
Restart ESP32 on network disconnect enabled: No
MQTT Timeout until restart (s): 60

------------ MQTT ------------
MQTT connected: Yes
MQTT broker address: homeassistant.local
MQTT broker port: 1883
MQTT username: ***
MQTT password: ***
MQTT base topic: nukihub
MQTT SSL CA: Not set
MQTT SSL CRT: Not set
MQTT SSL Key: Not set

------------ BLUETOOTH ------------
Bluetooth TX power (dB): 9
Bluetooth command nr of retries: 10
Bluetooth command retry delay (ms): 100
Seconds until reboot when no BLE beacons received: 60

------------ QUERY / PUBLISH SETTINGS ------------
Lock/Opener state query interval (s): 1800
Publish Nuki device authorization log: Yes
Max authorization log entries to retrieve: 5
Battery state query interval (s): 1800
Most non-JSON MQTT topics disabled: No
Publish Nuki device config: Yes
Config query interval (s): 3600
Publish Keypad info: No
Keypad query interval (s): 1800
Enable Keypad control: Yes
Publish Keypad topic per entry: No
Publish Keypad codes: No
Allow checking Keypad codes: No
Max keypad entries to retrieve: 10
Publish timecontrol info: Yes
Keypad query interval (s): 1800
Enable timecontrol control: Yes
Publish timecontrol topic per entry: Yes
Max timecontrol entries to retrieve: 10

------------ HOME ASSISTANT ------------
Home Assistant auto discovery enabled: Yes
Home Assistant auto discovery topic: homeassistant/
Nuki Hub configuration URL for HA: http://192.168.5.181

------------ NUKI LOCK ------------
Lock enabled: No

------------ NUKI OPENER ------------
Opener enabled: Yes
Paired: Yes
Nuki Hub device ID: 444256864
Nuki device ID: ***
Firmware version: 1.10.1
Hardware version: 4.17
Opener valid PIN set: Yes
Opener has keypad: No
Timecontrol highest entries count: 0
Register as: Bridge
Nuki Opener Lock/Unlock action set to Continuous mode in Home Assistant: No

------------ NUKI OPENER ACL ------------
Activate Ring-to-Open: Allowed
Deactivate Ring-to-Open: Allowed
Electric Strike Actuation: Allowed
Activate Continuous Mode: Allowed
Deactivate Continuous Mode: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed

------------ NUKI OPENER CONFIG ACL ------------
Name: Allowed
Latitude: Allowed
Longitude: Allowed
Pairing enabled: Allowed
Button enabled: Allowed
LED flash enabled: Allowed
Timezone offset: Allowed
DST mode: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed
Operating Mode: Allowed
Advertising Mode: Allowed
Timezone ID: Allowed
Intercom ID: Allowed
BUS mode Switch: Allowed
Short Circuit Duration: Allowed
Eletric Strike Delay: Allowed
Random Electric Strike Delay: Allowed
Electric Strike Duration: Allowed
Disable RTO after ring: Allowed
RTO timeout: Allowed
Doorbell suppression: Allowed
Doorbell suppression duration: Allowed
Sound Ring: Allowed
Sound Open: Allowed
Sound RTO: Allowed
Sound CM: Allowed
Sound confirmation: Allowed
Sound level: Allowed
Single button press action: Allowed
Double button press action: Allowed
Battery type: Allowed
Automatic battery type detection: Allowed
Reboot Nuki: Allowed

------------ GPIO ------------

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Flash Firmware, direct or OTA.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Booting up, after regular update it should connect to the Network and start working.
After Flash over wire, Provide a AP to set up everything.
Does neither

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

Also, after 9.02 the Opener gets more and more erratic. RTO does only work every third or fourth time and if i trigger it directli to unlock i have long delays (have not yet measured the time, but if its cold out it and i have to pee and standing in front of my locked door it feels veeery long ;-))
But maybe this is gone with the latest improvements in the firmware, but this version wont run...

**(Please, remember to close the issue when the problem has been addressed)**

technyon, 2025-01-10:
Hi. A few thins

- You're mentioning two issue, Nuk Hub not booting and problem with the Opener. Please split it into two issues so we can keep track of each without confusing things.
- In those topics please make clear what exactly you tried on which device.
- With 9.05/06 you mean Nuki Hub versions 9.05 and 9.06 ?

dutzi88, 2025-01-10:
ok, yes, makes sense. Sorry, my first issue openen on Github, there is some space to be more efficient on my side.
Disregard the opener-issue. if that persists, i will open a new issue.

Regarding not Booting:
Yes, i tried to Flash 9.05 and 9.06 on both my ESP32-S1, and they would not boot anymore.

technyon, 2025-01-10:
ok you're using one device, I misunderstood.

Yes you should be able to get logs. Just wire it up the way you did when you flashed it, and use a program like HTERM to get logs. You have to set a speed of 115200.

iranl, 2025-01-10:
What firmware version are you flashing?

The title of your issue mentions ESP32-Solo1 and WT32-ETH01.

Note that the WT32-ETH01 has a regular dual core ESP32(-S1) chip and not a single core ESP32-Solo1 chip (This is the WT32-ETH02).

You should not flash the ESP32-Solo1 binary to the WT32-ETH01, just flash the regular ESP32 binary.

dutzi88, 2025-01-10:
Well, dang. OTA-Flash of the ESP32(S1) firmware, and it just works. Sorry, my bad.
Thanks for your time and excuse for the false alarm.

closed 2025-01-10 by dutzi88