technyon/nuki_hub #607: Constant authentication on web interface since NH v9.07 (at least Firefox)

bcutter, 2025-01-13:
### PROBLEM DESCRIPTION
Since NH v9.07 on Firefox browser after providing the initial credential set in `/get?page=cred` (webserver access), the authentication pops up repeatedly. Not happening in Chrome, EDGE and Safari (iOS/iPadOS). For Chrome I discovered a 2nd authentication is only necessary once browsing another page (randomly: "Access Level Configuration" or "Credentials", doesn't really matter).

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [X] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [X] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [ ] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [X] Device used (e.g., ESP32-S3): LOLIN32-Lite
- [X] Nuki Hub binary firmware version number used: 12722791725.9.1
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Provide the output of http://nukihubIP/info:
```
System Information
------------ NUKI HUB ------------
Device: ESP32
Version: 9.07
Build: 12722791725.9.1
Build type: Release
Build date: 2025-01-11
Updater version: 9.07
Updater build: 12722791725.9.1
Updater build date: 2025-01-10
Uptime (min): 821
Config version: 907
Last restart reason FW: OTACompleted
Last restart reason ESP: ESP_RST_SW: Software reset via esp_restart.
Free internal heap: 73436
Total internal heap: 252700
PSRAM Available: No
Network task stack high watermark: 8320
Nuki task stack high watermark: 5124

------------ GENERAL SETTINGS ------------
Network task stack size: 12288
Nuki task stack size: 8192
Check for updates: Yes
Latest version: 9.07
Allow update from MQTT: No
Update NukiHub and Nuki devices time using NTP: No
Web configurator username: ***
Web configurator password: ***
Web configurator enabled: Yes
HTTP SSL: Disabled
Publish debug information enabled: No
MQTT log enabled: No
Webserial enabled: No
Bootloop protection enabled: No

------------ NETWORK ------------
Network device: Built-in Wi-Fi
Network connected: Yes
IP Address: xxx.xxx.xxx.xxx
SSID: XXX
BSSID of AP: xx:xx:xx:xx:xx:xx
ESP32 MAC address: xx:xx:xx:xx:xx:xx

------------ NETWORK SETTINGS ------------
Nuki Hub hostname: ESP-Nuki-Hub-Door-A
DHCP enabled: Yes
RSSI Publish interval (s): Disabled
Find WiFi AP with strongest signal: Yes
Restart ESP32 on network disconnect enabled: Yes
MQTT Timeout until restart (s): Disabled

------------ MQTT ------------
MQTT connected: Yes
MQTT broker address: xxx.xxx.xxx.xxx
MQTT broker port: 1883
MQTT username: ***
MQTT password: ***
MQTT base topic: nuki_door_a
MQTT SSL: Disabled

------------ BLUETOOTH ------------
Bluetooth TX power (dB): 9
Bluetooth command nr of retries: 5
Bluetooth command retry delay (ms): 100
Seconds until reboot when no BLE beacons received: 60

------------ QUERY / PUBLISH SETTINGS ------------
Lock/Opener state query interval (s): 3600
Publish Nuki device authorization log: Yes
Max authorization log entries to retrieve: 5
Battery state query interval (s): 7200
Most non-JSON MQTT topics disabled: No
Publish Nuki device config: Yes
Config query interval (s): 3600
Publish Keypad info: No
Keypad query interval (s): 1800
Enable Keypad control: Yes
Publish Keypad topic per entry: No
Publish Keypad codes: No
Allow checking Keypad codes: No
Max keypad entries to retrieve: 10
Publish timecontrol info: No
Keypad query interval (s): 1800
Enable timecontrol control: No
Publish timecontrol topic per entry: No
Max timecontrol entries to retrieve: 10

------------ HOME ASSISTANT ------------
Home Assistant auto discovery enabled: Yes
Home Assistant auto discovery topic: homeassistant/
Nuki Hub configuration URL for HA: http://xxx.xxx.xxx.xxx
Nuki Hub ID: 176852922780104

------------ NUKI LOCK ------------
Lock enabled: Yes
Paired: Yes
Nuki Hub device ID: 2782726096
Nuki device ID: ***
Firmware version: 3.10.7
Hardware version: 5.11
Valid PIN set: Yes
Has door sensor: No
Has keypad: Yes
Keypad highest entries count: 0
Timecontrol highest entries count: 0
Register as: Bridge

------------ HYBRID MODE ------------
Hybrid mode enabled: No

------------ NUKI LOCK ACL ------------
Lock: Allowed
Unlock: Allowed
Unlatch: Allowed
Lock N Go: Allowed
Lock N Go Unlatch: Allowed
Full Lock: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed

------------ NUKI LOCK CONFIG ACL ------------
Name: Disallowed
Latitude: Disallowed
Longitude: Disallowed
Auto Unlatch: Disallowed
Pairing enabled: Allowed
Button enabled: Allowed
LED flash enabled: Allowed
LED brightness: Allowed
Timezone offset: Disallowed
DST mode: Disallowed
Fob Action 1: Disallowed
Fob Action 2: Disallowed
Fob Action 3: Disallowed
Single Lock: Allowed
Advertising Mode: Disallowed
Timezone ID: Disallowed
Unlocked Position Offset Degrees: Disallowed
Locked Position Offset Degrees: Disallowed
Single Locked Position Offset Degrees: Disallowed
Unlocked To Locked Transition Offset Degrees: Disallowed
Lock n Go timeout: Allowed
Single button press action: Allowed
Double button press action: Allowed
Detached cylinder: Disallowed
Battery type: Disallowed
Automatic battery type detection: Disallowed
Unlatch duration: Allowed
Auto lock timeout: Disallowed
Auto unlock disabled: Disallowed
Nightmode enabled: Disallowed
Nightmode start time: Disallowed
Nightmode end time: Disallowed
Nightmode auto lock enabled: Disallowed
Nightmode auto unlock disabled: Disallowed
Nightmode immediate lock on start: Disallowed
Auto lock enabled: Disallowed
Immediate auto lock enabled: Disallowed
Auto update enabled: Disallowed
Reboot Nuki: Allowed

------------ NUKI OPENER ------------
Opener enabled: No

------------ GPIO ------------
```

### TO REPRODUCE
Update NH to v9.07 and access the web interface. Try using various browsers and experience at least a 2nd, for Firefox a constant never-ending authentication.

### EXPECTED BEHAVIOUR
Login once.

### SCREENSHOTS
Firefox
[image]

### ADDITIONAL CONTEXT
/

**(Please, remember to close the issue when the problem has been addressed)**

iranl, 2025-01-15:
For NH 9.07 we moved to a new version of the HTTP server library that we use. This probably plays a part in this.

NH by default uses HTTP Basic authentication. 
The client (browser) decides how this is handled. 
Apparently Firefox doesn't handle this the same way as browsers based on the Chromium engine.

You can try changing to Digest authentication (on the credentials page) and see if this works better for you on Firefox.

I'm also looking into implementing form based session/cookie authentication as a better alternative that will also play nice with password managers (extensions).

As the market share of Firefox is a couple of percent I won't be investing time into improving basic auth for Firefox.

bcutter, 2025-01-15:
Regarding Firefox: that's a slap in the face of all Firefox users.

More important: Firefox is the browser which handles the change the worse (basically pushing a new authentication request every 1 to 2 seconds), making NH web interface completely unusable. BUT all the other browsers trigger authentication again too.

So it's a basic problem, a bug - which requires a fix. Just want to make sure we're on the same page.

---

Edit: enabled "Use Digest Authentication (more secure)" and everything behaves like before (NH until v9.06), no additional/constant authentication popup. Only did a quick test in Firefox and on EDGE. Chrome and Safari to be tested/confirmed.

mundschenk-at, 2025-01-15:
> Regarding Firefox: that's a slap in the face of all Firefox users.

So I gather you just volunteered as the QA tester for future releases?

mundschenk-at, 2025-01-15:
The idea would be to test them before the release (i.e. the betas or nightlies). 

In general, I think your feedback would be appreciated more if you were less confrontational about things when developer priorities are different from your own.

bcutter, 2025-01-15:
I don't see anything confrontational here.
How many people are testing the betas or even nightlys currently before a new version gets released?

iranl, 2025-01-17:
Closing with #612 which adds HTTP form authentication as an option.
Use Digest or Form for Firefox.

PR #612 (2025-01-17): HTTP Form auth
## Description:

**Related issue (if applicable):** Closes #607 

## Checklist:
  - [x] The pull request is done against the latest master branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works
  - [x] I accept the [CLA](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md#contributor-license-agreement-cla).
closed 2025-01-17 by iranl