technyon/nuki_hub #608: Increased connection issues since NH v9.07

bcutter, 2025-01-13:
### PROBLEM DESCRIPTION
Since NH v9.07 one NH takes significant more time to connect to the lock, perform actions and update status (lock --> NH --> MQTT --> HA). That particular setup always had a delay e. g. when locking/unlocking the door it took ~ 10 to 15 seconds for the status in HA to update (locked/unlocked) which was fairly acceptable.
- Since NH v9.07 this takes at least x2, sometimes almost x4 the time as before (up to a minute ⚠️).
- Also few lock/unlock commands sent via HA are not operated (received and performed by the lock?) at all.
- At a glance: connection is not that reliable as it was with NH v9.06 before.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [X] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [X] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [ ] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [X] Device used (e.g., ESP32-S3): LOLIN32-Lite
- [X] Nuki Hub binary firmware version number used: 12722791725.9.1
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Provide the output of http://nukihubIP/info:
```
System Information
------------ NUKI HUB ------------
Device: ESP32
Version: 9.07
Build: 12722791725.9.1
Build type: Release
Build date: 2025-01-11
Updater version: 9.07
Updater build: 12722791725.9.1
Updater build date: 2025-01-10
Uptime (min): 821
Config version: 907
Last restart reason FW: OTACompleted
Last restart reason ESP: ESP_RST_SW: Software reset via esp_restart.
Free internal heap: 73436
Total internal heap: 252700
PSRAM Available: No
Network task stack high watermark: 8320
Nuki task stack high watermark: 5124

------------ GENERAL SETTINGS ------------
Network task stack size: 12288
Nuki task stack size: 8192
Check for updates: Yes
Latest version: 9.07
Allow update from MQTT: No
Update NukiHub and Nuki devices time using NTP: No
Web configurator username: ***
Web configurator password: ***
Web configurator enabled: Yes
HTTP SSL: Disabled
Publish debug information enabled: No
MQTT log enabled: No
Webserial enabled: No
Bootloop protection enabled: No

------------ NETWORK ------------
Network device: Built-in Wi-Fi
Network connected: Yes
IP Address: xxx.xxx.xxx.xxx
SSID: XXX
BSSID of AP: xx:xx:xx:xx:xx:xx
ESP32 MAC address: xx:xx:xx:xx:xx:xx

------------ NETWORK SETTINGS ------------
Nuki Hub hostname: ESP-Nuki-Hub-Door-B
DHCP enabled: Yes
RSSI Publish interval (s): Disabled
Find WiFi AP with strongest signal: Yes
Restart ESP32 on network disconnect enabled: Yes
MQTT Timeout until restart (s): Disabled

------------ MQTT ------------
MQTT connected: Yes
MQTT broker address: xxx.xxx.xxx.xxx
MQTT broker port: 1883
MQTT username: ***
MQTT password: ***
MQTT base topic: nuki_door_b
MQTT SSL: Disabled

------------ BLUETOOTH ------------
Bluetooth TX power (dB): 9
Bluetooth command nr of retries: 5
Bluetooth command retry delay (ms): 100
Seconds until reboot when no BLE beacons received: 60

------------ QUERY / PUBLISH SETTINGS ------------
Lock/Opener state query interval (s): 3600
Publish Nuki device authorization log: Yes
Max authorization log entries to retrieve: 5
Battery state query interval (s): 7200
Most non-JSON MQTT topics disabled: No
Publish Nuki device config: Yes
Config query interval (s): 3600
Publish Keypad info: No
Keypad query interval (s): 1800
Enable Keypad control: Yes
Publish Keypad topic per entry: No
Publish Keypad codes: No
Allow checking Keypad codes: No
Max keypad entries to retrieve: 10
Publish timecontrol info: No
Keypad query interval (s): 1800
Enable timecontrol control: No
Publish timecontrol topic per entry: No
Max timecontrol entries to retrieve: 10

------------ HOME ASSISTANT ------------
Home Assistant auto discovery enabled: Yes
Home Assistant auto discovery topic: homeassistant/
Nuki Hub configuration URL for HA: http://xxx.xxx.xxx.xxx
Nuki Hub ID: 247297435473132

------------ NUKI LOCK ------------
Lock enabled: Yes
Paired: Yes
Nuki Hub device ID: 3247846687
Nuki device ID: ***
Firmware version: 
Hardware version: 
Valid PIN set: Yes
Has door sensor: No
Has keypad: No
Timecontrol highest entries count: 0
Register as: Bridge

------------ HYBRID MODE ------------
Hybrid mode enabled: No

------------ NUKI LOCK ACL ------------
Lock: Allowed
Unlock: Allowed
Unlatch: Allowed
Lock N Go: Allowed
Lock N Go Unlatch: Allowed
Full Lock: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed

------------ NUKI LOCK CONFIG ACL ------------
Name: Disallowed
Latitude: Disallowed
Longitude: Disallowed
Auto Unlatch: Disallowed
Pairing enabled: Allowed
Button enabled: Allowed
LED flash enabled: Allowed
LED brightness: Allowed
Timezone offset: Disallowed
DST mode: Disallowed
Fob Action 1: Disallowed
Fob Action 2: Disallowed
Fob Action 3: Disallowed
Single Lock: Allowed
Advertising Mode: Disallowed
Timezone ID: Disallowed
Unlocked Position Offset Degrees: Disallowed
Locked Position Offset Degrees: Disallowed
Single Locked Position Offset Degrees: Disallowed
Unlocked To Locked Transition Offset Degrees: Disallowed
Lock n Go timeout: Allowed
Single button press action: Allowed
Double button press action: Allowed
Detached cylinder: Disallowed
Battery type: Disallowed
Automatic battery type detection: Disallowed
Unlatch duration: Allowed
Auto lock timeout: Disallowed
Auto unlock disabled: Disallowed
Nightmode enabled: Disallowed
Nightmode start time: Disallowed
Nightmode end time: Disallowed
Nightmode auto lock enabled: Disallowed
Nightmode auto unlock disabled: Disallowed
Nightmode immediate lock on start: Disallowed
Auto lock enabled: Disallowed
Immediate auto lock enabled: Disallowed
Auto update enabled: Disallowed
Reboot Nuki: Allowed

------------ NUKI OPENER ------------
Opener enabled: No

------------ GPIO ------------
```

### TO REPRODUCE
Update NH to v9.07.

### EXPECTED BEHAVIOUR
Connection is reliable in terms of "sent commands are received and operated", status update from lock to NH to MQTT to HA is roughly taking as long as with NH 9.06.

### SCREENSHOTS
yellow = initiated lock/unlock progress
green/red = finished lock/unlock progress (updated status)
read from bottom to top
[image]
[image]
[image]

### ADDITIONAL CONTEXT
/

**(Please, remember to close the issue when the problem has been addressed)**

technyon, 2025-01-13:
For me it's pretty mich instant. If it takes longer, it sounds like there's a retry or two necessary to send the command. As always, serial logs would be helpful to see what's going on.

sjors-lemniscap, 2025-01-15:
> As mentioned that behaviour for the second lock/NH always was like this - and it is mostly fine, while not perfect (the other lock/NH works really instantly). The point is something has changed in NH 9.07 which leads to a dramatic increase of these likely communication based status update delay.
> 
> I was searching https://github.com/technyon/nuki_hub?tab=readme-ov-file for "how to provide serial/MQTT logs". Can you point me in the right direction please.

Serial / MQTT logs can now be published through MQTT and directly in Home Assistant. You can navigate to `http://IP/get?page=advanced` and enable (most likely) both `Nuki connect debug logging` and `Nuki communication debug logging`. After a reboot you should be able to see advanced logs in the Logbook of the NukiHub ESP32 in Home Assistant.

This is a advanced feature and might break stuff so please only use this in a development environment where you've backed-up your NukiHub config and are able to re-flash the ESP32 if needed.

bcutter, 2025-01-15:
> Serial / MQTT logs can now be published through MQTT and directly in Home Assistant. You can navigate to `http://IP/get?page=advanced` and enable (most likely) both `Nuki connect debug logging` and [...]

Oh that sounds easy.

> This is a advanced feature and might break stuff so please only use this in a development environment where you've backed-up your NukiHub config and are able to re-flash the ESP32 if needed.

Okay that's a no-go.

technyon, 2025-01-15:
First of all, navigate to "MQTT configuration" and check "Enable MQTT logging". This should provide basic logs so we can confirm wether or no NH needs multiple retries. From all the advanced features, the "debug logging" is not that critical. Make sure especially not to change the "Task size" and the "Char buffer size" options and keep bootlop prevention enabled.

iranl, 2025-01-17:
Enabling Nuki connect debug logging or Nuki communication debug logging poses no risks.

Logging can be retrieved using a program like hterm or screen over (USB) serial or using MQTT by enabling the setting on the MQTT configuration page of NukiHub.

Please provide the requested information, otherwise no follow up can and will be done

iranl, 2025-01-23:
Closing due to inactivity. Please reopen after providing the requested information

closed 2025-01-23 by iranl