technyon/nuki_hub #617: NH v9.08: topic "last_action_authorization" removed

bcutter, 2025-01-27:
### PROBLEM DESCRIPTION
Updating from NH 9.07 to 9.08 makes the HA entity `sensor.lock_last_action_authorization` being not available anymore. Compared to another NH which was still running 9.07.

[image]

The auto discovery topic formerly looked like this:
```
{"dev":{"ids":["nuki_XXXXXXXX"]},"~":"nuki_door_a/lock","name":"Last action authorization","unique_id":"XXXXXXXX_last_action_authorization","stat_t":"~/log","ent_cat":"diagnostic","avty":{"t":"nuki_door_a/maintenance/mqttConnectionState"},"ic":"mdi:format-list-bulleted","val_tpl":"{{ (value_json|selectattr('type', 'eq', 'LockAction')|selectattr('action', 'in', ['Lock', 'Unlock', 'Unlatch'])|first|default).authorizationName|default }}"}
```

Updated the 2nd NH now too to 9.08 - same behaviour, MQTT topic and HA entity is gone too.

Pardon me if I get a bit lazy with provided details. Every update has a new issue so yeah :-)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [ ] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [ ] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [ ] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [ ] Device used (e.g., ESP32-S3): _____
- [ ] Nuki Hub binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Provide the output of http://nukihubIP/get?page=info:
```

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

iranl, 2025-01-27:
Not an issue.
Removed by design.
Outlined in the release notes.

bcutter, 2025-01-27:
Oh indeed, I must have skipped that line (as it started with something not interesting to me/my setup). Ah well, keypad status is gone too. Everything as outlined in the release notes.

[image]

While I can follow https://github.com/technyon/nuki_hub/pull/614 and see that it was done by design, may I asked **_why_** they have been removed?

Thx!