technyon/nuki_hub #618: New firmware finally upload but in HA MQTT I have most of the settings but no toggle button for lock/unlock

bertybassett, 2025-01-27:
### PROBLEM DESCRIPTION
finally flashed the new firm 9.09-master1 to the w32-eth01 and everything is OK however HomeAssistant is missing a lock button like the old firmware, instead I have just 3 entries

Lock 'n' Go
Lock 'n' Go with unlatch
Open

Where is the 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ X] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [ X] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [ X] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [ X] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [ X] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [ X] Device used (e.g., ESP32-S3): w32-eth01
- [ X] Nuki Hub binary firmware version number used: 9.09-master1
  - [ X] Pre-compiled
  - [ ] Self-compiled
- [ ] Provide the output of http://nukihubIP/get?page=info:
```
<h3 style="caret-color: rgb(0, 0, 0); color: var(--nc-tx-1); font-style: normal; font-variant-caps: normal; letter-spacing: normal; orphans: auto; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: auto; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration: none; margin: 0px 0px 8px; padding: 0.875rem 0px 2px; line-height: 1; border-bottom: 1px solid var(--nc-bg-2); font-size: 1.55rem; font-family: Inter, -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, Roboto, Oxygen, Ubuntu, Cantarell, &quot;Open Sans&quot;, &quot;Helvetica Neue&quot;, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;, &quot;Segoe UI Symbol&quot;; background-color: rgb(0, 0, 0);">Info</h3><br style="font-style: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: auto; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: auto; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration: none; margin: 0px; padding: 0px; caret-color: rgb(238, 238, 238); color: rgb(238, 238, 238); font-family: Inter, -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, Roboto, Oxygen, Ubuntu, Cantarell, &quot;Open Sans&quot;, &quot;Helvetica Neue&quot;, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;, &quot;Segoe UI Symbol&quot;; font-size: 16.48px; background-color: rgb(0, 0, 0);">
Hostname | nukihub
-- | --
MQTT Connected | Yes
Nuki Lock paired | Yes (BLE Address 54:d2:72:30:fc:95)
Nuki Lock state | unlocked
Nuki Lock PIN status | PIN valid
Firmware | 9.09-master1
Latest Firmware | 9.09-master1

Info

Hostname	nukihub
MQTT Connected	Yes
Nuki Lock paired	Yes (BLE Address 54:d2:72:30:fc:95)
Nuki Lock state	unlocked
Nuki Lock PIN status	PIN valid
Firmware	[9.09-master1](http://192.168.0.15/get?page=info)
Latest Firmware	[9.09-master1](http://192.168.0.15/get?page=ota)
```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
a lock toggle button within MQTT in home assistant that I can press or at lease an open button and a lock button (got open)

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

iranl, 2025-01-28:
Not sure where you updated from but Home Assistant had multiple changes in the last few versions.

This has caused duplicate devices in Home Assistant in many cases. Make sure that there you are looking at the "right/latest" device in Home Assistant. There should be a device for Nuki Hub and for each Nuki device (lock and/or opener) you have paired to Nuki Hub.

Also you haven't posted your info page but the main page, this is not what we ask for.
Please provide the correct information. You can access the info page by clicking on the firmware version on the main page or by entering the correct webaddress as outlined in the issue template to access it directly

bertybassett, 2025-01-28:
and there is no lock button in any of my HA MQTT screens as you can see from my screenshots

[image]
[image]
[image]
[image]

iranl, 2025-01-28:
What version of Home Assistant are you running?

bertybassett, 2025-01-29:
HA is version 2025.1.4!

[image]

iranl, 2025-01-29:
Please post the content of the MQTT topic `homeassistant/lock/NUKI_ID/smartlock/config` and `homeassistant/switch/15_DIGIT_NUMERIC_RANDOM_ID/reset/config` (check using MQTT explorer or similar)

And check your HA logbooks for errors related to Nuki Hub MQTT discovery

bertybassett, 2025-01-29:
There was no topic for Home assistant

mundschenk-at, 2025-01-29:
You probably did not subscribe to it when connecting in the MQTT Explorer frontend.

bertybassett, 2025-01-29:
I subscripted to #.  Is that not the right thing to do?

[image]

bertybassett, 2025-01-29:
Enable Home Assistant auto discovery	was not ticked  in NukieBridge

Fixed that, now grabbing the screenshot

bertybassett, 2025-01-29:
homeassistant/switch/101307317986208/reset/config

{"dev":{"ids":["nuki_101307317986208"],"mf":"Technyon","mdl":"Nuki Hub","name":"nukihub","sw":"9.09-master1","hw":"ESP32","cu":"192.168.0.22:8123"},"~":"nukihub","name":"Restart Nuki Hub","unique_id":"101307317986208_reset","avty":[{"t":"~/maintenance/mqttConnectionState"}],"opt":"false","stat_t":"~/maintenance/reset","ent_cat":"diagnostic","cmd_t":"~/maintenance/reset","ic":"mdi:restart","pl_on":"1","pl_off":"0","stat_on":"1","stat_off":"0"}

mundschenk-at, 2025-01-29:
It is, but I have found that it does not always work in MQTT explorer. if there was nothing under `homeassistant`, no entities would show up in HA for MQTT devices at all (unless you configured them manually in YAML).

bertybassett, 2025-01-29:
exactly I have no other MQTT devices just the nukiESP32 bridge

bertybassett, 2025-01-29:
[image]

See the picture I now have a topic in MQTT

mundschenk-at, 2025-01-29:
There should also be `homeassistant/status` published by HA, but maybe it is possible to disable that when setting up the MQTT integration (I don't want to check).

Anyway, look for topics under `homeassistanr/lock`

bertybassett, 2025-01-29:
I can see a lock topic and a button topic (with lock and Go, etc) but no close or actual toggle lock button

[image]

bertybassett, 2025-01-29:
toggling these on/off

[image]

Made a open button appear in HA

but in the old firmware before the hard flash via the USB cable I had a single toggle button for the lock.  Is that still possible?

bertybassett, 2025-01-29:
[image]

Open has appeared but was disabled in HA.

bertybassett, 2025-01-29:
[image]

Toggling the Lock N Go option makes and removes the option in HA MQTT devices NukiLock

Toggling lock  or unlock or full unlock does nothing in HA.

Let me know what to check next

bertybassett, 2025-01-30:
[image]

iranl, 2025-01-30:
So your issue was first that you didn't enable Home Assistant discovery in Nuki Hub.
Any device you saw in Home Assistant before that time was a leftover of an earlier version of Nuki Hub.

Your MQTT topics shows Nuki Hub has correctly created all topics. The lock device / button is under the homeassistant/lock/1830fc95/smartlock/config topic and the Nuki Hub device is primarily under the homeassistant/switch/101307317986208/reset/config topic.

If there are still issues this is on the side of your Home Assistant instance and not on Nuki Hub's side.

You should look for a device named "nukihub" and a device named "Frontdoor" in your list of MQTT devices in Home Assistant.

bertybassett, 2025-01-30:
where from MQTT get the name frontdoor from?

bertybassett, 2025-01-30:
fixed it by factory resetting the nuki ESP32 hub setting.  Most likely something I click or did in the setting.

bertybassett, 2025-01-31:
figured out my issue

Old firmware device was called nuki
new firmware device is called frontdoor

HA already had a device called frontdoor.

closed 2025-01-31 by bertybassett