technyon/nuki_hub #619: BLE issues with v9.08

leo-b, 2025-01-27:
### PROBLEM DESCRIPTION

I upgraded to from 9.07 to 9.08 yesterday. Since then I am noticing much more frequent watchdog reboots.

The first watchdog reboot after the update happened after 18 hours but after that, the restarts occured more often:

[image]

Besides my ping monitor reports regular packet losses (every 5 to 10 minutes), I suppose because of hanging tasks?

### REQUESTED INFORMATION

[ble-issues.log](https://github.com/user-attachments/files/18564335/ble-issues.log)

- [x] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [x] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [x] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [x] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [x] Device used (e.g., ESP32-S3): ESP32
- [x] Nuki Hub binary firmware version number used: 9.08
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Provide the output of http://nukihubIP/get?page=info:
```
System Information

------------ NUKI HUB ------------
Device: ESP32
Version: 9.08
Build: 12974430884.10.1
Build type: Release
Build date: 2025-01-26
Updater version: 9.08
Updater build: 12974430884.10.1
Updater build date: 2025-01-23
Uptime (min): 35
Config version: 908
Last restart reason FW: BLEBeaconWatchdog
Last restart reason ESP: ESP_RST_SW: Software reset via esp_restart.
Free internal heap: 66056
Total internal heap: 246356
PSRAM Available: No
Network task stack high watermark: 8472
Nuki task stack high watermark: 5020

------------ GENERAL SETTINGS ------------
Network task stack size: 12288
Nuki task stack size: 8192
Check for updates: No
Latest version: 9.08
Allow update from MQTT: No
Update Nuki Hub and Nuki devices time using NTP: No
Web configurator username: Not set
Web configurator password: Not set
Web configurator bypass for proxy IP: Not set
Web configurator authentication: Basic
Session validity (in seconds): 3600
Session validity remember (in hours): 720
Duo Push MFA enabled: No
Web configurator enabled: Yes
HTTP SSL: Disabled
Advanced menu enabled: No
Publish free heap over MQTT: No
Nuki connect debug logging enabled: Yes
Nuki communication debug logging enabled: Yes
Nuki readable data debug logging enabled: Yes
Nuki hex data debug logging enabled: Yes
Nuki command debug logging enabled: Yes
MQTT log enabled: Yes
Webserial enabled: No
Bootloop protection enabled: Yes

------------ NETWORK ------------
Network device: Built-in Wi-Fi
Network connected: Yes
IP Address: 10.0.1.92
SSID: ms14-iot
BSSID of AP: B2:B9:8A:60:14:91
ESP32 MAC address: C8:F0:9E:4E:7B:2C

------------ NETWORK SETTINGS ------------
Nuki Hub hostname: nuki-garten
DHCP enabled: No
Static IP address: 10.0.1.92
Static IP subnet: 255.255.255.0
Static IP gateway: 10.0.1.1
Static IP DNS server: 10.0.1.1
RSSI Publish interval (s): 10000
Find WiFi AP with strongest signal: Yes
Restart ESP32 on network disconnect enabled: No
Disable Network if not connected within 60s: No
MQTT Timeout until restart (s): 120

------------ MQTT ------------
MQTT connected: Yes
MQTT broker address: 10.0.1.1
MQTT broker port: 1883
MQTT username: ***
MQTT password: ***
MQTT base topic: nuki-garten
MQTT SSL: Disabled

------------ BLUETOOTH ------------
Bluetooth connection mode: New
Bluetooth TX power (dB): 9
Bluetooth command nr of retries: 3
Bluetooth command retry delay (ms): 200
Seconds until reboot when no BLE beacons received: 120

------------ QUERY / PUBLISH SETTINGS ------------
Lock/Opener state query interval (s): 1800
Publish Nuki device authorization log: No
Max authorization log entries to retrieve: 5
Battery state query interval (s): 1800
Most non-JSON MQTT topics disabled: Yes
Publish Nuki device config: Yes
Config query interval (s): 3600
Publish Keypad info: No
Keypad query interval (s): 1800
Enable Keypad control: Yes
Publish Keypad topic per entry: No
Publish Keypad codes: No
Allow checking Keypad codes: No
Max keypad entries to retrieve: 10
Publish timecontrol info: No
Keypad query interval (s): 1800
Enable timecontrol control: No
Publish timecontrol topic per entry: No
Max timecontrol entries to retrieve: 10
Enable authorization control: No
Publish authorization topic per entry: No
Max authorization entries to retrieve: 10

------------ HOME ASSISTANT ------------
Home Assistant auto discovery enabled: Yes
Home Assistant auto discovery topic: homeassistant/
Nuki Hub configuration URL for HA: http://10.0.1.92
Nuki Hub ID: 48908111638728

------------ NUKI LOCK ------------
Lock enabled: Yes
Lock Ultra enabled: No
Paired: Yes
Nuki Hub device ID: 4224542623
Nuki device ID: ***
Firmware version: 2.15.3
Hardware version: 11.2
Valid PIN set: Yes
Has door sensor: No
Has keypad: No
Timecontrol highest entries count: 0
Authorizations highest entries count: 0
Register as: Bridge

------------ HYBRID MODE ------------
Hybrid mode enabled: No
Force Lock ID: No
Force Lock Keypad: No
Force Lock Doorsensor: No

------------ NUKI LOCK ACL ------------
Lock: Allowed
Unlock: Allowed
Unlatch: Allowed
Lock N Go: Allowed
Lock N Go Unlatch: Allowed
Full Lock: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed

------------ NUKI LOCK CONFIG ACL ------------
Name: Allowed
Latitude: Allowed
Longitude: Allowed
Auto Unlatch: Allowed
Pairing enabled: Allowed
Button enabled: Allowed
LED flash enabled: Allowed
LED brightness: Allowed
Timezone offset: Allowed
DST mode: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed
Single Lock: Allowed
Advertising Mode: Allowed
Timezone ID: Allowed
Unlocked Position Offset Degrees: Allowed
Locked Position Offset Degrees: Allowed
Single Locked Position Offset Degrees: Allowed
Unlocked To Locked Transition Offset Degrees: Allowed
Lock n Go timeout: Allowed
Single button press action: Allowed
Double button press action: Allowed
Detached cylinder: Allowed
Battery type: Allowed
Automatic battery type detection: Allowed
Unlatch duration: Allowed
Auto lock timeout: Allowed
Auto unlock disabled: Allowed
Nightmode enabled: Allowed
Nightmode start time: Allowed
Nightmode end time: Allowed
Nightmode auto lock enabled: Allowed
Nightmode auto unlock disabled: Allowed
Nightmode immediate lock on start: Allowed
Auto lock enabled: Allowed
Immediate auto lock enabled: Allowed
Auto update enabled: Allowed
Reboot Nuki: Allowed
Motor speed: Allowed
Enable slow speed during nightmode: Allowed

------------ NUKI OPENER ------------
Opener enabled: No

------------ GPIO ------------

Retain Input GPIO MQTT state: No
```

### TO REPRODUCE
No activity needed. It just reboots.

### EXPECTED BEHAVIOUR
No watchdog restart.

**(Please, remember to close the issue when the problem has been addressed)**

leo-b, 2025-01-27:
[ble-issues.log](https://github.com/user-attachments/files/18564345/ble-issues.log)

iranl, 2025-01-28:
Why do you have Nuki debug logging enabled?

This is almost guaranteed to give issues on its own on an ESP32 without PSRAM, especially while also having MQTT logging enabled.
This will in its own cause unnecessary stress on the BLE and MQTT client that is not meant for production devices.
These options are only for development and advanced debugging and only than when asked to enable these settings by a developer.

Please disable all debug logging and observe if you have issues afterwards.
If so provide your info page with these settings disabled and information about what watchdogs triggers and other restartreasons you observe.

leo-b, 2025-01-28:
> Why do you have Nuki debug logging enabled?

Because you told me to do so in order to find out the reason for the daily watchdog restarts:
https://github.com/technyon/nuki_hub/issues/547#issuecomment-2571684031

The regular watchdogs resets are still present, that's why I am still reporting the issues in #547.

> This is almost guaranteed to give issues on its own on an ESP32 without PSRAM, especially while also having MQTT logging enabled.

Debug logging via syslog (UDP) would be a nice feature and much more convenient than MQTT logging.

iranl, 2025-02-05:
Closing due to inactivity and seems complete. Please reopen if the issue reappears.

closed 2025-02-05 by iranl