technyon/nuki_hub #623: opener doorbellSuppression/noValidPinSet

tca22, 2025-02-02:
### PROBLEM DESCRIPTION
I would like to change the doorbellSuppression from "Off" to "Ring" by sending a MQTT instruction like
/opener/configuration/action {"doorbellSuppression":"Ring"}
The result is 
/opener/configuration/commandResult {"general":"noValidPinSet"}
However, the PIN for the opener is set in the Credentials section, but appears to be not stored or saved. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [X] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [X] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [X] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [X] Device used (e.g., ESP32-S3): Mini ESP32
- [X] Nuki Hub binary firmware version number used: 9.08
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Provide the output of http://nukihubIP/get?page=info:
```
System Information
------------ NUKI HUB ------------
Device: ESP32
Version: 9.08
Build: 12974430884.10.1
Build type: Release
Build date: 2025-01-26
Updater version: 9.08
Updater build: 12974430884.10.1
Updater build date: 2025-01-23
Uptime (min): 13
Config version: 908
Last restart reason FW: RequestedViaWebServer
Last restart reason ESP: ESP_RST_SW: Software reset via esp_restart.
Free internal heap: 58308
Total internal heap: 246356
PSRAM Available: No
Network task stack high watermark: 6784
Nuki task stack high watermark: 5140

------------ GENERAL SETTINGS ------------
Network task stack size: 12288
Nuki task stack size: 8192
Check for updates: Yes
Latest version: 9.08
Allow update from MQTT: Yes
Update Nuki Hub and Nuki devices time using NTP: Yes
Web configurator username: Not set
Web configurator password: Not set
Web configurator bypass for proxy IP: Not set
Web configurator authentication: Basic
Session validity (in seconds): 3600
Session validity remember (in hours): 720
Duo Push MFA enabled: No
Web configurator enabled: Yes
HTTP SSL: Disabled
Advanced menu enabled: No
Publish free heap over MQTT: No
Nuki connect debug logging enabled: No
Nuki communication debug logging enabled: No
Nuki readable data debug logging enabled: No
Nuki hex data debug logging enabled: No
Nuki command debug logging enabled: No
MQTT log enabled: No
Webserial enabled: No
Bootloop protection enabled: No

------------ NETWORK ------------
Network device: Built-in Wi-Fi
Network connected: Yes
IP Address: 192.168.0.29
SSID: office-24
BSSID of AP: E2:B3:70:17:AD:CA
ESP32 MAC address: 30:C9:22:32:18:78

------------ NETWORK SETTINGS ------------
Nuki Hub hostname: nukihub
DHCP enabled: Yes
RSSI Publish interval (s): 60
Find WiFi AP with strongest signal: Yes
Restart ESP32 on network disconnect enabled: No
Disable Network if not connected within 60s: No
MQTT Timeout until restart (s): 60

------------ MQTT ------------
MQTT connected: Yes
MQTT broker address: 192.168.0.2
MQTT broker port: 1883
MQTT username: Not set
MQTT password: Not set
MQTT base topic: nukihub
MQTT SSL: Disabled

------------ BLUETOOTH ------------
Bluetooth connection mode: New
Bluetooth TX power (dB): 9
Bluetooth command nr of retries: 3
Bluetooth command retry delay (ms): 100
Seconds until reboot when no BLE beacons received: 60

------------ QUERY / PUBLISH SETTINGS ------------
Lock/Opener state query interval (s): 1800
Publish Nuki device authorization log: Yes
Max authorization log entries to retrieve: 5
Battery state query interval (s): 1800
Most non-JSON MQTT topics disabled: No
Publish Nuki device config: Yes
Config query interval (s): 3600
Publish Keypad info: No
Keypad query interval (s): 1800
Enable Keypad control: Yes
Publish Keypad topic per entry: No
Publish Keypad codes: No
Allow checking Keypad codes: No
Max keypad entries to retrieve: 10
Publish timecontrol info: Yes
Keypad query interval (s): 1800
Enable timecontrol control: Yes
Publish timecontrol topic per entry: No
Max timecontrol entries to retrieve: 10
Enable authorization control: Yes
Publish authorization topic per entry: No
Max authorization entries to retrieve: 5

------------ HOME ASSISTANT ------------
Home Assistant auto discovery enabled: No

------------ NUKI LOCK ------------
Lock enabled: Yes
Lock Ultra enabled: No
Paired: Yes
Nuki Hub device ID: 2793350671
Nuki device ID: ***
Firmware version: 3.10.7
Hardware version: 6.11
Valid PIN set: No
Has door sensor: No
Has keypad: No
Timecontrol highest entries count: 0
Authorizations highest entries count: 0
Register as: App

------------ HYBRID MODE ------------
Hybrid mode enabled: No
Force Lock ID: No
Force Lock Keypad: No
Force Lock Doorsensor: No

------------ NUKI LOCK ACL ------------
Lock: Allowed
Unlock: Allowed
Unlatch: Allowed
Lock N Go: Allowed
Lock N Go Unlatch: Allowed
Full Lock: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed

------------ NUKI LOCK CONFIG ACL ------------
Name: Allowed
Latitude: Allowed
Longitude: Allowed
Auto Unlatch: Allowed
Pairing enabled: Allowed
Button enabled: Allowed
LED flash enabled: Allowed
LED brightness: Allowed
Timezone offset: Allowed
DST mode: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed
Single Lock: Allowed
Advertising Mode: Allowed
Timezone ID: Allowed
Unlocked Position Offset Degrees: Allowed
Locked Position Offset Degrees: Allowed
Single Locked Position Offset Degrees: Allowed
Unlocked To Locked Transition Offset Degrees: Allowed
Lock n Go timeout: Allowed
Single button press action: Allowed
Double button press action: Allowed
Detached cylinder: Allowed
Battery type: Allowed
Automatic battery type detection: Allowed
Unlatch duration: Allowed
Auto lock timeout: Allowed
Auto unlock disabled: Allowed
Nightmode enabled: Allowed
Nightmode start time: Allowed
Nightmode end time: Allowed
Nightmode auto lock enabled: Allowed
Nightmode auto unlock disabled: Allowed
Nightmode immediate lock on start: Allowed
Auto lock enabled: Allowed
Immediate auto lock enabled: Allowed
Auto update enabled: Allowed
Reboot Nuki: Allowed
Motor speed: Allowed
Enable slow speed during nightmode: Allowed

------------ NUKI OPENER ------------
Opener enabled: Yes
Paired: Yes
Nuki Hub device ID: 2793350671
Nuki device ID: ***
Firmware version: 1.11.0
Hardware version: 4.17
Opener valid PIN set: No
Opener has keypad: No
Timecontrol highest entries count: 0
Authorizations highest entries count: 0
Register as: Bridge
Nuki Opener Lock/Unlock action set to Continuous mode in Home Assistant: No
Force Opener ID: No
Force Opener Keypad: No

------------ NUKI OPENER ACL ------------
Activate Ring-to-Open: Allowed
Deactivate Ring-to-Open: Allowed
Electric Strike Actuation: Allowed
Activate Continuous Mode: Allowed
Deactivate Continuous Mode: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed

------------ NUKI OPENER CONFIG ACL ------------
Name: Allowed
Latitude: Allowed
Longitude: Allowed
Pairing enabled: Allowed
Button enabled: Allowed
LED flash enabled: Allowed
Timezone offset: Allowed
DST mode: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed
Operating Mode: Allowed
Advertising Mode: Allowed
Timezone ID: Allowed
Intercom ID: Allowed
BUS mode Switch: Allowed
Short Circuit Duration: Allowed
Eletric Strike Delay: Allowed
Random Electric Strike Delay: Allowed
Electric Strike Duration: Allowed
Disable RTO after ring: Allowed
RTO timeout: Allowed
Doorbell suppression: Allowed
Doorbell suppression duration: Allowed
Sound Ring: Allowed
Sound Open: Allowed
Sound RTO: Allowed
Sound CM: Allowed
Sound confirmation: Allowed
Sound level: Allowed
Single button press action: Allowed
Double button press action: Allowed
Battery type: Allowed
Automatic battery type detection: Allowed
Reboot Nuki: Allowed

------------ GPIO ------------

Retain Input GPIO MQTT state: No
```

### TO REPRODUCE
_send MQTT command to nukihub: "nukihub/opener/configuration/action {"doorbellSuppression":"Ring"}"_

### EXPECTED BEHAVIOUR
_The MQTT published result should be something like /opener/configuration/commandResult {"general":"success"}_

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

technyon, 2025-02-03:
Hi, this result doesn't necessarily mean that the PIN isn't saved, it can also mean that the configured PIN doesn't work. Did you double check that the PIN is exactly the same as configured in the smartphone app?

tca22, 2025-02-03:
Yes, I kept it simple, it's just "0000". However, I just changed it to "1111" - same error.
Then I changed it to "1234" and it worked, means in Nukihub the PIN gets stored. Is the there any check in Nukihub to avoid Pins being too trivial?

Howerver, I carry on submitting `nukihub/opener/configuration/action:{"doorbellSuppression":"Ring"}` which now leads to a `ESP_RST_WDT: Reset due to other watchdogs.`
Anything else I could try from here?

technyon, 2025-02-03:
Hi. I'm not sure about that, it's possible. There's for example a check for keypad PINs in the lock firmwre that they aren't allowed to start with "12" (probably to avoid "123456" as a PIN). I wasn't aware of this at the time, and used "123456" in the example code, which led to some confusion. Good to know though, in case this happens again we can give users better advice.

closed 2025-02-03 by tca22