technyon/nuki_hub #624: Installation on Lilygo T-POE ESP32 board results in reset loop

ohadbenita, 2025-02-07:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X ] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [X ] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [X ] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [X ] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [X ] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [X ] Device used (e.g., ESP32-S3): Lilygo T-POE
- [X ] Nuki Hub binary firmware version number used: 9.08
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Provide the output of http://nukihubIP/get?page=info:
```

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
1. Install Nuki hub on a Lilygo T-POE ESP32 board using the web installer
2. Reboot and connect to serial adapter to see logs.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

The Wi-Fi AP opens and is available for connection

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

Logs:

[14:31:57]mode:DIO, clock div:2
[14:31:57]load:0x3fff0030,len:4644
[14:31:57]load:0x40078000,len:15312
[14:31:57]ho 0 tail 12 room 4
[14:31:57]load:0x40080400,len:4
[14:31:57]load:0x40080404,len:3220
[14:31:57]entry 0x400805a4
[14:31:57]Last config version: 908
[14:31:57]Current config version: 908
[14:31:57]Partition size: 2555904
[14:31:57]Partition subtype: 16
[14:31:57]Nuki Hub version 9.08
[14:31:57]Nuki Hub build 12974430884.10.1
[14:31:57]Load GPIO configuration
[14:31:57]IP configuration: DHCP
[14:31:57]Hardware detect: 1
[14:31:57]MQTT without TLS.
[14:31:57]Network device: Built-in Wi-Fi
[14:31:57]No SSID or Wifi password saved, opening AP
[14:31:58]Wi-Fi async scan started
[14:31:58][WiFi-event] event: 101
WiFi interface ready
[14:31:58]ets Jul 29 2019 12:21:46
[14:31:58]
[14:31:58]rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[14:31:58]configsip: 0, SPIWP:0xee
[14:31:58]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[14:31:58]mode:DIO, clock div:2

**(Please, remember to close the issue when the problem has been addressed)**

technyon, 2025-02-08:
Hi,

The ESP crashes after "WiFi interface ready". It's not clear what exactly causes the crash, but one common is is a bad power supply. Are you powering your ESP via POE, USB or some other power source? Maybe first try a good USB power supply with enough Amps (>= 1A), and a good USB cable.

If that doesn't fix the problem, try doing a factory reset and reflash the ESP using the esptool (preferred) or the Espressif flash dowload tool.

iranl, 2025-02-08:
Could very well be the ESP doing RF calibration, which asks more of the power supply than during usual operation. Agree with @technyon proposals.

ohadbenita, 2025-02-08:
I tried powering using POE which caused the same boot loop - would that be considered a good power supply?
If not, PLMK and I'll try with a good USB power supply and check back here.

technyon, 2025-02-09:
Hi. It should power fine on POE, but let's make sure it works on USB first before going to POE.

ohadbenita, 2025-02-11:
I was able to flash using the esptool, however the boot loops remain.
The thing is that when I try to power the device through the on-board USB port - it doesn't turn on (no blue light indicator).
The boot loop is true both for when powering through POE and through the USB to serial adapter (connected to my Macbook's USB port).

Any suggestions would be highly recommended.

ohadbenita, 2025-02-12:
Haven't tried, will give it a go today

ohadbenita, 2025-02-15:
USB couldn't power both devices. I ended up returning both of them. What would you recommend for a reliable POE ESP32 device?

iranl, 2025-02-15:
As per the readme:

An ESP32-S3 with PSRAM in combination with a SPI Ethernet module (W5500) and PoE to Ethernet and USB type B/C splitter or the **LilyGO-T-ETH ELite**, LilyGO-T-ETH-Lite-ESP32S3 or **M5stack Atom S3R with the M5stack AtomPoe W5500 module**

Bold choices would probably be my preferred options (note the R in **S3R**, regular Atom S3 doesn't have PSRAM).

Disclaimer though that I don't own any of these devices myself. But based on specs these should be the preferred options (S3 with PSRAM, builtin or addin ethernet port with PoE).

@rforro recently added the ELite support. Not sure if he owns one himself though.

technyon, 2025-02-16:
@ohadbenita Do you need encryption. If not Olimex boards are a good option from build quality, I'm running one myself although powered via USB and not POE. Comes with an external antenna too. If you do need encryption, or want it as an option in the future, some device with PSRAM is recommended.

rforro, 2025-02-16:
I own the LilyGO-T-ETH ELite and it's powered by PoE from TL-SG108PE POE+. This is fairly new product from Lilygo, you probably just got unlucky, because they are iterating their dev boards very often.

ohadbenita, 2025-02-16:
> I own the LilyGO-T-ETH ELite and it's powered by PoE from TL-SG108PE POE+. This is fairly new product from Lilygo, you probably just got unlucky, because they are iterating their dev boards very often.

I just ordered the M5Stack Atom POE board, hoping I'll get more lucky with this one...
@iranl @technyon any other board you'd recommend?

closed 2025-02-16 by ohadbenita