technyon/nuki_hub #625: [REQUEST][>=9.08] Please share your coredump crash logs

iranl, 2025-02-09:
We are looking for users that run Nuki Hub 9.08 or higher and have experienced crashes of Nuki Hub due to `ESP_RST_PANIC: Software reset due to exception/panic.`

If your ESP crashes due to this reason on 9.08 or higher Nuki Hub will create a Coredump crash log on the internal ESP storage.
This log can be downloaded from the "Import/Export configuration" page using the "Export Coredump" button.

If you have experienced crashes please post your coredump.txt file(s) in this issue so we can decode the information and maybe reduce the amount of crashes in the future.

Note that coredump files include **no privacy sensitive information**, see the example decoded coredump file below
<details>
<summary>Decoding your own coredump.txt files (NOT REQUIRED)</summary>
<br>

**Requirements**
- Ubuntu (VM/WSL/Bare-metal)
- Follow [these](https://gist.github.com/abobija/2721427f18d78e439fa9875cac960ffa) steps once to install ESP-IDF on Ubuntu
<br>

**Steps**
- Download the coredump.txt file from the "Import/Export configuration" page (`http://NUKIHUBIP/get?page=impexpcfg`)
- Open the file
- Find and download the Nuki Hub firmware package by browsing to `https://github.com/technyon/nuki_hub/actions/runs/INTEGER_BEFORE_THE_FIRST_DOT_MENTIONED_ON_SECOND_LINE_OF_COREDUMP_FILE` and selecting the assets that correspond to your device (mentioned on the first line of coredump file)
- Extract the nuki_hub_esp32.elf file from the archive
- Copy the contents of the coredump file starting from the third line into [https://cryptii.com/pipes/hex-to-base64](https://cryptii.com/pipes/hex-to-base64)
- Overwrite the entire contents of your coredump.txt file with the output from the hex to base64 conversion
- Run  `esp-coredump info_corefile -c coredump.txt -t b64 -rom-elf nuki_hub_esp32.elf`

</details>

<details>
<summary>Example decoded coredump</summary>

```

===============================================================
==================== ESP32 CORE DUMP START ====================

Crashed task handle: 0x3ffd5d78, name: 'nimble_host', GDB name: 'process 1073569144'
Crashed task is not in the interrupt context

================== CURRENT THREAD REGISTERS ===================
exccause       0xffff (InvalidCauseRegister)
excvaddr       0x0
pc             0x20000000          0x20000000
lbeg           0x4009197c          1074338172
lend           0x40091998          1074338200
lcount         0xffffffff          4294967295
sar            0x1a                26
ps             0x30                48
threadptr      <unavailable>
br             <unavailable>
scompare1      <unavailable>
acclo          <unavailable>
acchi          <unavailable>
m0             <unavailable>
m1             <unavailable>
m2             <unavailable>
m3             <unavailable>
expstate       <unavailable>
f64r_lo        <unavailable>
f64r_hi        <unavailable>
f64s           <unavailable>
fcr            <unavailable>
fsr            <unavailable>
a0             0x20000070          536871024
a1             0x0                 0
a2             0x0                 0
a3             0x0                 0
a4             0x60820             395296
a5             0x3f4330fc          1061368060
a6             0x3ffc6610          1073505808
[... 205 lines cut ...]
.coredump.tasks.data 0x3ffde340 0x210 RW
.coredump.tasks.data 0x3ffda45c 0x154 RW
.coredump.tasks.data 0x3ffd9e30 0x620 RW
.coredump.tasks.data 0x3ffdf40c 0x154 RW
.coredump.tasks.data 0x3ffcf010 0x2b0 RW
.coredump.tasks.data 0x3ffd60f4 0x154 RW
.coredump.tasks.data 0x3ffd7ee0 0x360 RW
.coredump.tasks.data 0x3ffdf2b4 0x154 RW
.coredump.tasks.data 0x3ffdf080 0x220 RW
.coredump.tasks.data 0x3ffb74e4 0x154 RW
.coredump.tasks.data 0x3ffb7310 0x1c0 RW
.coredump.tasks.data 0x3ffb7e40 0x154 RW
.coredump.tasks.data 0x3ffb7c70 0x1c0 RW
.coredump.tasks.data 0x3ffbae24 0x154 RW
.coredump.tasks.data 0x3ffbac50 0x1c0 RW
.coredump.tasks.data 0x3ffb5cb4 0x154 RW
.coredump.tasks.data 0x3ffd3840 0x200 RW
.coredump.tasks.data 0x3ffd8300 0x154 RW
.coredump.tasks.data 0x3ffe7200 0x4b0 RW

===================== ESP32 CORE DUMP END =====================
===============================================================
```

</details>

VDRainer, 2025-02-19:
Had that a few minutes ogo.

[image]

but needed to pull the plug and another restart to downlad the file. 
Hope thats ok?

[coredump.txt](https://github.com/user-attachments/files/18868873/coredump.txt)

VDRainer, 2025-02-19:
Here's one more. 

[coredump.txt](https://github.com/user-attachments/files/18875349/coredump.txt)

iranl, 2025-02-19:
@VDRainer: Thanks. Your crashdumps essentially show the same thing as eachother and as my example. This seems to be the main cause for (random) panics on the ESP32. Interestingly I don't see these on the ESP32-S3.

Based on the crashdumps it looks like a stack overflow somewhere, unfortunately the crashdumps don't point to the exact thread that overflows. I am going to change some settings in 9.09 that will hopefully help point to the offending task when crashes occur on 9.09.

<details>
<summary>Your decoded coredump</summary>

```
===============================================================
==================== ESP32 CORE DUMP START ====================

Crashed task handle: 0x3ffdfda4, name: 'nimble_host', GDB name: 'process 1073610148'
Crashed task is not in the interrupt context

================== CURRENT THREAD REGISTERS ===================
exccause       0xffff (InvalidCauseRegister)
excvaddr       0x0
pc             0x20000000          0x20000000
lbeg           0x4009197c          1074338172
lend           0x40091998          1074338200
lcount         0xffffffff          4294967295
sar            0x1a                26
ps             0x30                48
threadptr      <unavailable>
br             <unavailable>
scompare1      <unavailable>
acclo          <unavailable>
acchi          <unavailable>
m0             <unavailable>
m1             <unavailable>
m2             <unavailable>
m3             <unavailable>
expstate       <unavailable>
f64r_lo        <unavailable>
f64r_hi        <unavailable>
f64s           <unavailable>
fcr            <unavailable>
fsr            <unavailable>
a0             0x20000070          536871024
a1             0x0                 0
a2             0x0                 0
a3             0x0                 0
a4             0x60c20             396320
a5             0x3f4330fc          1061368060
a6             0x3ffc6610          1073505808
a7             0x3ffc6614          1073505812
a8             0x3ff000e0          1072693472
a9             0x1                 1
a10            0x3ffbdf30          1073471280
a11            0xffffffff          -1
a12            0xb33fffff          -1287651329
a13            0x60c23             396323
a14            0x1                 1
a15            0xabab              43947

==================== CURRENT THREAD STACK =====================
#0  0x20000000 in ?? ()

======================== THREADS INFO =========================
  Id   Target Id          Frame
* 1    process 1073610148 0x20000000 in ?? ()
  2    process 1073578156 0x4000bff0 in ?? ()
  3    process 1073466168 0x40094de2 in esp_cpu_wait_for_intr () at /home/runner/.platformio/packages/framework-espidf/components/esp_hw_support/cpu.c:64
  4    process 1073468052 0x40094de2 in esp_cpu_wait_for_intr () at /home/runner/.platformio/packages/framework-espidf/components/esp_hw_support/cpu.c:64
  5    process 1073560116 0x4000bff0 in ?? ()
  6    process 1073685328 0x4000bff0 in ?? ()
  7    process 1073568784 0x4000bff0 in ?? ()
  8    process 1073444420 0x4000bff0 in ?? ()
  9    process 1073564340 0x4000bff0 in ?? ()
  10   process 1073655172 0x4000bff0 in ?? ()
[... 180 lines cut ...]
.coredump.tasks.data 0x3ffd3810 0x210 RW
.coredump.tasks.data 0x3fff2350 0x154 RW
.coredump.tasks.data 0x3fff1d20 0x620 RW
.coredump.tasks.data 0x3ffd5c10 0x154 RW
.coredump.tasks.data 0x3ffd5950 0x2b0 RW
.coredump.tasks.data 0x3ffb7644 0x154 RW
.coredump.tasks.data 0x3ffb7470 0x1c0 RW
.coredump.tasks.data 0x3ffd4ab4 0x154 RW
.coredump.tasks.data 0x3ffd4880 0x220 RW
.coredump.tasks.data 0x3ffead84 0x154 RW
.coredump.tasks.data 0x3ffecb70 0x360 RW
.coredump.tasks.data 0x3ffb9d04 0x154 RW
.coredump.tasks.data 0x3ffb7dd0 0x1c0 RW
.coredump.tasks.data 0x3ffbaf8c 0x154 RW
.coredump.tasks.data 0x3ffbadc0 0x1c0 RW
.coredump.tasks.data 0x3ffb5cb4 0x154 RW
.coredump.tasks.data 0x3ffdf1f0 0x200 RW
.coredump.tasks.data 0x3fff01f4 0x154 RW
.coredump.tasks.data 0x3ffefd30 0x4b0 RW

===================== ESP32 CORE DUMP END =====================
===============================================================

```
</details>

VDRainer, 2025-02-26:
Another one from 9.09-master6

[coredump.txt](https://github.com/user-attachments/files/18993759/coredump.txt)

h1ghrise, 2025-03-03:
[coredump.txt](https://github.com/user-attachments/files/19052633/coredump.txt)
9.08. Uptime: 20h

Mkyuu10, 2025-03-05:
Two cordumps:
first from Monday after 21,5h uptime
second from tonight after 38h updtime

[coredump_1.txt](https://github.com/user-attachments/files/19088239/coredump_1.txt)
[coredump_2.txt](https://github.com/user-attachments/files/19088238/coredump_2.txt)

iranl, 2025-03-05:
Thanks for all the crashdumps.
These all point in basically the same direction, being a LoadProhibited error in nimble_host.

For now these are enough dumps

Ra72xx, 2025-03-06:
Looking forward to a solution. For me it seems to get worse every day, NukiHub seems to be rebooting all the time :-( .

Seestern93, 2025-04-04:
Hi - having 2 coredumps from v9.10-beta2 here, one after 900min and one after 1400min uptime

[coredump_250404_2236.txt](https://github.com/user-attachments/files/19610144/coredump_250404_2236.txt)
[coredump_250404_0754.txt](https://github.com/user-attachments/files/19610143/coredump_250404_0754.txt)

VDRainer, 2025-04-05:
As recommended here and in other issues, I switched to a ESP32-S3 device ([Waveshare ESP32-S3-ETH / ESP32-S3-POE-ETH](https://www.waveshare.com/esp32-s3-eth.htm?sku=28771))
No problems since then.
Uptime now 26 days 20 hours

iranl, 2025-04-07:
Thanks for all the coredumps. 

In general the coredump logs show crashes only on the original esp32dev, all related to the bluetooth controller and/or BLE library. Part of that code is closed-source. At this time I don't see any options to debug this further and for now we will have to accept and expect crashes on the esp32dev. If this is unacceptable upgrade to a ESP32-S3, as these crashes are not seen there.

I will close the issue for now.