technyon/nuki_hub #627: very bad network availability with nuki_hub 9.0x

leo-b, 2025-02-13:
### PROBLEM DESCRIPTION
I am experiencing very bad network availability issues with nuki_hub versions 9.0x.
Pinging the device shows a high packet loss and web requests often block für many seconds or even timeout:

```
# ping nukihub2
PING nuki-hub2.ms14.kloburg.at (10.0.1.92) 56(84) bytes of data.
64 bytes from nuki-hub2.ms14.kloburg.at (10.0.1.92): icmp_seq=6 ttl=64 time=4212 ms
64 bytes from nuki-hub2.ms14.kloburg.at (10.0.1.92): icmp_seq=7 ttl=64 time=3194 ms
64 bytes from nuki-hub2.ms14.kloburg.at (10.0.1.92): icmp_seq=8 ttl=64 time=2172 ms
64 bytes from nuki-hub2.ms14.kloburg.at (10.0.1.92): icmp_seq=9 ttl=64 time=1157 ms
64 bytes from nuki-hub2.ms14.kloburg.at (10.0.1.92): icmp_seq=10 ttl=64 time=133 ms
64 bytes from nuki-hub2.ms14.kloburg.at (10.0.1.92): icmp_seq=13 ttl=64 time=6978 ms
64 bytes from nuki-hub2.ms14.kloburg.at (10.0.1.92): icmp_seq=14 ttl=64 time=5957 ms
64 bytes from nuki-hub2.ms14.kloburg.at (10.0.1.92): icmp_seq=15 ttl=64 time=4942 ms
64 bytes from nuki-hub2.ms14.kloburg.at (10.0.1.92): icmp_seq=16 ttl=64 time=3918 ms
64 bytes from nuki-hub2.ms14.kloburg.at (10.0.1.92): icmp_seq=17 ttl=64 time=2894 ms
[...]
--- nuki-hub2.ms14.kloburg.at ping statistics ---
815 packets transmitted, 295 received, 63.8037% packet loss, time 832733ms
rtt min/avg/max/mdev = 3.000/2265.762/9921.020/1694.991 ms, pipe 10
```

IMHO it looks like the networking code is blocked by some other task for some time and is not able to handle requests.?

Until yesterday the problems only appeared after some uptime and I was having the impression that after a reboot of the lock, the problems would disappear for some time again. (I suspected it to be a BLE issue, see https://github.com/technyon/nuki_hub/issues/619)
But since yesterday, the problems even outlast a reboot of the lock and the hub.

Looking at the MQTT log (without debugging enabled), I cannot see any suspicous entries. (See below.)

I have a second nuki_hub (connected to another lock) running 9.00 that doesn't show that problems.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [x] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [x] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [x] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [x] Device used (e.g., ESP32-S3): ESP32
- [x] Nuki Hub binary firmware version number used: 9.08
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Provide the output of http://nukihubIP/get?page=info:
```
System Information

------------ NUKI HUB ------------
Device: ESP32
Version: 9.08
Build: 12974430884.10.1
Build type: Release
Build date: 2025-01-26
Updater version: 9.08
Updater build: 12974430884.10.1
Updater build date: 2025-01-23
Uptime (min): 75
Config version: 908
Last restart reason FW: NotApplicable
Last restart reason ESP: ESP_RST_POWERON: Reset due to power-on event.
Free internal heap: 64760
Total internal heap: 246356
PSRAM Available: No
Network task stack high watermark: 5240
Nuki task stack high watermark: 5020

------------ GENERAL SETTINGS ------------
Network task stack size: 12288
Nuki task stack size: 8192
[... 237 lines cut ...]
2025-02-13T18:58:04+0100 nuki-garten/maintenance/log: Result (attempt 1): success
2025-02-13T18:58:04+0100 nuki-garten/lock/battery/advancedJson: {"batteryDrain":3599,"batteryVoltage":5.183,"critical":156,"lockAction":"Unknown","startVoltage":5.176,"lowestVoltage":3.678,"lockDistance":0.081,"startTemperature":15,"maxTurnCurrent":3.098,"batteryResistance":0.515}
2025-02-13T18:58:04+0100 nuki-garten/maintenance/log: Done querying lock battery state
2025-02-13T18:58:57+0100 nuki-garten/maintenance/uptime: 31
2025-02-13T18:59:45+0100 nuki-garten/maintenance/log: networkTask is running
2025-02-13T18:59:45+0100 nuki-garten/maintenance/log: nukiTask is running
2025-02-13T18:59:57+0100 nuki-garten/maintenance/uptime: 32
2025-02-13T19:00:57+0100 nuki-garten/maintenance/uptime: 33
2025-02-13T19:01:45+0100 nuki-garten/maintenance/log: networkTask is running
2025-02-13T19:01:45+0100 nuki-garten/maintenance/log: nukiTask is running
2025-02-13T19:01:57+0100 nuki-garten/maintenance/uptime: 34
2025-02-13T19:02:57+0100 nuki-garten/maintenance/uptime: 35
2025-02-13T19:03:45+0100 nuki-garten/maintenance/log: networkTask is running
2025-02-13T19:03:45+0100 nuki-garten/maintenance/log: nukiTask is running
2025-02-13T19:03:57+0100 nuki-garten/maintenance/uptime: 36
2025-02-13T19:04:55+0100 nuki-garten/configuration/action: 
2025-02-13T19:04:57+0100 nuki-garten/maintenance/uptime: 37
2025-02-13T19:05:45+0100 nuki-garten/maintenance/log: networkTask is running
2025-02-13T19:05:45+0100 nuki-garten/maintenance/log: nukiTask is running
2025-02-13T19:05:57+0100 nuki-garten/maintenance/uptime: 38
2025-02-13T19:06:57+0100 nuki-garten/maintenance/uptime: 39
2025-02-13T19:07:45+0100 nuki-garten/maintenance/log: networkTask is running
2025-02-13T19:07:45+0100 nuki-garten/maintenance/log: nukiTask is running
2025-02-13T19:07:57+0100 nuki-garten/maintenance/uptime: 40
```

iranl, 2025-02-19:
You can try (temporarily) disabling your Nuki lock in Nuki Hub from the "Nuki configuration" page.
If both Lock and Opener are disabled all BLE functionality will be disabled.

You can check if this solves your network/ping issues.
If this does not solve the issues they are not related to BLE.
If the issues are solved this way you can try enabling the lock again and disabling "New Nuki Bluetooth connection mode (disable if there are connection issues)" frpm the "Nuki configuration" page

leo-b, 2025-02-22:
> You can try (temporarily) disabling your Nuki lock in Nuki Hub from the "Nuki configuration" page.
If both Lock and Opener are disabled all BLE functionality will be disabled.

Disabling lock and opener didn't improve the packet loss.

I flashed another device (ESP32-S3 with PSRAM). The new device doesn't show wifi connection quality issues - no packet loss.
I exported the device settings (w/ redacted) for later reference.

After importing the settings from the problematic device, the connectivity issues also appear on the new device.

Interestingly the connectivity issues remain even after re-importing the saved factory default settings.
I had to erase and flash the device again in order to obtain a stable wifi connection again.

I'll now try to configure my settings step by step via web-interface to see which one causes the problems.

leo-b, 2025-02-22:
> I'll now try to configure my settings step by step via web-interface to see which one causes the problems.

I could not relate the connectivity issues to specific settings. After some time the issues came back and didn't disappear after reverting the latest settings.

I checked the wifi station statistics on the openwrt access point and noticed a very high number of tx errors:

```
# iw dev wlan1-1 station get c8:f0:9e:4e:7b:2c
Station c8:f0:9e:4e:7b:2c (on wlan1-1)
        inactive time:  800 ms
        rx bytes:       190806
        rx packets:     1448
        tx bytes:       209025
        tx packets:     1514
        tx retries:     0
        tx failed:      884
        rx drop misc:   54
        signal:         -47 [-64, -47, -55, -65] dBm
        signal avg:     -49 [-66, -49, -56, -67] dBm
        tx bitrate:     18.0 MBit/s
        tx duration:    419543 us
        rx bitrate:     6.0 MBit/s
        rx duration:    0 us
        airtime weight: 256
        authorized:     yes
        authenticated:  yes
        associated:     yes
        preamble:       short
        WMM/WME:        no
        MFP:            no
        TDLS peer:      no
        DTIM period:    2
        beacon interval:100
        short preamble: yes
        connected time: 2321 seconds
        associated at [boottime]:       1747232.352s
        associated at:  1740040331677 ms
        current time:   1740042652727 ms
```

The other nukihub (running firmware 9.00), which is connected to the same AP didn't show such a high error rate:

```
# iw dev wlan1-1 station get c8:f0:9e:50:1b:d0
Station c8:f0:9e:50:1b:d0 (on wlan1-1)
        inactive time:  2410 ms
        rx bytes:       21325744
        rx packets:     375650
        tx bytes:       6135464
        tx packets:     68666
        tx retries:     0
        tx failed:      27
        rx drop misc:   3859
        signal:         -51 [-53, -54, -54, -68] dBm
        signal avg:     -51 [-53, -54, -54, -68] dBm
        tx bitrate:     54.0 MBit/s
        tx duration:    14016942 us
        rx bitrate:     6.0 MBit/s
        rx duration:    0 us
        airtime weight: 256
        authorized:     yes
        authenticated:  yes
        associated:     yes
        preamble:       short
        WMM/WME:        no
        MFP:            no
        TDLS peer:      no
        DTIM period:    2
        beacon interval:100
        short preamble: yes
        connected time: 165769 seconds
        associated at [boottime]:       1583884.906s
        associated at:  1739876984231 ms
        current time:   1740042752547 ms
```

I rebooted the access point and for now the connectivity issues are gone.

I'd really like to believe that the cause of the issue simply had been a hickup of my access point and everything is fine now. But some of my observations don't fit to that theory:

- the other nukihub (with FW 9.00) and about 5 other devices (mostly ESP 8266 running esphome) did not show the issue
- another  device (ESP32-S3 with PSRAM) with a different mac address developed the same issue after only a short time of testing

However, since I cannot currently reproduce the issue, I'll close this ticket.
Cheers,
--leo

closed 2025-02-22 by leo-b