technyon/nuki_hub #631: M5Stack PoESP32 connected via ethernet stays connected to WiFi

kooskaspers, 2025-02-20:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

The device keeps connecting via WiFi, even though I've configured the device to make use of the ethernet interface.
This is the same problem as mentioned here: https://github.com/technyon/nuki_hub/issues/152
But in that case a different esp32 is used (but same brand though)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [x] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [x] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [x] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [x] Device used (e.g., ESP32-S3): M5Stack PoESP32
- [ ] Nuki Hub binary firmware version number used: NukiHub-9.08-ESP32
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Provide the output of http://nukihubIP/get?page=info:
```
System Information

------------ NUKI HUB ------------
Device: ESP32
Version: 9.08
Build: 12974430884.10.1
Build type: Release
Build date: 2025-01-26
Updater version: 
Updater build: 
Updater build date: 
Uptime (min): 0
Config version: 908
Last restart reason FW: RequestedViaWebServer
Last restart reason ESP: ESP_RST_SW: Software reset via esp_restart.
Free internal heap: 64936
Total internal heap: 246356
PSRAM Available: No
Network task stack high watermark: 8292
Nuki task stack high watermark: 5016

------------ GENERAL SETTINGS ------------
Network task stack size: 12288
Nuki task stack size: 8192
Check for updates: Yes
Latest version: 9.08
Allow update from MQTT: No
Update Nuki Hub and Nuki devices time using NTP: No
Web configurator username: Not set
Web configurator password: Not set
Web configurator bypass for proxy IP: Not set
Web configurator authentication: Basic
Session validity (in seconds): 3600
Session validity remember (in hours): 720
Duo Push MFA enabled: No
Web configurator enabled: Yes
HTTP SSL: Disabled
Advanced menu enabled: No
Publish free heap over MQTT: No
Nuki connect debug logging enabled: No
Nuki communication debug logging enabled: No
Nuki readable data debug logging enabled: No
Nuki hex data debug logging enabled: No
Nuki command debug logging enabled: No
MQTT log enabled: No
Webserial enabled: No
Bootloop protection enabled: No

------------ NETWORK ------------
Network device: Built-in Wi-Fi
Network connected: Yes
IP Address: 192.168.1.144
SSID: Kaspers
BSSID of AP: 78:8A:20:2D:3E:46
ESP32 MAC address: C4:D8:D5:A0:41:64

------------ NETWORK SETTINGS ------------
Nuki Hub hostname: nukihub
DHCP enabled: Yes
RSSI Publish interval (s): 60
Find WiFi AP with strongest signal: Yes
Restart ESP32 on network disconnect enabled: No
Disable Network if not connected within 60s: No
MQTT Timeout until restart (s): 60

------------ MQTT ------------
MQTT connected: Yes
MQTT broker address: kubernetes.lan
MQTT broker port: 1883
MQTT username: ***
MQTT password: ***
MQTT base topic: nukihub
MQTT SSL: Disabled

------------ BLUETOOTH ------------
Bluetooth connection mode: New
Bluetooth TX power (dB): 9
Bluetooth command nr of retries: 3
Bluetooth command retry delay (ms): 100
Seconds until reboot when no BLE beacons received: 60

------------ QUERY / PUBLISH SETTINGS ------------
Lock/Opener state query interval (s): 1800
Publish Nuki device authorization log: No
Max authorization log entries to retrieve: 5
Battery state query interval (s): 1800
Most non-JSON MQTT topics disabled: No
Publish Nuki device config: Yes
Config query interval (s): 3600
Publish Keypad info: No
Keypad query interval (s): 1800
Enable Keypad control: Yes
Publish Keypad topic per entry: No
Publish Keypad codes: No
Allow checking Keypad codes: No
Max keypad entries to retrieve: 10
Publish timecontrol info: No
Keypad query interval (s): 1800
Enable timecontrol control: No
Publish timecontrol topic per entry: No
Max timecontrol entries to retrieve: 10
Enable authorization control: No
Publish authorization topic per entry: No
Max authorization entries to retrieve: 5

------------ HOME ASSISTANT ------------
Home Assistant auto discovery enabled: Yes
Home Assistant auto discovery topic: homeassistant/
Nuki Hub configuration URL for HA: http://192.168.1.144
Nuki Hub ID: 110233034021060

------------ NUKI LOCK ------------
Lock enabled: Yes
Lock Ultra enabled: No
Paired: Yes
Nuki Hub device ID: 2880165731
Nuki device ID: ***
Firmware version: 3.10.7
Hardware version: 5.12
Valid PIN set: Yes
Has door sensor: No
Has keypad: No
Timecontrol highest entries count: 0
Authorizations highest entries count: 0
Register as: Bridge

------------ HYBRID MODE ------------
Hybrid mode enabled: No
Force Lock ID: No
Force Lock Keypad: No
Force Lock Doorsensor: No

------------ NUKI LOCK ACL ------------
Lock: Allowed
Unlock: Allowed
Unlatch: Disallowed
Lock N Go: Disallowed
Lock N Go Unlatch: Disallowed
Full Lock: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed

------------ NUKI LOCK CONFIG ACL ------------
Name: Allowed
Latitude: Allowed
Longitude: Allowed
Auto Unlatch: Allowed
Pairing enabled: Allowed
Button enabled: Allowed
LED flash enabled: Allowed
LED brightness: Allowed
Timezone offset: Allowed
DST mode: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed
Single Lock: Allowed
Advertising Mode: Allowed
Timezone ID: Allowed
Unlocked Position Offset Degrees: Allowed
Locked Position Offset Degrees: Allowed
Single Locked Position Offset Degrees: Allowed
Unlocked To Locked Transition Offset Degrees: Allowed
Lock n Go timeout: Allowed
Single button press action: Allowed
Double button press action: Allowed
Detached cylinder: Allowed
Battery type: Allowed
Automatic battery type detection: Allowed
Unlatch duration: Allowed
Auto lock timeout: Allowed
Auto unlock disabled: Allowed
Nightmode enabled: Allowed
Nightmode start time: Allowed
Nightmode end time: Allowed
Nightmode auto lock enabled: Allowed
Nightmode auto unlock disabled: Allowed
Nightmode immediate lock on start: Allowed
Auto lock enabled: Allowed
Immediate auto lock enabled: Allowed
Auto update enabled: Allowed
Reboot Nuki: Allowed
Motor speed: Allowed
Enable slow speed during nightmode: Allowed

------------ NUKI OPENER ------------
Opener enabled: No

------------ GPIO ------------

Retain Input GPIO MQTT state: No
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Flashed the ESP, configured via WIFI and changed the network config to:

Afterwards I'm able to boot the device via Power-over-Ethernet, but the device keeps connecting over wifi.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

I expected the device to be connected over the ethernet interface.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

[image]

### ADDITIONAL CONTEXT
I tried to disable DHCP and add a static network config, but this didn't solve the issue.

**(Please, remember to close the issue when the problem has been addressed)**

kooskaspers, 2025-02-20:
> Try to set the device to GL-S10.

[image]

Same problem :(

```
System Information

------------ NUKI HUB ------------
Device: ESP32
Version: 9.08
Build: 12974430884.10.1
Build type: Release
Build date: 2025-01-26
Updater version: 
Updater build: 
Updater build date: 
Uptime (min): 0
Config version: 908
Last restart reason FW: RequestedViaWebServer
Last restart reason ESP: ESP_RST_SW: Software reset via esp_restart.
Free internal heap: 64820
Total internal heap: 246356
PSRAM Available: No
Network task stack high watermark: 8100
Nuki task stack high watermark: 5016

------------ GENERAL SETTINGS ------------
Network task stack size: 12288
Nuki task stack size: 8192
Check for updates: Yes
Latest version: 9.08
Allow update from MQTT: No
Update Nuki Hub and Nuki devices time using NTP: No
Web configurator username: Not set
Web configurator password: Not set
Web configurator bypass for proxy IP: Not set
Web configurator authentication: Basic
Session validity (in seconds): 3600
Session validity remember (in hours): 720
Duo Push MFA enabled: No
Web configurator enabled: Yes
HTTP SSL: Disabled
Advanced menu enabled: No
Publish free heap over MQTT: No
Nuki connect debug logging enabled: No
Nuki communication debug logging enabled: No
Nuki readable data debug logging enabled: No
Nuki hex data debug logging enabled: No
Nuki command debug logging enabled: No
MQTT log enabled: No
Webserial enabled: No
Bootloop protection enabled: No

------------ NETWORK ------------
Network device: Built-in Wi-Fi
Network connected: Yes
IP Address: 192.168.1.144
SSID: Kaspers
BSSID of AP: 78:8A:20:2D:3E:46
ESP32 MAC address: C4:D8:D5:A0:41:64

------------ NETWORK SETTINGS ------------
Nuki Hub hostname: nukihub
DHCP enabled: Yes
RSSI Publish interval (s): 60
Find WiFi AP with strongest signal: Yes
Restart ESP32 on network disconnect enabled: No
Disable Network if not connected within 60s: No
MQTT Timeout until restart (s): 60

------------ MQTT ------------
MQTT connected: Yes
MQTT broker address: kubernetes.lan
MQTT broker port: 1883
MQTT username: ***
MQTT password: ***
MQTT base topic: nukihub
MQTT SSL: Disabled

------------ BLUETOOTH ------------
Bluetooth connection mode: New
Bluetooth TX power (dB): 9
Bluetooth command nr of retries: 3
Bluetooth command retry delay (ms): 100
Seconds until reboot when no BLE beacons received: 60

------------ QUERY / PUBLISH SETTINGS ------------
Lock/Opener state query interval (s): 1800
Publish Nuki device authorization log: No
Max authorization log entries to retrieve: 5
Battery state query interval (s): 1800
Most non-JSON MQTT topics disabled: No
Publish Nuki device config: Yes
Config query interval (s): 3600
Publish Keypad info: No
Keypad query interval (s): 1800
Enable Keypad control: Yes
Publish Keypad topic per entry: No
Publish Keypad codes: No
Allow checking Keypad codes: No
Max keypad entries to retrieve: 10
Publish timecontrol info: No
Keypad query interval (s): 1800
Enable timecontrol control: No
Publish timecontrol topic per entry: No
Max timecontrol entries to retrieve: 10
Enable authorization control: No
Publish authorization topic per entry: No
Max authorization entries to retrieve: 5

------------ HOME ASSISTANT ------------
Home Assistant auto discovery enabled: Yes
Home Assistant auto discovery topic: homeassistant/
Nuki Hub configuration URL for HA: http://192.168.1.144
Nuki Hub ID: 110233034021060

------------ NUKI LOCK ------------
Lock enabled: Yes
Lock Ultra enabled: No
Paired: Yes
Nuki Hub device ID: 2880165731
Nuki device ID: ***
Firmware version: 3.10.7
Hardware version: 5.12
Valid PIN set: Yes
Has door sensor: No
Has keypad: No
Timecontrol highest entries count: 0
Authorizations highest entries count: 0
Register as: Bridge

------------ HYBRID MODE ------------
Hybrid mode enabled: No
Force Lock ID: No
Force Lock Keypad: No
Force Lock Doorsensor: No

------------ NUKI LOCK ACL ------------
Lock: Allowed
Unlock: Allowed
Unlatch: Disallowed
Lock N Go: Disallowed
Lock N Go Unlatch: Disallowed
Full Lock: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed

------------ NUKI LOCK CONFIG ACL ------------
Name: Allowed
Latitude: Allowed
Longitude: Allowed
Auto Unlatch: Allowed
Pairing enabled: Allowed
Button enabled: Allowed
LED flash enabled: Allowed
LED brightness: Allowed
Timezone offset: Allowed
DST mode: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed
Single Lock: Allowed
Advertising Mode: Allowed
Timezone ID: Allowed
Unlocked Position Offset Degrees: Allowed
Locked Position Offset Degrees: Allowed
Single Locked Position Offset Degrees: Allowed
Unlocked To Locked Transition Offset Degrees: Allowed
Lock n Go timeout: Allowed
Single button press action: Allowed
Double button press action: Allowed
Detached cylinder: Allowed
Battery type: Allowed
Automatic battery type detection: Allowed
Unlatch duration: Allowed
Auto lock timeout: Allowed
Auto unlock disabled: Allowed
Nightmode enabled: Allowed
Nightmode start time: Allowed
Nightmode end time: Allowed
Nightmode auto lock enabled: Allowed
Nightmode auto unlock disabled: Allowed
Nightmode immediate lock on start: Allowed
Auto lock enabled: Allowed
Immediate auto lock enabled: Allowed
Auto update enabled: Allowed
Reboot Nuki: Allowed
Motor speed: Allowed
Enable slow speed during nightmode: Allowed

------------ NUKI OPENER ------------
Opener enabled: No

------------ GPIO ------------

Retain Input GPIO MQTT state: No

```

iranl, 2025-02-20:
Try the following:

- Set network hardware to custom and save:

[image]
<br>
<br>

- Go to custom ethernet configuration:

[image]
<br>
<br>

- Set the custom settings to the following and save:

[image]
<br>
<br>

- Reboot

kooskaspers, 2025-02-22:
> Try the following:

I set the configuration as requested:

[image]

Stil the device is connecting over Wi-Fi:

[image]

technyon, 2025-02-23:
Hi, I think that was bug. It's strange, it should just work. I have one of those my own, just need to get it that sucker to flash ... a bit annoying it doesn't come with USB.

kooskaspers, 2025-02-23:
> Hi, I think that was bug. It's strange, it should just work. I have one of those my own, just need to get it that sucker to flash ... a bit annoying it doesn't come with USB.
> 
> P.S.: It'll have to wait a bit, need to order proper jumper cables first.

No worries, thanks for looking into this!
I’ve been a bit lazy if it comes to the wiring haha:

[image]

ohadbenita, 2025-02-28:
Same here, just got the same type of unit, tried to use the POEESP32 setting, GL-S10 & custom settings suggested by @iranl with no luck, still connected through the built-in WiFi.

mfrelink, 2025-03-01:
Exactly the same problem with a comparable device the Waveshare ESP32-S3-POE-ETH
 
https://www.waveshare.com/esp32-s3-eth.htm?sku=28771 https://www.waveshare.com/esp32-s3-eth.htm?sku=28771

technyon, 2025-03-02:
Hi, the code for initialization is incorrect. I need to figure how to do it correctly. Flashing that board is quite unreliable even with an espprog board, I think due to limited power from the USB port of my laptop. If I manage top flash it, it fails to boot due to insufficient power.

technyon, 2025-03-02:
Please check this build:

https://github.com/technyon/nuki_hub/actions/runs/13612835910

I think there was some copy and paste error from refactoring the code, I've restored the parameters from release 8.35.

ohadbenita, 2025-03-02:
I can confirm the uploaded release fixed the issue @technyon 👍 🎉 

[image]

kooskaspers, 2025-03-02:
Great to hear @technyon and @ohadbenita! Could this fix be merged with the latest codebase?

technyon, 2025-03-03:
Fixed with PR #633

technyon, 2025-03-03:
@mfrelink Supporting that board should be possible, but the hardware is different so let's not track this in the same issue. Need to figure out the PIN mapping, I'll have a look at their documentation.

closed 2025-03-03 by technyon

kooskaspers, 2025-03-12:
@technyon I can confirm as well your fix works!:

[image]

Important to mention for others: I was not able to switch from WiFi to LAN by only changing the network hardware to:

[image]
and rebooting the device: I had to fully disconnect and reconnect the ESP from the POE port, forcing it to completely reboot, instead of just a 'software' reboot. Otherwise it kept connecting via WiFi.

technyon, 2025-03-13:
@kooskaspers Thank you