technyon/nuki_hub #634: Add Waveshare ESP32-S3-ETH support

technyon, 2025-03-03:
Issue to track support for Waveshare ESP32-S3-ETH.

mfrelink, 2025-03-03:
Maybe this helps :)

[image]

technyon, 2025-03-03:
Please check this:

https://github.com/technyon/nuki_hub/actions/runs/13631492253

They have an ethernet example online, I just copied their PIN defintions from their example, so this just might work.

technyon, 2025-03-03:
Get serial logs from startup

mfrelink, 2025-03-04:
I got hand on antoher Waveshare ESP32 and low and behold,  this one works!

------------ NETWORK ------------
Network device: Waveshare ESP32-S3-ETH / ESP32-S3-ETH-POE
Network connected: Yes
IP Address: 10.100.10.245

I will reset my Waveshare ESP32 tonight and flash it again and report back to you.

technyon, 2025-03-04:
Great! Strange though, does the other one maybe have a hardware defect?

mfrelink, 2025-03-04:
> Great! Strange though, does the other one maybe have a hardware defect?

It works! It was a simple fix. Rebooting didn’t help, but power cycling did. So thanks for adding the device 👍

technyon, 2025-03-05:
#636 merged to master. will be included in the next release.

closed 2025-03-05 by technyon