technyon/nuki_hub #637: Nukihub 9.09 disconnects from MQTT server and the Web UI still shows "connected: yes"

ssorgatem, 2025-03-07:
### PROBLEM DESCRIPTION
NukiHub disappears from Home Assistant 2 minutes after rebooting, and won't appear until I manually reboot the nukihub. In the web UI MQTT appears as "connected" both before and after disappearing from Home Assistant (however int he info page MQTT appears as "disconnected", so t the very least this is a bug concerning the UI not agreeing witht he info page)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [x] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [x] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [x] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [x] Device used (e.g., ESP32-S3): ESP32-C3
- [x] Nuki Hub binary firmware version number used: 9.09
  - [x] Pre-compiled (Build: 13675437614.11.1)
  - [ ] Self-compiled
- [x] Provide the output of http://nukihubIP/get?page=info:
```
System Information
------------ NUKI HUB ------------
Device: ESP32-C3
Version: 9.09
Build: 13675437614.11.1
Build type: Release
Build date: 2025-03-05
Updater version: 9.09
Updater build: 13675437614.11.1
Updater build date: 2025-03-03
Uptime (min): 1320
Config version: 909
Last restart reason FW: RestartOnDisconnectWatchdog
Last restart reason ESP: ESP_RST_SW: Software reset via esp_restart.
Free internal heap: 60944
Total internal heap: 255368
PSRAM Available: No
Network task stack high watermark: 7652
Nuki task stack high watermark: 5484

------------ SPIFFS ------------
SPIFFS Total Bytes: 113201
SPIFFS Used Bytes: 27359
SPIFFS Free Bytes: 85842

------------ GENERAL SETTINGS ------------
Network task stack size: 12288
Nuki task stack size: 8192
Check for updates: Yes
Latest version: 9.09
Allow update from MQTT: Yes
Update Nuki Hub and Nuki devices time using NTP: Yes
Web configurator username: ***
Web configurator password: ***
Web configurator bypass for proxy IP: Not set
Web configurator authentication: Basic
Session validity (in seconds): 3600
Session validity remember (in hours): 720
Duo Push MFA enabled: No
TOTP MFA enabled: No
Web configurator enabled: Yes
HTTP SSL: Disabled
Advanced menu enabled: No
Publish free heap over MQTT: No
Nuki connect debug logging enabled: No
Nuki communication debug logging enabled: No
Nuki readable data debug logging enabled: No
Nuki hex data debug logging enabled: No
Nuki command debug logging enabled: No
MQTT log enabled: Yes
Webserial enabled: No
Bootloop protection enabled: No

------------ NETWORK ------------
Network device: Built-in Wi-Fi
Network connected: Yes
IP Address: 192.168.7.199
SSID: Shub-Niggurath
BSSID of AP: 24:2F:D0:32:9B:D2
ESP32 MAC address: 64:E8:33:88:63:48

------------ NETWORK SETTINGS ------------
Nuki Hub hostname: nukihub
DHCP enabled: No
Static IP address: 192.168.7.199
Static IP subnet: 255.255.255.0
Static IP gateway: 192.168.7.1
Static IP DNS server: 192.168.7.214
RSSI Publish interval (s): 60
Find WiFi AP with strongest signal: Yes
Restart ESP32 on network disconnect enabled: Yes
Disable Network if not connected within 60s: No
MQTT Timeout until restart (s): 60

------------ MQTT ------------
MQTT connected: No
MQTT broker address: 192.168.7.215
MQTT broker port: 1883
MQTT username: ***
MQTT password: ***
MQTT base topic: nukihub
MQTT SSL: Disabled

------------ BLUETOOTH ------------
Bluetooth connection mode: New
Bluetooth TX power (dB): 9
Bluetooth command nr of retries: 3
Bluetooth command retry delay (ms): 100
Seconds until reboot when no BLE beacons received: 60

------------ QUERY / PUBLISH SETTINGS ------------
Lock/Opener state query interval (s): 1800
Publish Nuki device authorization log: No
Max authorization log entries to retrieve: 5
Battery state query interval (s): 1800
Most non-JSON MQTT topics disabled: Yes
Publish Nuki device config: Yes
Config query interval (s): 3600
Publish Keypad info: No
Keypad query interval (s): 1800
Enable Keypad control: Yes
Publish Keypad topic per entry: No
Publish Keypad codes: No
Allow checking Keypad codes: No
Max keypad entries to retrieve: 10
Publish timecontrol info: No
Keypad query interval (s): 1800
Enable timecontrol control: Yes
Publish timecontrol topic per entry: No
Max timecontrol entries to retrieve: 10
Enable authorization control: No
Publish authorization topic per entry: No
Max authorization entries to retrieve: 5

------------ HOME ASSISTANT ------------
Home Assistant auto discovery enabled: Yes
Home Assistant auto discovery topic: homeassistant/
Nuki Hub configuration URL for HA: http://192.168.7.199
Nuki Hub ID: 79592324065380

------------ NUKI LOCK ------------
Lock enabled: Yes
Lock Ultra enabled: No
Paired: Yes
Nuki Hub device ID: 774882908
Nuki device ID: ***
Firmware version: 4.4.9
Hardware version: 8.3
Valid PIN set: Yes
Has door sensor: No
Has keypad: No
Timecontrol highest entries count: 0
Authorizations highest entries count: 0
Register as: App

------------ HYBRID MODE ------------
Hybrid mode enabled: Yes
Hybrid mode connected: Yes
Reboot Nuki lock on official MQTT failure: Yes
Sending actions through official MQTT enabled: Yes
Retry actions through BLE enabled: Yes
Time between status updates when official MQTT is offline (s): 600
Force Lock ID: No
Force Lock Keypad: No
Force Lock Doorsensor: No

------------ NUKI LOCK ACL ------------
Lock: Allowed
Unlock: Allowed
Unlatch: Allowed
Lock N Go: Allowed
Lock N Go Unlatch: Allowed
Full Lock: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed

------------ NUKI LOCK CONFIG ACL ------------
Name: Allowed
Latitude: Allowed
Longitude: Allowed
Auto Unlatch: Allowed
Pairing enabled: Allowed
Button enabled: Allowed
LED flash enabled: Allowed
LED brightness: Allowed
Timezone offset: Allowed
DST mode: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed
Single Lock: Allowed
Advertising Mode: Allowed
Timezone ID: Allowed
Unlocked Position Offset Degrees: Allowed
Locked Position Offset Degrees: Allowed
Single Locked Position Offset Degrees: Allowed
Unlocked To Locked Transition Offset Degrees: Allowed
Lock n Go timeout: Allowed
Single button press action: Allowed
Double button press action: Allowed
Detached cylinder: Allowed
Battery type: Allowed
Automatic battery type detection: Allowed
Unlatch duration: Allowed
Auto lock timeout: Allowed
Auto unlock disabled: Allowed
Nightmode enabled: Allowed
Nightmode start time: Allowed
Nightmode end time: Allowed
Nightmode auto lock enabled: Allowed
Nightmode auto unlock disabled: Allowed
Nightmode immediate lock on start: Allowed
Auto lock enabled: Allowed
Immediate auto lock enabled: Allowed
Auto update enabled: Allowed
Reboot Nuki: Allowed
Motor speed: Allowed
Enable slow speed during nightmode: Allowed

------------ NUKI OPENER ------------
Opener enabled: No

------------ GPIO ------------

Retain Input GPIO MQTT state: No
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Just boot nukihub as always, but with version 9.09. It didn't happen with 9.08.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
To keep the connection to the MQTT broker

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

[image]
The UI shows MQTT connected, but the info says it's not, and it's "unavailable" in Home Assistant

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
The MQTT broker is mosquitto on a Debian stable server, and there's a bunch of other stuff using it. Among these, notably, a Nuki V4 lock, which I can still control through the official MQTT over thread.

**(Please, remember to close the issue when the problem has been addressed)**

Ra72xx, 2025-03-07:
Same here. I had the problem already with previous versions, but with 9.09 the hub is offline for homeassistant most of the time. Mosquitto log shows very often messages like `Client nukihub has exceeded timeout, disconnecting.` .

h1ghrise, 2025-03-07:
Same here. Had to revert back to 9.08

technyon, 2025-03-08:
At this point I really have no idea what could be causing this. Can you get serial logs while this is happening?

bcutter, 2025-03-08:
Thanks for filing this issue. That way one can skip this 9.09 release or revert to 9.08. So far nearly every 9.x release had similar surprises unfortunately.

Did you all have issues with previous versions too (just not that excessively)?

technyon, 2025-03-09:
At this point it's a guessing game about what's happening. As a start, I'd need serial logs.

ssorgatem, 2025-03-09:
> At this point it's a guessing game about what's happening. As a start, I'd need serial logs.

All I seem to get from the serial is this:

```
ESP-ROM:esp32c3-api1-20210207␍␊
Build:Feb  7 2021␍␊
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)␍␊
Saved PC:0x40382388␍␊
SPIWP:0xee␍␊
mode:DIO, clock div:1␍␊
load:0x3fcd5820,len:0x1024␍␊
load:0x403cc710,len:0x968␍␊
load:0x403ce710,len:0x2cc4␍␊
entry 0x403cc710␍␊
```

and nothing more until I reboot and I get the same

technyon, 2025-03-10:
So after this there's no more serial output? This is from serial, not MQTT logs?

ssorgatem, 2025-03-10:
Yes, this is from serial, USB directly plugged into my PC and reading with cutecom.

Tried enabling all kinds of debug logs in the advanced settings, and rebooting, and nothing changed

technyon, 2025-03-10:
Something is wrong here, with this kind of output there shouldn't be a web UI. At the very least you should get regular messages that the Nuki and Network task are running. Without the latter, there's no web interface.

ssorgatem, 2025-03-10:
I flashed the 9.08 binary from releases, and looked at the serial output through minicom (just in case it was a cutcom problem), and this is all the output I get:
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40382388
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x1024
load:0x403cc710,len:0x968
load:0x403ce710,len:0x2cc4
entry 0x403cc710
```
Looks like the only serial output is from the bootloader. Nukihub 9.08 works without problem... but there is nothing logged to the serial.

ssorgatem, 2025-03-10:
I'm trying to bisect this but I'm having trouble building some of the commits with this error:

```
*** [.pio/build/esp32-c3/.pio/build/esp32-c3/x509_crt_bundle.S.o] Source `.pio/build/esp32-c3/x509_crt_bundle.S' not found, needed by target `.pio/build/esp32-c3/.pio/build/esp32-c3/x509_crt_bundle.S.o'.

```

ssorgatem, 2025-03-10:
The aforementioned error starts appearing after commit 59a5f02 ("Rename root CA bundle", 2025-02-16).
The commit right before that builds and it works.

ssorgatem, 2025-03-11:
I built 9.09 myself and... it kinda works. It reboots itself every 6-7 minutes with a `ESP_RST_PANIC: Software reset due to exception/panic.` but other than it works just fine, unlike the binary downloaded from releases.

I still can't get anything useful from the serial though.

Ra72xx, 2025-03-14:
Anything one can do to help solve this problem?
I reverted to 9.07, which works flawlessly, and then to 9.08. 9.08 works, but strangely after a restart of Homeassistant Nukihub has always to be restarted, too, so that HA recognizes it again. So there seems to be some kind of MQTT problem also with 9.08.

technyon, 2025-03-14:
I don't see any obvious changes regarding MQTT between 9.08 and 9.09, at this point I'm out of ideas, especially since there are no serial logs.

@ssorgatem You're using 115200 baud to get the serial logs?

ssorgatem, 2025-03-14:
Yes, and all I can see is bootloader stuff.

More puzzling still, I rebuilt 9.09 with no changes to code at all, and it now works. After a day or so it has stopped rebooting and it just works. 

So maybe the issue is not in the source code but on the build itself...?

technyon, 2025-03-14:
It sounds quite unlikely to me, unless you changed something about the build configuration. You could post your binary here for other users to test.

iranl, 2025-03-14:
Imho a very bad idea to suggest users try a binary from an untrusted source.

technyon, 2025-03-16:
@ssorgatem I'd think the problem is somewhere else anyway. Maybe just start over, factory reset your ESP and flash the official Nuki Hub binary. You're using the C3 which is a single core chip, maybe it's worth trying an S3 preferrably with PSRAM. Sorry I can't give you more specific help, but without proper serial logs it's impossible to tell.

Ra72xx, 2025-03-30:
I never managed to get my old ESP working properly again, so I upgraded to an ESP32-S3 with more RAM and the new one runs without problems. For me this bug can be closed.

iranl, 2025-03-30:
Problem is probably due to stack issues occuring in certain situations and configurations (Enabling Home assistant discovery puts strain on the MQTT client for example) so getting a device with PSRAM is likely to solve this.

I'm looking at some changes for 9.10 that might help for devices without PSRAM.

swann05, 2025-04-02:
Hello, I think I have the same problem. I upgraded from 9.05 to 9.09 the 26th of march. Surprisingly, it happen only today. The lock is fine and in the ui too but there is no contact between the hub and the lock. Everything is down in home assistant.
I use a esp32 from azdelivery wroom type. I think it's dual core?
If I restart the esp by unplugging it works again for some hours but then it's down again.

iranl, 2025-04-02:
> > getting a device with PSRAM is likely to solve this.
> 
> If hardware being capable to run NH for years would need to be replaced because of a software update which introduced an issue... that would in general not be great. Just sayin.

Your Pentium II runs Windows 11?

Adding more and more features to NukiHub stretches older hardware to its limits, as does ESP-IDF/Arduino moving from 4.4 to 5.4 and 2.0.0 to 3.2.0.  While we are continuouslyworking on keeping the stack usage in check this is getting harder and harder as NukiHub gains new features and keeps up to date with the latest ESP-IDF and Arduino version. If some more recent features or functionality doesn't work bceause of limits of your chosen hardware you can either stay on the older version of NH that works or accept that newer versions of NH will require more capable hardware and upgrade to a recommended device.

bcutter, 2025-04-02:
Interesting to read what is done in the backend. I only noticed features getting removed like presence tracking with v9.0.

Staying on well-working hardware and not updating usually is not working for long; once a dependency thing pops up (HA/MQTT/Nuki device firmware update/whatever), one probably needs to upgrade NH sooner or later - it's part of an ecosystem, not working fully independently.

Why can't we check for physical limitations (is it really the RAM? Did we track down this here without a doubt?) and work around those like "if critical value X is reached, we'll automatically reboot NH“? That seems to be what works for all here, besides the fact a hard reset is needed after exceeding a certain point.

Thanks for keeping up the good work.

iranl, 2025-04-02:
> Interesting to read what is done in the backend. I only noticed features getting removed like presence tracking with v9.0.

Our [release notes](https://github.com/technyon/nuki_hub/releases) beg to differ about only features getting removed

> Staying on well-working hardware and not updating usually is not working for long; once a dependency thing pops up (HA/MQTT/Nuki device firmware update/whatever), one probably needs to upgrade NH sooner or later - it's part of an ecosystem, not working fully independently.

Fully agree. And part of this is moving to newer versions of ESP-IDF and Arduino (which you can compare to OS upgrades) which adds support for things such as WPA3. But all these additions and features (either in NH or the frameworks) eat stack. WPA3 alone adds 1k of RAM usage.
 
> Why can't we check for physical limitations (is it really the RAM? Did we track down this here without a doubt?) and work around those like "if critical value X is reached, we'll automatically [...]

Issues such as this, that seemingly occur at random and don't occur on PSRAM enabled devices are very likely RAM related. The fact that the MQTT library thinks it is still connected to the MQTT server also points to this. It is hard to react to such issues as reconnect code is not triggered when there is no signal the MQTT connection has dropped. It is also pretty hard to set a hard number at which time you would trigger a reboot as there does seem to be enough stack available looking at the numbers from the info page.

I have just pushed a PR that improves available RAM by about 33% (25kb to 50kb depending on device model). Hopefully this will help improve the situation regarding RAM. It will need rigorous testing before a release though.

iranl, 2025-04-08:
Please report if this is fixed with the release of 9.10

iranl, 2025-04-15:
Closing due to inactivity

closed 2025-04-15 by iranl

bcutter, 2025-04-15:
> Closing due to inactivity

Maybe because updating to 9.10 needs a flash - again ;-)

Hopefully @ssorgatem will reply once 9.10 was tested.