technyon/nuki_hub #639: Nukihub + Home Assistant - central manage keypad codes?

vsatmydynipnet, 2025-03-10:
I have a Homeassistant  / MQTT / Nuki_Hub setup runnig and basically this works pretty well.

The final setup should be 4 locks with keypads and access codes for 4 people and some temporary changing people.

I saw your not in regards of the JSON stuff to send out codes to the locks, but I imagine my wife, when I am not at home, changing the codes for temporary access, maybe only to 2 of the 4 locks. I assume this goes wrong. :-)

Is there any WebUI to Nuki_Hub or some Addon to Home Assistant where non techies are able to centrally manage the keys for the keypads/locks?

Thank you very much for any hint.

technyon, 2025-03-10:
Hi, Nuki Hub has no web interface to manage keys, it only exposes the Nuki Bluetooth API via MQTT. About HA someone else has to answer, since I don't use it.

iranl, 2025-03-10:
Not an issue, moving to discussions