technyon/nuki_hub #642: [Feature] Nuki Smart Lock Pro 5th gen support

iranl, 2025-03-11:
We are looking for owners of the new [Nuki Smart Lock Pro 5th gen](https://nuki.io/products/smart-lock-pro-5th-gen) that can confirm if NukiHub works with the new device as expected.

Testing will include confirming:
- NukiHub pairs with the device correctly
- NukiHub can query the correct device lock state
- NukiHub can query additional parameters of the device state (auth log, etc.)
- NukiHub can change the lock state as expected
- NukiHub can query and change all supported device settings
- NukiHub can query and change keypad settings and codes for keypads connected through the device
- NukiHub can query and change authorization settings
- NukiHub can query and change time control settings

If you use NukiHub with the Nuki Smart Lock Pro 5th gen please let us know if any or all of the above works as expected or not in this issue.

We expect that the instructions for pairing with the Smart Lock Ultra need to be followed: https://github.com/technyon/nuki_hub#pairing-with-a-nuki-lock-ultra

leonverhoeven, 2025-03-25:
Hi, 

I installed the Nuki Smart Lock Pro 5th Gen with door sensor and keypad pro yesterday. Had a little trouble getting the WiFi connected. But after that connected successfully, the device works as expected. 
I installed NukiHub on a ESP32-WROOM-32 (my first ESP usage). Looks like everything is working. Settings are the same as the Ultra.
MQTT-ing to the MQTT-Broker of Home Assistant (not Zigbee2MQTT). 

Tested so far: 

- [x] NukiHub pairs with the device correctly
- [x] NukiHub can query the correct device lock state
- [x] NukiHub can query additional parameters of the device state (auth log, etc.)
- [x] NukiHub can change the lock state as expected
- [ ] NukiHub can query and change all supported device settings
- [x] NukiHub can query and change keypad settings and codes for keypads connected through the device
- [x] NukiHub can query and change authorization settings
- [x] NukiHub can query and change time control settings

I haven't tested everything yet. But it looks like, it's fine!

Thanks for providing us with this fine piece of software. Used it a couple of hours so far, and i am very enthusiastic about it! Great works!

technyon, 2025-03-26:
@leonverhoeven Thanks for your feedback

PR #650 (2025-04-01): Various improvements and fixes
## Description:

- Update Arduino to 3.2.0 and ESP-IDF to 5.4
- Update esp-nimble-cpp to lastest master
- Create P4 builds
- Add support for 5th gen (Go/Pro) locks
- Fix OTA updating because of changed GitHub SSL certificates. Include entire CA root certificates bundle from now on.
- Update partition table to accommodate including the CA bundle in the updater
- Add LilyGO T-ETH-Lite-ESP32S3 to Ethernet devices
- Fix TOTP setup
- Generate unique hostnames based on MAC to prevent collision in MQTT client id
- Allow MQTT client to use internal task in PSRAM devices
 [... cut]
closed 2025-04-01 by iranl