technyon/nuki_hub #643: [Feature] Nuki Smart Lock Go support

iranl, 2025-03-11:
We are looking for owners of the new [Nuki Smart Lock Go](https://nuki.io/products/smart-lock-go) that can confirm if NukiHub works with the new device as expected.

Testing will include confirming:
- NukiHub pairs with the device correctly
- NukiHub can query the correct device lock state
- NukiHub can query additional parameters of the device state (auth log, etc.)
- NukiHub can change the lock state as expected
- NukiHub can query and change all supported device settings
- NukiHub can query and change keypad settings and codes for keypads connected through the device
- NukiHub can query and change authorization settings
- NukiHub can query and change time control settings

If you use NukiHub with the Nuki Smart Lock Go please let us know if any or all of the above works as expected or not in this issue.

We expect that the instructions for pairing with the regular Smart Lock need to be followed (as the Go basically looks like a rebranded non-pro 4.0 lock): https://github.com/technyon/nuki_hub#pairing-with-a-nuki-lock-10-40-or-opener

But if this does not work as expected it might be needed to follow the instructions for the Smart Lock Ultra: https://github.com/technyon/nuki_hub#pairing-with-a-nuki-lock-ultra

JakobRauch, 2025-03-27:
Hey! I got my Nuki Go today and would like to contribute to this wonderful project.

##  NukiHub pairs with the device correctly
Yes, pairing worked after setting the correct PIN.
````
------------ NUKI LOCK ------------
Lock enabled: Yes
Lock Ultra enabled: Yes
Paired: Yes
Nuki Hub device ID: 3672009111
Nuki device ID: ***
Firmware version: 5.2.2
Hardware version: 9.0
Valid PIN set: Yes
Has door sensor: No
Has keypad: No
Timecontrol highest entries count: 0
Authorizations highest entries count: 0
Register as: Bridge
````

## NukiHub can query the correct device lock state
Yes, working like it should.

## NukiHub can query additional parameters of the device state (auth log, etc.)
I am not sure where I can find this information.

## NukiHub can change the lock state as expected
Yes.

Maybe I can find the time to test the other functions in the next days. Let me know if you need something specific.

iranl, 2025-03-27:
Thanks for confirming the Go works. Seeing as we already have full reports for the 5th gen pro and ultra I don't expect any suprises in  functionality for the Go and do not need any other information other than the following:

Your status page (`Lock Ultra enabled: Yes`) and mention of setting the PIN for pairing suggests you followed the instructions for the Smart Lock Ultra and not for the 4th gen and earlier devices. Can you confirm if your "Nuki configuration" settings look like this:

[image]

Or this:

[image]

JakobRauch, 2025-03-27:
> Your status page (`Lock Ultra enabled: Yes`) and mention of setting the PIN for pairing suggests you followed the instructions for the Smart Lock Ultra and not for the 4th gen and earlier devices. Can you confirm if your "Nuki configuration" settings look like this:

Yes, thats true. If I try to pair using the default config, I get these errors
````
[23:04:58]No pairing PIN code set, not pairing with Nuki SmartLock Ultra
[23:04:58]Nuki lock start pairing
[23:04:58]Pairing as bridge
[23:04:58]Error getting data from NVS
[23:04:58]No pairing PIN code set, not pairing with Nuki SmartLock Ultra
[23:04:58]No pairing PIN code set, not pairing with Nuki SmartLock Ultra
[23:04:58]No pairing PIN code set, not pairing with Nuki SmartLock Ultra
[23:04:58]No pairing PIN code set, not pairing with Nuki SmartLock Ultra
[23:04:59]No pairing PIN code set, not pairing with Nuki SmartLock Ultra
[23:04:59]No pairing PIN code set, not pairing with Nuki SmartLock Ultra
[23:04:59]No pairing PIN code set, not pairing with Nuki SmartLock Ultra
[23:05:00]No pairing PIN code set, not pairing with Nuki SmartLock Ultra
[23:05:00]MQTT Broker not configured, aborting connection attempt.
[23:05:00]No pairing PIN code set, not pairing with Nuki SmartLock Ultra
[23:05:00]Nuki lock start pairing
[23:05:00]Pairing as bridge
[23:05:00]Error getting data from NVS
[23:05:00]No pairing PIN code set, not pairing with Nuki SmartLock Ultra
[23:05:01]No pairing PIN code set, not pairing with Nuki SmartLock Ultra
[23:05:01]No pairing PIN code set, not pairing with Nuki SmartLock Ultra
[23:05:01]No pairing PIN code set, not pairing with Nuki SmartLock Ultra
[23:05:01]No pairing PIN code set, not pairing with Nuki SmartLock Ultra
[23:05:02]No pairing PIN code set, not pairing with Nuki SmartLock Ultra
````

After enabling `Nuki Smartlock Ultra enabled` and setting the `Nuki Lock PIN`, pairing works:

````
[23:08:02]Pairing as bridge
[23:08:02]Error getting data from NVS
[23:08:05]Nuki lock start pairing
[23:08:05]Pairing as bridge
[23:08:05]Error getting data from NVS
[23:08:05][NukiHub] Connect timeout 3000 ms
[23:08:07]MQTT Broker not configured, aborting connection attempt.
[23:08:09]Nuki paired
[23:08:09]Updating Lock state based on status, timer or query
[23:08:09]Querying lock state
[23:08:10]Result (attempt 1): locked
[23:08:10]Done querying lock state
[23:08:12]KeyTurnerStatusUpdated
[23:08:12]Updating Lock state based on status, timer or query
[23:08:12]Querying lock state
[23:08:12]Result (attempt 1): locked
[23:08:12]Done querying lock state
[23:08:13]MQTT Broker not configured, aborting connection attempt.
[23:08:15]KeyTurnerStatusUpdated
[23:08:15]Updating Lock state based on status, timer or query
[23:08:15]Querying lock state
[23:08:15]Result (attempt 1): locked
[23:08:15]Done querying lock state
[23:08:18]KeyTurnerStatusUpdated
[23:08:18]Updating Lock state based on status, timer or query
[23:08:18]Querying lock state
````

btw: Should I worry about those `Error getting data from NVS` log entries?

iranl, 2025-03-27:
Thanks for clearing this up. I'll update the README for 9.10 to reflect the Go requires the same setup as the Ultra and 5th gen pro.

No, you don't have to worry about the NVS errors, these are to be expected and are harmless.

iranl, 2025-03-30:
Follow up question: Have your purchased the remote access? If not are you able to setup WiFi at all? I'm yet to find many reviews of the Go or a manual of the device but the review I have seen suggests WiFi (and thus official MQTT API over WiFi) is not available without purchasing the remote access addon. Would seem a bit ridiculous if this is the case. 

Changing WiFi settings of Nuki devices using NukiHub is on my roadmap for one of the next releases and it would take priority if the Go does not allow WiFi setup without the remote access addon and we might be able to bypass this requirement.

JakobRauch, 2025-03-30:
The app allows me to activate Wi-Fi for security updates, but remote control is not possible without buying remote access for 49 €. I have not purchased it.

PR #650 (2025-04-01): Various improvements and fixes
## Description:

- Update Arduino to 3.2.0 and ESP-IDF to 5.4
- Update esp-nimble-cpp to lastest master
- Create P4 builds
- Add support for 5th gen (Go/Pro) locks
- Fix OTA updating because of changed GitHub SSL certificates. Include entire CA root certificates bundle from now on.
- Update partition table to accommodate including the CA bundle in the updater
- Add LilyGO T-ETH-Lite-ESP32S3 to Ethernet devices
- Fix TOTP setup
- Generate unique hostnames based on MAC to prevent collision in MQTT client id
- Allow MQTT client to use internal task in PSRAM devices
 [... cut]