technyon/nuki_hub #646: Error upgrading from 9.08 to 9.09

flashbackpa, 2025-03-17:
### PROBLEM DESCRIPTION
Auto upgrade fails with a "Confirm code is invalid" message.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [ ] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [ ] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [ ] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [ ] Device used (e.g., ESP32-S3): ESP32_____
- [ ] Nuki Hub binary firmware version number used: _____
  - [X ] Pre-compiled
  - [ ] Self-compiled
- [ ] Provide the output of http://nukihubIP/get?page=info:
```
------------ NUKI HUB ------------
Device: ESP32
Version: 9.08
Build: 12974430884.10.1
Build type: Release
Build date: 2025-01-26
Updater version: 9.08
Updater build: 12974430884.10.1
Updater build date: 2025-01-23
Uptime (min): 328
Config version: 908
Last restart reason FW: OTAAborted
Last restart reason ESP: ESP_RST_SW: Software reset via esp_restart.
Free internal heap: 66320
Total internal heap: 246356
PSRAM Available: No
Network task stack high watermark: 8464
Nuki task stack high watermark: 5012

------------ GENERAL SETTINGS ------------
Network task stack size: 12288
Nuki task stack size: 8192
Check for updates: Yes
Latest version: 9.09
Allow update from MQTT: Yes
Update Nuki Hub and Nuki devices time using NTP: No
Web configurator username: Not set
Web configurator password: Not set
Web configurator bypass for proxy IP: Not set
Web configurator authentication: Basic
Session validity (in seconds): 3600
Session validity remember (in hours): 720
Duo Push MFA enabled: No
Web configurator enabled: Yes
HTTP SSL: Disabled
Advanced menu enabled: No
Publish free heap over MQTT: No
Nuki connect debug logging enabled: No
Nuki communication debug logging enabled: No
Nuki readable data debug logging enabled: No
Nuki hex data debug logging enabled: No
Nuki command debug logging enabled: No
MQTT log enabled: No
Webserial enabled: No
Bootloop protection enabled: No

------------ NETWORK ------------
Network device: Built-in Wi-Fi
Network connected: Yes
IP Address: 192.168.1.13
SSID: wifispoofer
BSSID of AP: 78:D6:D6:41:F6:45
ESP32 MAC address: A0:B7:65:49:99:F4

------------ NETWORK SETTINGS ------------
Nuki Hub hostname: nukihub
DHCP enabled: Yes
RSSI Publish interval (s): 60
Find WiFi AP with strongest signal: No
Restart ESP32 on network disconnect enabled: Yes
Disable Network if not connected within 60s: No
MQTT Timeout until restart (s): 60

------------ MQTT ------------
MQTT connected: Yes
MQTT broker address: 192.168.1.6
MQTT broker port: 1883
MQTT username: Not set
MQTT password: Not set
MQTT base topic: home/nuki
MQTT SSL: Disabled

------------ BLUETOOTH ------------
Bluetooth connection mode: New
Bluetooth TX power (dB): 9
Bluetooth command nr of retries: 3
Bluetooth command retry delay (ms): 100
Seconds until reboot when no BLE beacons received: 600

------------ QUERY / PUBLISH SETTINGS ------------
Lock/Opener state query interval (s): 1800
Publish Nuki device authorization log: No
Max authorization log entries to retrieve: 5
Battery state query interval (s): 3600
Most non-JSON MQTT topics disabled: No
Publish Nuki device config: Yes
Config query interval (s): 3600
Publish Keypad info: No
Keypad query interval (s): 1800
Enable Keypad control: Yes
Publish Keypad topic per entry: No
Publish Keypad codes: No
Allow checking Keypad codes: No
Max keypad entries to retrieve: 10
Publish timecontrol info: Yes
Keypad query interval (s): 1800
Enable timecontrol control: No
Publish timecontrol topic per entry: No
Max timecontrol entries to retrieve: 10
Enable authorization control: No
Publish authorization topic per entry: No
Max authorization entries to retrieve: 10

------------ HOME ASSISTANT ------------
Home Assistant auto discovery enabled: Yes
Home Assistant auto discovery topic: home/nuki/
Nuki Hub configuration URL for HA: http://192.168.1.13
Nuki Hub ID: 268939198576544

------------ NUKI LOCK ------------
Lock enabled: Yes
Lock Ultra enabled: No
Paired: Yes
Nuki Hub device ID: 3282602332
Nuki device ID: ***
Firmware version: 4.5.5
Hardware version: 8.3
Valid PIN set: No
Has door sensor: No
Has keypad: No
Timecontrol highest entries count: 0
Authorizations highest entries count: 0
Register as: Bridge

------------ HYBRID MODE ------------
Hybrid mode enabled: No
Force Lock ID: No
Force Lock Keypad: No
Force Lock Doorsensor: No

------------ NUKI LOCK ACL ------------
Lock: Allowed
Unlock: Allowed
Unlatch: Allowed
Lock N Go: Allowed
Lock N Go Unlatch: Allowed
Full Lock: Allowed
Fob Action 1: Disallowed
Fob Action 2: Disallowed
Fob Action 3: Disallowed

------------ NUKI LOCK CONFIG ACL ------------
Name: Allowed
Latitude: Allowed
Longitude: Allowed
Auto Unlatch: Allowed
Pairing enabled: Allowed
Button enabled: Allowed
LED flash enabled: Allowed
LED brightness: Allowed
Timezone offset: Allowed
DST mode: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed
Single Lock: Allowed
Advertising Mode: Allowed
Timezone ID: Allowed
Unlocked Position Offset Degrees: Allowed
Locked Position Offset Degrees: Allowed
Single Locked Position Offset Degrees: Allowed
Unlocked To Locked Transition Offset Degrees: Allowed
Lock n Go timeout: Allowed
Single button press action: Allowed
Double button press action: Allowed
Detached cylinder: Allowed
Battery type: Allowed
Automatic battery type detection: Allowed
Unlatch duration: Allowed
Auto lock timeout: Allowed
Auto unlock disabled: Allowed
Nightmode enabled: Allowed
Nightmode start time: Allowed
Nightmode end time: Allowed
Nightmode auto lock enabled: Allowed
Nightmode auto unlock disabled: Allowed
Nightmode immediate lock on start: Allowed
Auto lock enabled: Allowed
Immediate auto lock enabled: Allowed
Auto update enabled: Allowed
Reboot Nuki: Allowed
Motor speed: Allowed
Enable slow speed during nightmode: Allowed

------------ NUKI OPENER ------------
Opener enabled: No

------------ GPIO ------------

Retain Input GPIO MQTT state: No
```

### TO REPRODUCE
Click on Latest Firmware and then Update to latest release

### EXPECTED BEHAVIOUR
Update to last available release

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Dlay2017, 2025-03-24:
Same here. . clicking on OTA Update, ESP reboots to update mode, but afterwards 9.08 is still installed.

Ra72xx, 2025-03-25:
FWIW: To recover from #637, I also tried to update again from 9.08 to 9.09. This didn't work with the integrated ota updater, I had to flash over USB (and in the process, all pairing info was lost although I had the right kind of backup). However, it seems as if #637 might be solved for me now, I'll wait and see.

iranl, 2025-03-25:
OTA updating is broken due to Github switching intermediate SSL certificate.

This will be fixed in 9.10 which will require a manual flash because all current releases of NukiHub will not accept the current GitHub SSL certificate.

iranl, 2025-03-26:
Disabling SSL check on OTA can only be changed with compile flags so this is not as simple as a checkmark on the OTA page.
I'll see if including the common SSL bundle is feasable, with this NukiHub should be able to connect over SSL with 99% of domains.

If this isn't feasable we could disable SSL check for OTA using compile flags and do a check of our own to determine if the cert is valid and only start the update process if it is (with an option to bypass this check somewhere to prevent the need of a manual flash in cases such as this). But imho this is a less preferable option.

pilot1981, 2025-03-29:
what it happen now using OTA update? it fail and thaen it run again previous firmware or it fails to boots? I tryed to upgrade to lastest version using OTA, but now ESP don't connect again to WIFI network...

technyon, 2025-03-30:
Yes it should fall back to the previously flashed firmware.

PR #650 (2025-04-01): Various improvements and fixes
## Description:

- Update Arduino to 3.2.0 and ESP-IDF to 5.4
- Update esp-nimble-cpp to lastest master
- Create P4 builds
- Add support for 5th gen (Go/Pro) locks
- Fix OTA updating because of changed GitHub SSL certificates. Include entire CA root certificates bundle from now on.
- Update partition table to accommodate including the CA bundle in the updater
- Add LilyGO T-ETH-Lite-ESP32S3 to Ethernet devices
- Fix TOTP setup
- Generate unique hostnames based on MAC to prevent collision in MQTT client id
- Allow MQTT client to use internal task in PSRAM devices
 [... cut]
closed 2025-04-01 by iranl