technyon/nuki_hub #647: T-ETH-Lite-ESP32S3 with POE Shield not connecting via ethernet

jo-ket, 2025-03-24:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I installed the latest firmware (NukiHub-9.09-ESP32S3OCT) on two LilyGO-T-ETH-Lite-ESP32S3 with PoE shield. Both show the same behavior. Connection via Wifi works well. No connection via ethernet. Network settings were changed according to readme. Tried different settings under network hardware (LilyGO-T-ETH-POE, LilyGO-T Elite, Generic W5500.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [x] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [x] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [x] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [x] Device used (e.g., ESP32-S3): LilyGO-T-ETH-Lite-ESP32S3 with PoE shield
- [x] Nuki Hub binary firmware version number used: 9.09
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Provide the output of http://nukihubIP/get?page=info:
```
------------ NUKI HUB ------------
Device: ESP32-S3 (Octal PSRAM)
Version: 9.09
Build: 13675437614.11.1
Build type: Release
Build date: 2025-03-05
Updater version: 9.09
Updater build: 13675437614.11.1
Updater build date: 2025-03-03
Uptime (min): 12
Config version: 909
Last restart reason FW: NotApplicable
Last restart reason ESP: ESP_RST_POWERON: Reset due to power-on event.
Free internal heap: 138319
Total internal heap: 367967
PSRAM Available: Yes
Free usable PSRAM: 8341588
Total usable PSRAM: 8388608
Total PSRAM: 8388608
Total free heap: 8429944
Network task stack high watermark: 10140
Nuki task stack high watermark: 6464

------------ SPIFFS ------------
SPIFFS Total Bytes: 233681
SPIFFS Used Bytes: 0
SPIFFS Free Bytes: 233681

------------ GENERAL SETTINGS ------------
Network task stack size: 12288
Nuki task stack size: 8192
Check for updates: Yes
Latest version: 
Allow update from MQTT: No
Update Nuki Hub and Nuki devices time using NTP: No
Web configurator username: Not set
Web configurator password: Not set
Web configurator bypass for proxy IP: Not set
Web configurator authentication: Basic
Session validity (in seconds): 3600
Session validity remember (in hours): 720
Duo Push MFA enabled: No
TOTP MFA enabled: No
Web configurator enabled: Yes
HTTP SSL: Disabled
Nuki Hub FQDN for HTTP redirect: Not set
Advanced menu enabled: No
Publish free heap over MQTT: No
Nuki connect debug logging enabled: No
Nuki communication debug logging enabled: No
Nuki readable data debug logging enabled: No
Nuki hex data debug logging enabled: No
Nuki command debug logging enabled: No
MQTT log enabled: No
Webserial enabled: No
Bootloop protection enabled: No

------------ NETWORK ------------
Network device: Built-in Wi-Fi
Network connected: Yes
IP Address: 192.168.3.203
SSID: MyWifiNetwork
BSSID of AP: 7A:11:41:B1:AF:A6
ESP32 MAC address: 64:E8:00:47:CD:C8

------------ NETWORK SETTINGS ------------
Nuki Hub hostname: nukihub_keller
DHCP enabled: Yes
RSSI Publish interval (s): 60
Find WiFi AP with strongest signal: No
Restart ESP32 on network disconnect enabled: No
Disable Network if not connected within 60s: No
MQTT Timeout until restart (s): 60

------------ MQTT ------------
MQTT connected: No
MQTT broker address: 
MQTT broker port: 1883
MQTT username: Not set
MQTT password: Not set
MQTT base topic: nukihub
MQTT SSL: Disabled

------------ BLUETOOTH ------------
Bluetooth connection mode: New
Bluetooth TX power (dB): 9
Bluetooth command nr of retries: 3
Bluetooth command retry delay (ms): 100
Seconds until reboot when no BLE beacons received: 60

------------ QUERY / PUBLISH SETTINGS ------------
Lock/Opener state query interval (s): 1800
Publish Nuki device authorization log: No
Max authorization log entries to retrieve: 5
Battery state query interval (s): 1800
Most non-JSON MQTT topics disabled: No
Publish Nuki device config: Yes
Config query interval (s): 3600
Publish Keypad info: No
Keypad query interval (s): 1800
Enable Keypad control: Yes
Publish Keypad topic per entry: No
Publish Keypad codes: No
Allow checking Keypad codes: No
Max keypad entries to retrieve: 10
Publish timecontrol info: No
Keypad query interval (s): 1800
Enable timecontrol control: No
Publish timecontrol topic per entry: No
Max timecontrol entries to retrieve: 10
Enable authorization control: No
Publish authorization topic per entry: No
Max authorization entries to retrieve: 5

------------ HOME ASSISTANT ------------
Home Assistant auto discovery enabled: No

------------ NUKI LOCK ------------
Lock enabled: Yes
Lock Ultra enabled: No
Paired: No
Nuki Hub device ID: 2764633919
Nuki device ID: Not set
Firmware version: 
Hardware version: 
Valid PIN set: -
Has door sensor: No
Has keypad: No
Timecontrol highest entries count: 0
Authorizations highest entries count: 0
Register as: Bridge

------------ HYBRID MODE ------------
Hybrid mode enabled: No
Force Lock ID: No
Force Lock Keypad: No
Force Lock Doorsensor: No

------------ NUKI LOCK ACL ------------
Lock: Allowed
Unlock: Allowed
Unlatch: Allowed
Lock N Go: Allowed
Lock N Go Unlatch: Allowed
Full Lock: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed

------------ NUKI LOCK CONFIG ACL ------------
Name: Allowed
Latitude: Allowed
Longitude: Allowed
Auto Unlatch: Allowed
Pairing enabled: Allowed
Button enabled: Allowed
LED flash enabled: Allowed
LED brightness: Allowed
Timezone offset: Allowed
DST mode: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed
Single Lock: Allowed
Advertising Mode: Allowed
Timezone ID: Allowed
Unlocked Position Offset Degrees: Allowed
Locked Position Offset Degrees: Allowed
Single Locked Position Offset Degrees: Allowed
Unlocked To Locked Transition Offset Degrees: Allowed
Lock n Go timeout: Allowed
Single button press action: Allowed
Double button press action: Allowed
Detached cylinder: Allowed
Battery type: Allowed
Automatic battery type detection: Allowed
Unlatch duration: Allowed
Auto lock timeout: Allowed
Auto unlock disabled: Allowed
Nightmode enabled: Allowed
Nightmode start time: Allowed
Nightmode end time: Allowed
Nightmode auto lock enabled: Allowed
Nightmode auto unlock disabled: Allowed
Nightmode immediate lock on start: Allowed
Auto lock enabled: Allowed
Immediate auto lock enabled: Allowed
Auto update enabled: Allowed
Reboot Nuki: Allowed
Motor speed: Allowed
Enable slow speed during nightmode: Allowed

------------ NUKI OPENER ------------
Opener enabled: No

------------ GPIO ------------

Retain Input GPIO MQTT state: No
```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Ethernet adapter is used

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

iranl, 2025-03-24:
There currently is no predefined setting for the LilyGO T-ETH-Lite-ESP32S3.

I'll add in in 9.10 as we have it mentioned in the README as a supported device.

In 9.09 you will need to select "Custom LAN module" on the "Network configuration" page and the following on the "Custom Ethernet Configuration" page:

[image]

jo-ket, 2025-03-24:
Works perfectly well. Thanks for the quick-fix. Looking forward to 9.10. Should I leave the issue open until 9.10 or close it?

PR #650 (2025-04-01): Various improvements and fixes
## Description:

- Update Arduino to 3.2.0 and ESP-IDF to 5.4
- Update esp-nimble-cpp to lastest master
- Create P4 builds
- Add support for 5th gen (Go/Pro) locks
- Fix OTA updating because of changed GitHub SSL certificates. Include entire CA root certificates bundle from now on.
- Update partition table to accommodate including the CA bundle in the updater
- Add LilyGO T-ETH-Lite-ESP32S3 to Ethernet devices
- Fix TOTP setup
- Generate unique hostnames based on MAC to prevent collision in MQTT client id
- Allow MQTT client to use internal task in PSRAM devices
 [... cut]
closed 2025-04-01 by iranl