technyon/nuki_hub #648: Bootloop after power loss

Seestern93, 2025-03-28:
### PROBLEM DESCRIPTION
Since using nukihub i had always the problem that, after power loss nukihub starts with a bootloop which does not allow to access web page or use its functions. 
It is needed to reset the ESP (ESP-WROOM-32) over the RST button. After updating to latest version 9.09 the problem still exists, so I documented the boot loop with the web flashing tool which provides a logger. 
It seems that the core 0 is panic'ing

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [x] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [x] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [ ] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [ ] Device used (e.g., ESP32-S3): ESP-WROOM-32 (ESP32-C)
- [ ] Nuki Hub binary firmware version number used: 9.09
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Provide the output of http://nukihubIP/get?page=info:
System Information
------------ NUKI HUB ------------
Device: ESP32
Version: 9.09
Build: 13675437614.11.1
Build type: Release
Build date: 2025-03-05
Updater version: 
Updater build: 
Updater build date: 
Uptime (min): 297
Config version: 909
Last restart reason FW: NotApplicable
Last restart reason ESP: ESP_RST_PANIC: Software reset due to exception/panic.
Free internal heap: 55744
Total internal heap: 243624
PSRAM Available: No
Network task stack high watermark: 8400
Nuki task stack high watermark: 5140

------------ SPIFFS ------------
SPIFFS Total Bytes: 233681
SPIFFS Used Bytes: 51455
SPIFFS Free Bytes: 182226

------------ GENERAL SETTINGS ------------
Network task stack size: 12288
Nuki task stack size: 8192
Check for updates: Yes
Latest version: 
Allow update from MQTT: No
Update Nuki Hub and Nuki devices time using NTP: No
Web configurator username: Not set
Web configurator password: Not set
Web configurator bypass for proxy IP: Not set
Web configurator authentication: Basic
Session validity (in seconds): 3600
Session validity remember (in hours): 720
Duo Push MFA enabled: No
TOTP MFA enabled: No
Web configurator enabled: Yes
HTTP SSL: Disabled
Advanced menu enabled: No
Publish free heap over MQTT: No
Nuki connect debug logging enabled: No
Nuki communication debug logging enabled: No
Nuki readable data debug logging enabled: No
Nuki hex data debug logging enabled: No
Nuki command debug logging enabled: No
MQTT log enabled: No
Webserial enabled: No
Bootloop protection enabled: No

------------ NETWORK ------------
Network device: Built-in Wi-Fi
Network connected: Yes
IP Address: 10.69.0.138
SSID: Seestern-Homenet
BSSID of AP: 74:42:7F:13:3E:A2
ESP32 MAC address: 08:D1:F9:CB:D4:B0

------------ NETWORK SETTINGS ------------
Nuki Hub hostname: nukihub
DHCP enabled: Yes
RSSI Publish interval (s): 60
Find WiFi AP with strongest signal: Yes
Restart ESP32 on network disconnect enabled: No
Disable Network if not connected within 60s: No
MQTT Timeout until restart (s): 60

------------ MQTT ------------
MQTT connected: Yes
MQTT broker address: 10.69.0.10
MQTT broker port: 1883
MQTT username: ***
MQTT password: ***
MQTT base topic: nuki
MQTT SSL: Disabled

------------ BLUETOOTH ------------
Bluetooth connection mode: New
Bluetooth TX power (dB): 9
Bluetooth command nr of retries: 3
Bluetooth command retry delay (ms): 100
Seconds until reboot when no BLE beacons received: 60

------------ QUERY / PUBLISH SETTINGS ------------
Lock/Opener state query interval (s): 1800
Publish Nuki device authorization log: No
Max authorization log entries to retrieve: 5
Battery state query interval (s): 1800
Most non-JSON MQTT topics disabled: No
Publish Nuki device config: Yes
Config query interval (s): 3600
Publish Keypad info: No
Keypad query interval (s): 1800
Enable Keypad control: Yes
Publish Keypad topic per entry: No
Publish Keypad codes: No
Allow checking Keypad codes: No
Max keypad entries to retrieve: 10
Publish timecontrol info: No
Keypad query interval (s): 1800
Enable timecontrol control: No
Publish timecontrol topic per entry: No
Max timecontrol entries to retrieve: 10
Enable authorization control: No
Publish authorization topic per entry: No
Max authorization entries to retrieve: 5

------------ HOME ASSISTANT ------------
Home Assistant auto discovery enabled: No

------------ NUKI LOCK ------------
Lock enabled: Yes
Lock Ultra enabled: No
Paired: Yes
Nuki Hub device ID: 473277539
Nuki device ID: ***
Firmware version: 3.10.7
Hardware version: 6.11
Valid PIN set: No
Has door sensor: Yes
Has keypad: No
Timecontrol highest entries count: 0
Authorizations highest entries count: 0
Register as: Bridge

------------ HYBRID MODE ------------
Hybrid mode enabled: No
Force Lock ID: No
Force Lock Keypad: No
Force Lock Doorsensor: No

------------ NUKI LOCK ACL ------------
Lock: Allowed
Unlock: Allowed
Unlatch: Allowed
Lock N Go: Allowed
Lock N Go Unlatch: Allowed
Full Lock: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed

------------ NUKI LOCK CONFIG ACL ------------
Name: Allowed
Latitude: Allowed
Longitude: Allowed
Auto Unlatch: Allowed
Pairing enabled: Allowed
Button enabled: Allowed
LED flash enabled: Allowed
LED brightness: Allowed
Timezone offset: Allowed
DST mode: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed
Single Lock: Allowed
Advertising Mode: Allowed
Timezone ID: Allowed
Unlocked Position Offset Degrees: Allowed
Locked Position Offset Degrees: Allowed
Single Locked Position Offset Degrees: Allowed
Unlocked To Locked Transition Offset Degrees: Allowed
Lock n Go timeout: Allowed
Single button press action: Allowed
Double button press action: Allowed
Detached cylinder: Allowed
Battery type: Allowed
Automatic battery type detection: Allowed
Unlatch duration: Allowed
Auto lock timeout: Allowed
Auto unlock disabled: Allowed
Nightmode enabled: Allowed
Nightmode start time: Allowed
Nightmode end time: Allowed
Nightmode auto lock enabled: Allowed
Nightmode auto unlock disabled: Allowed
Nightmode immediate lock on start: Allowed
Auto lock enabled: Allowed
Immediate auto lock enabled: Allowed
Auto update enabled: Allowed
Reboot Nuki: Allowed
Motor speed: Allowed
Enable slow speed during nightmode: Allowed

------------ NUKI OPENER ------------
Opener enabled: No

------------ GPIO ------------

Retain Input GPIO MQTT state: No

### TO REPRODUCE
Steps to reproduce the behavior:
1.) Plugin Power
2.) Wait -> bootloop is happening
3.) to resolve: press RST button for soft reset

### EXPECTED BEHAVIOUR
Nukihub should start up without pressing a physical button after power loss

### SCREENSHOTS

### ADDITIONAL CONTEXT
I attach the log file from web installer ("https://technyon.github.io/nuki_hub/")
Short description of what happens in the log:

[18:30:58] --> start of log
[18:31:13] --> failed to connect to WiFi and waited 15s (dont know why, WiFi itself works and credentials are OK)
[18:31:24] --> second attempt to connect to WiFi works after 6s
[18:31:25] --> trying to connect to MQTT and immediate core panic
[18:31:25] --> Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
[18:31:35] --> after boot, again WiFi connection is tryed
[18:31:50] --> and failed at first attempt
[18:32:09] --> second WiFi connection attempt fails
[18:32:19] --> third attempt works after 5s
[18:32:20] --> trying to connect to MQTT and immediate core panic
[18:32:32] --> physical RST button is pressed, after that booting happens normally
[18:32:37] --> trying to connect to WiFi
[18:32:47] --> successfully after 10s (again - should work faster?)
[18:32:48] --> MQTT connection successfully

==> thoughts from my side: 
  why does WiFi connection takes so long? Signal is good (RSSI: -65(70.00 %) )
  it panic's directly at MQTT connection, but I dont know if that is maybe just co-incidence 
  when I do not press RST button it does not work and loops all day long, so it is not a random problem.. but still it seems to only affect my environment since that problem would have been known / solved for long before.

**(Please, remember to close the issue when the problem has been addressed)**

iranl, 2025-03-30:
This is probably an issue with your board and/or power supply as this is not an issue seen by others.

We can look take a look at the loadProhibited panic but you will have to post a coredump log (from the export page) and/or your complete serial logs when the panic happens.

Seestern93, 2025-03-31:
Hi,

Seems like I totally forgot to attach the log file 😆
Here it is: 

[esp-web-tools-logs.txt](https://github.com/user-attachments/files/19535349/esp-web-tools-logs.txt)

Regarding power supply: the issue occurs only after unplugging and replugging the board or if power loss occurs (which happens for example when I turn off the fuse of the house an turn it on again).
It does NOT occur during "normal" usage and for me that means it is not specific to bad power supply or low voltage.
Apart from that I tried with different usb chargers and also from usb port of my pc, same behavior.

Thanks for analyzing :) I tried to convert the core panic to human readable format but I am missing the a2l file since I did not built it on my own. So I cant read where in the program the core panic occurs.

iranl, 2025-04-01:
Your coredump signals issues with MDNS. Not sure why this would cause an issue after a panic but it was a change that was introduced between 9.08 and 9.09.

Because MDNS gave me other issues while developing 9.10 I have decided to disable it for now in 9.10.

This could very well fix your issues. Please try the 9.10-beta1 or wait for 9.10 final and report if this is fixed.

Note: 9.10 needs to be flashed using webflash or esptool.

Seestern93, 2025-04-03:
@iranl: Thanks for looking at the problem :) 
Since the update to 9.09 I did to check if it works better with that, causes now a lot of reboots every few minutes, I decided to update to 9.10 beta.
It wasnt that easy :D 
* because of #646 I couldnt update to beta version over OTA (OTA aborts with message that it could not download the file from server).
* I then tried to update over the nuki hub web flash tool -> but there you cannot choose version so it always updates to latest (=9.09)
* I tried with Espressif Flash download tool .exe as mentioned in .txt for first time flashing but that seems to be buggy since it does not start flashing
* finally I could flash with the esptool.py as CLI version :)

=> took me ~ 3h but now the nukihub runs with 9.10-beta2 and boots without problems :) it runs now for 1h without reboot, will check the next days if it runs without reboot.

iranl, 2025-04-08:
I'm going to mark this as completed with the release of 9.10

closed 2025-04-08 by iranl