technyon/nuki_hub #649: ESP don't connect to WIFI network after OTA upgrade to the lastest version

pilot1981, 2025-03-29:
### PROBLEM DESCRIPTION
I tryed to upgrade to lastest version using OTA, but now ESP don't connect again to WIFI network...

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ X] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [ X] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [ X] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [ X] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [ X] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [ X] Device used (e.g., ESP32-S3): ESP32
- [ X] Nuki Hub binary firmware version number used: 9.00
  - [ X] Pre-compiled
  - [ ] Self-compiled
- [ ] Provide the output of http://nukihubIP/get?page=info:
```
NA
```

### TO REPRODUCE
starting from 9.00 version, use OTA update to lastest version.

### EXPECTED BEHAVIOUR
it upgraded to lastest version

### SCREENSHOTS
NA

### ADDITIONAL CONTEXT
NA

**(Please, remember to close the issue when the problem has been addressed)**

technyon, 2025-03-30:
Do you get the access point from the ESP to configure Wifi?

If not please get serial logs during boot.

iranl, 2025-03-30:
OTA has seen many improvements since 9.00 including boot loop protections when OTA fails (which will always be the case untill we release 9.10 with changes to the SSL bundle).

The fix either way will be to update over serial/USB to 9.10 when it is released or 9.09 in the meantime (and again when 9.10 is released)

pilot1981, 2025-03-30:
ESP connected  again to AP WIFI after about 30 minutes…version 9.00 again…no updated!

iranl, 2025-03-30:
Well that is to be expected.

PR #650 (2025-04-01): Various improvements and fixes
## Description:

- Update Arduino to 3.2.0 and ESP-IDF to 5.4
- Update esp-nimble-cpp to lastest master
- Create P4 builds
- Add support for 5th gen (Go/Pro) locks
- Fix OTA updating because of changed GitHub SSL certificates. Include entire CA root certificates bundle from now on.
- Update partition table to accommodate including the CA bundle in the updater
- Add LilyGO T-ETH-Lite-ESP32S3 to Ethernet devices
- Fix TOTP setup
- Generate unique hostnames based on MAC to prevent collision in MQTT client id
- Allow MQTT client to use internal task in PSRAM devices
 [... cut]
closed 2025-04-01 by iranl