technyon/nuki_hub #666: Uptime decreased since 9.10

VDRainer, 2025-04-26:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

As almost always, not a big thing, but since 9.10 uptimes decreased, and just want to report.
I installed nukihub on a Waveshare ESP32-S3-POE-ETH on 20250309 and since then i had an uptime of 30 days, 21:11 without any problem.
After installing 9.10 on 20250409 i had uptimes from 3 hours to  max 3 days.
9.11-masterX did not fix this.

Measured with a HA automation that triggers on the uptime sensor.
```
20250409_214017 - Uptime 12:06
20250410_110356 - Uptime 13:21
20250412_083655 - Uptime 1 day, 21:26
20250412_152136 - Uptime 06:16
...
20250417_051123 - Uptime 2 days, 01:59
20250417_082828 - Uptime 03:15
20250417_122028 - Uptime 03:51
20250421_022045 - Uptime 3 days, 14:00
20250421_073856 - Uptime 05:15
...

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [x] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [x] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [x] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [x] Device used (e.g., ESP32-S3): _____
- [x] Nuki Hub binary firmware version number used: _____
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Provide the output of http://nukihubIP/get?page=info:
```
System Information

------------ NUKI HUB ------------
Device: ESP32-S3
Version: 9.11-master6
Build: 14555077456.342.1
Build type: Release
Build date: 2025-04-20
Updater version: 9.11-master6
Updater build: 14555077456.342.1
Updater build date: 2025-04-19
Uptime (min): 16
Config version: 911
Last restart reason FW: RequestedViaWebServer
Last restart reason ESP: ESP_RST_SW: Software reset via esp_restart.
Free internal heap: 166140
Total internal heap: 341784
PSRAM Available: No
Network task stack high watermark: 8112
Nuki task stack high watermark: 4864

------------ SPIFFS ------------
SPIFFS Total Bytes: 233681
SPIFFS Used Bytes: 59738
SPIFFS Free Bytes: 173943

------------ GENERAL SETTINGS ------------
Network task stack size: 12288
Nuki task stack size: 8192
Check for updates: No
Latest version: 9.11-master6
Allow update from MQTT: No
Update Nuki Hub and Nuki devices time using NTP: Yes
Web configurator username: ***
Web configurator password: ***
Web configurator bypass for proxy IP: Not set
Web configurator authentication: Basic
Session validity (in seconds): 3600
Session validity remember (in hours): 720
Duo Push MFA enabled: No
TOTP MFA enabled: No
Web configurator enabled: Yes
HTTP SSL: Disabled
Advanced menu enabled: No
Publish free heap over MQTT: No
Nuki connect debug logging enabled: No
Nuki communication debug logging enabled: No
Nuki readable data debug logging enabled: No
Nuki hex data debug logging enabled: No
Nuki command debug logging enabled: No
MQTT log enabled: No
Webserial enabled: No
Bootloop protection enabled: No

------------ NETWORK ------------
Network device: Waveshare ESP32-S3-ETH / ESP32-S3-ETH-POE
Network connected: Yes
IP Address: 192.168.77.42

------------ NETWORK SETTINGS ------------
Nuki Hub hostname: nukihub-eth
DHCP enabled: Yes
Restart ESP32 on network disconnect enabled: No
Disable Network if not connected within 60s: No
MQTT Timeout until restart (s): 60

------------ MQTT ------------
MQTT connected: Yes
MQTT broker address: 192.168.77.67
MQTT broker port: 1883
MQTT username: Not set
MQTT password: Not set
MQTT base topic: nuki
MQTT SSL: Disabled

------------ BLUETOOTH ------------
Bluetooth connection mode: New
Bluetooth TX power (dB): 9
Bluetooth command nr of retries: 3
Bluetooth command retry delay (ms): 100
Seconds until reboot when no BLE beacons received: -1

------------ QUERY / PUBLISH SETTINGS ------------
Lock/Opener state query interval (s): 1800
Publish Nuki device authorization log: No
Max authorization log entries to retrieve: 5
Battery state query interval (s): 1800
Most non-JSON MQTT topics disabled: No
Publish Nuki device config: Yes
Config query interval (s): 3600
Publish Keypad info: No
Keypad query interval (s): 1800
Enable Keypad control: Yes
Publish Keypad topic per entry: No
Publish Keypad codes: No
Allow checking Keypad codes: No
Max keypad entries to retrieve: 10
Publish timecontrol info: No
Keypad query interval (s): 1800
Enable timecontrol control: No
Publish timecontrol topic per entry: No
Max timecontrol entries to retrieve: 10
Enable authorization control: No
Publish authorization topic per entry: No
Max authorization entries to retrieve: 10

------------ HOME ASSISTANT ------------
Home Assistant auto discovery enabled: Yes
Home Assistant auto discovery topic: homeassistant/
Nuki Hub configuration URL for HA: http://192.168.77.42
Nuki Hub ID: 242704585308876

------------ NUKI LOCK ------------
Lock enabled: Yes
Lock Ultra/Go/5th gen enabled: No
Paired: Yes
Nuki Hub device ID: 1782990656
Nuki device ID: ***
Firmware version: 3.10.7
Hardware version: 4.10
Valid PIN set: Yes
Has door sensor: No
Has keypad: No
Timecontrol highest entries count: 0
Authorizations highest entries count: 0
Register as: Bridge

------------ HYBRID MODE ------------
Hybrid mode enabled: No
Force Lock ID: No
Force Lock Keypad: No
Force Lock Doorsensor: No

------------ NUKI LOCK ACL ------------
Lock: Allowed
Unlock: Allowed
Unlatch: Allowed
Lock N Go: Disallowed
Lock N Go Unlatch: Disallowed
Full Lock: Disallowed
Fob Action 1: Disallowed
Fob Action 2: Disallowed
Fob Action 3: Disallowed

------------ NUKI LOCK CONFIG ACL ------------
Name: Disallowed
Latitude: Disallowed
Longitude: Disallowed
Auto Unlatch: Disallowed
Pairing enabled: Disallowed
Button enabled: Disallowed
LED flash enabled: Disallowed
LED brightness: Disallowed
Timezone offset: Disallowed
DST mode: Disallowed
Fob Action 1: Disallowed
Fob Action 2: Disallowed
Fob Action 3: Disallowed
Single Lock: Disallowed
Advertising Mode: Disallowed
Timezone ID: Disallowed
Unlocked Position Offset Degrees: Disallowed
Locked Position Offset Degrees: Disallowed
Single Locked Position Offset Degrees: Disallowed
Unlocked To Locked Transition Offset Degrees: Disallowed
Lock n Go timeout: Disallowed
Single button press action: Disallowed
Double button press action: Disallowed
Detached cylinder: Disallowed
Battery type: Disallowed
Automatic battery type detection: Allowed
Unlatch duration: Disallowed
Auto lock timeout: Disallowed
Auto unlock disabled: Disallowed
Nightmode enabled: Disallowed
Nightmode start time: Disallowed
Nightmode end time: Disallowed
Nightmode auto lock enabled: Disallowed
Nightmode auto unlock disabled: Disallowed
Nightmode immediate lock on start: Disallowed
Auto lock enabled: Disallowed
Immediate auto lock enabled: Disallowed
Auto update enabled: Disallowed
Reboot Nuki: Allowed
Motor speed: Allowed
Enable slow speed during nightmode: Allowed

------------ NUKI OPENER ------------
Opener enabled: No

------------ GPIO ------------

Retain Input GPIO MQTT state: YesGPIO 1 : General input (Pull-up)

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Here a coredump from the last restart

[coredump.txt](https://github.com/user-attachments/files/19921073/coredump.txt)

**(Please, remember to close the issue when the problem has been addressed)**

technyon, 2025-04-26:
For me the opposite is true, I have an uptime of almost 15000 minutes now.

VDRainer, 2025-04-27:
Found https://github.com/technyon/nuki_hub#memory-constraints in the docs and switched to 'the other S3 binary'.
Now i also see PSRAM on the info page.
```
PSRAM Available: Yes
Free usable PSRAM: 8344760
Total usable PSRAM: 8388608
Total PSRAM: 8388608
```
Sorry for not reading TFM. 😄 
Will report back and close this in a few days if this was the culprit.

VDRainer, 2025-05-04:
I wanted to close this issue today, but unfortunately I just noticed that the nukihub restarted after 6 days, 18:59. :cry:
```
------------ NUKI HUB ------------
Device: ESP32-S3 (Octal PSRAM)
Version: 9.10
Build: 14318502393.12.1
Build type: Release
Build date: 2025-04-07
Updater version: 9.10
Updater build: 14318502393.12.1
Updater build date: 2025-04-06
Uptime (min): 241
Config version: 911
Last restart reason FW: NotApplicable
Last restart reason ESP: ESP_RST_PANIC: Software reset due to exception/panic.
Free internal heap: 206635
Total internal heap: 392807
PSRAM Available: Yes
Free usable PSRAM: 8344760
Total usable PSRAM: 8388608
Total PSRAM: 8388608
Total free heap: 8501432
Network task stack high watermark: 8324
Nuki task stack high watermark: 4868
```
I think we have to live with it?

Here another coredump:

[coredump.txt](https://github.com/user-attachments/files/20027069/coredump.txt)

Seestern93, 2025-05-04:
I have similar issue again - but for me it restarts due to BLEException.. every few hours..

[coredump.txt (1).txt](https://github.com/user-attachments/files/20027890/coredump.txt.1.txt)

Itsmadhatter, 2025-05-08:
Hello, my ESP32-S3 also crashes every few hours with the BLEError. Previously, I used an ESP32 together with the Opener, and everything worked fine. The last message in the MQTT log before every crash is always: "Unable to subscribe to USDIO characteristic."

```
System Information
------------ NUKI HUB ------------
Device: ESP32-S3 (Octal PSRAM)
Version: 9.10
Build: 14318502393.12.1
Build type: Release
Build date: 2025-04-07
Updater version: 
Updater build: 
Updater build date: 
Uptime (min): 98
Config version: 910
Last restart reason FW: BLEError
Last restart reason ESP: ESP_RST_SW: Software reset via esp_restart.
Free internal heap: 182591
Total internal heap: 392807
PSRAM Available: Yes
Free usable PSRAM: 8337264
Total usable PSRAM: 8388608
Total PSRAM: 8388608
Total free heap: 8469892
Network task stack high watermark: 6836
Nuki task stack high watermark: 4868

------------ SPIFFS ------------
SPIFFS Total Bytes: 233681
SPIFFS Used Bytes: 0
SPIFFS Free Bytes: 233681

------------ GENERAL SETTINGS ------------
Network task stack size: 12288
Nuki task stack size: 8192
Check for updates: Yes
Latest version: 9.10
Allow update from MQTT: Yes
Update Nuki Hub and Nuki devices time using NTP: Yes
Web configurator username: Not set
Web configurator password: Not set
Web configurator bypass for proxy IP: Not set
Web configurator authentication: Basic
Session validity (in seconds): 3600
Session validity remember (in hours): 720
Duo Push MFA enabled: No
TOTP MFA enabled: No
Web configurator enabled: Yes
HTTP SSL: Disabled
Nuki Hub FQDN for HTTP redirect: Not set
Advanced menu enabled: No
Publish free heap over MQTT: No
Nuki connect debug logging enabled: No
Nuki communication debug logging enabled: No
Nuki readable data debug logging enabled: No
Nuki hex data debug logging enabled: No
Nuki command debug logging enabled: No
MQTT log enabled: No
Webserial enabled: No
Bootloop protection enabled: No

------------ NETWORK ------------
Network device: Built-in Wi-Fi
Network connected: Yes
IP Address: 192.168.178.27
SSID: Fritz_WLAN_Ketty
BSSID of AP: 5C:49:79:4C:A0:7E
ESP32 MAC address: 94:A9:90:2B:F8:D4

------------ NETWORK SETTINGS ------------
[... 180 lines cut ...]
BUS mode Switch: Allowed
Short Circuit Duration: Allowed
Eletric Strike Delay: Allowed
Random Electric Strike Delay: Allowed
Electric Strike Duration: Allowed
Disable RTO after ring: Allowed
RTO timeout: Allowed
Doorbell suppression: Allowed
Doorbell suppression duration: Allowed
Sound Ring: Allowed
Sound Open: Allowed
Sound RTO: Allowed
Sound CM: Allowed
Sound confirmation: Allowed
Sound level: Allowed
Single button press action: Allowed
Double button press action: Allowed
Battery type: Allowed
Automatic battery type detection: Allowed
Reboot Nuki: Disallowed

------------ GPIO ------------

Retain Input GPIO MQTT state: No
```

iranl, 2025-05-08:
Last time major work was done on the Bluetooth side was before 9.08.
So also can't explain BLE seems to cause much more issues in 9.10/9.11 based on changes from our side.

I do see some chatter about issues with BLE caused  by IDF 5.4.1 / Arduino 3.2.0 which we upgraded to in 9.10

I've done a build of current master against Arduino 3.1.3 which can be found on https://github.com/iranl/nuki_hub/actions/runs/14914564401

Please report if uptime changes for the better with this build.

Itsmadhatter, 2025-05-09:
@iranl Unfortunately, I'm also getting the BLEError with your build (9.11/Arduino 3.1.3).

VDRainer, 2025-05-20:
> Please report if uptime changes for the better with this build.

9.11 was up for 11 days, 05:13 👍 

Now i also had the first ever BLEError
```
Last restart reason FW: BLEError
Last restart reason ESP: ESP_RST_SW: Software reset via esp_restart.
```

CH-1967, 2025-05-20:
I also have the problem with the irregular reboot.
Sometimes 1 day. Sometimes just 6 hours.
I have already tested several ESP models.
Each one showed the same problem.
I am now with the ESP32-S3 (Octal PSRAM).

The last reboot shows this.

Last restart reason FW: NotApplicable
Last restart reason ESP: ESP_RST_PANIC: Software reset due to exception/panic.

Such irregular reboots are a bit ugly.

I will test the build (9.11/Arduino 3.1.3).

Translated with DeepL.com (free version)

`System Information
------------ NUKI HUB ------------
Device: ESP32-S3 (Octal PSRAM)
Version: 9.10
Build: 14318502393.12.1
Build type: Release
Build date: 2025-04-07
Updater version: 
Updater build: 
Updater build date: 
Uptime (min): 5
Config version: 910
Last restart reason FW: NotApplicable
Last restart reason ESP: ESP_RST_PANIC: Software reset due to exception/panic.
Free internal heap: 182923
Total internal heap: 392807
PSRAM Available: Yes
Free usable PSRAM: 8338480
Total usable PSRAM: 8388608
Total PSRAM: 8388608
Total free heap: 8471440
Network task stack high watermark: 8328
Nuki task stack high watermark: 4872

------------ SPIFFS ------------
SPIFFS Total Bytes: 233681
SPIFFS Used Bytes: 58734
SPIFFS Free Bytes: 174947

------------ GENERAL SETTINGS ------------
Network task stack size: 12288
Nuki task stack size: 8192
Check for updates: Yes
Latest version: 9.10
Allow update from MQTT: No
Update Nuki Hub and Nuki devices time using NTP: No
Web configurator username: Not set
Web configurator password: Not set
Web configurator bypass for proxy IP: Not set
Web configurator authentication: Basic
Session validity (in seconds): 3600
Session validity remember (in hours): 720
Duo Push MFA enabled: No
TOTP MFA enabled: No
Web configurator enabled: Yes
HTTP SSL: Disabled
Nuki Hub FQDN for HTTP redirect: Not set
Advanced menu enabled: No
Publish free heap over MQTT: No
Nuki connect debug logging enabled: No
Nuki communication debug logging enabled: No
Nuki readable data debug logging enabled: No
Nuki hex data debug logging enabled: No
[... 185 lines cut ...]
Intercom ID: Allowed
BUS mode Switch: Allowed
Short Circuit Duration: Allowed
Eletric Strike Delay: Allowed
Random Electric Strike Delay: Allowed
Electric Strike Duration: Allowed
Disable RTO after ring: Allowed
RTO timeout: Allowed
Doorbell suppression: Allowed
Doorbell suppression duration: Allowed
Sound Ring: Allowed
Sound Open: Allowed
Sound RTO: Allowed
Sound CM: Allowed
Sound confirmation: Allowed
Sound level: Allowed
Single button press action: Allowed
Double button press action: Allowed
Battery type: Allowed
Automatic battery type detection: Allowed
Reboot Nuki: Allowed

------------ GPIO ------------

Retain Input GPIO MQTT state: No`

Seestern93, 2025-06-11:
As it is no option to dowbgrade to 9.09 or lower because of the github certificate, is it possible to get a build of 9.08 or 9.09 but with the newer certificate as a fast and stable workaround for the ESP32-S3? Until we have a permanent solution?

Seestern93, 2025-06-12:
> [@Seestern93](https://github.com/Seestern93) I know this not optimal, but is it an actual issue for you in normal use? As far as I can see although the uptime decreased for some users, the ESP does recover, and a restart usually takes only a few seconds. It's quite unlikely this interferes with normal usage.

Definitely its not optimal, although I understand that finding the issue is not easy.. but if you have several restarts per hour, the chance to try to open the lock while the nukihub restarts increases a lot.. I had already a few occasions where it did not work at all, because the command came at a moment where it seems that it discarded the command completely..

My thought was, rebuilding the old version only with new certificates would be the fastest workaround to have a working system.

StefanX13, 2025-06-12:
Same here. The restarts itself aren't a problem itself the  frequency is. E.g. I use an automation which locks the door when home assistant recognizes that everybody's leaves an no one's home. I had multiple times now where the lock command wasn't fulfilled since there was one of the restarts at this time. 
Just for your info why this is important to me. although I understand that finding the issue hard. Thanks for your great work btw.

technyon, 2025-06-14:
I've updated nimble, the bluetooth library. You can try the latest development build, this may or may not improve things.

I updated my ESP32 (Olimex board non-S3) to the latest dev version as well, before updating I had an uptime of 15619 minutes.

iranl, 2025-06-14:
I'm at 35.000 minutes on my main S3 with OCTAL PSRAM running release 9.10.

The specific 2.4ghz environment probably has an effect here where the antenna, distance esp-lock, having both opener and lock, amount of other (Zigbee, Bluetooth/BLE, WiFi) devices in range all matter for the stability of the ESP.

Debugging this is, as previously stated, very difficult. I don't expect newest nimble to change much. Next ESP-IDF / Arduino in a month might. 

And the most recent nimble does have some fixes on the deinit side of thing. I might take another look at unloading and reloading the BLE lib when issues are detected instead of doing a reset as we do now.

technyon, 2025-06-15:
Two more things:

- I know we often say this, but I can't stress it enough, make sure you have a good power supply and a good USB cable. ESPs are very sensitive to this. And yes this can vary from firmware version to firmware version, because it can increase peak power usage. I had the case that I could run an ESP just fine, and after an update to a new firmware it boot-looped because it now required more power during boot.
- I see you in all the configurations that the BLE transmit power is configured to 9 or higher. Try setting it to 0, so RX and TX is on par. More is not always better when it comes to radio communication, and these issues seem to be caused by some BLE error.

iranl, 2025-06-15:
Anyone with restart issues can try a build from here: https://github.com/iranl/nuki_hub/actions/runs/15666583145

Note that these are builds against an alpha build of ESP-IDF 5.5 and Arduino Core 3.3.0.

In this build I have enabled automatic restarting of the BLE controller in case of issues instead of just restarting the entire ESP32 as we do now.

In testing the process of restarting the BLE controller does seem to work fine, although I can't reproduce the problems reported in this issue so I can't really test if restarting the BLE controller will be enough to fix the issues that occur.

Seestern93, 2025-07-03:
> Anyone with restart issues can try a build from here: https://github.com/iranl/nuki_hub/actions/runs/15666583145
>
> Note that these are builds against an alpha build of ESP-IDF 5.5 and Arduino Core [...]

This build works for me like a charm :) since installing I had no reset anymore (7400 minutes uptime since then)
Thank you!!

closed 2025-07-23 by iranl