technyon/nuki_hub #668: Long Delay issue with Opener

riyas1980, 2025-05-07:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Hello,

First, thank you for your work on the Nuki Hub project.

I am using Nuki Hub (version 9.10, build 14318502393.12.1) on an ESP32 (no PSRAM) to control a Nuki Opener (firmware 1.11.0, hardware 4.17) with Home Assistant via MQTT. My MQTT broker is local. 

My configuration works, but I am experiencing a significant delay between pressing the button in Home Assistant and the action being executed by the Opener. The delay is usually around 5 to 7 seconds. This happens consistently. I have checked the Opener’s settings and set the opening delay to minimum.

Is this delay normal with the current architecture, or are there any settings or hardware recommendations that could help reduce the response time? Would using an ESP32-S3 with PSRAM make a noticeable difference? Any advice or insight would be greatly appreciated.

Thank you for your help!

### REQUESTED INFORMATION
Yes I did read your github to find a solution. 

System Information
------------ NUKI HUB ------------
Device: ESP32
Version: 9.10
Build: 14318502393.12.1
Build type: Release
Build date: 2025-04-07
Updater version: 
Updater build: 
Updater build date: 
Uptime (min): 871
Config version: 910
Last restart reason FW: NotApplicable
Last restart reason ESP: ESP_RST_WDT: Reset due to other watchdogs.
Free internal heap: 80020
Total internal heap: 266332
PSRAM Available: No
Network task stack high watermark: 8532
Nuki task stack high watermark: 5144

------------ SPIFFS ------------
SPIFFS Total Bytes: 233681
SPIFFS Used Bytes: 48192
SPIFFS Free Bytes: 185489

------------ GENERAL SETTINGS ------------
Network task stack size: 12288
Nuki task stack size: 8192
Check for updates: Yes
Latest version: 9.10
Allow update from MQTT: No
Update Nuki Hub and Nuki devices time using NTP: Yes
Web configurator username: Not set
Web configurator password: Not set
Web configurator bypass for proxy IP: Not set
Web configurator authentication: Basic
Session validity (in seconds): 3600
Session validity remember (in hours): 720
Duo Push MFA enabled: No
TOTP MFA enabled: No
Web configurator enabled: Yes
HTTP SSL: Disabled
Advanced menu enabled: No
Publish free heap over MQTT: No
Nuki connect debug logging enabled: No
Nuki communication debug logging enabled: No
Nuki readable data debug logging enabled: No
Nuki hex data debug logging enabled: No
Nuki command debug logging enabled: No
MQTT log enabled: No
Webserial enabled: No
Bootloop protection enabled: No

------------ NETWORK ------------
Network device: Built-in Wi-Fi
Network connected: Yes
IP Address: 192.168.0.95
SSID: home
BSSID of AP: 30:16:9D:20:4B:FA
ESP32 MAC address: 08:3A:F2:51:3B:24

------------ NETWORK SETTINGS ------------
Nuki Hub hostname: NH9227352523725224456
DHCP enabled: Yes
RSSI Publish interval (s): 60
Find WiFi AP with strongest signal: Yes
Restart ESP32 on network disconnect enabled: No
Disable Network if not connected within 60s: No
MQTT Timeout until restart (s): 60

------------ MQTT ------------
MQTT connected: Yes
MQTT broker address: 192.168.0.100
MQTT broker port: 1883
MQTT username: ***
MQTT password: ***
MQTT base topic: nukihub
MQTT SSL: Disabled

------------ BLUETOOTH ------------
Bluetooth connection mode: New
Bluetooth TX power (dB): 9
Bluetooth command nr of retries: 3
Bluetooth command retry delay (ms): 100
Seconds until reboot when no BLE beacons received: 60

------------ QUERY / PUBLISH SETTINGS ------------
Lock/Opener state query interval (s): 1800
Publish Nuki device authorization log: No
Max authorization log entries to retrieve: 5
Battery state query interval (s): 1800
Most non-JSON MQTT topics disabled: No
Publish Nuki device config: Yes
Config query interval (s): 3600
Publish Keypad info: No
Keypad query interval (s): 1800
Enable Keypad control: Yes
Publish Keypad topic per entry: No
Publish Keypad codes: No
Allow checking Keypad codes: No
Max keypad entries to retrieve: 10
Publish timecontrol info: No
Keypad query interval (s): 1800
Enable timecontrol control: No
Publish timecontrol topic per entry: No
Max timecontrol entries to retrieve: 10
Enable authorization control: No
Publish authorization topic per entry: No
Max authorization entries to retrieve: 5

------------ HOME ASSISTANT ------------
Home Assistant auto discovery enabled: Yes
Home Assistant auto discovery topic: homeassistant/
Nuki Hub configuration URL for HA: https://ha.malim.fr
Nuki Hub ID: 39837196499464

------------ NUKI LOCK ------------
Lock enabled: No

------------ NUKI OPENER ------------
Opener enabled: Yes
Paired: Yes
Nuki Hub device ID: 1556462785
Nuki device ID: ***
Firmware version: 1.11.0
Hardware version: 4.17
Opener valid PIN set: Yes
Opener has keypad: No
Timecontrol highest entries count: 0
Authorizations highest entries count: 0
Register as: Bridge
Nuki Opener Lock/Unlock action set to Continuous mode in Home Assistant: No
Force Opener ID: No
Force Opener Keypad: No

------------ NUKI OPENER ACL ------------
Activate Ring-to-Open: Allowed
Deactivate Ring-to-Open: Allowed
Electric Strike Actuation: Allowed
Activate Continuous Mode: Allowed
Deactivate Continuous Mode: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed

------------ NUKI OPENER CONFIG ACL ------------
Name: Allowed
Latitude: Allowed
Longitude: Allowed
Pairing enabled: Allowed
Button enabled: Allowed
LED flash enabled: Allowed
Timezone offset: Allowed
DST mode: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed
Operating Mode: Allowed
Advertising Mode: Allowed
Timezone ID: Allowed
Intercom ID: Allowed
BUS mode Switch: Allowed
Short Circuit Duration: Allowed
Eletric Strike Delay: Allowed
Random Electric Strike Delay: Allowed
Electric Strike Duration: Allowed
Disable RTO after ring: Allowed
RTO timeout: Allowed
Doorbell suppression: Allowed
Doorbell suppression duration: Allowed
Sound Ring: Allowed
Sound Open: Allowed
Sound RTO: Allowed
Sound CM: Allowed
Sound confirmation: Allowed
Sound level: Allowed
Single button press action: Allowed
Double button press action: Allowed
Battery type: Allowed
Automatic battery type detection: Allowed
Reboot Nuki: Allowed

------------ GPIO ------------

Retain Input GPIO MQTT state: No
```

### TO REPRODUCE
_Steps to reproduce the behavior: The issue is there every time

### EXPECTED BEHAVIOUR
A delay between 1 to 3 second would be perfect

### SCREENSHOTS
No

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

technyon, 2025-05-07:
It can take a moment due to bluetooth communication, but 5 to 7 seconds sounds a bit on the long side. Often this is caused by bluetooth issues. How far apart are your ESP and the opener?

You can:
- check the RSSI value. I think a reasonable value is below -60.
- You can enable MQTT logs and check the logs while you initiate an action. If there are retries, you know why it takes longer than usual.

adrianipopescu, 2025-05-28:
Happens to me too, running bleeding edge (9.11m7)

RSSI for the opener is -58

BUT:
```

Topic: nuki/opener/rssiQoS: 0
-59
2025-05-29 00:39:15:148

Topic: nuki/opener/actionQoS: 0
activateRTO
2025-05-29 00:39:55:226

Topic: nuki/maintenance/logQoS: 0
Opener action received: activateRTO
2025-05-29 00:39:55:307

Topic: nuki/opener/actionQoS: 0
ack
2025-05-29 00:39:55:324

Topic: nuki/opener/commandResultQoS: 0
success
2025-05-29 00:39:59:497

Topic: nuki/maintenance/logQoS: 0
Opener action result: success
2025-05-29 00:39:59:519

Topic: nuki/opener/retryQoS: 0
--
2025-05-29 00:39:59:527

Topic: nuki/maintenance/logQoS: 0
Opener: updating status after action
2025-05-29 00:39:59:532

Topic: nuki/opener/query/lockstateCommandResultQoS: 0
success
2025-05-29 00:39:59:777

Topic: nuki/maintenance/logQoS: 0
Result (attempt 1): Publishing auth data
2025-05-29 00:39:59:788

Topic: nuki/maintenance/logQoS: 0
************************ COMMAND FAILED TIMEOUT ************************
2025-05-29 00:40:10:041
```

which adds to the lag but the command does inevitably go through and status is refreshed correctly, it just takes ±5-7sec